zigpy/zha-device-handlers #939: [Device Support Request] Woox R7060 Irrigation Controller

blakadder, 2021-06-26:
Device: Woox R7060 Irrigation Controller - https://wooxhome.com/products-c10/other-c4/woox-r7060-smart-garden-irrigation-control-p61

Tuya device, unable to finish interview during pairing. Using ZbBridge via tcp bridge

**Describe the solution you'd like**
To pair successfully

**Device signature - this can be acquired by removing the device from ZHA and pairing it again from the add devices screen. Be sure to add the entire content of the log panel after pairing the device to a code block below this line.**

```
[0x0000:zdo] ZDO request ZDOCmd.Mgmt_Permit_Joining_req: [60, <Bool.false: 0>]
[0x0000](EZSP): started initialization
[0x0000:ZDO](EZSP): 'async_initialize' stage succeeded
[0x0000](EZSP): power source: Mains
[0x0000](EZSP): completed initialization
[0x0000:zdo] ZDO request ZDOCmd.Mgmt_Permit_Joining_req: [60, <Bool.false: 0>]
New device 0x7eed (84:71:27:ff:fe:d4:28:29) joined the network
[0x7eed] Scheduling initialization
Tries remaining: 3
[0x7eed] Requesting 'Node Descriptor'
Tries remaining: 2
[0x7eed] Extending timeout for 0x0d request
Device 0x7eed (84:71:27:ff:fe:d4:28:29) joined the network
[0x7eed] Scheduling initialization
[0x7eed] Canceling old initialize call
Received frame on uninitialized device <Device model=None manuf=None nwk=0x7EED ieee=84:71:27:ff:fe:d4:28:29 is_initialized=False> from ep 0 to ep 0, cluster 19: b"\x81\xed~)(\xd4\xfe\xff'q\x84\x80"
[0x7eed:zdo] ZDO request ZDOCmd.Device_annce: [0x7EED, 84:71:27:ff:fe:d4:28:29, 128]
Tries remaining: 3
[0x7eed] Requesting 'Node Descriptor'
Tries remaining: 2
[0x7eed] Extending timeout for 0x0f request
Received frame on uninitialized device <Device model=None manuf=None nwk=0x7EED ieee=84:71:27:ff:fe:d4:28:29 is_initialized=False> from ep 0 to ep 0, cluster 32770: b'\r\x00\xed~\x02@\x80\x02\x10RR\x00\x00,R\x00\x00'
[0x7eed:zdo] ZDO request ZDOCmd.Node_Desc_rsp: [<Status.SUCCESS: 0>, 0x7EED, NodeDescriptor(logical_type=<LogicalType.EndDevice: 2>, complex_descriptor_available=0, user_descriptor_available=0, reserved=0, aps_flags=0, frequency_band=<FrequencyBand.Freq2400MHz: 8>, mac_capability_flags=<MACCapabilityFlags.AllocateAddress: 128>, manufacturer_code=4098, maximum_buffer_size=82, maximum_incoming_transfer_size=82, server_mask=11264, maximum_outgoing_transfer_size=82, descriptor_capability_field=<DescriptorCapability.0: 0>, *allocate_address=True, *is_alternate_pan_coordinator=False, *is_coordinator=False, *is_end_device=True, *is_full_function_device=False, *is_mains_powered=False, *is_receiver_on_when_idle=False, *is_router=False, *is_security_capable=False)]
[0x7eed:zdo] Unsupported ZDO request:ZDOCmd.Node_Desc_rsp
Received frame on uninitialized device <Device model=None manuf=None nwk=0x7EED ieee=84:71:27:ff:fe:d4:28:29 is_initialized=False> from ep 1 to ep 1, cluster 10: b'\x00\x00\x00\x07\x00'
[0x7eed] Received ZCL while uninitialized on endpoint id 1, cluster 0x000a id, hdr: <ZCLHeader frame_control=<FrameControl frame_type=GLOBAL_COMMAND manufacturer_specific=False is_reply=False disable_default_response=False> manufacturer=None tsn=0 command_id=Command.Read_Attributes>, payload: b'\x07\x00'
[0x7eed] Uninitialized device command 'Command.Read_Attributes' args: [[7]]
Received frame on uninitialized device <Device model=None manuf=None nwk=0x7EED ieee=84:71:27:ff:fe:d4:28:29 is_initialized=False> from ep 0 to ep 0, cluster 32770: b'\x0f\x00\xed~\x02@\x80\x02\x10RR\x00\x00,R\x00\x00'
[0x7eed] Got Node Descriptor: NodeDescriptor(logical_type=<LogicalType.EndDevice: 2>, complex_descriptor_available=0, user_descriptor_available=0, reserved=0, aps_flags=0, frequency_band=<FrequencyBand.Freq2400MHz: 8>, mac_capability_flags=<MACCapabilityFlags.AllocateAddress: 128>, manufacturer_code=4098, maximum_buffer_size=82, maximum_incoming_transfer_size=82, server_mask=11264, maximum_outgoing_transfer_size=82, descriptor_capability_field=<DescriptorCapability.0: 0>, *allocate_address=True, *is_alternate_pan_coordinator=False, *is_coordinator=False, *is_end_device=True, *is_full_function_device=False, *is_mains_powered=False, *is_receiver_on_when_idle=False, *is_router=False, *is_security_capable=False)
[0x7eed] Discovering endpoints
Tries remaining: 3
[0x7eed] Extending timeout for 0x11 request
Received frame on uninitialized device <Device model=None manuf=None nwk=0x7EED ieee=84:71:27:ff:fe:d4:28:29 is_initialized=False> from ep 1 to ep 1, cluster 10: b'\x00\x00\x00\x07\x00'
[0x7eed] Received ZCL while uninitialized on endpoint id 1, cluster 0x000a id, hdr: <ZCLHeader frame_control=<FrameControl frame_type=GLOBAL_COMMAND manufacturer_specific=False is_reply=False disable_default_response=False> manufacturer=None tsn=0 command_id=Command.Read_Attributes>, payload: b'\x07\x00'
[0x7eed] Uninitialized device command 'Command.Read_Attributes' args: [[7]]
Received frame on uninitialized device <Device model=None manuf=None nwk=0x7EED ieee=84:71:27:ff:fe:d4:28:29 is_initialized=False> from ep 0 to ep 0, cluster 32773: b'\x11\x00\xed~\x01\x01'
[0x7eed] Discovered endpoints: [1]
[0x7eed] Initializing endpoints [<Endpoint id=1 in=[] out=[] status=0>]
[0x7eed:1] Discovering endpoint information
Tries remaining: 3
[0x7eed] Extending timeout for 0x13 request
Received frame on uninitialized device <Device model=None manuf=None nwk=0x7EED ieee=84:71:27:ff:fe:d4:28:29 is_initialized=False> from ep 1 to ep 1, cluster 10: b'\x00\x00\x00\x07\x00'
[0x7eed] Received ZCL while uninitialized on endpoint id 1, cluster 0x000a id, hdr: <ZCLHeader frame_control=<FrameControl frame_type=GLOBAL_COMMAND manufacturer_specific=False is_reply=False disable_default_response=False> manufacturer=None tsn=0 command_id=Command.Read_Attributes>, payload: b'\x07\x00'
[0x7eed] Uninitialized device command 'Command.Read_Attributes' args: [[7]]
Received frame on uninitialized device <Device model=None manuf=None nwk=0x7EED ieee=84:71:27:ff:fe:d4:28:29 is_initialized=False> from ep 0 to ep 0, cluster 32772: b'\x13\x00\xed~\x18\x01\x04\x01Q\x00\x01\x06\x00\x00\x04\x00\x05\x00\x06\x00\x00\xef\x01\x00\x02\x19\x00\n\x00'
[0x7eed:1] Discovered endpoint information: SizePrefixedSimpleDescriptor(endpoint=1, profile=260, device_type=81, device_version=1, input_clusters=[0, 4, 5, 6, 61184, 1], output_clusters=[25, 10])
Unknown cluster 61184
[0x7eed] Extending timeout for 0x15 request
[0x7eed:1:0x000a] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=GLOBAL_COMMAND manufacturer_specific=False is_reply=False disable_default_response=False> manufacturer=None tsn=0 command_id=Command.Read_Attributes>
[0x7eed:1:0x000a] ZCL request 0x0000: [[7]]
[0x7eed:1:0x0000] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=GLOBAL_COMMAND manufacturer_specific=False is_reply=True disable_default_response=True> manufacturer=None tsn=21 command_id=Command.Read_Attributes_rsp>
Device failed to initialize due to unexpected error
Error handling '_save_attribute' event with (84:71:27:ff:fe:d4:28:29, 1, 0, 4, '_TZ3210_eymunffl') params: FOREIGN KEY constraint failed
Error handling '_save_attribute' event with (84:71:27:ff:fe:d4:28:29, 1, 0, 5, 'TS0101') params: FOREIGN KEY constraint failed
[0x7eed:zdo] ZDO request ZDOCmd.Node_Desc_req: [0x0000]
[0x7eed:zdo] Unsupported ZDO request:ZDOCmd.Node_Desc_req
[0x7eed:1:0x000a] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=GLOBAL_COMMAND manufacturer_specific=False is_reply=False disable_default_response=False> manufacturer=None tsn=1 command_id=Command.Read_Attributes>
[0x7eed:1:0x000a] ZCL request 0x0000: [[7]]
[0x7eed:1:0x0006] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=GLOBAL_COMMAND manufacturer_specific=False is_reply=True disable_default_response=False> manufacturer=None tsn=2 command_id=Command.Report_Attributes>
[0x7eed:1:0x0006] ZCL request 0x000a: [[Attribute(attrid=0, value=<TypeValue type=Bool, value=Bool.false>), Attribute(attrid=16386, value=<TypeValue type=uint16_t, value=0>), Attribute(attrid=16385, value=<TypeValue type=uint16_t, value=0>)]]
[... 40 lines cut ...]
[0x7eed:1:0xef00] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=CLUSTER_COMMAND manufacturer_specific=False is_reply=True disable_default_response=False> manufacturer=None tsn=3 command_id=6>
[0x7eed:1:0xef00] Unknown cluster-specific command 6
[0x7eed:1:0xef00] ZCL request 0x0006: b'\x00\x01f\x04\x00\x01\x02'
[0x7eed:1:0xef00] No handler for cluster command 6
[... cut ...]
[0x7eed:1:0xef00] ZCL request 0x0006: b'\x00\x02k\x04\x00\x01\x00'
[0x7eed:1:0xef00] No handler for cluster command 6
[0x7eed:1:0xef00] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=CLUSTER_COMMAND manufacturer_specific=False is_reply=True disable_default_response=False> manufacturer=None tsn=5 command_id=6>
[0x7eed:1:0xef00] Unknown cluster-specific command 6
[0x7eed:1:0xef00] ZCL request 0x0006: b'\x00\x03j\x03\x000000000000000000000000000000000000000000000000000'
[0x7eed:1:0xef00] No handler for cluster command 6
```

Adminiuga, 2021-06-27:
```
NodeDescriptor(logical_type=<LogicalType.EndDevice: 2>, complex_descriptor_available=0, user_descriptor_available=0, reserved=0, aps_flags=0, frequency_band=<FrequencyBand.Freq2400MHz: 8>, mac_capability_flags=<MACCapabilityFlags.AllocateAddress: 128>, manufacturer_code=4098, maximum_buffer_size=82, maximum_incoming_transfer_size=82, server_mask=11264, maximum_outgoing_transfer_size=82, descriptor_capability_field=<DescriptorCapability.0: 0>, *allocate_address=True, *is_alternate_pan_coordinator=False, *is_coordinator=False, *is_end_device=True, *is_full_function_device=False, *is_mains_powered=False, *is_receiver_on_when_idle=False, *is_router=False, *is_security_capable=False)

SizePrefixedSimpleDescriptor(endpoint=1, profile=260, device_type=81, device_version=1, input_clusters=[0, 4, 5, 6, 61184, 1], output_clusters=[25, 10])

manufacturer: '_TZ3210_eymunffl'
model: 'TS0101'
```

Is it really an end device? as in "battery operated" device?

Adminiuga, 2021-07-07:
Do you have any contact with the vendor and any chance of getting their UART protocol or data points explained?
Are you on 2021.7 beta or newer?
Did you enable custom quirks folder in ZHA configuration? something like:
```
zha:
  custom_quirks_path: /config/my_quirks
```

If yes to all, can you add the following to `/config/my_quirks`: an empty `__init__.py` just in case and a `ts0101_woox.py` containing:
```
"""Woox irrigation controller."""

from zigpy.profiles import zha
from zigpy.quirks import CustomDevice
from zigpy.zcl.clusters.general import (
    Basic,
    Groups,
    Ota,
    OnOff,
    PowerConfiguration,
    Scenes,
    Time,
)

from zhaquirks.const import (
    DEVICE_TYPE,
    ENDPOINTS,
    INPUT_CLUSTERS,
    MODELS_INFO,
    OUTPUT_CLUSTERS,
    PROFILE_ID,
)
from zhaquirks.tuya import DPToAttributeMapping, TuyaNewManufCluster

class WooxR7060(CustomDevice):
    """Woox R7060 Irrigation Controller."""

    signature = {
        MODELS_INFO: [("_TZ3210_eymunffl", "TS0101")],
        ENDPOINTS: {
            1: {
                PROFILE_ID: zha.PROFILE_ID,
                DEVICE_TYPE: zha.DeviceType.SMART_PLUG,
                INPUT_CLUSTERS: [
                    Basic.cluster_id,
                    PowerConfiguration.cluster_id,
                    Groups.cluster_id,
                    Scenes.cluster_id,
                    OnOff.cluster_id,
                    TuyaNewManufCluster.cluster_id,
                ],
                OUTPUT_CLUSTERS: [Time.cluster_id, Ota.cluster_id],
            }
        },
    }

    replacement = {
        ENDPOINTS: {
            1: {
                DEVICE_TYPE: zha.DeviceType.SMART_PLUG,
                INPUT_CLUSTERS: [
                    Basic.cluster_id,
                    PowerConfiguration.cluster_id,
                    Groups.cluster_id,
                    Scenes.cluster_id,
                    OnOff.cluster_id,
                    TuyaNewManufCluster,
                ],
                OUTPUT_CLUSTERS: [Time.cluster_id, Ota.cluster_id],
            }
        }
    }
```

enable debug. Join device and see if it assigned quirks to the device. Leave it running and try to operate the device. Would interested to see all the traffic from the device and if able to correlate the traffic to operations of the device. This quirk should print the datapoints received from the device.

blakadder, 2021-07-14:
paired on latest HA dev

closed 2021-07-14 by blakadder