zigpy/zha-device-handlers #940: [Device Support Request] Aqara H1 wireless switch double rocker (model WRS-R02)

pavaroto, 2021-06-26:
**Is your feature request related to a problem? Please describe.**
A clear and concise description of what the problem is. Ex. I'm always frustrated when [...]
The device is not yet supported in ZHA but I saw that it received support in zigbee2mqtt https://www.zigbee2mqtt.io/devices/WRS-R02.html

**Describe the solution you'd like**
A clear and concise description of what you want to happen.
Support for single/double/triple/long press of left/right & both buttons.

**Device signature - this can be acquired by removing the device from ZHA and pairing it again from the add devices screen. Be sure to add the entire content of the log panel after pairing the device to a code block below this line.**
```
{
  "node_descriptor": "NodeDescriptor(byte1=2, byte2=64, mac_capability_flags=128, manufacturer_code=4447, maximum_buffer_size=127, maximum_incoming_transfer_size=100, server_mask=11264, maximum_outgoing_transfer_size=100, descriptor_capability_field=0, *allocate_address=True, *complex_descriptor_available=False, *is_alternate_pan_coordinator=False, *is_coordinator=False, *is_end_device=True, *is_full_function_device=False, *is_mains_powered=False, *is_receiver_on_when_idle=False, *is_router=False, *is_security_capable=False, *is_valid=True, *logical_type=<LogicalType.EndDevice: 2>, *user_descriptor_available=False)",
  "endpoints": {
    "1": {
      "profile_id": 260,
      "device_type": "0x0103",
      "in_clusters": [
        "0x0000",
        "0x0001",
        "0x0003"
      ],
      "out_clusters": [
        "0x0003",
        "0x0006"
      ]
    },
    "2": {
      "profile_id": null,
      "device_type": "",
      "in_clusters": [],
      "out_clusters": []
    },
    "3": {
      "profile_id": 260,
      "device_type": "0x0103",
      "in_clusters": [
        "0x0003"
      ],
      "out_clusters": [
        "0x0006"
      ]
    },
    "4": {
      "profile_id": null,
      "device_type": "",
      "in_clusters": [],
      "out_clusters": []
    },
    "5": {
      "profile_id": null,
      "device_type": "",
      "in_clusters": [],
      "out_clusters": []
    },
    "6": {
      "profile_id": null,
      "device_type": "",
      "in_clusters": [],
      "out_clusters": []
    }
  },
  "manufacturer": "LUMI",
  "model": "lumi.remote.b28ac1",
  "class": "zigpy.device.Device"
}
```
**Additional context**
Add any other context or screenshots about the feature request here.

The device was paired in home assistant's ZHA integration but not working properly. Both buttons behave as the same button.
Below is the result of zha_event listening

Right button single press
```
{
    "event_type": "zha_event",
    "data": {
        "device_ieee": "EDITED OUT",
        "unique_id": "EDITED OUT",
        "device_id": "EDITED OUT",
        "endpoint_id": 1,
        "cluster_id": 6,
        "command": "toggle",
        "args": []
    },
    "origin": "LOCAL",
    "time_fired": "2021-06-26T13:43:42.499260+00:00",
    "context": {
        "id": "b9d5f1afde358915f8619d1153832a07",
        "parent_id": null,
        "user_id": null
    }
}
```
Left button single press
```
{
    "event_type": "zha_event",
    "data": {
        "device_ieee": "EDITED OUT",
        "unique_id": "EDITED OUT",
        "device_id": "EDITED OUT",
        "endpoint_id": 1,
        "cluster_id": 6,
        "command": "toggle",
        "args": []
    },
    "origin": "LOCAL",
    "time_fired": "2021-06-26T13:43:32.947572+00:00",
    "context": {
        "id": "d35c686ec88e197fa8f7adad07ad1b72",
        "parent_id": null,
        "user_id": null
    }
}
```

Adminiuga, 2021-07-07:
what radio are you using?

pavaroto, 2021-07-07:
I am using EZSP on a Xiaomi Gateway 3.
I should mention that I am not a programmer and new to HA so I hope I managed to provide the correct info. Below is the extracted log lines for each button single press from the log file after enabling logger.

Left button press:
```2021-07-07 20:42:19 DEBUG (MainThread) [homeassistant.core] Bus:Handling <Event zha_event[L]: device_ieee=54:ef:44:10:00:06:5c:96, unique_id=54:ef:44:10:00:06:5c:96:1:0x0006, device_id=5d00d3d3912f69bf88c9bdeaea66fd3f, endpoint_id=1, cluster_id=6, command=toggle, args=[]>```

Right button press:
```2021-07-07 20:42:22 DEBUG (MainThread) [homeassistant.core] Bus:Handling <Event zha_event[L]: device_ieee=54:ef:44:10:00:06:5c:96, unique_id=54:ef:44:10:00:06:5c:96:1:0x0006, device_id=5d00d3d3912f69bf88c9bdeaea66fd3f, endpoint_id=1, cluster_id=6, command=toggle, args=[]>```

I tried to reconnect the switch in ZHA in order to get the log as well:

```[0x0000:zdo] ZDO request ZDOCmd.Mgmt_Permit_Joining_req: [60, <Bool.false: 0>]
[0x93a3:1:0x0400] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=GLOBAL_COMMAND manufacturer_specific=False is_reply=True disable_default_response=True> manufacturer=None tsn=235 command_id=Command.Report_Attributes>
[0x93a3:1:0x0400] ZCL request 0x000a: [[Attribute(attrid=0, value=<TypeValue type=uint16_t, value=35>)]]
[0x93a3:1:0x0400] Attribute report received: measured_value=35
[0x93a3:1:0x0406] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=GLOBAL_COMMAND manufacturer_specific=False is_reply=True disable_default_response=True> manufacturer=None tsn=236 command_id=Command.Report_Attributes>
[0x93a3:1:0x0406] ZCL request 0x000a: [[Attribute(attrid=0, value=<TypeValue type=bitmap8, value=bitmap8.1>)]]
[0x93a3:1:0x0406] Attribute report received: occupancy=1
[0x93A3:1:0x0500]: Updated alarm state: 1
Received frame on uninitialized device 54:ef:44:10:00:06:5c:96 (Status.NEW) for endpoint: 1
[0xc6e6] Received ZCL while uninitialized on endpoint id 1, cluster 0x0006 id, hdr: <ZCLHeader frame_control=<FrameControl frame_type=CLUSTER_COMMAND manufacturer_specific=False is_reply=False disable_default_response=True> manufacturer=None tsn=55 command_id=2>, payload: b''
Device 0xc6e6 (54:ef:44:10:00:06:5c:96) left the network
[0x4ded:zdo] ZDO request ZDOCmd.Active_EP_req: [0x0000]
[0x4ded:zdo] Unsupported ZDO request:ZDOCmd.Active_EP_req
Device 0xc6e6 (54:ef:44:10:00:06:5c:96) joined the network
[0xc6e6] Requesting 'Node Descriptor'
Tries remaining: 2
[0xc6e6] Extending timeout for 0x1f request
Device 0xc6e6 (54:ef:44:10:00:06:5c:96) joined the network
Skip initialization for existing device 54:ef:44:10:00:06:5c:96
[0xc6e6:zdo] ZDO request ZDOCmd.Device_annce: [0xC6E6, 54:ef:44:10:00:06:5c:96, 128]
[0xc6e6:zdo] ZDO request ZDOCmd.Node_Desc_req: [0x0000]
[0xc6e6:zdo] Unsupported ZDO request:ZDOCmd.Node_Desc_req
[0xc6e6] Node Descriptor: NodeDescriptor(byte1=2, byte2=64, mac_capability_flags=128, manufacturer_code=4447, maximum_buffer_size=127, maximum_incoming_transfer_size=100, server_mask=11264, maximum_outgoing_transfer_size=100, descriptor_capability_field=0, *allocate_address=True, *complex_descriptor_available=False, *is_alternate_pan_coordinator=False, *is_coordinator=False, *is_end_device=True, *is_full_function_device=False, *is_mains_powered=False, *is_receiver_on_when_idle=False, *is_router=False, *is_security_capable=False, *is_valid=True, *logical_type=<LogicalType.EndDevice: 2>, *user_descriptor_available=False)
[0xc6e6] Discovering endpoints
Tries remaining: 3
[0xc6e6] Extending timeout for 0x21 request
[0xc6e6] Discovered endpoints: [1, 2, 3, 4, 5, 6]
[0xc6e6:1] Discovering endpoint information
Tries remaining: 3
[0xc6e6] Extending timeout for 0x23 request
[0xc6e6:zdo] ZDO request ZDOCmd.Mgmt_Permit_Joining_req: [180, <Bool.true: 1>]
[0xc6e6:1] Discovered endpoint information: SizePrefixedSimpleDescriptor(endpoint=1, profile=260, device_type=259, device_version=1, input_clusters=[0, 3, 1], output_clusters=[3, 6])
[0xc6e6] Extending timeout for 0x25 request
[0xc6e6:1:0x0000] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=GLOBAL_COMMAND manufacturer_specific=False is_reply=True disable_default_response=True> manufacturer=None tsn=37 command_id=Command.Read_Attributes_rsp>
[0xc6e6:1] Manufacturer: LUMI
[0xc6e6:1] Model: lumi.remote.b28ac1
[0xc6e6:2] Discovering endpoint information
Tries remaining: 3
[0xc6e6] Extending timeout for 0x27 request
[0xc6e6:3] Discovering endpoint information
Tries remaining: 3
[0xc6e6] Extending timeout for 0x29 request
[0xc6e6:3] Discovered endpoint information: SizePrefixedSimpleDescriptor(endpoint=3, profile=260, device_type=259, device_version=1, input_clusters=[3], output_clusters=[6])
[0xc6e6:4] Discovering endpoint information
Tries remaining: 3
[0xc6e6] Extending timeout for 0x2b request
[0xc6e6:5] Discovering endpoint information
Tries remaining: 3
[0xc6e6] Extending timeout for 0x2d request
[0xc6e6:6] Discovering endpoint information
Tries remaining: 3
[0xc6e6] Extending timeout for 0x2f request
Checking quirks for LUMI lumi.remote.b28ac1 (54:ef:44:10:00:06:5c:96)
Considering <class 'zhaquirks.xbee.xbee_io.XBeeSensor'>
Fail because endpoint list mismatch: {232, 230} {1, 2, 3, 4, 5, 6}
Considering <class 'zhaquirks.xbee.xbee3_io.XBee3Sensor'>
Fail because endpoint list mismatch: {232, 230} {1, 2, 3, 4, 5, 6}
Considering <class 'zhaquirks.smartthings.tag_v4.SmartThingsTagV4'>
[... cut ...] es_rsp>
[0xC6E6:1:0x0001]: finished channel configuration
[0xC6E6:1:0x0000]: 'async_initialize' stage succeeded
[0xC6E6:1:0x0001]: 'async_initialize' stage succeeded
[0xC6E6:1:0x0006]: 'async_initialize' stage succeeded
[0xC6E6](lumi.remote.b28ac1): power source: Battery or Unknown
[0xC6E6](lumi.remote.b28ac1): completed initialization
[0x93A3:1:0x0500]: Updated alarm state: 0
[0x4ded:zdo] ZDO request ZDOCmd.Active_EP_req: [0x0000]
[0x4ded:zdo] Unsupported ZDO request:ZDOCmd.Active_EP_req```

''

Adminiuga, 2021-07-11:
Leave the same debug log running, but press the left the. Right  button and let me know the bed address of the device.

pavaroto, 2021-07-11:
I don't know what or where is the bed address of the device so I extracted the records for each button press from the home-assistant.log file. I hope it has the information you need. If not please give me more details on how to get it.

Left button press
```2021-07-11 11:08:07 DEBUG (bellows.thread_0) [bellows.uart] Data frame: b'577fb157546f15b658924a24ab5593499cded8770b089874fed7b58b2cab7e'
2021-07-11 11:08:07 DEBUG (bellows.thread_0) [bellows.uart] Sending: b'8610be7e'
2021-07-11 11:08:07 DEBUG (MainThread) [bellows.ezsp.protocol] Application frame 69 (incomingMessageHandler) received: b'000401060001010001000090ffdce6c6ffff0311d602'
2021-07-11 11:08:07 DEBUG (MainThread) [bellows.zigbee.application] Received incomingMessageHandler frame with [<EmberIncomingMessageType.INCOMING_UNICAST: 0>, EmberApsFrame(profileId=260, clusterId=6, sourceEndpoint=1, destinationEndpoint=1, options=<EmberApsOption.APS_OPTION_ENABLE_ROUTE_DISCOVERY: 256>, groupId=0, sequence=144), 255, -36, 0xc6e6, 255, 255, b'\x11\xd6\x02']
2021-07-11 11:08:07 DEBUG (MainThread) [zigpy.zcl] [0xc6e6:1:0x0006] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=CLUSTER_COMMAND manufacturer_specific=False is_reply=False disable_default_response=True> manufacturer=None tsn=214 command_id=2>
2021-07-11 11:08:07 DEBUG (MainThread) [zigpy.zcl] [0xc6e6:1:0x0006] ZCL request 0x0002: []
2021-07-11 11:08:07 DEBUG (MainThread) [zigpy.zcl] [0xc6e6:1:0x0006] No handler for cluster command 2
```

Right button press
```2021-07-11 11:10:03 DEBUG (bellows.thread_0) [bellows.uart] Data frame: b'460eb157546f15b658924a24ab5593499cdcd8770b089874fed7bb8b34017e'
2021-07-11 11:10:03 DEBUG (bellows.thread_0) [bellows.uart] Sending: b'8520dd7e'
2021-07-11 11:10:03 DEBUG (MainThread) [bellows.ezsp.protocol] Application frame 69 (incomingMessageHandler) received: b'000401060001010001000092ffdce6c6ffff0311d802'
2021-07-11 11:10:03 DEBUG (MainThread) [bellows.zigbee.application] Received incomingMessageHandler frame with [<EmberIncomingMessageType.INCOMING_UNICAST: 0>, EmberApsFrame(profileId=260, clusterId=6, sourceEndpoint=1, destinationEndpoint=1, options=<EmberApsOption.APS_OPTION_ENABLE_ROUTE_DISCOVERY: 256>, groupId=0, sequence=146), 255, -36, 0xc6e6, 255, 255, b'\x11\xd8\x02']
2021-07-11 11:10:03 DEBUG (MainThread) [zigpy.zcl] [0xc6e6:1:0x0006] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=CLUSTER_COMMAND manufacturer_specific=False is_reply=False disable_default_response=True> manufacturer=None tsn=216 command_id=2>
2021-07-11 11:10:03 DEBUG (MainThread) [zigpy.zcl] [0xc6e6:1:0x0006] ZCL request 0x0002: []
2021-07-11 11:10:03 DEBUG (MainThread) [zigpy.zcl] [0xc6e6:1:0x0006] No handler for cluster command 2
```

Adminiuga, 2021-07-11:
I don't know if xiaomi does some extra, manufacturer specific configuration, but it is not possible to discern between left and right clicks, as incoming data is virtually the same. 
In ZigBee spec compliant devices, button traffic comes from different endpoints, e.g endpoint 1 and endpoint 2. The commands are the same but endpoints are different. In this case the endpoint is always the same. Not sure what could be done here.

MattWestb, 2021-07-11:
Z2M looks having the possibility changing operating mode between LO and HA.
Perhaps its sending the commands on different endpoints if being in LO mode ?

Adminiuga, 2021-07-13:
What is LO and what is HA?

MattWestb, 2021-07-13:
LO = Light and occupancy (sending light and commands) and  HA = sending "clicks".

oliver989, 2021-09-10:
Any hope getting this device to work? I have two of these sitting in my drawer.

thomasfr, 2021-09-15:
Would love to see support too. I could pair it on HA with ZHA on A RPi with Raspbee2 but it does not show any entities except a malfunctioning power/battery entity.

sy-per, 2021-11-04:
Same for me.
To many, maybe things will change.

d-bourdon, 2021-11-24:
Same here, but I'm adding some information:

From the User Manual:
- They explain on the product overview that the switch have two modes: a """ single press in the high speed click mode and 7 functions in the multi-function mode including single press ... etc.""" (the feature that we all want)
- On the switch modes section: they explain that basically you need to connect to their app (so through the bridge) to set the mode of the device

So I guess here we just need to find a way to send this command to the switch, things that zigbee2mqtt seems to allow https://www.zigbee2mqtt.io/devices/WRS-R02.html#operation-mode-enum

MattWestb, 2021-11-24:
The code used in Z2M:
https://github.com/Koenkk/zigbee-herdsman-converters/blob/fa1faf7ec578df488a95dbb187aa0d18482a0f3f/devices/xiaomi.js#L1500-L1532
End point 1 (basic cluster i think) and attribute `0x0125` `0x01` or `0x02` with manufacturerCode: `0x115f`

I dont knowing if its using more endpoint for sending commands then the 1 or the rest is being used at all.

SergiTello, 2021-11-26:
Hi! I am exactly the same as you, model b28ac1, CC2531 and I can only do an automation that applies to the two buttons, there is no way to be able to do different automation with the buttons.
It also does not detect double press or long press ...

Have you been able to find something else?

`{
    "event_type": "zha_event",
    "data": {
        "device_ieee": "54:ef:44:10:00:1a:e8:4b",
        "unique_id": "54:ef:44:10:00:1a:e8:4b:1:0x0006",
        "device_id": "dd6ee77957cc4d2c1ccbe1f76d59d14g",
        "endpoint_id": 1,
        "cluster_id": 6,
        "command": "toggle",
        "args": []
    },
    "origin": "LOCAL",
    "time_fired": "2021-11-26T17:58:37.452616+00:00",
    "context": {
        "id": "41740772fb9801e9c719797000c7fe4d",
        "parent_id": null,
        "user_id": null
    }
}`

I can use this automatization: 

`platform: event
event_type: zha_event
event_data:
  device_id: dd6ee77957cc4d2c1ccbe1f76d59d14g
  endpoint_id: 1`

NickDevoctomy, 2021-12-20:
Would be great to get this working with ZHA, is it on the roadmap?

xuxiaoqiao, 2021-12-25:
I recently bought an Aqara H1 wireless switch single rocker version (`lumi.remote.b18ac1`), and might be able to help.

Here's my custom_quirk file (https://gist.github.com/xuxiaoqiao/6ed8dcb950f26337218d0507733970a0) which works for me, and it might be a starting point for the double rocker version.

I'm a bit confused about the "operation mode" setting (the `0x9` in `0xfcc0` cluster), and the LO/HA mode.
I don't think I've ever observed the "LO" behavior. No matter what value I set to `0x9`, it never sends the toggle command. Instead, it sends click events via `MultistateInputCluster` just like AqaraOpple switch.
This is different from the behavior described by @pavaroto . And I don't know why.

I'm using Conbee 2 with ZHA. Maybe it's relevant because it seems aqara devices "require the coodinator to have a Xiaomi manufacturer code to eventually reveal their full set of clusters" ([source](https://github.com/dresden-elektronik/deconz-rest-plugin/issues/4842#issuecomment-845438855)), or maybe my remote has a different firmware version.

xuxiaoqiao, 2021-12-26:
Hey everyone,

I've finished my [prototype zhaquirk](https://gist.github.com/xuxiaoqiao/6ed8dcb950f26337218d0507733970a0) which (hopefully) will work with WRS-R02. 

Just in case you don't have experience with custom zha quirks before, here are the steps:

1. create a `custom_zha_quirks` folder under the home assistant config folder
2. add the configurations below to enable zha quirks in the home assistant `configuration.yaml`
```
zha:
  enable_quirks: True
  custom_quirks_path: custom_zha_quirks
```
3. put the `aqara_h1_remote.py` https://gist.github.com/xuxiaoqiao/6ed8dcb950f26337218d0507733970a0#file-aqara_h1_remote-py in the `custom_zha_quirks` folder
4. reload the ZHA home assistant integration, or just reboot the whole home assistant
5. Home Assistant ➡️ Configuration ➡️ Devices & Services ➡️ Find the H1 Wireless Remote Device. You should see `Quirk: aqara_h1_remote.RemoteH1DoubleRocker` in the Device Info section. If it's not the case, maybe I've made some mistake in the quirk signature.
6. (Just to be safe) remove the H1 remote device and pair again.
7. Now in Home Assistant ➡️ Developer Tools ➡️ Events ➡️ Listen to events, fill zha_event and press start listening. When you press the Aqara H1 remote, you should see something.

8. Then you might want to set the H1 Remote to your desired mode:
   1. Home Assistant ➡️ Configuration ➡️ Devices & Services ➡️ (Find your H1 Remote device) ➡️ Manage Clusters ➡️ Select `AqaraRemoteManuSpecificCluster (Endpoint id: 1, Id: 0xfcc0, Type: in)` ➡️ See "Cluster Attributes"
   2. For `operation_mode (id: 0x0009)`: 0 means "command" mode, aka LO (Light and occupancy) mentioned by @MattWestb . 1 means "event" mode, aka "sending clicks". Fill your desired value, set "Manufacturer Code Override" to 4447, press "Set Zigbee Attribute". You may need to press H1 remote five times quickly to wake it up.
      * in the "command" mode, the remote will send commands like "toggle" or "off". You can bind the device with a light (just like ikea tradfri dimmers). Pro: when you bind to a specific light, the button works even when home assistant is down. Con: Limited customization.
      * in the "event" mode, the remote will send generic click events to HA, and HA is responsible for the automation. When HA is down, the automation is down. But it allows more customization. I'm using this mode.
   3. For `click_mode (id: 0x0125)`, set it similarly.
      * value `1` means single click mode. This mode has lower delay but only detects single click events. This is the default mode after factor reset.
      * value `2` means multiple click mode. This mode has a slightly higher delay but detects single/double/triple click and long press. I'm using this mode.

9. Done. When you press the button, a `zha_event` will be triggered, which can be used in automation.

cray19003, 2021-12-26:
> Hey everyone,
> 
> I've finished my [prototype zhaquirk](https://gist.github.com/xuxiaoqiao/6ed8dcb950f26337218d0507733970a0) which (hopefully) will work with WRS-R02.
> 
> Just in case you don't have experience with custom zha quirks before, here are the steps:

Thank you very much for sharing your work. Unfortunately I was not able to integrate my double b28ac1. It still appears as LUMI lumi.remote.b28ac1 and I can not find the right Cluster 
(AqaraRemoteManuSpecificCluster). I am not sure that the quirks are working. Is there another option to pu in my config file ?
Does anyone have the same problem or can help me ?
Thanks.
Cedric
______________
Here is my config.yaml

`# Configure a default setup of Home Assistant (frontend, api, etc)
default_config:

group: !include groups.yaml
automation: !include automations.yaml
script: !include scripts.yaml
scene: !include scenes.yaml
#cover: !include cover.yaml

http:
  ssl_certificate: /ssl/fullchain.pem
  ssl_key: /ssl/privkey.pem
  ip_ban_enabled: true
  login_attempts_threshold: 5

sensor:
  - platform: systemmonitor
    resources:
      - type: disk_use_percent
        arg: /config
      - type: memory_free

# Text to speech
tts:
  - platform: google_translate
    language: 'fr'
    base_url: https://*******

recorder:
  db_url: mysql://******

alexa:
  smart_home:

zha:
  enable_quirks: True
  custom_quirks_path: custom_zha_quirks

`

xuxiaoqiao, 2021-12-28:
@cray19003 thanks for helping test it!

I'm not sure why it's not loading. If you have already create the `aqara_h1_remote.py` it should be loaded automatically. Here're a few questions that might help. You can also ping me on the HA discord (I'm xiaoqiao#8505 )

1. Just to confirm, is your device signature the same as what @pavaroto has [posted above](https://github.com/zigpy/zha-device-handlers/issues/940#issue-930700124)? You can see the signature at "Device -> Zigbee Device Signature".

<details><summary>signature (click to expand)</summary>
<pre>
{
  "node_descriptor": "NodeDescriptor(byte1=2, byte2=64, mac_capability_flags=128, manufacturer_code=4447, maximum_buffer_size=127, maximum_incoming_transfer_size=100, server_mask=11264, maximum_outgoing_transfer_size=100, descriptor_capability_field=0, *allocate_address=True, *complex_descriptor_available=False, *is_alternate_pan_coordinator=False, *is_coordinator=False, *is_end_device=True, *is_full_function_device=False, *is_mains_powered=False, *is_receiver_on_when_idle=False, *is_router=False, *is_security_capable=False, *is_valid=True, *logical_type=<LogicalType.EndDevice: 2>, *user_descriptor_available=False)",
  "endpoints": {
    "1": {
      "profile_id": 260,
      "device_type": "0x0103",
      "in_clusters": [
        "0x0000",
        "0x0001",
        "0x0003"
      ],
      "out_clusters": [
        "0x0003",
        "0x0006"
      ]
    },
    "2": {
      "profile_id": null,
      "device_type": "",
      "in_clusters": [],
      "out_clusters": []
    },
    "3": {
      "profile_id": 260,
      "device_type": "0x0103",
      "in_clusters": [
        "0x0003"
      ],
      "out_clusters": [
        "0x0006"
      ]
    },
    "4": {
      "profile_id": null,
      "device_type": "",
      "in_clusters": [],
      "out_clusters": []
    },
    "5": {
      "profile_id": null,
      "device_type": "",
      "in_clusters": [],
      "out_clusters": []
    },
    "6": {
      "profile_id": null,
      "device_type": "",
      "in_clusters": [],
      "out_clusters": []
    }
  },
  "manufacturer": "LUMI",
  "model": "lumi.remote.b28ac1",
  "class": "zigpy.device.Device"
}
</pre>
</details>

2. Can you set log level to debug and see what's happening?
<details>
<summary>Set log level to debug in configration.yaml by adding the following block and reboot (click to expand)</summary>
<pre>
logger:
  default: warning
  logs:
    homeassistant.components.zha: debug
    bellows.zigbee.application: debug
    bellows.ezsp: debug
    zigpy: debug
    zigpy_deconz.zigbee.application: debug
    zigpy_deconz.api: debug
    zigpy_xbee.zigbee.application: debug
    zigpy_xbee.api: debug
    zigpy_zigate: debug
    zigpy_znp: debug
    zhaquirks: debug
</pre>
</details>

For me, it's working, and I can find the following content in the log, if yours doesn't work, it'll be good to know at which step it went wrong.

```
DEBUG (MainThread) [zhaquirks] Loading custom quirks from custom_zha_quirks
DEBUG (MainThread) [zhaquirks] Loading custom quirks module aqara_h1_remote
...
DEBUG (MainThread) [zigpy.quirks.registry]  Checking quirks for LUMI lumi.remote.b18ac1 (xx:xx:xx:xx:xx:xx:xx:xx)
DEBUG (MainThread) [zigpy.quirks.registry]  Considering <class 'aqara_h1_remote.RemoteH1SingleRocker'>
DEBUG (MainThread) [zigpy.quirks.registry]  Found custom device replacement for xx:xx:xx:xx:xx:xx:xx:xx: <class 'aqara_h1_remote.RemoteH1SingleRocker'>
```

PaulMofers, 2021-12-28:
The same for me, i am not getting "Quirk: aqara_h1_remote.RemoteH1DoubleRocker". The device info only shows lumi.remote.b28ac1

mmatecki, 2021-12-28:
This is my device signature:
```
{
  "node_descriptor": "NodeDescriptor(logical_type=<LogicalType.EndDevice: 2>, complex_descriptor_available=0, user_descriptor_available=0, reserved=0, aps_flags=0, frequency_band=<FrequencyBand.Freq2400MHz: 8>, mac_capability_flags=<MACCapabilityFlags.AllocateAddress: 128>, manufacturer_code=4447, maximum_buffer_size=127, maximum_incoming_transfer_size=100, server_mask=11264, maximum_outgoing_transfer_size=100, descriptor_capability_field=<DescriptorCapability.NONE: 0>, *allocate_address=True, *is_alternate_pan_coordinator=False, *is_coordinator=False, *is_end_device=True, *is_full_function_device=False, *is_mains_powered=False, *is_receiver_on_when_idle=False, *is_router=False, *is_security_capable=False)",
  "endpoints": {
    "1": {
      "profile_id": 260,
      "device_type": "0x0103",
      "in_clusters": [
        "0x0000",
        "0x0001",
        "0x0003"
      ],
      "out_clusters": [
        "0x0003",
        "0x0006"
      ]
    },
    "3": {
      "profile_id": 260,
      "device_type": "0x0103",
      "in_clusters": [
        "0x0003"
      ],
      "out_clusters": [
        "0x0006"
      ]
    }
  },
  "manufacturer": "LUMI",
  "model": "lumi.remote.b28ac1",
  "class": "zigpy.device.Device"
}
```

So I had to modify the quirk a little bit: remove in `signature` and `replacement` keys 2,4,5 and 6. 

Now I have I think the last issue. I'm sending successfully to `AqaraRemoteManuSpecificCluster` -> `selected_cluster: click_mode`, `value: 1`, `manufacturer_code_override: 4447` but still in Developers->Events when I push button I can see only `toggle` zha_event. Any advise how to have sole it?

xiaoqiaoxu-db, 2021-12-28:
> @mmatecki I'm sending successfully to AqaraRemoteManuSpecificCluster -> selected_cluster: click_mode, value: 1, manufacturer_code_override: 4447 but still in Developers->Events when I push button I can see only toggle zha_event

When`click_mode=1 and operation_mode=0`, only toggle zha_event will be sent. I don’t know what’s your desired behavior but you probably want to set operation mode to 1? And if you need double/triple/long click you’ll want to set click_mode to 2

xuxiaoqiao, 2021-12-28:
> @mmatecki I'm sending successfully to AqaraRemoteManuSpecificCluster -> selected_cluster: click_mode, value: 1, manufacturer_code_override: 4447 but still in Developers->Events when I push button I can see only toggle zha_event

When`click_mode=1 and operation_mode=0`, only toggle zha_event will be sent. I don’t know what’s your desired behavior but you probably want to set operation mode to 1? And if you need double/triple/long click you’ll want to set click_mode to 2

xuxiaoqiao, 2021-12-28:
FYI here's the behavior of my aqara h1 single button remote under different settings.

* `click_mode=1 and operation_mode=0`
   * single click: sends `toggle` zha_event
* `click_mode=2 and operation_mode=0`
   * single click: `toggle`
   * double click: `off`
* `click_mode=1 and operation_mode=1`
  * single click: `1_single` zha_event
* `click_mode=2 and operation_mode=1`
  * single click: `1_single` zha_event
  * double click: `1_double` zha_event
  * triple click: `1_triple` zha_event
  * long press: `1_hold` zha_event

mmatecki, 2021-12-28:
Thank you @xuxiaoqiao. Below you can find payloads. I have one more issue - the only left button is working. The right one is not generating any events for options when in payload you can see button in args.

click_mode: 1, operation_mode: 0

```
{
    "event_type": "zha_event",
    "data": {
        "device_ieee": "54:ef:44:10:00:2a:32:32",
        "unique_id": "54:ef:44:10:00:2a:32:32:1:0x0006",
        "device_id": "e8426aef5111750da533e3aa802f4407",
        "endpoint_id": 1,
        "cluster_id": 6,
        "command": "toggle",
        "args": []
    },
    "origin": "LOCAL",
    "time_fired": "2021-12-28T18:13:58.981917+00:00",
    "context": {
        "id": "50eed57c846823fe11e662e2f68645b7",
        "parent_id": null,
        "user_id": null
    }
}
```

click_mode: 2, operation_mode: 0

```
{
    "event_type": "zha_event",
    "data": {
        "device_ieee": "54:ef:44:10:00:2a:32:32",
        "unique_id": "54:ef:44:10:00:2a:32:32:1:0x0006",
        "device_id": "e8426aef5111750da533e3aa802f4407",
        "endpoint_id": 1,
        "cluster_id": 6,
        "command": "toggle",
        "args": []
    },
    "origin": "LOCAL",
    "time_fired": "2021-12-28T18:14:41.799728+00:00",
    "context": {
        "id": "1556b026018efcb25d059bc8ae47b113",
        "parent_id": null,
        "user_id": null
    }
}
```

no off command on double click.

click_mode: 1, operation_mode: 1

```
{
    "event_type": "zha_event",
    "data": {
        "device_ieee": "54:ef:44:10:00:2a:32:32",
        "unique_id": "54:ef:44:10:00:2a:32:32:1:0x0012",
        "device_id": "e8426aef5111750da533e3aa802f4407",
        "endpoint_id": 1,
        "cluster_id": 18,
        "command": "1_single",
        "args": {
            "button": 1,
            "press_type": "single",
            "attr_id": 85,
            "value": 1
        }
    },
    "origin": "LOCAL",
    "time_fired": "2021-12-28T18:10:32.077832+00:00",
    "context": {
        "id": "79e96481cc43de077344fea9532ba65d",
        "parent_id": null,
        "user_id": null
    }
}
```

click_mode: 2, operation_mode: 1

```
{
    "event_type": "zha_event",
    "data": {
        "device_ieee": "54:ef:44:10:00:2a:32:32",
        "unique_id": "54:ef:44:10:00:2a:32:32:1:0x0012",
        "device_id": "e8426aef5111750da533e3aa802f4407",
        "endpoint_id": 1,
        "cluster_id": 18,
        "command": "1_single",
        "args": {
            "button": 1,
            "press_type": "single",
            "attr_id": 85,
            "value": 1
        }
    },
    "origin": "LOCAL",
    "time_fired": "2021-12-28T18:12:13.290049+00:00",
    "context": {
        "id": "2dae12a19e5985fa449535d96c222510",
        "parent_id": null,
        "user_id": null
    }
}
```

```
{
    "event_type": "zha_event",
    "data": {
        "device_ieee": "54:ef:44:10:00:2a:32:32",
        "unique_id": "54:ef:44:10:00:2a:32:32:1:0x0012",
        "device_id": "e8426aef5111750da533e3aa802f4407",
        "endpoint_id": 1,
        "cluster_id": 18,
        "command": "1_double",
        "args": {
            "button": 1,
            "press_type": "double",
            "attr_id": 85,
            "value": 2
        }
    },
    "origin": "LOCAL",
    "time_fired": "2021-12-28T18:11:34.563156+00:00",
    "context": {
        "id": "3efc35ca0ebea44d6d22758ba6c9d0fe",
        "parent_id": null,
        "user_id": null
    }
}
```

```
{
    "event_type": "zha_event",
    "data": {
        "device_ieee": "54:ef:44:10:00:2a:32:32",
        "unique_id": "54:ef:44:10:00:2a:32:32:1:0x0012",
        "device_id": "e8426aef5111750da533e3aa802f4407",
        "endpoint_id": 1,
        "cluster_id": 18,
        "command": "1_triple",
        "args": {
            "button": 1,
            "press_type": "triple",
            "attr_id": 85,
            "value": 3
        }
    },
    "origin": "LOCAL",
    "time_fired": "2021-12-28T18:12:33.824133+00:00",
    "context": {
        "id": "c0b3e77599a3cf9a42fbc83da614231d",
        "parent_id": null,
        "user_id": null
    }
}
```
```
{
    "event_type": "zha_event",
    "data": {
        "device_ieee": "54:ef:44:10:00:2a:32:32",
        "unique_id": "54:ef:44:10:00:2a:32:32:1:0x0012",
        "device_id": "e8426aef5111750da533e3aa802f4407",
        "endpoint_id": 1,
        "cluster_id": 18,
        "command": "1_hold",
        "args": {
            "button": 1,
            "press_type": "hold",
            "attr_id": 85,
            "value": 0
        }
    },
    "origin": "LOCAL",
    "time_fired": "2021-12-28T18:12:53.595980+00:00",
    "context": {
        "id": "f046affc742eef3660b731925ddebe0b",
        "parent_id": null,
        "user_id": null
    }
}
```

xuxiaoqiao, 2021-12-28:
@mmatecki Great, seems we've made big progress. Thank you for your time.

My guess is that we should add `MultistateInputCluster` to endpoint 2.
I probably need to look at a few examples first to figure out implementation details. But it shouldn't be too hard.

BTW, it seems if you press two buttons simultaneously in `click_mode: 2, operation_mode: 1`, it might trigger events from endpoint 3.

xuxiaoqiao, 2021-12-28:
Can you try https://gist.githubusercontent.com/xuxiaoqiao/6ed8dcb950f26337218d0507733970a0/raw/7861c08cc19893b7b7d2fd2a56400d76d72df7da/aqara_h1_remote_exp1.py and see if it works? @mmatecki

mmatecki, 2021-12-28:
@xuxiaoqiao works perfectly well! Thank you.

Event for the left button:

```
{
    "event_type": "zha_event",
    "data": {
        "device_ieee": "54:ef:44:10:00:2a:32:32",
        "unique_id": "54:ef:44:10:00:2a:32:32:1:0x0012",
        "device_id": "e8426aef5111750da533e3aa802f4407",
        "endpoint_id": 1,
        "cluster_id": 18,
        "command": "1_single",
        "args": {
            "button": 1,
            "press_type": "single",
            "attr_id": 85,
            "value": 1
        }
    },
    "origin": "LOCAL",
    "time_fired": "2021-12-28T20:08:03.390331+00:00",
    "context": {
        "id": "138605fe76db9b7bc5a762e457d2c505",
        "parent_id": null,
        "user_id": null
    }
}
```

Event for the right button:

```
{
    "event_type": "zha_event",
    "data": {
        "device_ieee": "54:ef:44:10:00:2a:32:32",
        "unique_id": "54:ef:44:10:00:2a:32:32:2:0x0012",
        "device_id": "e8426aef5111750da533e3aa802f4407",
        "endpoint_id": 2,
        "cluster_id": 18,
        "command": "2_single",
        "args": {
            "button": 2,
            "press_type": "single",
            "attr_id": 85,
            "value": 1
        }
    },
    "origin": "LOCAL",
    "time_fired": "2021-12-28T20:08:19.047120+00:00",
    "context": {
        "id": "a22ebe6a00391050ada8af61b19111a3",
        "parent_id": null,
        "user_id": null
    }
}
```

griphos, 2021-12-29:
Hi,
I got the double button version and your quirk works fine, however i can't seem to change click mode and operation mode. I only get single click on each button.

And for those who wonder how to create an automation with those events, below is an example to toggle a light with the left button.
Replace the device_ieee, unique_id and device_id with the values you get when listening for a zha_event.
The button action is also found is the zha_event ( as seen in the post above )

```yaml
- id: '1640791149833'
  alias: H1 Right button
  description: ''
  trigger:
  - platform: event
    event_type: zha_event
    event_data:
      device_ieee: 54:ef:44:10:00:1b:09:dd
      unique_id: 54:ef:44:10:00:1b:09:dd:2:0x0012
      device_id: dea4120a7d912d1eafa097413bba89f5
  condition:
  - condition: template
    value_template: '{{ trigger.event.data.command == ''2_single'' }}'
  action:
  - type: toggle
    device_id: 62f7271352ef226d301c78ec023524e9
    entity_id: light.shelly_shdm_2_40f5200038b7
    domain: light
  mode: single
```
[image]

xuxiaoqiao, 2021-12-29:
> however i can't seem to change click mode and operation mode. @griphos 

Did you quickly press the button five times when setting mode?

xuxiaoqiao, 2021-12-29:
Hey everyone, I have an update and a request.

First, I've just added `device_automation_triggers` section ([ latest quirk](https://gist.github.com/xuxiaoqiao/6ed8dcb950f26337218d0507733970a0#file-aqara_h1_remote-py) ). Which means you can set "Device" triggers like below
[image]

**Please let me know if it works or not. I need more feedback before I create a PR.**

---

And can anyone help confirm if the H1 remote supports "both buttons pressed at the same time" event? And if it does, does it support single/double/triple/long press? You might want to set mode to `click_mode: 2, operation_mode: 1`

griphos, 2021-12-30:
Hi @xuxiaoqiao I pressed 5 time but only for the action, i had to press 5 time before applying each settings. Now it's fine

I installed your new quirk and i see all the trigger for the device.
I tested:
- right button, 1 press, double click, triple click, long press => Everything work
- left button, 1 press, double click, triple click, long press => Ok as well
- double button, 1 press, double click => Ok, so i assumed the triple and long press where ok too

Thanks for your work! As it is now, unless the action and click mode can be set with a simple button instead of the cluster managment, i don't know what kind of improvment can be done =)

I'm happy to finally be able to use the aqara double rocker

NickDevoctomy, 2021-12-30:
That's awesome news, I noticed the email notifications, you guys put a lot of work into this.  Thanks a million, one last thing though, for the less homeassistant savvy amongst us, could you please summarise how to get to that state of full functionality as you have just mentioned?  i.e. installation to configuration.

Sorry to be a pain, I'd like to try it out too but as there are so many messages here I'm not entirely sure what to be doing.

xuxiaoqiao, 2021-12-30:
@NickDevoctomy Follow the instruction at https://github.com/zigpy/zha-device-handlers/issues/940#issuecomment-1001107154 . It has been updated.

xuxiaoqiao, 2021-12-30:
> As it is now, unless the action and click mode can be set with a simple button instead of the cluster managment, i don't know what kind of improvment can be done =) @griphos 

Agree, I wish I can make the device mode setting process easier. But based on my knowledge, unless we make dramatic changes to ZHA/zigpy/zha-device-handler stack, this is what we can have at best.

griphos, 2021-12-30:
A workourand could be, if it is possible via a quirk, to add more information on the device page. You could then display the instructions to change click mode and action.

xuxiaoqiao, 2021-12-31:
> if it is possible via a quirk, to add more information on the device page. @griphos

I'm not aware of any APIs that could allow me to do this in zhaquirks. I've asked this question https://github.com/zigpy/zha-device-handlers/pull/1243 .

One thing I can think of: add a link in https://zigbee.blakadder.com/Aqara_WRS-R02.html and point to the instructions. Hopefully, users would search in the blakadder's website and be aware of this.

MattWestb, 2021-12-31:
For the moment its only possible making "new" Zigbee cluster for adding functions in ZHA.
We have doing it with tuya TRVs for adding functions like local temperature compensation an switches for ati calk and window open detection on/off but its not so nice.
@dmulcahey have stating implanting some "default" functions for using attributes / commands in ZHA (I think power on state and more things) but its for "real" ZCL device. 
It shall being possible making the same with not normal attributes like Xiaomi and tuya is having dome but i think only the devs is knowing what is coming in the pipe and how its possible adopting it for no standard attributes.

Wanted function that normal user need having in the GUI is power on state and light level and perhaps power on color / CT.
Switch modes for devices that can changing how switches is working (switching between 2 or more modes) (Used by tuya, Philips HUE, Xiaomi and likely more devices).

NickDevoctomy, 2021-12-31:
> @NickDevoctomy Follow the instruction at [#940 (comment)](https://github.com/zigpy/zha-device-handlers/issues/940#issuecomment-1001107154) . It has been updated.

awesome, many thanks, I'll give it a try.

OriolOMM, 2022-01-01:
Hi,

Thanks for that great work! Have H! with 2 buttons. I have been able to change the operation mode, but I'm unable to change click mode. Should I use manufacturer code 4447 also for click mode? Should I click 5 times one button and then click "Set Zigbee Attribute"?

Thanks!

xuxiaoqiao, 2022-01-01:
@OriolOMM yes. You need 4447 and quickly pressing 5 times.

OriolOMM, 2022-01-02:
I have this:

[image]

And I click 5 times one button on the switch and then click on Set Zigbee attribute, but I don't know why is not working. 

Looks like that now it is configured with click_mode=1 and operation_mode=1.

I'm unable to go back to operation_mode=0 or set value 2 to click_mode.

When I click 5 times on the button I can see this on the event listener:

{
    "event_type": "zha_event",
    "data": {
        "device_ieee": "54:ef:44:10:00:06:60:b0",
        "unique_id": "54:ef:44:10:00:06:60:b0:1:0x0000",
        "device_id": "744a6b531244895f9cde4280a4389c43",
        "endpoint_id": 1,
        "cluster_id": 0,
        "command": "attribute_updated",
        "args": {
            "attribute_id": 5,
            "attribute_name": "model",
            "value": "lumi.remote.b28ac1"
        }
    },
    "origin": "LOCAL",
    "time_fired": "2022-01-02T11:22:00.621304+00:00",
    "context": {
        "id": "cbd09df2c84300b040033277c5b3876e",
        "parent_id": null,
        "user_id": null
    }
}

Regards

griphos, 2022-01-02:
Did the two button flashed blue after the 5 press?
In my case set the action then the click mode.
I did it like this;
-I prepared the cluster command for action ( like you did for click mode )
- Pressed a button five time
- The two button flashed blue
- I clicked on "Set attribute" and the set attribute button turned green instantaneously ( if it is not and you have to wait more than 2 seconds for the "set attribute" button to turn green, that's a tell sign that something went wrong )

And i repeated the exact same process for click mode
-I prepared the cluster command for click mode
- Pressed a button five time
- The two button flashed blue
- I clicked on "Set attribute" and the set attribute button turned green instantaneously

upkanet, 2022-01-03:
Might this quirk be added to home assistant ZHA one day (if merged) or is it necessary to add it with a custom quirk folder anyway ?

Thanks for this solution !

xuxiaoqiao, 2022-01-03:
@upkanet Yes. After #1243 is merged and HA bumps its dependency library version, we won't need to add this custom quirk manually.

PaulMofers, 2022-01-05:
I can not get this one to work

DEBUG (MainThread) [zigpy.quirks.registry] Checking quirks for LUMI lumi.remote.b28ac1 (54:ef:44:10:00:2a:21:35)
DEBUG (MainThread) [zigpy.quirks.registry] Considering <class 'aqara_h1_remote.RemoteH1DoubleRocker'>
DEBUG (MainThread) [zigpy.quirks.registry] Fail because endpoint list mismatch: {1, 3} {1, 2, 3}

any ideas?

xuxiaoqiao, 2022-01-05:
@PaulMofers whats your “Zigbee Device Signature”? And what’s your Zigbee hub?

PaulMofers, 2022-01-05:
Device Signature
{
  "node_descriptor": "NodeDescriptor(logical_type=<LogicalType.EndDevice: 2>, complex_descriptor_available=0, user_descriptor_available=0, reserved=0, aps_flags=0, frequency_band=<FrequencyBand.Freq2400MHz: 8>, mac_capability_flags=<MACCapabilityFlags.AllocateAddress: 128>, manufacturer_code=4447, maximum_buffer_size=127, maximum_incoming_transfer_size=100, server_mask=11264, maximum_outgoing_transfer_size=100, descriptor_capability_field=<DescriptorCapability.NONE: 0>, *allocate_address=True, *is_alternate_pan_coordinator=False, *is_coordinator=False, *is_end_device=True, *is_full_function_device=False, *is_mains_powered=False, *is_receiver_on_when_idle=False, *is_router=False, *is_security_capable=False)",
  "endpoints": {
    "1": {
      "profile_id": 260,
      "device_type": "0x0103",
      "in_clusters": [
        "0x0000",
        "0x0001",
        "0x0003"
      ],
      "out_clusters": [
        "0x0003",
        "0x0006"
      ]
    },
    "2": {
      "profile_id": 260,
      "device_type": "0x0103",
      "in_clusters": [
        "0x0003"
      ],
      "out_clusters": [
        "0x0003",
        "0x0006"
      ]
    },
    "3": {
      "profile_id": 260,
      "device_type": "0x0103",
      "in_clusters": [
        "0x0003"
      ],
      "out_clusters": [
        "0x0006"
      ]
    }
  },
  "manufacturer": "LUMI",
  "model": "lumi.remote.b28ac1",
  "class": "zigpy.device.Device"
}

zigbee coordinator 
ZNP = Texas Instruments Z-Stack ZNP protocol: CC253x, CC26x2, CC13x2
door ZHA

xuxiaoqiao, 2022-01-06:
@PaulMofers Thanks, I'll update the signature accordingly later. I'm a bit tied up with other stuff right now so I'm aiming to reach back this weekend. 

In the meantime, can you capture the SimpleDescriptor log entries for each endpoint on the device, as is described in https://github.com/zigpy/zha-device-handlers#contribution-guidelines ?

> Capture the SimpleDescriptor log entries for each endpoint on the device. These can be found in the HA logs after joining a device and they look like this: `<SimpleDescriptor endpoint=1 profile=260 device_type=1026 device_version=0 input_clusters=[0, 1, 3, 32, 1026, 1280, 2821] output_clusters=[25]>`(*). This information can also be obtained from the zigbee.db if you want to take the time to query the tables and reconstitute the log entry. I find it easier to just remove and rejoin the device. ZHA entity ids are stable for the most part so it shouldn't disrupt anything you have configured. These need to match what the device reports EXACTLY or zigpy will not match them when a device joins and the handler will not be used for the device. You can also obtain this information from the device screen in HA for the device. The Zigbee Device Signature button will launch a dialog that contains all of the information necessary to create quirks.

(*): On my conbee2 it's something like `SizePrefixedSimpleDescriptor(endpoint=1, profile=260, device_type=259, device_version=1, input_clusters=[0, 3, 1], output_clusters=[3, 6])`

I think I'm supposed to include this information for the pull request.

PaulMofers, 2022-01-07:
i do not even see the device sending events when i listen to zha_event, but no idea why, in the past it did work. Now clueless on how to continue with this remote switch (btw, i have 2 of them and they both behave the same)

xuxiaoqiao, 2022-01-08:
@PaulMofers can you try this updated version of quirk? https://gist.github.com/xuxiaoqiao/6ed8dcb950f26337218d0507733970a0#file-aqara_h1_remote-py

PaulMofers, 2022-01-10:
Yess, that did it, thanks!

PaulMofers, 2022-01-10:
This switch has no hold release action?

xuxiaoqiao, 2022-01-10:
@PaulMofers i don’t think so. I didn’t see any events when I release the button after long press. 

If my memories are correct I don’t think the “release” actually work on zigbee2mqtt either (it seems they fire the release event 4 seconds after long press, regardless of when you actually release).

Actually I don’t think any of the Aqara D1/E1/H1 remotes supports “release” event, given that a 10-second press means factory reset. But correct me if I’m wrong.

PaulMofers, 2022-01-10:
I do not have a D1 or E1, but i have a opple and a sensor switch and they both have long press release

griphos, 2022-01-10:
@xuxiaoqiao Do you need more testing on the new version of your quirk?

xuxiaoqiao, 2022-01-10:
@griphos that would be nice, if you can.  Thank you!

deify, 2022-01-12:
I've received one of the double rocker switches today! I will try to get your quirk up and running the next few days! Nice work and thanks for your efforts 👍

xuxiaoqiao, 2022-01-14:
@deify cool, thanks! I’ve spent a good amount of time searching but I’ve never see anyone who said their D1/E1/H1 supports the “release” event. I’m leaning to believe that the hardware doesn’t support such event.

PaulMofers, 2022-01-14:
> @deify cool, thanks! I’ve spent a good amount of time searching but I’ve never see anyone who said their D1/E1/H1 supports the “release” event. I’m leaning to believe that the hardware doesn’t support such event.
Probably not indeed, these ones can be paired with one of their switches, the other ones have dedicated buttons for pairing.

[167 more replies]

closed 2024-05-29 by TheJulianJES