zigpy/zha-device-handlers #941: [Device Support Request] Calex 421782 Smart Wall Switch, wall mounted RGB controller

Emile86, 2021-06-26:
**Is your feature request related to a problem? Please describe.**
A clear and concise description of what the problem is. Ex. I'm always frustrated when [...]
Device is recognized by ZHA but there are no entities at all and what i can see is not listed as supported device in ZHA but there is support in zigbee2mqtt.

**Describe the solution you'd like**
A clear and concise description of what you want to happen.
Add proper support to be able to use it in automation like On/Off/brightness + -/ White color/ RGB color select and battery status.

**Device signature - this can be acquired by removing the device from ZHA and pairing it again from the add devices screen. Be sure to add the entire content of the log panel after pairing the device to a code block below this line.**

{
  "node_descriptor": "NodeDescriptor(byte1=2, byte2=64, mac_capability_flags=128, manufacturer_code=4509, maximum_buffer_size=127, maximum_incoming_transfer_size=100, server_mask=10752, maximum_outgoing_transfer_size=100, descriptor_capability_field=0, *allocate_address=True, *complex_descriptor_available=False, *is_alternate_pan_coordinator=False, *is_coordinator=False, *is_end_device=True, *is_full_function_device=False, *is_mains_powered=False, *is_receiver_on_when_idle=False, *is_router=False, *is_security_capable=False, *is_valid=True, *logical_type=<LogicalType.EndDevice: 2>, *user_descriptor_available=False)",
  "endpoints": {
    "1": {
      "profile_id": 260,
      "device_type": "0x0810",
      "in_clusters": [
        "0x0000",
        "0x0003"
      ],
      "out_clusters": [
        "0x0003",
        "0x0004",
        "0x0005",
        "0x0006",
        "0x0008",
        "0x0019",
        "0x0300",
        "0x1000"
      ]
    }
  },
  "manufacturer": "Calex ",
  "model": "Smart Wall Switch ",
  "class": "zigpy.device.Device"
}

**Additional context**
Add any other context or screenshots about the feature request here.
[image]

zigbee2mqtt hass already support for this remote
https://www.zigbee2mqtt.io/devices/421782.html

Thank you!

Adminiuga, 2021-06-27:
This is a remote. Remotes do not have entities. Either listen to `zha_events` and use automations based on those, or bind the remote to a group or light you want to control.

Emile86, 2021-06-27:
Yes my mistake.
It's not about entities it about possiblity to select action from automation like : when preset on than do that.
And i was able to make automation in node red based on zha_events but the problem is that al of actions are usable except color select because it is "enhanced_move_to_hue_and_saturation" and what and there is no usable data in it.

Does ZHA support "enhanced_move_to_hue_and_saturation"?

Emile86, 2021-06-27:
Closed by mistake

closed 2021-12-31 by github-actions[bot]