zigpy/zha-device-handlers #949: [Device Support Request] Develco Frient ZHEMI101 Electricity meter reader

KieranDrain, 2021-07-07:
**Is your feature request related to a problem? Please describe.**
Device is supported but only exposes a single entity for Instantaneous measured power in Watts. 

**Describe the solution you'd like**
Support to be added for Energy monitoring in KWh and if possible battery level

**Device signature - this can be acquired by removing the device from ZHA and pairing it again from the add devices screen. Be sure to add the entire content of the log panel after pairing the device to a code block below this line.**
`{
  "node_descriptor": "NodeDescriptor(logical_type=<LogicalType.EndDevice: 2>, complex_descriptor_available=0, user_descriptor_available=0, reserved=0, aps_flags=0, frequency_band=<FrequencyBand.0: 0>, mac_capability_flags=<MACCapabilityFlags.0: 0>, manufacturer_code=0, maximum_buffer_size=0, maximum_incoming_transfer_size=0, server_mask=0, maximum_outgoing_transfer_size=0, descriptor_capability_field=<DescriptorCapability.0: 0>, *allocate_address=False, *is_alternate_pan_coordinator=False, *is_coordinator=False, *is_end_device=True, *is_full_function_device=False, *is_mains_powered=False, *is_receiver_on_when_idle=False, *is_router=False, *is_security_capable=False)",
  "endpoints": {
    "1": {
      "profile_id": 49353,
      "device_type": "0x0001",
      "in_clusters": [
        "0x0002",
        "0x0003",
        "0x0005"
      ],
      "out_clusters": [
        "0x0001"
      ]
    },
    "2": {
      "profile_id": 260,
      "device_type": "0x0007",
      "in_clusters": [
        "0x0000",
        "0x0003",
        "0x0702"
      ],
      "out_clusters": [
        "0x000a"
      ]
    }
  },
  "manufacturer": "Develco",
  "model": "ZHEMI101",
  "class": "zigpy.device.Device"
}`

**Additional context**
zigbee2mqtt appears to support energy monitoring for this device already:
https://www.zigbee2mqtt.io/devices/ZHEMI101.html

Also the OEM manufacturer has documentation on their webpage.
https://www.develcoproducts.com/products/meter-interfaces/external-meter-interface/

KieranDrain, 2021-08-14:
Likely a limitation of ZHA to be addressed by 
https://github.com/home-assistant/core/pull/50471

Hedda, 2021-10-06:
FYI, it looks like support for more clusters attributes is now being worked on for [ZHA integration](https://www.home-assistant.io/integrations/zha) in Home Assistant core, see:

https://github.com/home-assistant/core/pull/56909

and

https://github.com/home-assistant/core/pull/56666

Also, note the last comment from Adminiuga in PR for ZHA in Home Assistant core -> https://github.com/home-assistant/core/pull/56909

"**_Support for other attributes could be extended later, as I couldn't find too many devices other than Sinopé RM3250ZB Electrical Load Controller – 50A – Zigbee https://www.amazon.com/dp/B07CGJ6XH2/ reporting anything interesting on the electrical measurement cluster._**"

So maybe now you could try to get this working with the latest Home Assistant beta or wait for the next upcoming release:

https://www.home-assistant.io/faq/release/

PS: There is much more discussion about the current ZHA limitation of only supporting 1 sensor per cluster here -> https://github.com/zigpy/zha-device-handlers/issues/605

KieranDrain, 2021-10-08:
> 
> 
> FYI, it looks like support for more Zigbee clusters attributes is now being worked on for [ZHA integration](https://www.home-assistant.io/integrations/zha) in Home Assistant core, see:
> 
> [home-assistant/core#56909](https://github.com/home-assistant/core/pull/56909)
> 
> and
> 
> [home-assistant/core#56666](https://github.com/home-assistant/core/pull/56666)
> 
> Also, note the last comment from Adminiuga in PR for ZHA in Home Assistant core -> [home-assistant/core#56909](https://github.com/home-assistant/core/pull/56909)
> 
> "**_Support for other attributes could be extended later, as I couldn't find too many devices other than Sinopé RM3250ZB Electrical Load Controller – 50A – Zigbee https://www.amazon.com/dp/B07CGJ6XH2/ reporting anything interesting on the electrical measurement cluster._**"
> 
> So maybe now you could try to get this working with the latest Home Assistant beta or wait for the next upcoming release:
> 
> https://www.home-assistant.io/faq/release/
> 
> PS: There is much more discussion about the current ZHA limitation of only supporting 1 sensor per cluster here -> #605 and #1045

Thanks - I've been quietly following the developments and appreciate the effort thats gone in to a comprehensive solution rather than the quick fix I had referenced - As soon as the commits end up in the RC branch I think I'll be able to close this bug as I'm not on the beta branch at the moment.

Lyr3x, 2021-12-14:
The device pairing and entity creation were successful, however both sensors do not offer units. As the customizing menu was removed, its not even easy to add correct units