zigpy/zha-device-handlers #953: [Device Support Request] TS0601 _TZE200_yvx5lh6k (Smart Air Box)

depl0y, 2021-07-10:
**Is your feature request related to a problem? Please describe.**
Bought a Smart Air Box which should measure CO2, Formaldehyde, temperature and humidity. It does pair, but it does not expose any status.

**Describe the solution you'd like**
Read the available values.

**Device signature - this can be acquired by removing the device from ZHA and pairing it again from the add devices screen. Be sure to add the entire content of the log panel after pairing the device to a code block below this line.**
```
{
  "node_descriptor": "NodeDescriptor(logical_type=<LogicalType.Router: 1>, complex_descriptor_available=0, user_descriptor_available=0, reserved=0, aps_flags=0, frequency_band=<FrequencyBand.Freq2400MHz: 8>, mac_capability_flags=<MACCapabilityFlags.AllocateAddress|RxOnWhenIdle|MainsPowered|FullFunctionDevice: 142>, manufacturer_code=4098, maximum_buffer_size=82, maximum_incoming_transfer_size=82, server_mask=11264, maximum_outgoing_transfer_size=82, descriptor_capability_field=<DescriptorCapability.0: 0>, *allocate_address=True, *is_alternate_pan_coordinator=False, *is_coordinator=False, *is_end_device=False, *is_full_function_device=True, *is_mains_powered=True, *is_receiver_on_when_idle=True, *is_router=True, *is_security_capable=False)",
  "endpoints": {
    "1": {
      "profile_id": 260,
      "device_type": "0x0051",
      "in_clusters": [
        "0x0000",
        "0x0004",
        "0x0005",
        "0xef00"
      ],
      "out_clusters": [
        "0x000a",
        "0x0019"
      ]
    }
  },
  "manufacturer": "_TZE200_yvx5lh6k",
  "model": "TS0601",
  "class": "zigpy.device.Device"
}
```

**Additional context**
Product: https://www.aliexpress.com/item/1005002511043244.html
Blakadder link: https://zigbee.blakadder.com/Tuya_RSH-AirBox01.html

depl0y, 2021-07-10:
Also, it seems like another variant of this: https://github.com/zigpy/zha-device-handlers/blob/dev/zhaquirks/tuya/air/ts0601_air_quality.py

depl0y, 2021-07-10:
Updating the file mentioned above to include the following line: 

```
 MODELS_INFO: [("_TZE200_8ygsuhe1", "TS0601"), ("_TZE200_yvx5lh6k", "TS0601")],
```

Fixes the issue, and exposes 3 entities, contain the (I presume) right numbers.

depl0y, 2021-07-10:
This is mainly a conversation with myself, I know ;) . But it seems that the box should expose VoC and Formaldehyde as well, but maybe it isn't a straight "just add the model number and be done with it".

[image]

MattWestb, 2021-07-10:
@depl0y Great work done !!
Can you making one PR so other user can using the  device without problems ?

Indeed Z2M have squeezing out little more information from the tuya MCU.
Hopefully the devs can doing the same in ZHA but its depends is they having time looking on it 👍

depl0y, 2021-07-10:
Of course, and I pushed a PR this morning. I have it running now for a couple of hours and everything seems fine. CO2 was the reason why I bought the sensor, so that work, but still, would be cool if all sensors were supported.

MattWestb, 2021-07-10:
Great !!

Our master dev is having one and hi was doing the quirk perhaps hi taking one more look on it then hi having time (For the moment problems with HA 2021.7) but tuya is using there own proprietary protocol for there sensors so its more or less one must having one device for getting them working well.

Thanks for contributing !!

depl0y, 2021-07-10:
No problem, if you (or anyone else) needs any more debugging info, just let me know.

akatsuki-meigetsu, 2021-08-10:
> This is mainly a conversation with myself, I know ;) . But it seems that the box should expose VoC and Formaldehyde as well, but maybe it isn't a straight "just add the model number and be done with it".
> 
> [image]

Thank you very much! Successfully work with my air box. And in my homeassistant(2021.8.4), the entities also include VoC and Formaldehyde, it works very well and no problems until now.

akatsuki-meigetsu, 2021-08-10:
> This is mainly a conversation with myself, I know ;) . But it seems that the box should expose VoC and Formaldehyde as well, but maybe it isn't a straight "just add the model number and be done with it".
> 
> [image]

Thank you very much! Successfully work with my air box. And in my homeassistant(2021.8.4), the entities also include VoC and Formaldehyde, it works very well and no problems until now.

> This is mainly a conversation with myself, I know ;) . But it seems that the box should expose VoC and Formaldehyde as well, but maybe it isn't a straight "just add the model number and be done with it".
> 
> [image]

This is status of my entities. All sensors were supported.
(I just follow your method to do 
[image]
it

Idskov, 2021-08-10:
Worked like a charm.
Thanks for that.
[image]

Idskov, 2021-08-18:
There is a slight problem with the unit of measurement for formaldehyde.
I just crosschecked with 2 identical devices. 1 joined to a factory Tuya Gateway and SmartLife, the other to ZHA via a tasmotized SonOff gateway.

I get very close to the same measurements but it seems the conversions made in  TuyaCO2ManufCluster is fine and actually it seems that SmarLife is converting this wrong from the input in g/m³.
TuyaCO2ManufCluster converts from g/m³ to µg/m³ ("measured_value", lambda x: x * 1e-6,)
Unfortunately only VOC is reported as µg/m³ and formaldehyde (CH20) is reported as unit PPM but without converting the value.

If PPM is the goal then the formula should be something like ("measured_value", lambda x: x * 24450 / 30.026,) to account for molecular weight and an average temperature and pressure.
The easiest and most correct would be to stay with µg/m³

I have yet to understand and trace where this unit-mapping is and whether or not this is just a problem for this device or for the class FormaldehydeConcentration (quite new to python)

Any input is most welcome.

Screenshots below shows the HCHO/CH20 of 0.03 mg/m3 in Tuya = 30µg/m³ = 0.024 ppm but in HA it's shown as 3 ppm = 3.684 mg/m3
Measured_value is 3e-06 = 3 µg/m³ not ppm

[image] [image]

Currently I have just overridden the unit of measurement in HA customizations, but why not get it right by default?
[image]

Adminiuga, 2021-08-18:
In home-assistant/components/zha/sensor.py there should be the formaldehyde sensor

Idskov, 2021-08-18:
> In home-assistant/components/zha/sensor.py there should be the formaldehyde sensor

Yup.
There it is.
Probably not going to get that changed...

[LINK](https://github.com/home-assistant/core/blob/60f8e24bde91d50cdb588fcab66c2a6c5dccd492/homeassistant/components/zha/sensor.py#L337)
```
@STRICT_MATCH(channel_names="formaldehyde_concentration")
class FormaldehydeConcentration(Sensor):
    """Formaldehyde Concentration sensor."""

    SENSOR_ATTR = "measured_value"
    _decimals = 0
    _multiplier = 1e6
    _unit = CONCENTRATION_PARTS_PER_MILLION
```

Idskov, 2021-08-18:
> Yup.
> There it is.
> Probably not going to get that changed...
>
> [LINK](https://github.com/home-assistant/core/blob/60f8e24bde91d50cdb588fcab66c2a6c5dccd492/homeassistant/components/zha/sensor.py#L337) [...]

Confirmed that changing _unit for class FormaldehydeConcentration(Sensor) in /usr/src/homeassistant/homeassistant/components/zha/sensor.py fixes the issue, but it might break other sensors...

MattWestb, 2021-08-18:
In ZCL V7 `4.13.1.3 Cluster Identifiers` is looks like this:

| Cluster Id | Substance Measured | Ambient Substance | Units | Notes (not normative or prescribed) |
|-|-|-|-|-|
| 0x042B | Formaldehyde | Air | Volume | |

And no clue wot units shall being used for the Volume :-(

MattWestb, 2021-08-18:
I think the same its left open end for doing wot you like and that is wrong (and our Chinese friends is very creative imprinting there own things).

I looking in `project-chip` git that have updated some xmls with ZCL V8 if i can finding some more clue wot is expected being handled of the cluster 0x042B.

MattWestb, 2021-08-18:
Have not fund and Silans SDK is only having cluster until 0x0428 so no clue is found.
Then its more how do professional equipment measuring Formaldehyde ?

Idskov, 2021-08-18:
As far as I can tell µg/m³ og mg/m³ is the measured value and PPM is derived from that with most equipment letting you choose between the two.
However to correctly calculate PPM you must include air pressure and temperature in the equation, or an average factor.
As far as I can tell scientific reports and various guidelines, exposure limits and so on primarily use µg/m³ or mg/m³.
See for instance [WHO Guidelines for Indoor Air Quality: Selected Pollutants - 3 Formaldehyde](https://www.ncbi.nlm.nih.gov/books/NBK138711/)

joaoasilva, 2021-11-22:
Trying to add a device with _TZE200_yvx5lh6k but I can only see the voc_level sensor, is there anything I need to do to get the other sensors?
Thanks
[image]

MattWestb, 2021-11-22:
I have seen that ZHA is not loading quirks OK late mounts but then its not getting any extra sensors at all.

Try restarting HA and see if the quirk is making all sensors OK.
If not working delete the sensor, wait one minute and restarting HA and adding the sensor then ZHA is stable.

joaoasilva, 2021-11-23:
That worked! Thanks!
Does anyone have alerts for the levels? I have no idea what's the level I should measure to fire alerts ex like the Nest Protect smoke and co alarm?

closed 2022-05-29 by github-actions[bot]