zigpy/zha-device-handlers #956: [Device Support Request] TS0041 _TZ3000_iklyvnq3  - Zigbee 1-Button Scene Switch

jmlankford, 2021-07-11:
``Is your feature request related to a problem? Please describe.
As is the case with many TUYA scene switches, it pairs, but exposes a battery life entity and a switch entity that is not controlled by the switch.

Describe the solution you'd like
I need an entity exposed that will read when the scene switch is toggled.

Device signature - this can be acquired by removing the device from ZHA and pairing it again from the add devices screen. Be sure to add the entire content of the log panel after pairing the device to a code block below this line.

`{
  "node_descriptor": "NodeDescriptor(logical_type=<LogicalType.EndDevice: 2>, complex_descriptor_available=0, user_descriptor_available=0, reserved=0, aps_flags=0, frequency_band=<FrequencyBand.Freq2400MHz: 8>, mac_capability_flags=<MACCapabilityFlags.AllocateAddress: 128>, manufacturer_code=4098, maximum_buffer_size=82, maximum_incoming_transfer_size=82, server_mask=11264, maximum_outgoing_transfer_size=82, descriptor_capability_field=<DescriptorCapability.0: 0>, *allocate_address=True, *is_alternate_pan_coordinator=False, *is_coordinator=False, *is_end_device=True, *is_full_function_device=False, *is_mains_powered=False, *is_receiver_on_when_idle=False, *is_router=False, *is_security_capable=False)",
  "endpoints": {
    "1": {
      "profile_id": 260,
      "device_type": "0x0000",
      "in_clusters": [
        "0x0000",
        "0x0001",
        "0x0006"
      ],
      "out_clusters": [
        "0x000a",
        "0x0019"
      ]
    }
  },
  "manufacturer": "_TZ3000_iklyvnq3",
  "model": "TS0041",
  "class": "zigpy.device.Device"
}`

Additional context
Product: https://www.aliexpress.com/item/1005002799050710.html
Blakadder (closest I could find): https://zigbee.blakadder.com/Tuya_TS0041.html

MattWestb, 2021-07-11:
Its 95% one new version if the [ts0041_zemismart.py](https://github.com/zigpy/zha-device-handlers/blob/dev/zhaquirks/tuya/ts0041_zemismart.py).
You can trying adding the new ID in the quirk in on local quirk folder like this:
```
"""Tuya Zemismart 1 Button Remote."""

from zigpy.profiles import zha
from zigpy.quirks import CustomDevice
from zigpy.zcl.clusters.general import Basic, OnOff, Ota, PowerConfiguration, Time

from zhaquirks.const import (
    BUTTON_1,
    COMMAND,
    DEVICE_TYPE,
    DOUBLE_PRESS,
    ENDPOINT_ID,
    ENDPOINTS,
    INPUT_CLUSTERS,
    LONG_PRESS,
    MODELS_INFO,
    OUTPUT_CLUSTERS,
    PROFILE_ID,
    SHORT_PRESS,
)
from zhaquirks.tuya import TuyaSmartRemoteOnOffCluster

class TuyaZemismartSmartRemote0041(CustomDevice):
    """Tuya Zemismart 1-button remote device."""

    signature = {
        # SizePrefixedSimpleDescriptor(endpoint=1, profile=260, device_type=0, device_version=1, input_clusters=[0, 1, 6], output_clusters=[25, 10])
        MODELS_INFO: [
            ("_TZ3000_tk3s5tyg", "TS0041"),
            ("_TZ3000_q68478x7", "TS0041"),
            ("_TZ3400_keyjqthh", "TS0041"),
            ("_TZ3000_iklyvnq3", "TS0041"),
        ],
        ENDPOINTS: {
            1: {
                PROFILE_ID: zha.PROFILE_ID,
                DEVICE_TYPE: zha.DeviceType.ON_OFF_SWITCH,
                INPUT_CLUSTERS: [
                    Basic.cluster_id,
                    PowerConfiguration.cluster_id,
                    OnOff.cluster_id,
                ],
                OUTPUT_CLUSTERS: [Ota.cluster_id, Time.cluster_id],
            },
        },
    }
    replacement = {
        ENDPOINTS: {
            1: {
                PROFILE_ID: zha.PROFILE_ID,
                DEVICE_TYPE: zha.DeviceType.REMOTE_CONTROL,
                INPUT_CLUSTERS: [
                    Basic.cluster_id,
                    PowerConfiguration.cluster_id,
                    TuyaSmartRemoteOnOffCluster,
                ],
                OUTPUT_CLUSTERS: [Ota.cluster_id, Time.cluster_id],
            },
        },
    }

    device_automation_triggers = {
        (SHORT_PRESS, BUTTON_1): {ENDPOINT_ID: 1, COMMAND: SHORT_PRESS},
        (LONG_PRESS, BUTTON_1): {ENDPOINT_ID: 1, COMMAND: LONG_PRESS},
        (DOUBLE_PRESS, BUTTON_1): {ENDPOINT_ID: 1, COMMAND: DOUBLE_PRESS},
    }
```

MattWestb, 2021-07-11:
@Adminiuga Its 3 quirk with 99 % the same code (tuya/TS0041X). The change is one type have time as in cluster and the other as out cluster plus one with only one new ID.

Is it good putting all in one quirk and making 2 typed one "time out" and one "time in"  or do you like having multiple quirk for more or less the same device ?

MattWestb, 2021-07-11:
I was looking on the 2, 3 and 4 button quirks and some is made with only model info  for triggering the quirk and then the cluster with time on in or out cluster.
```
MODEL: "TS0042",
```
I think  its very OK then its looks tuya is keeping the switches in with the same model name then one new custom is "making" one new switch and users is getting they new switches working out of the box.

Is it OK doing the same for the TS0041 switches ?

jmlankford, 2021-07-11:
Apologies for the extra comment in this thread. 

The suggestion to use the Zemismart TS0041 Quirk was unsuccessful, and only exposed a battery entity.  I realized that I had already modified a TS0041 quirk for this particular model, and in my original post, it may have provided misleading or incorrect signature information. 

Please see below the information that is presented when NO quirks are modified to match this model:

```
{
  "node_descriptor": "NodeDescriptor(logical_type=<LogicalType.EndDevice: 2>, complex_descriptor_available=0, user_descriptor_available=0, reserved=0, aps_flags=0, frequency_band=<FrequencyBand.Freq2400MHz: 8>, mac_capability_flags=<MACCapabilityFlags.AllocateAddress: 128>, manufacturer_code=4098, maximum_buffer_size=82, maximum_incoming_transfer_size=82, server_mask=11264, maximum_outgoing_transfer_size=82, descriptor_capability_field=<DescriptorCapability.0: 0>, *allocate_address=True, *is_alternate_pan_coordinator=False, *is_coordinator=False, *is_end_device=True, *is_full_function_device=False, *is_mains_powered=False, *is_receiver_on_when_idle=False, *is_router=False, *is_security_capable=False)",
  "endpoints": {
    "1": {
      "profile_id": 260,
      "device_type": "0x0000",
      "in_clusters": [
        "0x0000",
        "0x0001",
        "0x0006"
      ],
      "out_clusters": [
        "0x000a",
        "0x0019"
      ]
    }
  },
  "manufacturer": "_TZ3000_iklyvnq3",
  "model": "TS0041",
  "class": "zigpy.device.Device"
}
```

MattWestb, 2021-07-11:
The line  ` "class": "zigpy.device.Device"` is saying the quirk is not matching and is not being loaded.

MattWestb, 2021-07-11:
You can trying putting in 
```
       MODEL: "TS0041",
```
instead if the block with 
```
        MODELS_INFO: [
            ("_TZ3000_tk3s5tyg", "TS0041"),
            ("_TZ3000_q68478x7", "TS0041"),
            ("_TZ3400_keyjqthh", "TS0041"),
            ("_TZ3000_iklyvnq3", "TS0041"),
        ],
```

jmlankford, 2021-07-11:
Correct - sorry - I purposely did that after removing the model from all quirks just to see if this device would provide any other information if it was 100% unmatched. 

***

I re-modified the quirk you suggested (ts0041_zemismart.py) to include the model number of this device. It comes up with only a battery entity. 

```
 [0x0000:zdo] ZDO request ZDOCmd.Mgmt_Permit_Joining_req: [60, <Bool.false: 0>]
[0xA67C](ZLL Light): Attempting to checkin with device - missed checkins: 1
Device 0x80d3 (60:a4:23:ff:fe:6e:c1:eb) left the network
[0xa67c] Delivery error for seq # 0x3c, on endpoint id 1 cluster 0x0000: message send failure
[0x9337](LWB014): Attempting to checkin with device - missed checkins: 1
[0x05C4](LWB014): Attempting to checkin with device - missed checkins: 1
Device 0x24b7 (60:a4:23:ff:fe:6e:c1:eb) joined the network
Device 60:a4:23:ff:fe:6e:c1:eb changed id (0x24b7 => 0x24b7)
[0x24b7] Scheduling initialization
Tries remaining: 3
[0x24b7] Discovering endpoints
Tries remaining: 3
[0x24b7] Extending timeout for 0x42 request
Device 0x24b7 (60:a4:23:ff:fe:6e:c1:eb) joined the network
[0x24b7] Scheduling initialization
[0x24b7] Canceling old initialize call
Received frame on uninitialized device <Device model=None manuf=None nwk=0x24B7 ieee=60:a4:23:ff:fe:6e:c1:eb is_initialized=False> from ep 0 to ep 0, cluster 19: b'\x84\xb7$\xeb\xc1n\xfe\xff#\xa4`\x80'
[0x24b7:zdo] ZDO request ZDOCmd.Device_annce: [0x24B7, 60:a4:23:ff:fe:6e:c1:eb, 128]
Tries remaining: 3
[0x24b7] Discovering endpoints
Tries remaining: 3
[0x24b7] Extending timeout for 0x44 request
Received frame on uninitialized device <Device model=None manuf=None nwk=0x24B7 ieee=60:a4:23:ff:fe:6e:c1:eb is_initialized=False> from ep 0 to ep 0, cluster 32773: b'B\x00\xb7$\x01\x01'
[0x24b7:zdo] ZDO request ZDOCmd.Active_EP_rsp: [<Status.SUCCESS: 0>, 0x24B7, [1]]
[0x24b7:zdo] Unsupported ZDO request:ZDOCmd.Active_EP_rsp
Received frame on uninitialized device <Device model=None manuf=None nwk=0x24B7 ieee=60:a4:23:ff:fe:6e:c1:eb is_initialized=False> from ep 0 to ep 0, cluster 32773: b'D\x00\xb7$\x01\x01'
[0x24b7] Discovered endpoints: [1]
[0x24b7] Initializing endpoints [<Endpoint id=1 in=[] out=[] status=<Status.NEW: 0>>]
[0x24b7:1] Discovering endpoint information
Tries remaining: 3
[0x24b7] Extending timeout for 0x46 request
Received frame on uninitialized device <Device model=None manuf=None nwk=0x24B7 ieee=60:a4:23:ff:fe:6e:c1:eb is_initialized=False> from ep 0 to ep 0, cluster 32772: b'F\x00\xb7$\x12\x01\x04\x01\x00\x00\x01\x03\x00\x00\x01\x00\x06\x00\x02\x19\x00\n\x00'
[0x24b7:1] Discovered endpoint information: SizePrefixedSimpleDescriptor(endpoint=1, profile=260, device_type=0, device_version=1, input_clusters=[0, 1, 6], output_clusters=[25, 10])
[0x24b7] Extending timeout for 0x48 request
[0x24b7:1:0x0000] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=GLOBAL_COMMAND manufacturer_specific=False is_reply=True disable_default_response=True> manufacturer=None tsn=72 command_id=Command.Read_Attributes_rsp>
[0x24b7] Read model 'TS0041' and manufacturer '_TZ3000_iklyvnq3' from <Endpoint id=1 in=[basic:0x0000, power:0x0001, on_off:0x0006] out=[ota:0x0019, time:0x000A] status=<Status.ZDO_INIT: 1>>
[0x24b7] Discovered basic device information for <Device model='TS0041' manuf='_TZ3000_iklyvnq3' nwk=0x24B7 ieee=60:a4:23:ff:fe:6e:c1:eb is_initialized=True>
Device is initialized <Device model='TS0041' manuf='_TZ3000_iklyvnq3' nwk=0x24B7 ieee=60:a4:23:ff:fe:6e:c1:eb is_initialized=True>
Checking quirks for _TZ3000_iklyvnq3 TS0041 (60:a4:23:ff:fe:6e:c1:eb)
Considering <class 'tuya.ts0041_zemismart.TuyaZemismartSmartRemote0041'>
Found custom device replacement for 60:a4:23:ff:fe:6e:c1:eb: <class 'tuya.ts0041_zemismart.TuyaZemismartSmartRemote0041'>
device - 0x24B7:60:a4:23:ff:fe:6e:c1:eb entering async_device_initialized - is_new_join: True
device - 0x24B7:60:a4:23:ff:fe:6e:c1:eb has joined the ZHA zigbee network
[0x24B7](TS0041): started configuration
[0x24B7:ZDO](TS0041): 'async_configure' stage succeeded
[0x24B7:1:0x0000]: finished channel configuration
[0x24b7] Extending timeout for 0x4a request
[0x24B7:1:0x0019]: finished channel configuration
Error handling '_save_attribute' event with (60:a4:23:ff:fe:6e:c1:eb, 1, 0, 4, '_TZ3000_iklyvnq3') params: FOREIGN KEY constraint failed
Error handling '_save_attribute' event with (60:a4:23:ff:fe:6e:c1:eb, 1, 0, 5, 'TS0041') params: FOREIGN KEY constraint failed
[0x24B7:1:0x0001]: bound 'power' cluster: Status.SUCCESS
[0x24b7] Extending timeout for 0x4c request
[0x24b7:1:0x0001] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=GLOBAL_COMMAND manufacturer_specific=False is_reply=True disable_default_response=True> manufacturer=None tsn=76 command_id=Command.Configure_Reporting_rsp>
[0x24B7:1:0x0001]: reporting 'battery_voltage' attr on 'power' cluster: 3600/10800/1: Result: '[[ConfigureReportingResponseRecord(status=0)]]'
[0x24b7] Extending timeout for 0x4e request
[0x24b7:1:0x0001] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=GLOBAL_COMMAND manufacturer_specific=False is_reply=True disable_default_response=True> manufacturer=None tsn=78 command_id=Command.Configure_Reporting_rsp>
[0x24B7:1:0x0001]: reporting 'battery_percentage_remaining' attr on 'power' cluster: 3600/10800/1: Result: '[[ConfigureReportingResponseRecord(status=0)]]'
[0x24B7:1:0x0001]: finished channel configuration
[0x24B7:1:0x0000]: 'async_configure' stage succeeded
[... cut ...]
[0x24b7] Delivery error for seq # 0x38, on endpoint id 0 cluster 0x0034: message send failure
Sending 'zdo_leave_req' failed: [0x24b7:0:0x0034]: Message send failure
[0x05c4] Delivery error for seq # 0x40, on endpoint id 11 cluster 0x0000: message send failure
[0x1b48] Delivery error for seq # 0x54, on endpoint id 11 cluster 0x0000: message send failure
[0x5ff1:zdo] ZDO request ZDOCmd.NWK_addr_req: [00:0d:6f:00:0b:56:82:81, 0, 0]
[0x5ff1:zdo] ZDO request ZDOCmd.NWK_addr_req: [00:0d:6f:00:0b:56:82:81, 0, 0]
```

jmlankford, 2021-07-11:
> You can trying putting in
>
> ```
>        MODEL: "TS0041",
> ```
>
> instead if the block with
>
> ```
>         MODELS_INFO: [
>             ("_TZ3000_tk3s5tyg", "TS0041"),
>             ("_TZ3000_q68478x7", [...]

Do you mean so it looks like this? I think my syntax is messed up because it's throwing up an error.

```
    signature = {
        # SizePrefixedSimpleDescriptor(endpoint=1, profile=260, device_type=0, device_version=1, input_clusters=[0, 1, 6], output_clusters=[25, 10])
       MODEL: "TS0041",
        ENDPOINTS: {
            1: {
                PROFILE_ID: zha.PROFILE_ID,
                DEVICE_TYPE: zha.DeviceType.ON_OFF_SWITCH,
                INPUT_CLUSTERS: [
                    Basic.cluster_id,
                    PowerConfiguration.cluster_id,
                    OnOff.cluster_id,
                ],
                OUTPUT_CLUSTERS: [Ota.cluster_id, Time.cluster_id],
            },
        },
    }
```

MattWestb, 2021-07-11:
Switches is only exposing battery sensors and the "clicks" is being events that you can see in HA event viewer.

Your device is falling in sleep before its being finished paired so you  was getting timeout errors.

Yes i is trying putting all 3 different quirk to one.

MattWestb, 2021-07-11:
One combined version with both types of cluster configs and only model type:
```
"""Tuya tuya 1 Button Remote."""

from zigpy.profiles import zha
from zigpy.quirks import CustomDevice
from zigpy.zcl.clusters.general import Basic, OnOff, Ota, PowerConfiguration, Time

from zhaquirks.const import (
    BUTTON_1,
    COMMAND,
    DEVICE_TYPE,
    DOUBLE_PRESS,
    ENDPOINT_ID,
    ENDPOINTS,
    INPUT_CLUSTERS,
    LONG_PRESS,
    MODELS_INFO,
    OUTPUT_CLUSTERS,
    PROFILE_ID,
    SHORT_PRESS,
)
from zhaquirks.tuya import TuyaSmartRemoteOnOffCluster

	class TuyaSmartRemote0041TO(CustomDevice):
    """Tuya 1-button remote device with time on out."""

    signature = {
        # SizePrefixedSimpleDescriptor(endpoint=1, profile=260, device_type=0, device_version=1, input_clusters=[0, 1, 6], output_clusters=[25, 10])
        MODEL: "TS0041",
        ENDPOINTS: {
            1: {
                PROFILE_ID: zha.PROFILE_ID,
                DEVICE_TYPE: zha.DeviceType.ON_OFF_SWITCH,
                INPUT_CLUSTERS: [
                    Basic.cluster_id,
                    PowerConfiguration.cluster_id,
                    OnOff.cluster_id,
                ],
                OUTPUT_CLUSTERS: [Ota.cluster_id, Time.cluster_id],
            },
        },
    }
    replacement = {
        ENDPOINTS: {
            1: {
                PROFILE_ID: zha.PROFILE_ID,
                DEVICE_TYPE: zha.DeviceType.REMOTE_CONTROL,
                INPUT_CLUSTERS: [
                    Basic.cluster_id,
                    PowerConfiguration.cluster_id,
                    TuyaSmartRemoteOnOffCluster,
                ],
                OUTPUT_CLUSTERS: [Ota.cluster_id, Time.cluster_id],
            },
        },
    }

    device_automation_triggers = {
        (SHORT_PRESS, BUTTON_1): {ENDPOINT_ID: 1, COMMAND: SHORT_PRESS},
        (LONG_PRESS, BUTTON_1): {ENDPOINT_ID: 1, COMMAND: LONG_PRESS},
        (DOUBLE_PRESS, BUTTON_1): {ENDPOINT_ID: 1, COMMAND: DOUBLE_PRESS},
    }

	class TuyaSmartRemote0041TI(CustomDevice):
    """Tuya 1-button remote device with time on in."""

    signature = {
        # SizePrefixedSimpleDescriptor(endpoint=1, profile=260, device_type=0, device_version=1, input_clusters=[0, 10, 1, 6], output_clusters=[25]))
        MODEL: "TS0041",
        ENDPOINTS: {
            1: {
                PROFILE_ID: zha.PROFILE_ID,
                DEVICE_TYPE: zha.DeviceType.ON_OFF_SWITCH,
                INPUT_CLUSTERS: [
                    Basic.cluster_id,
                    PowerConfiguration.cluster_id,
                    OnOff.cluster_id,
                    Time.cluster_id,
                ],
                OUTPUT_CLUSTERS: [Ota.cluster_id],
            },
        },
    }
    replacement = {
        ENDPOINTS: {
            1: {
                PROFILE_ID: zha.PROFILE_ID,
                DEVICE_TYPE: zha.DeviceType.REMOTE_CONTROL,
                INPUT_CLUSTERS: [
                    Basic.cluster_id,
                    PowerConfiguration.cluster_id,
                    TuyaSmartRemoteOnOffCluster,
                    Time.cluster_id,
                ],
                OUTPUT_CLUSTERS: [Ota.cluster_id],
            },
        },
    }

    device_automation_triggers = {
        (SHORT_PRESS, BUTTON_1): {ENDPOINT_ID: 1, COMMAND: SHORT_PRESS},
        (LONG_PRESS, BUTTON_1): {ENDPOINT_ID: 1, COMMAND: LONG_PRESS},
        (DOUBLE_PRESS, BUTTON_1): {ENDPOINT_ID: 1, COMMAND: DOUBLE_PRESS},
    }
```
It shall working OK if not having more types of remotes then the 2 other is still active in the "HA quirk".

jmlankford, 2021-07-11:
Thanks - I had to be sure to import "MODEL" from zaquirks.const and had to check the indent on line 24 (25 after MODEL is imported.

I re-added and no entities show anymore. This is my log:

`[0x0000:zdo] ZDO request ZDOCmd.Mgmt_Permit_Joining_req: [60, <Bool.false: 0>]
Device 0x24b7 (60:a4:23:ff:fe:6e:c1:eb) left the network
Device 0x24b7 (60:a4:23:ff:fe:6e:c1:eb) left the network
[0x24b7] Delivery error for seq # 0x3d, on endpoint id 0 cluster 0x0034: message send failure
Sending 'zdo_leave_req' failed: [0x24b7:0:0x0034]: Message send failure
New device 0xc25d (60:a4:23:ff:fe:6e:c1:eb) joined the network
[0xc25d] Scheduling initialization
Tries remaining: 3
[0xc25d] Requesting 'Node Descriptor'
Tries remaining: 2
[0xc25d] Extending timeout for 0x41 request
Device 0xc25d (60:a4:23:ff:fe:6e:c1:eb) joined the network
[0xc25d] Scheduling initialization
[0xc25d] Canceling old initialize call
Received frame on uninitialized device <Device model=None manuf=None nwk=0xC25D ieee=60:a4:23:ff:fe:6e:c1:eb is_initialized=False> from ep 0 to ep 0, cluster 19: b'\x82]\xc2\xeb\xc1n\xfe\xff#\xa4`\x80'
[0xc25d:zdo] ZDO request ZDOCmd.Device_annce: [0xC25D, 60:a4:23:ff:fe:6e:c1:eb, 128]
Tries remaining: 3
[0xc25d] Requesting 'Node Descriptor'
Tries remaining: 2
[0xc25d] Extending timeout for 0x43 request
Received frame on uninitialized device <Device model=None manuf=None nwk=0xC25D ieee=60:a4:23:ff:fe:6e:c1:eb is_initialized=False> from ep 0 to ep 0, cluster 32770: b'A\x00]\xc2\x02@\x80\x02\x10RR\x00\x00,R\x00\x00'
[0xc25d] Got Node Descriptor: NodeDescriptor(logical_type=<LogicalType.EndDevice: 2>, complex_descriptor_available=0, user_descriptor_available=0, reserved=0, aps_flags=0, frequency_band=<FrequencyBand.Freq2400MHz: 8>, mac_capability_flags=<MACCapabilityFlags.AllocateAddress: 128>, manufacturer_code=4098, maximum_buffer_size=82, maximum_incoming_transfer_size=82, server_mask=11264, maximum_outgoing_transfer_size=82, descriptor_capability_field=<DescriptorCapability.0: 0>, *allocate_address=True, *is_alternate_pan_coordinator=False, *is_coordinator=False, *is_end_device=True, *is_full_function_device=False, *is_mains_powered=False, *is_receiver_on_when_idle=False, *is_router=False, *is_security_capable=False)
[0xc25d] Discovering endpoints
Tries remaining: 3
[0xc25d] Extending timeout for 0x45 request
Received frame on uninitialized device <Device model=None manuf=None nwk=0xC25D ieee=60:a4:23:ff:fe:6e:c1:eb is_initialized=False> from ep 0 to ep 0, cluster 32770: b'C\x00]\xc2\x02@\x80\x02\x10RR\x00\x00,R\x00\x00'
Received frame on uninitialized device <Device model=None manuf=None nwk=0xC25D ieee=60:a4:23:ff:fe:6e:c1:eb is_initialized=False> from ep 0 to ep 0, cluster 32773: b'E\x00]\xc2\x01\x01'
[0xc25d] Got Node Descriptor: NodeDescriptor(logical_type=<LogicalType.EndDevice: 2>, complex_descriptor_available=0, user_descriptor_available=0, reserved=0, aps_flags=0, frequency_band=<FrequencyBand.Freq2400MHz: 8>, mac_capability_flags=<MACCapabilityFlags.AllocateAddress: 128>, manufacturer_code=4098, maximum_buffer_size=82, maximum_incoming_transfer_size=82, server_mask=11264, maximum_outgoing_transfer_size=82, descriptor_capability_field=<DescriptorCapability.0: 0>, *allocate_address=True, *is_alternate_pan_coordinator=False, *is_coordinator=False, *is_end_device=True, *is_full_function_device=False, *is_mains_powered=False, *is_receiver_on_when_idle=False, *is_router=False, *is_security_capable=False)
[0xc25d] Discovering endpoints
Tries remaining: 3
[0xc25d] Extending timeout for 0x47 request
[0xc25d] Discovered endpoints: [1]
[0xc25d] Initializing endpoints [<Endpoint id=1 in=[] out=[] status=<Status.NEW: 0>>]
[0xc25d:1] Discovering endpoint information
Tries remaining: 3
[0xc25d] Extending timeout for 0x49 request
Received frame on uninitialized device <Device model=None manuf=None nwk=0xC25D ieee=60:a4:23:ff:fe:6e:c1:eb is_initialized=False> from ep 0 to ep 0, cluster 32773: b'G\x00]\xc2\x01\x01'
[0xc25d] Discovered endpoints: [1]
[0xc25d] Initializing endpoints [<Endpoint id=1 in=[] out=[] status=<Status.NEW: 0>>]
[0xc25d:1] Discovering endpoint information
Tries remaining: 3
[0xc25d] Extending timeout for 0x4b request
Received frame on uninitialized device <Device model=None manuf=None nwk=0xC25D ieee=60:a4:23:ff:fe:6e:c1:eb is_initialized=False> from ep 0 to ep 0, cluster 32772: b'I\x00]\xc2\x12\x01\x04\x01\x00\x00\x01\x03\x00\x00\x01\x00\x06\x00\x02\x19\x00\n\x00'
[0xc25d:1] Discovered endpoint information: SizePrefixedSimpleDescriptor(endpoint=1, profile=260, device_type=0, device_version=1, input_clusters=[0, 1, 6], output_clusters=[25, 10])
[0xc25d] Read model None and manufacturer None from <Endpoint id=1 in=[] out=[] status=<Status.NEW: 0>>
[0xc25d] Discovered basic device information for <Device model=None manuf=None nwk=0xC25D ieee=60:a4:23:ff:fe:6e:c1:eb is_initialized=False>
Device is initialized <Device model=None manuf=None nwk=0xC25D ieee=60:a4:23:ff:fe:6e:c1:eb is_initialized=False>
Checking quirks for None None (60:a4:23:ff:fe:6e:c1:eb)
Considering <class 'xbee.xbee_io.XBeeSensor'>
Fail because endpoint list mismatch: {232, 230} {1}
Considering <class 'xbee.xbee3_io.XBee3Sensor'>
Fail because endpoint list mismatch: {232, 230} {1}
Considering <class 'smartthings.tag_v4.SmartThingsTagV4'>
Fail because profile_id mismatch on at least one endpoint
Considering <class 'smartthings.multi.SmartthingsMultiPurposeSensor'>
Fail because profile_id mismatch on at least one endpoint
[... cut ...]
                    Basic.cluster_id,
                    PowerConfiguration.cluster_id,
                    TuyaSmartRemoteOnOffCluster,
                ],
                OUTPUT_CLUSTERS: [Ota.cluster_id, Time.cluster_id],
            },
        },
    }

    device_automation_triggers = {
        (SHORT_PRESS, BUTTON_1): {ENDPOINT_ID: 1, COMMAND: SHORT_PRESS},
        (LONG_PRESS, BUTTON_1): {ENDPOINT_ID: 1, COMMAND: LONG_PRESS},
        (DOUBLE_PRESS, BUTTON_1): {ENDPOINT_ID: 1, COMMAND: DOUBLE_PRESS},
    }

```

MattWestb, 2021-07-11:
The line looks very good !!!
```
Found custom device replacement for 60:a4:23:ff:fe:6e:c1:eb: <class 'tuya.ts0041_zemismart.TuyaZemismartSmartRemote0041'>
```
Do you getting all events in HA ?

jmlankford, 2021-07-11:
I just did some checking and tweaking. Unfortunately I only get a battery entity with a "?" on the icon. I tried to listen for events with various button pushes using "zha_event" and nothing. 

Here's my log when I pair the device:

```
[0x0000:zdo] ZDO request ZDOCmd.Mgmt_Permit_Joining_req: [60, <Bool.false: 0>]
[0x5d9b:1:0x0006] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=GLOBAL_COMMAND manufacturer_specific=False is_reply=True disable_default_response=True> manufacturer=None tsn=80 command_id=Command.Report_Attributes>
[0x5d9b:1:0x0006] ZCL request 0x000a: [[Attribute(attrid=0, value=<TypeValue type=Bool, value=Bool.false>)]]
[0x5d9b:1:0x0006] Attribute report received: on_off=0
Device 0x2761 (60:a4:23:ff:fe:6e:c1:eb) left the network
New device 0xdc7b (60:a4:23:ff:fe:6e:c1:eb) joined the network
[0xdc7b] Scheduling initialization
Tries remaining: 3
[0xdc7b] Requesting 'Node Descriptor'
Tries remaining: 2
[0xdc7b] Extending timeout for 0x93 request
Device 0xdc7b (60:a4:23:ff:fe:6e:c1:eb) joined the network
[0xdc7b] Scheduling initialization
[0xdc7b] Canceling old initialize call
Received frame on uninitialized device <Device model=None manuf=None nwk=0xDC7B ieee=60:a4:23:ff:fe:6e:c1:eb is_initialized=False> from ep 0 to ep 0, cluster 19: b'\x8a{\xdc\xeb\xc1n\xfe\xff#\xa4`\x80'
[0xdc7b:zdo] ZDO request ZDOCmd.Device_annce: [0xDC7B, 60:a4:23:ff:fe:6e:c1:eb, 128]
Tries remaining: 3
[0xdc7b] Requesting 'Node Descriptor'
Tries remaining: 2
[0xdc7b] Extending timeout for 0x95 request
Received frame on uninitialized device <Device model=None manuf=None nwk=0xDC7B ieee=60:a4:23:ff:fe:6e:c1:eb is_initialized=False> from ep 0 to ep 0, cluster 32770: b'\x93\x00{\xdc\x02@\x80\x02\x10RR\x00\x00,R\x00\x00'
[0xdc7b:zdo] ZDO request ZDOCmd.Node_Desc_rsp: [<Status.SUCCESS: 0>, 0xDC7B, NodeDescriptor(logical_type=<LogicalType.EndDevice: 2>, complex_descriptor_available=0, user_descriptor_available=0, reserved=0, aps_flags=0, frequency_band=<FrequencyBand.Freq2400MHz: 8>, mac_capability_flags=<MACCapabilityFlags.AllocateAddress: 128>, manufacturer_code=4098, maximum_buffer_size=82, maximum_incoming_transfer_size=82, server_mask=11264, maximum_outgoing_transfer_size=82, descriptor_capability_field=<DescriptorCapability.0: 0>, *allocate_address=True, *is_alternate_pan_coordinator=False, *is_coordinator=False, *is_end_device=True, *is_full_function_device=False, *is_mains_powered=False, *is_receiver_on_when_idle=False, *is_router=False, *is_security_capable=False)]
[0xdc7b:zdo] Unsupported ZDO request:ZDOCmd.Node_Desc_rsp
Received frame on uninitialized device <Device model=None manuf=None nwk=0xDC7B ieee=60:a4:23:ff:fe:6e:c1:eb is_initialized=False> from ep 0 to ep 0, cluster 32770: b'\x95\x00{\xdc\x02@\x80\x02\x10RR\x00\x00,R\x00\x00'
[0xdc7b] Got Node Descriptor: NodeDescriptor(logical_type=<LogicalType.EndDevice: 2>, complex_descriptor_available=0, user_descriptor_available=0, reserved=0, aps_flags=0, frequency_band=<FrequencyBand.Freq2400MHz: 8>, mac_capability_flags=<MACCapabilityFlags.AllocateAddress: 128>, manufacturer_code=4098, maximum_buffer_size=82, maximum_incoming_transfer_size=82, server_mask=11264, maximum_outgoing_transfer_size=82, descriptor_capability_field=<DescriptorCapability.0: 0>, *allocate_address=True, *is_alternate_pan_coordinator=False, *is_coordinator=False, *is_end_device=True, *is_full_function_device=False, *is_mains_powered=False, *is_receiver_on_when_idle=False, *is_router=False, *is_security_capable=False)
[0xdc7b] Discovering endpoints
Tries remaining: 3
[0xdc7b] Extending timeout for 0x97 request
Received frame on uninitialized device <Device model=None manuf=None nwk=0xDC7B ieee=60:a4:23:ff:fe:6e:c1:eb is_initialized=False> from ep 0 to ep 0, cluster 32773: b'\x97\x00{\xdc\x01\x01'
[0xdc7b] Discovered endpoints: [1]
[0xdc7b] Initializing endpoints [<Endpoint id=1 in=[] out=[] status=<Status.NEW: 0>>]
[0xdc7b:1] Discovering endpoint information
Tries remaining: 3
[0xdc7b] Extending timeout for 0x99 request
Received frame on uninitialized device <Device model=None manuf=None nwk=0xDC7B ieee=60:a4:23:ff:fe:6e:c1:eb is_initialized=False> from ep 0 to ep 0, cluster 32772: b'\x99\x00{\xdc\x12\x01\x04\x01\x00\x00\x01\x03\x00\x00\x01\x00\x06\x00\x02\x19\x00\n\x00'
[0xdc7b:1] Discovered endpoint information: SizePrefixedSimpleDescriptor(endpoint=1, profile=260, device_type=0, device_version=1, input_clusters=[0, 1, 6], output_clusters=[25, 10])
[0xdc7b] Extending timeout for 0x9b request
[0xdc7b:1:0x0000] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=GLOBAL_COMMAND manufacturer_specific=False is_reply=True disable_default_response=True> manufacturer=None tsn=155 command_id=Command.Read_Attributes_rsp>
[0xdc7b] Read model 'TS0041' and manufacturer '_TZ3000_iklyvnq3' from <Endpoint id=1 in=[basic:0x0000, power:0x0001, on_off:0x0006] out=[ota:0x0019, time:0x000A] status=<Status.ZDO_INIT: 1>>
[0xdc7b] Discovered basic device information for <Device model='TS0041' manuf='_TZ3000_iklyvnq3' nwk=0xDC7B ieee=60:a4:23:ff:fe:6e:c1:eb is_initialized=True>
Device is initialized <Device model='TS0041' manuf='_TZ3000_iklyvnq3' nwk=0xDC7B ieee=60:a4:23:ff:fe:6e:c1:eb is_initialized=True>
Checking quirks for _TZ3000_iklyvnq3 TS0041 (60:a4:23:ff:fe:6e:c1:eb)
Considering <class 'tuya.ts0041_zemismart.TuyaSmartRemote0041TI'>
Fail because input cluster mismatch on at least one endpoint
Considering <class 'tuya.ts0041_zemismart.TuyaSmartRemote0041TO'>
[... cut ...]
                    PowerConfiguration.cluster_id,
                    TuyaSmartRemoteOnOffCluster,
                    Time.cluster_id,
                ],
                OUTPUT_CLUSTERS: [Ota.cluster_id],
            },
        },
    }

    device_automation_triggers = {
        (SHORT_PRESS, BUTTON_1): {ENDPOINT_ID: 1, COMMAND: SHORT_PRESS},
        (LONG_PRESS, BUTTON_1): {ENDPOINT_ID: 1, COMMAND: LONG_PRESS},
        (DOUBLE_PRESS, BUTTON_1): {ENDPOINT_ID: 1, COMMAND: DOUBLE_PRESS},
    }

```

MattWestb, 2021-07-11:
Its starts being added as 0xDC7B and then its being added as 0x2761  ;-(

Trye deleting it from the device card and waking it up then clicking "remove" and then waiting 2 minutes so the system is writing the cash in the DB and then adding it gen.

The quirk looks nice and its shall working for all  one buttons devices.

jmlankford, 2021-07-11:
I agree, this quirk should work based on everything going into it. After waiting a few minutes, I removed the device, waited a few minutes again, and paired it only the battery entity is connecting (and it's showing an accurate number for the first time). 

No other entity shows.

With that said, I am able to get some zha_events showing:

LONG PRESS:
```
{
    "event_type": "zha_event",
    "data": {
        "device_ieee": "60:a4:23:ff:fe:6e:c1:eb",
        "unique_id": "60:a4:23:ff:fe:6e:c1:eb:1:0x0006",
        "device_id": "4fdcdbe55f21859925b61d7030441135",
        "endpoint_id": 1,
        "cluster_id": 6,
        "command": "remote_button_long_press",
        "args": []
    },
    "origin": "LOCAL",
    "time_fired": "2021-07-11T16:12:51.904487+00:00",
    "context": {
        "id": "a79b7c8dde3ee6bcfe6a0ac0ede6ee4f",
        "parent_id": null,
        "user_id": null
    }
}
```

SHORT PRESS:
```
{
    "event_type": "zha_event",
    "data": {
        "device_ieee": "60:a4:23:ff:fe:6e:c1:eb",
        "unique_id": "60:a4:23:ff:fe:6e:c1:eb:1:0x0006",
        "device_id": "4fdcdbe55f21859925b61d7030441135",
        "endpoint_id": 1,
        "cluster_id": 6,
        "command": "remote_button_short_press",
        "args": []
    },
    "origin": "LOCAL",
    "time_fired": "2021-07-11T16:12:20.650798+00:00",
    "context": {
        "id": "05adf5d639ce9265facac305217b7d57",
        "parent_id": null,
        "user_id": null
    }
}
```
DOUBLE PRESS:
```

{
    "event_type": "zha_event",
    "data": {
        "device_ieee": "60:a4:23:ff:fe:6e:c1:eb",
        "unique_id": "60:a4:23:ff:fe:6e:c1:eb:1:0x0006",
        "device_id": "4fdcdbe55f21859925b61d7030441135",
        "endpoint_id": 1,
        "cluster_id": 6,
        "command": "remote_button_double_press",
        "args": []
    },
    "origin": "LOCAL",
    "time_fired": "2021-07-11T16:14:25.199430+00:00",
    "context": {
        "id": "6d4a1940a48278f8ddb7e3da23c344cb",
        "parent_id": null,
        "user_id": null
    }
}
```

MattWestb, 2021-07-11:
The zha_event looks normal for one switch.

You is using the "commands" as triggers in your automation.

And you is having battery status reported OK.

The problems with deleting and start adding one device is one PITA and you is getting half working device but the devs is in work refactoring the pairing and database storage for the devices so i hope its being better.

Is your last quirk working OK (with the TuyaSmartRemote0041TO and TI) ?
I like making one PR with it but i dont knowing how to deleting the 2 that is not needed after making the MODEL thing.

Great work done !!!

jmlankford, 2021-07-11:
Everything is working fine and I was able to add it to a working automation. Thank you so much for walking through putting this new device on a simple quirk!

The quirk it seems to be using is "TuyaSmartRemote0041TO"

```
{
  "node_descriptor": "NodeDescriptor(logical_type=<LogicalType.EndDevice: 2>, complex_descriptor_available=0, user_descriptor_available=0, reserved=0, aps_flags=0, frequency_band=<FrequencyBand.Freq2400MHz: 8>, mac_capability_flags=<MACCapabilityFlags.AllocateAddress: 128>, manufacturer_code=4098, maximum_buffer_size=82, maximum_incoming_transfer_size=82, server_mask=11264, maximum_outgoing_transfer_size=82, descriptor_capability_field=<DescriptorCapability.0: 0>, *allocate_address=True, *is_alternate_pan_coordinator=False, *is_coordinator=False, *is_end_device=True, *is_full_function_device=False, *is_mains_powered=False, *is_receiver_on_when_idle=False, *is_router=False, *is_security_capable=False)",
  "endpoints": {
    "1": {
      "profile_id": 260,
      "device_type": "0x0006",
      "in_clusters": [
        "0x0000",
        "0x0001",
        "0x0006"
      ],
      "out_clusters": [
        "0x000a",
        "0x0019"
      ]
    }
  },
  "manufacturer": "_TZ3000_iklyvnq3",
  "model": "TS0041",
  "class": "tuya.ts0041_zemismart.TuyaSmartRemote0041TO"
}
```

MattWestb, 2021-07-11:
I have making one "universal quirk" then its looks the TS004X is all the same only 2 types of cluster config and have deleting 2 "custom named quirks" that was 99 % the same ad the real one.
If you have time its being great if you can testing the quirk in the PR for verification that is working  OK before the devs is committing it.

Perhaps i doing the other TS004Xs with model so its working for all switches without adding id for id.

closed 2022-01-14 by github-actions[bot]

evlo, 2022-09-09:
Do you have issue with this device not reacting on pressing the button first time? Even though button led flashes, no action is registered on coordinator.