zigpy/zha-device-handlers #959: [Device Support Request] Ikea Trader bulbs power on behaviour

jonathanrobichaud4, 2021-07-12:
**Is your feature request related to a problem? Please describe.**
A couple days ago the power went out at night and then came back and turned my lights on. It is really annoying and a pain to deal with at 3 am lol. I did some research and found out that you can change ikea bulbs power on behaviour.

**Describe the solution you'd like**
Someone who is way more technically proficient then me should open a pull request on the quirk and add a custom attribute for 0x4003. This is in the OnOff cluster and controls the power on behaviour. I've changed it with deconz but it would be really nice to change it from the ha clusters screen. It should be a pretty simple fix as it is just on attribute but I don't really know much lol. If anyone would like to at least guide me how to add this attribute to the cluster that would be great as well!

Adminiuga, 2021-07-13:
you have to be more specific, like exact manufacturer and model and Zigbee device signature.

jonathanrobichaud4, 2021-07-13:
It is an Ikea bulb but I was on the homeassistant discord and it needs to first be implemented into zigpy first so for now this issue can be closed since the current issue is zigpy

closed 2021-07-13 by jonathanrobichaud4

jonathanrobichaud4, 2021-07-13:
@puddly made a PR in that repo here: https://github.com/zigpy/zigpy/pull/716. So this PR will need to be fixed and merged before this can be figured out.

MattWestb, 2021-08-13:
I think its in place but i have not testing it with my IKEA lights and outlets https://github.com/zigpy/zigpy/pull/774 but it  shall working OK with Philips HUE lights.