zigpy/zha-device-handlers #960: [Device Support Request] _TZE200_iossyxra (Battery Roller Blind Motor)

VladNord, 2021-07-14:
**Is your feature request related to a problem? Please describe.**
Bought ZigBee Lithium Battery Roller Blind Motor. It does pair, but it does not expose any status, so I can't control it from HA

**Describe the solution you'd like**
Possibility to control it from HA

**Device signature - this can be acquired by removing the device from ZHA and pairing it again from the add devices screen. Be sure to add the entire content of the log panel after pairing the device to a code block below this line.**
```
"NodeDescriptor(byte1=1, byte2=64, mac_capability_flags=142, manufacturer_code=4098, 
maximum_buffer_size=82, maximum_incoming_transfer_size=82, server_mask=11264, 
maximum_outgoing_transfer_size=82, descriptor_capability_field=0, 
  "endpoints": {
    "1": {
      "profile_id": 260,
      "device_type": "0x0051",
      "in_clusters": [
        "0x0000",
        "0x0004",
        "0x0005",
        "0xef00"
      ],
      "out_clusters": [
        "0x000a",
        "0x0019"
      ]
    }
  },
  "manufacturer": "_TZE200_iossyxra",
  "model": "TS0601",
  "class": "zigpy.device.Device"
}
```

**Additional context**
Product: https://aliexpress.ru/item/4000764153310.html?spm=a2g39.orderlist.0.0.47ce4aa67PXmvK&_ga=2.58309290.1271432921.1626213995-1279367481.1621973111

VladNord, 2021-07-14:
After dancing with a tambourine, I found half-working solution with adding such code to  
**__init__.py**      `"_TZE200_iossyxra": {0x0000: 0x0000, 0x0001: 0x0002, 0x0002: 0x0001},`into # TUYA Cover Custom Values block
and **ts0601.py** `("_TZE200_iossyxra", "TS0601"),`into class TuyaMoesCover0601(TuyaWindowCover)

After these device is detected correctly, but blind buttons functionality from HA doesn't working (Error calling service)

But it can works, using this code in dev tools

```
service: zha.issue_zigbee_cluster_command
data:
  ieee: '84:71:27:ff:fe:0e:18:00'
  endpoint_id: 1
  cluster_id: 258
  cluster_type: in
  command_type: server
  command: 6
  args:
    - 0
    - 0
    - 1025
    - 0
    - - 0
      - 0 or 1 or 2
```

MattWestb, 2021-07-28:
I think your quirk is OK but you need deleting the `__pycache__` in the tuya quirk folder or the system is using some old setting / code.

Also if the buttons is not working as expected try changing the settings for them in **init.py** so up is up and down is down and you can stopping the movement.

VladNord, 2021-08-10:
> I think your quirk is OK but you need deleting the `__pycache__` in the tuya quirk folder or the system is using some old setting / code.
> 
> Also if the buttons is not working as expected try changing the settings for them in **init.py** so up is up and down is down and you can stopping the movement.

Cache cleared, tried different combinations in init.py, but no luck.

With this `"_TZE200_iossyxra": {0x0000: 0x0000, 0x0001: 0x0002, 0x0002: 0x0001}` set in HA result following:

[image]

MattWestb, 2021-11-15:
Is it still not working ?
Its 2 PR that is fixing the inverting mode and some communication with the tuya MCU ?
https://github.com/zigpy/zha-device-handlers/pull/1130 https://github.com/zigpy/zha-device-handlers/pull/1115

mj1985, 2021-11-25:
I added this line:
`"_TZE200_iossyxra": {0x0000: 0x0000, 0x0001: 0x0002, 0x0002: 0x0001}`
to the file `ts0601_cover.py` and edited ´__init__.py´ file with the latest changes of the dev-branch, but I still get an error of calling the server. Anything I have missed?

MattWestb, 2021-11-25:
I think the combination in https://github.com/zigpy/zha-device-handlers/issues/960#issuecomment-895637580 is wrong then if the blind is 100 or 0 % only one button shall being possible activating and if its not 100 or 0% both up and down shall being possible activating. Then the blinds is "on the move" the stop shall being possible activating.

Can you posting the error from the log ?

Chreece, 2021-12-17:
Hello,

I want to confirm that this product works with zha:
https://zigbee.blakadder.com/Zemismart_AM15.html

Custom Quirk:
```from zigpy.profiles import zha
from zigpy.zcl.clusters.general import Basic, Groups, Ota, Scenes, Time

from zhaquirks.const import (
    DEVICE_TYPE,
    ENDPOINTS,
    INPUT_CLUSTERS,
    MODELS_INFO,
    OUTPUT_CLUSTERS,
    PROFILE_ID,
)
from zhaquirks.tuya import (
    TuyaManufacturerClusterOnOff,
    TuyaManufacturerWindowCover,
    TuyaManufCluster,
    TuyaOnOff,
    TuyaSwitch,
    TuyaWindowCover,
    TuyaWindowCoverControl,
)

class TuyaZemismartSmartCover0601_2(TuyaWindowCover):
    """Tuya Zemismart curtain cover motor."""

    signature = {
        # "node_descriptor": "<NodeDescriptor byte1=1 byte2=64 mac_capability_flags=142 manufacturer_code=4098
        #                       maximum_buffer_size=82 maximum_incoming_transfer_size=82 server_mask=11264
        #                       maximum_outgoing_transfer_size=82 descriptor_capability_field=0>",
        # input_clusters=[0x0000, 0x000a, 0x0004, 0x0005, 0xef00]
        # output_clusters=[0x0019]
        # <SimpleDescriptor endpoint=1 profile=260 device_type=81 input_clusters=[0, 10, 4, 5, 61184] output_clusters=[25]>
        MODELS_INFO: [
            ("_TZE200_3i3exuay", "TS0601"),
        ],
        ENDPOINTS: {
            1: {
                PROFILE_ID: zha.PROFILE_ID,
                DEVICE_TYPE: zha.DeviceType.SMART_PLUG,
                INPUT_CLUSTERS: [
                    Basic.cluster_id,
                    Time.cluster_id,
                    Groups.cluster_id,
                    Scenes.cluster_id,
                    TuyaManufCluster.cluster_id,
                ],
                OUTPUT_CLUSTERS: [Ota.cluster_id],
            },
        },
    }
    replacement = {
        ENDPOINTS: {
            1: {
                PROFILE_ID: zha.PROFILE_ID,
                DEVICE_TYPE: zha.DeviceType.WINDOW_COVERING_DEVICE,
                INPUT_CLUSTERS: [
                    Basic.cluster_id,
                    Groups.cluster_id,
                    Scenes.cluster_id,
                    Time.cluster_id,
                    TuyaManufacturerWindowCover,
                    TuyaWindowCoverControl,
                ],
                OUTPUT_CLUSTERS: [Ota.cluster_id],
            },
        },
    }

class TuyaMoesCover0601(TuyaWindowCover):
    """Tuya blind controller device."""

    signature = {
        # "node_descriptor": "NodeDescriptor(byte1=2, byte2=64, mac_capability_flags=128, manufacturer_code=4098,
        #                    maximum_buffer_size=82, maximum_incoming_transfer_size=82, server_mask=11264,
        #                    maximum_outgoing_transfer_size=82, descriptor_capability_field=0)",
        # "endpoints": {
        # "1": { "profile_id": 260, "device_type": "0x0051", "in_clusters": [ "0x0000", "0x0004","0x0005","0xef00"], "out_clusters": ["0x000a","0x0019"] }
        # },
        # "manufacturer": "_TZE200_zah67ekd",
        # "model": "TS0601",
        # "class": "zigpy.device.Device"
        # }
        MODELS_INFO: [
            ("_TZE200_zah67ekd", "TS0601"),
            ("_TZE200_xuzcvlku", "TS0601"),
            ("_TZE200_rddyvrci", "TS0601"),
            ("_TZE200_iossyxra", "TS0601"),
        ],
        ENDPOINTS: {
            1: {
                PROFILE_ID: zha.PROFILE_ID,
                DEVICE_TYPE: zha.DeviceType.SMART_PLUG,
                INPUT_CLUSTERS: [
                    Basic.cluster_id,
                    Groups.cluster_id,
                    Scenes.cluster_id,
                    TuyaManufCluster.cluster_id,
                ],
                OUTPUT_CLUSTERS: [Time.cluster_id, Ota.cluster_id],
            }
        },
    }

    replacement = {
        ENDPOINTS: {
            1: {
                DEVICE_TYPE: zha.DeviceType.WINDOW_COVERING_DEVICE,
                INPUT_CLUSTERS: [
                    Basic.cluster_id,
                    Groups.cluster_id,
                    Scenes.cluster_id,
                    TuyaManufacturerWindowCover,
                    TuyaWindowCoverControl,
                ],
                OUTPUT_CLUSTERS: [Time.cluster_id, Ota.cluster_id],
            }
        }
    }
```
   
Adding this in line __init__.py for TUYA_COVER_COMMAND:
```
"_TZE200_iossyxra": {0x0000: 0x0000, 0x0001: 0x0002, 0x0002: 0x0001},
```

Can someone implement this to work out of the box?

closed 2022-01-12 by dmulcahey

mgwoj, 2022-01-18:
From what I understand, this AM-15 Zigbee2RF dongle should mimic the original RF remote controller. This controller exists in versions supporting up to 16 different channels. Do you know how to provide channel ID in the Open/Close commands?

VladNord, 2022-01-19:
> 

you can check this article. its in Russian, but anyway

https://sprut.ai/client/article/3650

mgwoj, 2022-01-27:
Thank you @VladNord for this article, but this is how I have setup it currently using Sonoff RF Bridge.
I am looking how to control motors using Zigbee2RF USB dongle which can be bought with those motors - see the first link from this thread.
My question is how to provide channel when sending Up/Down commands.

mgwoj, 2022-01-28:
I have a answer to my previous question from the original instruction for ZM-AM15-02:
> Pair more motor
> It can be paired to several motors, but all motors should be control together.

So there is no ability to select only one motor, if you need to tune one motor, you have to turn off the other ones.

So now my question is how to initiate the pairing operation between this dongle and the motor without Tuya App?

PR #1278 (2022-01-12): Add support for _TZE200_iossyxra
Adds device support for [Zemismart_AM15](https://zigbee.blakadder.com/Zemismart_AM15.html).
Closes #960