zigpy/zha-device-handlers #963: [BUG] TS0210 Tuya Vibration Sensor state won't reset

jayden-jw, 2021-07-16:
**Describe the bug**
After added the Vibration Sensor, the sensor state always showed "Detected", tested with few units but all having the same issue.

**To Reproduce**
Steps to reproduce the behavior:
1. Just add the sensor in ZHA and the sensor state will show Detected by default.

**Expected behavior**
The sensor state should reset to false after timeout.

**Screenshots**
[image]

MattWestb, 2021-07-16:
Your device is needing one quirk for working OK in ZHA then the reset shall being made of the device but its not doing it.

Can you pleas posting the Zigbee device signature and debug log then the device is sending the commands to the coordinator.

jayden-jw, 2021-07-16:
Not sure if this is the debug log that you mentioned:
2021-07-16 22:38:44 DEBUG (MainThread) [zigpy.zcl] [0xed44:1:0x0500] ZCL request 0x0000: [<ZoneStatus.1024|Alarm_1: 1025>, <bitmap8.0: 0>, 0, 0]
2021-07-16 22:38:44 DEBUG (MainThread) [zigpy.zcl] [0xed44:1:0x0500] No handler for cluster command 0
2021-07-16 22:38:44 DEBUG (MainThread) [homeassistant.components.zha.core.channels.base] [0xED44:1:0x0500]: Updated alarm state: ZoneStatus.Alarm_1
2021-07-16 22:38:44 DEBUG (MainThread) [zigpy_znp.api] Received command: AF.IncomingMsg.Callback(GroupId=0x0000, ClusterId=1280, SrcAddr=0xED44, SrcEndpoint=1, DstEndpoint=1, WasBroadcast=<Bool.false: 0>, LQI=48, SecurityUse=<Bool.false: 0>, TimeStamp=15045879, TSN=0, Data=b'\x09\x59\x00\x01\x04\x00\x00\x00\x00', MacSrcAddr=0xED44, MsgResultRadius=29)
2021-07-16 22:38:44 DEBUG (MainThread) [zigpy.zcl] [0xed44:1:0x0500] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=CLUSTER_COMMAND manufacturer_specific=False is_reply=True disable_default_response=False> manufacturer=None tsn=89 command_id=0>
2021-07-16 22:38:44 DEBUG (MainThread) [zigpy.zcl] [0xed44:1:0x0500] ZCL request 0x0000: [<ZoneStatus.1024|Alarm_1: 1025>, <bitmap8.0: 0>, 0, 0]

Device signature:
{
  "node_descriptor": "NodeDescriptor(logical_type=<LogicalType.EndDevice: 2>, complex_descriptor_available=0, user_descriptor_available=0, reserved=0, aps_flags=0, frequency_band=<FrequencyBand.Freq2400MHz: 8>, mac_capability_flags=<MACCapabilityFlags.AllocateAddress: 128>, manufacturer_code=4098, maximum_buffer_size=82, maximum_incoming_transfer_size=512, server_mask=11264, maximum_outgoing_transfer_size=512, descriptor_capability_field=<DescriptorCapability.0: 0>, *allocate_address=True, *is_alternate_pan_coordinator=False, *is_coordinator=False, *is_end_device=True, *is_full_function_device=False, *is_mains_powered=False, *is_receiver_on_when_idle=False, *is_router=False, *is_security_capable=False)",
  "endpoints": {
    "1": {
      "profile_id": 260,
      "device_type": "0x0402",
      "in_clusters": [
        "0x0000",
        "0x0001",
        "0x000a",
        "0x0500"
      ],
      "out_clusters": [
        "0x0019"
      ]
    }
  },
  "manufacturer": "_TYZB01_3zv6oleo",
  "model": "TS0210",
  "class": "zigpy.device.Device"
}

craigjmidwinter, 2021-11-10:
I've got this device as well. I noticed this PR in another repo for this device: https://github.com/Koenkk/zigbee-herdsman-converters/issues/2609

craigjmidwinter, 2021-11-10:
Device Signature:
```
{
  "node_descriptor": "NodeDescriptor(logical_type=<LogicalType.EndDevice: 2>, complex_descriptor_available=0, user_descriptor_available=0, reserved=0, aps_flags=0, frequency_band=<FrequencyBand.Freq2400MHz: 8>, mac_capability_flags=<MACCapabilityFlags.AllocateAddress: 128>, manufacturer_code=4098, maximum_buffer_size=82, maximum_incoming_transfer_size=512, server_mask=11264, maximum_outgoing_transfer_size=512, descriptor_capability_field=<DescriptorCapability.NONE: 0>, *allocate_address=True, *is_alternate_pan_coordinator=False, *is_coordinator=False, *is_end_device=True, *is_full_function_device=False, *is_mains_powered=False, *is_receiver_on_when_idle=False, *is_router=False, *is_security_capable=False)",
  "endpoints": {
    "1": {
      "profile_id": 260,
      "device_type": "0x0402",
      "in_clusters": [
        "0x0000",
        "0x0001",
        "0x000a",
        "0x0500"
      ],
      "out_clusters": [
        "0x0019"
      ]
    }
  },
  "manufacturer": "_TYZB01_3zv6oleo",
  "model": "TS0210",
  "class": "zigpy.device.Device"
}
```

craigjmidwinter, 2021-11-11:
I was trying to fiddle with writing a quirk for this-- not sure if this is meaningful, but pairing doesn't have a SimpleDescriptor, but it does have this SizePrefixedSimpleDescriptor which looks similar: 
```
SizePrefixedSimpleDescriptor(endpoint=1, profile=260, device_type=1026, device_version=0, input_clusters=[0, 10, 1, 1280], output_clusters=[25])
```

craigjmidwinter, 2021-11-12:
Ok, I think I'm close to having a quirk working for this. In the logs I can see the event come through and then the reset happening, but the thing I'm running into now is that the device shows up in home assistant with only the battery entity, not the actual IAS zone. @MattWestb do you happen to have any ideas as to what I could be doing wrong? https://github.com/craigjmidwinter/zha-device-handlers/blob/TS0210/zhaquirks/tuya/ts0210.py is where I'm currently at

MattWestb, 2021-11-12:
I dont have experience of IAS devices but i thing your new cluster is wrong class.
Looking  on https://github.com/zigpy/zigpy/blob/7c168c1c2ca2a338676e5956d2277af7c40971d6/zigpy/zcl/clusters/security.py#L31 shall your command being one Vibration_Movement_Sensor = 0x002D from one cluster type 0x0500 and then is the zone status but i dont knowing how it shall working in real.

craigjmidwinter, 2021-11-12:
Ah ok cool thanks!

MattWestb, 2021-11-12:
One that patching one IAS device cluster https://github.com/zigpy/zha-device-handlers/blob/242d5f36b9d9bdd2ee3594e56fa4bac47c369de8/zhaquirks/develco/open_close.py#L30

MattWestb, 2021-11-12:
That was fast getting one PR for it :-))

closed 2021-11-13 by dmulcahey