zigpy/zha-device-handlers #964: [Device Support Request] Lonsonho Zigbee Smart Touch Dimmer [_TZE200_0nauxa0p]-[TS0601]

SkyFox277, 2021-07-16:
Is your feature request related to a problem? Please describe.
Device pairs but no actions (entities) are available.
Shop: [https://de.aliexpress.com/item/1005002247615037.html](url)
Seams to be same like: [https://zigbee.blakadder.com/Eardatek_EDM-1ZAA-EU.html](url)

Describe the solution you'd like
Support for this device

Zigbee device signature from HA device panel:

{
  "node_descriptor": "NodeDescriptor(logical_type=<LogicalType.Router: 1>, complex_descriptor_available=0, user_descriptor_available=0, reserved=0, aps_flags=0, frequency_band=<FrequencyBand.Freq2400MHz: 8>, mac_capability_flags=<MACCapabilityFlags.AllocateAddress|RxOnWhenIdle|MainsPowered|FullFunctionDevice: 142>, manufacturer_code=4098, maximum_buffer_size=82, maximum_incoming_transfer_size=82, server_mask=11264, maximum_outgoing_transfer_size=82, descriptor_capability_field=<DescriptorCapability.0: 0>, *allocate_address=True, *is_alternate_pan_coordinator=False, *is_coordinator=False, *is_end_device=False, *is_full_function_device=True, *is_mains_powered=True, *is_receiver_on_when_idle=True, *is_router=True, *is_security_capable=False)",
  "endpoints": {
    "1": {
      "profile_id": 260,
      "device_type": "0x0051",
      "in_clusters": [
        "0x0000",
        "0x0004",
        "0x0005",
        "0xef00"
      ],
      "out_clusters": [
        "0x000a",
        "0x0019"
      ]
    }
  },
  "manufacturer": "_TZE200_0nauxa0p",
  "model": "TS0601",
  "class": "zigpy.device.Device"
}

Logs:

New device 0x781e (60:a4:23:ff:fe:45:92:04) joined the network
[0x781e] Scheduling initialization
Received frame on uninitialized device <Device model=None manuf=None nwk=0x781E ieee=60:a4:23:ff:fe:45:92:04 is_initialized=False> from ep 0 to ep 0, cluster ZDOCmd.Device_annce: b'\xFF\x1E\x78\x04\x92\x45\xFE\xFF\x23\xA4\x60\x8E'
[0x781e:zdo] ZDO request ZDOCmd.Device_annce: [0x781E, 60:a4:23:ff:fe:45:92:04, 142]
Tries remaining: 3
[0x781e] Requesting 'Node Descriptor'
Tries remaining: 2
[0x781e] Extending timeout for 0x0c request
Received frame on uninitialized device <Device model=None manuf=None nwk=0x781E ieee=60:a4:23:ff:fe:45:92:04 is_initialized=False> from ep 0 to ep 0, cluster ZDOCmd.Node_Desc_rsp: b'\x0C\x00\x1E\x78\x01\x40\x8E\x02\x10\x52\x52\x00\x00\x2C\x52\x00\x00'
[0x781e] Got Node Descriptor: NodeDescriptor(logical_type=<LogicalType.Router: 1>, complex_descriptor_available=0, user_descriptor_available=0, reserved=0, aps_flags=0, frequency_band=<FrequencyBand.Freq2400MHz: 8>, mac_capability_flags=<MACCapabilityFlags.AllocateAddress|RxOnWhenIdle|MainsPowered|FullFunctionDevice: 142>, manufacturer_code=4098, maximum_buffer_size=82, maximum_incoming_transfer_size=82, server_mask=11264, maximum_outgoing_transfer_size=82, descriptor_capability_field=<DescriptorCapability.0: 0>, *allocate_address=True, *is_alternate_pan_coordinator=False, *is_coordinator=False, *is_end_device=False, *is_full_function_device=True, *is_mains_powered=True, *is_receiver_on_when_idle=True, *is_router=True, *is_security_capable=False)
[0x781e] Discovering endpoints
Tries remaining: 3
Received frame on uninitialized device <Device model=None manuf=None nwk=0x781E ieee=60:a4:23:ff:fe:45:92:04 is_initialized=False> from ep 0 to ep 0, cluster ZDOCmd.Active_EP_rsp: b'\x0D\x00\x1E\x78\x01\x01'
[0x781e] Discovered endpoints: [1]
[0x781e] Initializing endpoints [<Endpoint id=1 in=[] out=[] status=<Status.NEW: 0>>]
[0x781e:1] Discovering endpoint information
Tries remaining: 3
Received frame on uninitialized device <Device model=None manuf=None nwk=0x781E ieee=60:a4:23:ff:fe:45:92:04 is_initialized=False> from ep 0 to ep 0, cluster ZDOCmd.Simple_Desc_rsp: b'\x0E\x00\x1E\x78\x14\x01\x04\x01\x51\x00\x01\x04\x00\x00\x04\x00\x05\x00\x00\xEF\x02\x19\x00\x0A\x00'
[0x781e:1] Discovered endpoint information: SizePrefixedSimpleDescriptor(endpoint=1, profile=260, device_type=81, device_version=1, input_clusters=[0, 4, 5, 61184], output_clusters=[25, 10])
Unknown cluster 61184
[0x781e:1:0x0000] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=GLOBAL_COMMAND manufacturer_specific=False is_reply=True disable_default_response=True> manufacturer=None tsn=15 command_id=Command.Read_Attributes_rsp>
[0x781e] Read model 'TS0601' and manufacturer '_TZE200_0nauxa0p' from <Endpoint id=1 in=[basic:0x0000, groups:0x0004, scenes:0x0005, None:0xEF00] out=[ota:0x0019, time:0x000A] status=<Status.ZDO_INIT: 1>>
[0x781e] Discovered basic device information for <Device model='TS0601' manuf='_TZE200_0nauxa0p' nwk=0x781E ieee=60:a4:23:ff:fe:45:92:04 is_initialized=True>
Device is initialized <Device model='TS0601' manuf='_TZE200_0nauxa0p' nwk=0x781E ieee=60:a4:23:ff:fe:45:92:04 is_initialized=True>
Checking quirks for _TZE200_0nauxa0p TS0601 (60:a4:23:ff:fe:45:92:04)
Considering <class 'zhaquirks.xbee.xbee_io.XBeeSensor'>
Fail because endpoint list mismatch: {232, 230} {1}
Considering <class 'zhaquirks.xbee.xbee3_io.XBee3Sensor'>
Fail because endpoint list mismatch: {232, 230} {1}
Considering <class 'zhaquirks.smartthings.tag_v4.SmartThingsTagV4'>
Fail because device_type mismatch on at least one endpoint
Considering <class 'zhaquirks.smartthings.multi.SmartthingsMultiPurposeSensor'>
Fail because device_type mismatch on at least one endpoint
Considering <class 'zhaquirks.netvox.z308e3ed.Z308E3ED'>
Fail because device_type mismatch on at least one endpoint
[... 151 lines cut ...]
[0x781e:1:0xef00] No handler for cluster command 1
[0x781e:1:0xef00] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=CLUSTER_COMMAND manufacturer_specific=False is_reply=True disable_default_response=True> manufacturer=None tsn=106 command_id=36>
[0x781e:1:0xef00] Unknown cluster-specific command 36
[0x781e:1:0xef00] ZCL request 0x0024: b'\x008'
[0x781e:1:0xef00] No handler for cluster command 36
[0x781e:1:0xef00] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=CLUSTER_COMMAND manufacturer_specific=False is_reply=True disable_default_response=False> manufacturer=None tsn=107 command_id=17>
[0x781e:1:0xef00] Unknown cluster-specific command 17
[0x781e:1:0xef00] ZCL request 0x0011: b'\x009B'
[0x781e:1:0xef00] No handler for cluster command 17
[0x781e:1:0xef00] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=CLUSTER_COMMAND manufacturer_specific=False is_reply=True disable_default_response=False> manufacturer=None tsn=107 command_id=17>
[0x781e:1:0xef00] Unknown cluster-specific command 17
[0x781e:1:0xef00] ZCL request 0x0011: b'\x009B'
[0x781e:1:0xef00] No handler for cluster command 17
[0x781e:1:0xef00] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=CLUSTER_COMMAND manufacturer_specific=False is_reply=True disable_default_response=False> manufacturer=None tsn=107 command_id=17>
[0x781e:1:0xef00] Unknown cluster-specific command 17
[0x781e:1:0xef00] ZCL request 0x0011: b'\x009B'
[0x781e:1:0xef00] No handler for cluster command 17
[0x781e:1:0xef00] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=CLUSTER_COMMAND manufacturer_specific=False is_reply=True disable_default_response=False> manufacturer=None tsn=107 command_id=17>
[0x781e:1:0xef00] Unknown cluster-specific command 17
[0x781e:1:0xef00] ZCL request 0x0011: b'\x009B'
[0x781e:1:0xef00] No handler for cluster command 17
[0x781e:1:0xef00] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=CLUSTER_COMMAND manufacturer_specific=False is_reply=True disable_default_response=False> manufacturer=None tsn=107 command_id=17>
[0x781e:1:0xef00] Unknown cluster-specific command 17
[0x781e:1:0xef00] ZCL request 0x0011: b'\x009B'
[0x781e:1:0xef00] No handler for cluster command 17

MattWestb, 2021-07-16:
Can you trying this patched quirk:
`/tuya/ts0601.py`
```
"""Tuya based touch switch."""
from zigpy.profiles import zha
from zigpy.zcl.clusters.general import Basic, Groups, LevelControl, Ota, Scenes, Time

from zhaquirks.const import (
    DEVICE_TYPE,
    ENDPOINTS,
    INPUT_CLUSTERS,
    MODELS_INFO,
    OUTPUT_CLUSTERS,
    PROFILE_ID,
)
from zhaquirks.tuya import (
    TuyaDimmerSwitch,
    TuyaLevelControl,
    TuyaManufacturerClusterOnOff,
    TuyaManufacturerLevelControl,
    TuyaManufCluster,
    TuyaOnOff,
)

class TuyaSingleSwitchDimmer(TuyaDimmerSwitch):
    """Tuya touch switch device."""

    signature = {
        # "node_descriptor": "<NodeDescriptor byte1=1, byte2=64, mac_capability_flags=142, manufacturer_code=4098,
        # maximum_buffer_size=82, maximum_incoming_transfer_size=82, server_mask=11264,
        # maximum_outgoing_transfer_size=82, descriptor_capability_field=0>",
        # <SimpleDescriptor endpoint=1 profile=260 device_type=51 device_version=1 input_clusters=[0, 4, 5, 61184] output_clusters=[10, 25]>
        MODELS_INFO: [
            ("_TZE200_dfxkcots", "TS0601"),
            ("_TZE200_whpb9yts", "TS0601"),
            ("_TZE200_ebwgzdqq", "TS0601"),
            ("_TZE200_9i9dt8is", "TS0601"),
            ("_TZE200_swaamsoy", "TS0601"),
            ("_TZE200_0nauxa0p", "TS0601"),
        ],
        ENDPOINTS: {
            1: {
                PROFILE_ID: zha.PROFILE_ID,
                DEVICE_TYPE: zha.DeviceType.SMART_PLUG,
                INPUT_CLUSTERS: [
                    Basic.cluster_id,
                    Groups.cluster_id,
                    Scenes.cluster_id,
                    TuyaManufCluster.cluster_id,
                ],
                OUTPUT_CLUSTERS: [Time.cluster_id, Ota.cluster_id],
            }
        },
    }

    replacement = {
        ENDPOINTS: {
            1: {
                DEVICE_TYPE: zha.DeviceType.ON_OFF_LIGHT,
                INPUT_CLUSTERS: [
                    Basic.cluster_id,
                    Groups.cluster_id,
                    Scenes.cluster_id,
                    LevelControl.cluster_id,
                    TuyaManufacturerClusterOnOff,
                    TuyaOnOff,
                    TuyaManufacturerLevelControl,
                    TuyaLevelControl,
                ],
                OUTPUT_CLUSTERS: [Time.cluster_id, Ota.cluster_id],
            }
        }
    }
```
Enabling local quirk in your config and copy it in the directory and restarting HA.
You can see if its being loaded on the quirk string in the device info .

MattWestb, 2021-07-16:
For adding local quirks https://github.com/zigpy/zha-device-handlers/discussions/693#discussioncomment-857274 and one file version of the quirks for putting in the local quirk folder (unzip it). [ts0601_dimmer.zip](https://github.com/zigpy/zha-device-handlers/files/6832041/ts0601_dimmer.zip)

SkyFox277, 2021-07-21:
Thanks for your help. But this custom quirk doesn't seem to work. Not with the dimmer bound and not after removing and rebinding. I'll post logs in the next days

MattWestb, 2021-07-21:
Thanks for feed back and i was looking on this switch and was thinking ordering from Spain but i think i passing after you test ;-((

tuya was / is using  its own "MCU protocol " that  is more or less one tunneling commands (DP) over mqtt to there cloud. Then they is putting one WiFi chip (can being on ESP chip), BT, RF433 or Zigbee chip for tunneling the commands from the MCU with one comport. In Zigbee they is normally using cluster 0xef00 that also can being hidden and most device that is working this way is TS0601 with zigbee module TZE200 or TYST11.

Other tuya devises is "normal" zigbee devices like lights and some switches that is sending on/off light level up and down and can being groped and bounded (more or less good) and is working out of  the box and also then the coordinator is not online.

The only thing that is possible doing with "tuya cluster" devices is encoding the commands and using them in automatons like the on/off is by the automation being sent to one group with lights.

Us event listener in HA and listening to `zha_event` and hopefully you is getting the on/off and light level  commands from the dimmer.

I is also on the hunt of one TS004F switch that shall being  one normal zigbee dimmer switch with on/off and light level but looks having problems with encoding all the commands in  ZHA for doing automation. I hope its working good and can being bounded to groups and searing them OK also then the coordinator is off line.

I think all nice tuya touch switches (moes and co) is also tuya MCU ones and is useless for searing light then the coordinator is off line :-((((

Pleas do more test and reporting back !!

PR #1010 (2021-08-25): Adding some more tuya dimmers
Adding more tuya dimmers:
```
            ("_TZE200_whpb9yts", "TS0601"),
            ("_TZE200_ebwgzdqq", "TS0601"),
            ("_TZE200_9i9dt8is", "TS0601"),
            ("_TZE200_swaamsoy", "TS0601"),
            ("_TZE200_0nauxa0p", "TS0601"),
```
And fixes #964
closed 2021-08-25 by dmulcahey