zigpy/zha-device-handlers #965: [Device Support Request] FireAngel ZB-module for CO detectors and smoke detectors

JoeyGnarf, 2021-07-17:
**Is your feature request related to a problem? Please describe.**
I've bought a FireAngel NM-CO-10X-INT carbon monoxide detector with a FireAngel ZB-module. When adding the device it is only detected as a binary sensor and does not trigger a zha_event when pushing the test-button. I'm guessing functionality will be the same for the 2x FireAngel SM-F-1EU smoke detectors that should be delivered to me next week, which use the exact same ZB-module.

**Describe the solution you'd like**
I would like support for an event to be triggered when CO/smoke is detected (and the test-button is pushed)
If at all possible it would be fantastic to be able to trigger the alarm manually (since the devices can be linked (using their custom RF module instead of the ZB-module) I'm hoping this can be done)
Battery level readings would be great

**Device signature - this can be acquired by removing the device from ZHA and pairing it again from the add devices screen. Be sure to add the entire content of the log panel after pairing the device to a code block below this line.**
Device signature
`{
  "node_descriptor": "NodeDescriptor(logical_type=<LogicalType.EndDevice: 2>, complex_descriptor_available=0, user_descriptor_available=0, reserved=0, aps_flags=0, frequency_band=<FrequencyBand.Freq2400MHz: 8>, mac_capability_flags=<MACCapabilityFlags.AllocateAddress|RxOnWhenIdle|MainsPowered: 140>, manufacturer_code=4674, maximum_buffer_size=108, maximum_incoming_transfer_size=0, server_mask=10752, maximum_outgoing_transfer_size=0, descriptor_capability_field=<DescriptorCapability.0: 0>, *allocate_address=True, *is_alternate_pan_coordinator=False, *is_coordinator=False, *is_end_device=True, *is_full_function_device=False, *is_mains_powered=True, *is_receiver_on_when_idle=True, *is_router=False, *is_security_capable=False)",
  "endpoints": {
    "1": {
      "profile_id": 260,
      "device_type": "0x0403",
      "in_clusters": [
        "0x0000",
        "0x0001",
        "0x0003",
        "0x0020",
        "0x0500",
        "0x0502"
      ],
      "out_clusters": [
        "0x0019",
        "0x0502"
      ]
    }
  },
  "manufacturer": "Fireangel",
  "model": "Alarm_SD_Device",
  "class": "zigpy.device.Device"
}`

Log
`[zigpy.application] New device 0x822f (00:15:5f:00:dc:c8:ec:19) joined the network
[zigpy.device] [0x822f] Scheduling initialization
[zigpy.application] Received frame on uninitialized device <Device model=None manuf=None nwk=0x822F ieee=00:15:5f:00:dc:c8:ec:19 is_initialized=False> from ep 0 to ep 0, cluster 19: b'\x01/\x82\x19\xec\xc8\xdc\x00_\x15\x00\x80'
[zigpy.zdo] [0x822f:zdo] ZDO request ZDOCmd.Device_annce: [0x822F, 00:15:5f:00:dc:c8:ec:19, 128]
[zigpy.util] Tries remaining: 3
[zigpy.device] [0x822f] Requesting 'Node Descriptor'
[zigpy.util] Tries remaining: 2
[zigpy.device] [0x822f] Extending timeout for 0x83 request
[zigpy_deconz.zigbee.application] Sending Zigbee request with tsn 131 under 132 request id, data: b'832f82'
[zigpy_deconz.api] 'aps_data_indication' response from <DeconzAddress address_mode=ADDRESS_MODE.NWK address=0x822f>, ep: 0, profile: 0x0000, cluster_id: 0x0013, data: b'012f8219ecc8dc005f150080'
[zigpy_deconz.api] Command Command.aps_data_request (18, 132, 0, <DeconzAddressEndpoint address_mode=2 address=0x822F endpoint=0>, 0, <ZDOCmd.Node_Desc_req: 0x0002>, 0, b'\x83/\x82', 2, 0)
[zigpy_deconz.uart] Send: 0x12c300190012008400022f820000000200000300832f820200
[zigpy_deconz.uart] Frame received: 0x12c300090002002284
[zigpy_deconz.api] APS data request response: [2, <DeviceState.APSDE_DATA_REQUEST_SLOTS_AVAILABLE|2: 34>, 132]
[zigpy_deconz.uart] Frame received: 0x1cc4000c000500022f82fff3
[zigpy_deconz.uart] Frame received: 0x0ec5000700a600
[zigpy_deconz.api] Device state changed response: [<DeviceState.128|APSDE_DATA_REQUEST_SLOTS_AVAILABLE|APSDE_DATA_CONFIRM|2: 166>, 0]
[zigpy_deconz.uart] Frame received: 0x1cc6000c000500022f82fff3
[zigpy_deconz.api] Command Command.aps_data_confirm (0,)
[zigpy_deconz.uart] Send: 0x04c40007000000
[zigpy_deconz.uart] Frame received: 0x04c40013000c002284022f8200000000000000
[zigpy_deconz.api] APS data confirm response for request with id 132: 00
[zigpy_deconz.api] Request id: 0x84 'aps_data_confirm' for <DeconzAddressEndpoint address_mode=ADDRESS_MODE.NWK address=0x822f endpoint=0>, status: 0x00
[zigpy_deconz.uart] Frame received: 0x0ec5000700aa00
[zigpy_deconz.api] Device state changed response: [<DeviceState.128|APSDE_DATA_REQUEST_SLOTS_AVAILABLE|APSDE_DATA_INDICATION|2: 170>, 0]
[zigpy_deconz.uart] Frame received: 0x1cc6000c000500022f82fff3
[zigpy_deconz.api] Command Command.aps_data_indication (1, 1)
[zigpy_deconz.uart] Send: 0x17c5000800010001
[zigpy_deconz.uart] Frame received: 0x17c5002f0028002202000000022f820000000280110083002f8202408c42126c0000002a00000000afff96e40000f3
[zigpy_deconz.api] APS data indication response: [40, <DeviceState.APSDE_DATA_REQUEST_SLOTS_AVAILABLE|2: 34>, <DeconzAddress address_mode=ADDRESS_MODE.NWK address=0x0000>, 0, <DeconzAddress address_mode=ADDRESS_MODE.NWK address=0x822f>, 0, 0, 32770, b'\x83\x00/\x82\x02@\x8cB\x12l\x00\x00\x00*\x00\x00\x00', 0, 175, 255, 150, 228, 0, 0, -13]
[zigpy.application] Received frame on uninitialized device <Device model=None manuf=None nwk=0x822F ieee=00:15:5f:00:dc:c8:ec:19 is_initialized=False> from ep 0 to ep 0, cluster 32770: b'\x83\x00/\x82\x02@\x8cB\x12l\x00\x00\x00*\x00\x00\x00'
[zigpy_deconz.api] 'aps_data_indication' response from <DeconzAddress address_mode=ADDRESS_MODE.NWK address=0x822f>, ep: 0, profile: 0x0000, cluster_id: 0x8002, data: b'83002f8202408c42126c0000002a000000'
[zigpy.device] [0x822f] Got Node Descriptor: NodeDescriptor(logical_type=<LogicalType.EndDevice: 2>, complex_descriptor_available=0, user_descriptor_available=0, reserved=0, aps_flags=0, frequency_band=<FrequencyBand.Freq2400MHz: 8>, mac_capability_flags=<MACCapabilityFlags.AllocateAddress|RxOnWhenIdle|MainsPowered: 140>, manufacturer_code=4674, maximum_buffer_size=108, maximum_incoming_transfer_size=0, server_mask=10752, maximum_outgoing_transfer_size=0, descriptor_capability_field=<DescriptorCapability.0: 0>, *allocate_address=True, *is_alternate_pan_coordinator=False, *is_coordinator=False, *is_end_device=True, *is_full_function_device=False, *is_mains_powered=True, *is_receiver_on_when_idle=True, *is_router=False, *is_security_capable=False)
[zigpy.device] [0x822f] Discovering endpoints
[... 454 lines cut ...]
[zigpy_deconz.uart] Frame received: 0x1cfb000c000500022f82ffdf
[zigpy_deconz.uart] Frame received: 0x1cfc000c000500022f82ffdf
[zigpy_deconz.uart] Frame received: 0x1cfd000c000500022f82ffdf
[zigpy_deconz.uart] Frame received: 0x1cfe000c000500022f82ffde
[zigpy_deconz.uart] Frame received: 0x1cff000c000500022f82ffdf
[zigpy_deconz.uart] Frame received: 0x1c00000c000500022f82ffdf
[zigpy_deconz.uart] Frame received: 0x1c01000c000500022f82ffdf
[zigpy_deconz.uart] Frame received: 0x1c02000c000500022f82ffdf
[zigpy_deconz.uart] Frame received: 0x1c03000c000500022f82ffdf
[zigpy_deconz.uart] Frame received: 0x1c04000c000500022f82ffdf
[zigpy_deconz.uart] Frame received: 0x1c05000c000500022f82ffdf
[zigpy_deconz.uart] Frame received: 0x1c06000c000500022f82ffdf
[zigpy_deconz.uart] Frame received: 0x1c07000c000500022f82ffdf
[zigpy_deconz.uart] Frame received: 0x1c08000c000500022f82ffdf
[zigpy_deconz.uart] Frame received: 0x1c09000c000500022f82ffdf
[zigpy_deconz.uart] Frame received: 0x1c0a000c000500022f82ffdf
[zigpy_deconz.uart] Frame received: 0x1c0b000c000500022f82ffdf
[zigpy_deconz.uart] Frame received: 0x1c0c000c000500022f82ffdf
[zigpy_deconz.uart] Frame received: 0x1c0d000c000500022f82ffdf
[zigpy_deconz.uart] Frame received: 0x1c0e000c000500022f82ffdf`

**Additional context**
EAN 816317005078

Many thanks in advance!

MattWestb, 2021-07-17:
Pleas use:

` ``` `
Log text
` ``` `
for not loosing the code formatting and its possible reading the log OK ;-))

MattWestb, 2021-07-17:
GREAT !!!!

Thanks :-)

closed 2022-01-20 by github-actions[bot]

LanMart, 2022-02-12:
Hi All,

I hava a similar request. I started with the smoke detector, and did not unbox the CO detector yet, as they probably have the same Zigbee module. I did not succeed in adding the module at all, not even as a binary detector.

At Gamma I bought AngelEye/FireAngel smoke detectors and en CO-detector, all with a Zigbee module (P-line).

I'm using a Rasberry Pi with a ConBee stick and I cannot connect the smoke detector with my HomeAssistant.
Am I doing something wrong? Or is there a compatibility issue?
I was able to pair the smoke-detector with the ICS-2000 at my parent's, so it seems to work fine.

Is there an update on this topic?

Martijn

bramd, 2022-08-09:
@LanMart Did you solve this? I also have the same devices I think. At first, I got them paired to the Zigbee network and they showed up as alarm/siren devices. I never tried if I could get test events though. I recently had to rebuild my Zigbee network and now I can't get them connected anymore.

The manual for the Zigbee module states that I should reset the device by holding the pairing button for 5 sec. Based on the LED the device responds to this reset, but afterwards I can't add it in any way. I suspect the module has issues when it has been paired before and doesn't reset properly somehow. Did you pair your device to another Zigbee network before having this problem?

LanMart, 2022-08-10:
@bramd No, unfortunately I did not solve this. They do not show up at all, not even as a siren. I tested one of the units at my parents trust hub, and it dit work. It seems a compatibility problem. I opened a feature request : 

https://community.home-assistant.io/t/support-for-angeleye-fireangel-smoke-detectors-and-en-co-detector-all-with-a-zigbee-module-p-line/449050?u=martlan

bramd, 2022-10-11:
I contacted Fireangel support and got a somewhat confusing response. 
They suggest that resetting the Zigbee module is not enough, but that it 
should be disconnected from the alarm after reset and that you should 
pull the battery for 5 minutes. That is not in the manual, but I will 
give it a try when I'm back home and have some time.

That being said, my problem is probably different then yours, since my 
alarms worked the first time but now refuse to pair in any way.