zigpy/zha-device-handlers #966: [Device Support Request] TZE200_tz32mtza TS0601 (3 gang Moes no neutral wired switch)

Netatout, 2021-07-17:
**Is your feature request related to a problem? Please describe.**
Device pairs successfully, but no entities are created. 

**Describe the solution you'd like**
I would like a quirk that handles this device and exposes control over both gangs via ZHA. Each switch has a led that can be deactivated. If possible, also have access to this setting.

**Device signature - this can be acquired by removing the device from ZHA and pairing it again from the add devices screen. Be sure to add the entire content of the log panel after pairing the device to a code block below this line.**
Device 0x4286 (60:a4:23:ff:fe:aa:ec:42) joined the network
Device 60:a4:23:ff:fe:aa:ec:42 changed id (0x4286 => 0x4286)
[0x4286] Scheduling initialization
Received frame on uninitialized device <Device model=None manuf=None nwk=0x4286 ieee=60:a4:23:ff:fe:aa:ec:42 is_initialized=False> from ep 0 to ep 0, cluster ZDOCmd.Device_annce: b'\xFF\x86\x42\x42\xEC\xAA\xFE\xFF\x23\xA4\x60\x80'
[0x4286:zdo] ZDO request ZDOCmd.Device_annce: [0x4286, 60:a4:23:ff:fe:aa:ec:42, 128]
Tries remaining: 3
[0x4286] Requesting 'Node Descriptor'
Tries remaining: 2
[0x4286] Extending timeout for 0x47 request
Received frame on uninitialized device <Device model=None manuf=None nwk=0x4286 ieee=60:a4:23:ff:fe:aa:ec:42 is_initialized=False> from ep 0 to ep 0, cluster ZDOCmd.Node_Desc_rsp: b'\x47\x00\x86\x42\x02\x40\x80\x02\x10\x52\x52\x00\x00\x2C\x52\x00\x00'
[0x4286] Got Node Descriptor: NodeDescriptor(logical_type=<LogicalType.EndDevice: 2>, complex_descriptor_available=0, user_descriptor_available=0, reserved=0, aps_flags=0, frequency_band=<FrequencyBand.Freq2400MHz: 8>, mac_capability_flags=<MACCapabilityFlags.AllocateAddress: 128>, manufacturer_code=4098, maximum_buffer_size=82, maximum_incoming_transfer_size=82, server_mask=11264, maximum_outgoing_transfer_size=82, descriptor_capability_field=<DescriptorCapability.0: 0>, *allocate_address=True, *is_alternate_pan_coordinator=False, *is_coordinator=False, *is_end_device=True, *is_full_function_device=False, *is_mains_powered=False, *is_receiver_on_when_idle=False, *is_router=False, *is_security_capable=False)
[0x4286] Discovering endpoints
Tries remaining: 3
[0x4286] Extending timeout for 0x48 request
Received frame on uninitialized device <Device model=None manuf=None nwk=0x4286 ieee=60:a4:23:ff:fe:aa:ec:42 is_initialized=False> from ep 0 to ep 0, cluster ZDOCmd.Active_EP_rsp: b'\x48\x00\x86\x42\x01\x01'
[0x4286] Discovered endpoints: [1]
[0x4286] Initializing endpoints [<Endpoint id=1 in=[] out=[] status=<Status.NEW: 0>>]
[0x4286:1] Discovering endpoint information
Tries remaining: 3
[0x4286] Extending timeout for 0x49 request
Received frame on uninitialized device <Device model=None manuf=None nwk=0x4286 ieee=60:a4:23:ff:fe:aa:ec:42 is_initialized=False> from ep 0 to ep 0, cluster ZDOCmd.Simple_Desc_rsp: b'\x49\x00\x86\x42\x14\x01\x04\x01\x51\x00\x01\x04\x00\x00\x04\x00\x05\x00\x00\xEF\x02\x19\x00\x0A\x00'
[0x4286:1] Discovered endpoint information: SizePrefixedSimpleDescriptor(endpoint=1, profile=260, device_type=81, device_version=1, input_clusters=[0, 4, 5, 61184], output_clusters=[25, 10])
Unknown cluster 61184
[0x4286] Extending timeout for 0x4a request
[0x4286:1:0x0000] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=GLOBAL_COMMAND manufacturer_specific=False is_reply=True disable_default_response=True> manufacturer=None tsn=74 command_id=Command.Read_Attributes_rsp>
[0x4286] Read model 'TS0601' and manufacturer '_TZE200_tz32mtza' from <Endpoint id=1 in=[basic:0x0000, groups:0x0004, scenes:0x0005, None:0xEF00] out=[ota:0x0019, time:0x000A] status=<Status.ZDO_INIT: 1>>
[0x4286] Discovered basic device information for <Device model='TS0601' manuf='_TZE200_tz32mtza' nwk=0x4286 ieee=60:a4:23:ff:fe:aa:ec:42 is_initialized=True>
Device is initialized <Device model='TS0601' manuf='_TZE200_tz32mtza' nwk=0x4286 ieee=60:a4:23:ff:fe:aa:ec:42 is_initialized=True>
Checking quirks for _TZE200_tz32mtza TS0601 (60:a4:23:ff:fe:aa:ec:42)
Considering <class 'zhaquirks.xbee.xbee_io.XBeeSensor'>
Fail because endpoint list mismatch: {232, 230} {1}
Considering <class 'zhaquirks.xbee.xbee3_io.XBee3Sensor'>
Fail because endpoint list mismatch: {232, 230} {1}
Considering <class 'zhaquirks.smartthings.tag_v4.SmartThingsTagV4'>
Fail because device_type mismatch on at least one endpoint
Considering <class 'zhaquirks.smartthings.multi.SmartthingsMultiPurposeSensor'>
Fail because device_type mismatch on at least one endpoint
Considering <class 'zhaquirks.netvox.z308e3ed.Z308E3ED'>
Fail because device_type mismatch on at least one endpoint
Considering <class 'zhaquirks.gledopto.soposhgu10.SoposhGU10'>
Fail because endpoint list mismatch: {11, 13} {1}
Considering <class 'bellows.zigbee.application.EZSPCoordinator'>
Fail because device_type mismatch on at least one endpoint
device - 0x4286:60:a4:23:ff:fe:aa:ec:42 entering async_device_initialized - is_new_join: True
device - 0x4286:60:a4:23:ff:fe:aa:ec:42 has joined the ZHA zigbee network
[0x4286](TS0601): started configuration
[0x4286:ZDO](TS0601): 'async_configure' stage succeeded
[0x4286:1:0x0000]: finished channel configuration
[0x4286:1:0x0019]: finished channel configuration
Error handling '_save_attribute' event with (60:a4:23:ff:fe:aa:ec:42, 1, 0, 4, '_TZE200_tz32mtza') params: FOREIGN KEY constraint failed
[0x4286:1:0x0000]: 'async_configure' stage succeeded
[0x4286:1:0x0019]: 'async_configure' stage succeeded
Error handling '_save_attribute' event with (60:a4:23:ff:fe:aa:ec:42, 1, 0, 5, 'TS0601') params: FOREIGN KEY constraint failed
[0x4286](TS0601): completed configuration
[0x4286](TS0601): stored in registry: ZhaDeviceEntry(name='_TZE200_tz32mtza TS0601', ieee='60:a4:23:ff:fe:aa:ec:42', last_seen=1626517183.0988317)
[0x4286](TS0601): started initialization
[0x4286:ZDO](TS0601): 'async_initialize' stage succeeded
[0x4286:1:0x0000]: initializing channel: from_cache: False
[0x4286:1:0x0000]: finished channel configuration
[0x4286:1:0x0019]: initializing channel: from_cache: False
[0x4286:1:0x0019]: finished channel configuration
[0x4286:1:0x0000]: 'async_initialize' stage succeeded
[0x4286:1:0x0019]: 'async_initialize' stage succeeded
[... 95 lines cut ...]

**Additional context**
Signature of the Zigbee device in HA :
{
  "node_descriptor": "NodeDescriptor(logical_type=<LogicalType.EndDevice: 2>, complex_descriptor_available=0, user_descriptor_available=0, reserved=0, aps_flags=0, frequency_band=<FrequencyBand.Freq2400MHz: 8>, mac_capability_flags=<MACCapabilityFlags.AllocateAddress: 128>, manufacturer_code=4098, maximum_buffer_size=82, maximum_incoming_transfer_size=82, server_mask=11264, maximum_outgoing_transfer_size=82, descriptor_capability_field=<DescriptorCapability.0: 0>, *allocate_address=True, *is_alternate_pan_coordinator=False, *is_coordinator=False, *is_end_device=True, *is_full_function_device=False, *is_mains_powered=False, *is_receiver_on_when_idle=False, *is_router=False, *is_security_capable=False)",
  "endpoints": {
    "1": {
      "profile_id": 260,
      "device_type": "0x0051",
      "in_clusters": [
        "0x0000",
        "0x0004",
        "0x0005",
        "0xef00"
      ],
      "out_clusters": [
        "0x000a",
        "0x0019"
      ]
    }
  },
  "manufacturer": "_TZE200_tz32mtza",
  "model": "TS0601",
  "class": "zigpy.device.Device"
}

MattWestb, 2021-07-17:
Looks being one variant of one already supported tuya touch switch.

I have adding it and some more that is have the same function in Z2M in the ts0601.py.
Configure local quirk in ZHA https://github.com/zigpy/zha-device-handlers/discussions/693#discussioncomment-857274 and putting the new (unziped) ts0601.py in the local quirk folder and restarting HA.
[ts0601.zip](https://github.com/zigpy/zha-device-handlers/files/6834626/ts0601.zip)

Netatout, 2021-07-18:
Thank you @MattWestb. HA now recognizes one of the 3 buttons. How do I get access to the others.

MattWestb, 2021-07-18:
Im  not one code worrier so i cant digging so deep in the code but its looks the devs have implanting functions for the single switch and not for the 2-4 ones and i dont knowing how to patching the code for getting the events from the extra inputs.

Do you have some debug logs then using the switch 1 and the other switches on the device ?

The interesting the tuya DP ZHA is receiving from the device.

MattWestb, 2021-07-28:
I have adding the quirk and its merged in the dev branch.
I have one user with one 2 button switch  that have the same problem hi only getting events from the frost switch gang but hi is having "hardware problems" for the moment and cant doing some tests ;-((

Can you pleas enabling debug logging for:
```
    homeassistant.components.zha: debug
    zigpy: debug
    zhaquirks.tuya: debug
```
And pressing button 1 to 3 and look if you is getting tuya DP reports in the log ?

LunarstarPony, 2021-10-03:
I'm also having the same issue that only first gang can be controlled by Home Assistant (TZE200_wunufsil) 2 gang

MattWestb, 2021-10-03:
We need help from the core devs for fixing this and also we dont have getting any debug logs from devices then pressing the not working buttons with the debut setting in post https://github.com/zigpy/zha-device-handlers/issues/966#issuecomment-888172098.

I hope this can being fixed for multi gang switches and also for the multi dimmers from tuya with tuya MCU (TS0601 devices).

MPoppinga, 2021-11-06:
I have the same issue: The events are nearly the same. The 3 and 4  last message for each event are differing.
I pressed gang 1, 2 3 once each.

logs of zigpy
```
[zigpy.zcl] [0x5bc4:1:0xef00] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=CLUSTER_COMMAND manufacturer_specific=False is_reply=True disable_default_response=False> manufacturer=None tsn=5 command_id=1>
[zigpy.zcl] [0x5bc4:1:0xef00] ZCL request 0x0001: [Command(status=1, tsn=14, command_id=257, function=0, data=[1, 1])]
[zigpy.zcl] [0x5bc4:1:0xef00] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=CLUSTER_COMMAND manufacturer_specific=False is_reply=True disable_default_response=False> manufacturer=None tsn=5 command_id=1>
[zigpy.zcl] [0x5bc4:1:0xef00] ZCL request 0x0001: [Command(status=1, tsn=14, command_id=257, function=0, data=[1, 1])]
[zigpy.zcl] [0x5bc4:1:0xef00] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=CLUSTER_COMMAND manufacturer_specific=False is_reply=True disable_default_response=False> manufacturer=None tsn=5 command_id=1>
[zigpy.zcl] [0x5bc4:1:0xef00] ZCL request 0x0001: [Command(status=1, tsn=14, command_id=257, function=0, data=[1, 1])]
[zigpy.zcl] [0x5bc4:1:0xef00] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=CLUSTER_COMMAND manufacturer_specific=False is_reply=True disable_default_response=False> manufacturer=None tsn=5 command_id=1>
[zigpy.zcl] [0x5bc4:1:0xef00] ZCL request 0x0001: [Command(status=1, tsn=14, command_id=257, function=0, data=[1, 1])]
[zigpy.zcl] [0x5bc4:1:0xef00] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=CLUSTER_COMMAND manufacturer_specific=False is_reply=True disable_default_response=False> manufacturer=None tsn=5 command_id=1>
[zigpy.zcl] [0x5bc4:1:0xef00] ZCL request 0x0001: [Command(status=1, tsn=14, command_id=257, function=0, data=[1, 1])]
[zigpy.zcl] [0x5bc4:1:0xef00] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=CLUSTER_COMMAND manufacturer_specific=False is_reply=True disable_default_response=False> manufacturer=None tsn=5 command_id=1>
[zigpy.zcl] [0x5bc4:1:0xef00] ZCL request 0x0001: [Command(status=1, tsn=14, command_id=257, function=0, data=[1, 1])]
[zigpy.zcl] [0x5bc4:1:0xef00] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=CLUSTER_COMMAND manufacturer_specific=False is_reply=True disable_default_response=False> manufacturer=None tsn=5 command_id=1>
[zigpy.zcl] [0x5bc4:1:0xef00] ZCL request 0x0001: [Command(status=1, tsn=14, command_id=257, function=0, data=[1, 1])]
[zigpy.zcl] [0x5bc4:1:0xef00] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=CLUSTER_COMMAND manufacturer_specific=False is_reply=True disable_default_response=False> manufacturer=None tsn=5 command_id=1>
[zigpy.zcl] [0x5bc4:1:0xef00] ZCL request 0x0001: [Command(status=1, tsn=14, command_id=257, function=0, data=[1, 1])]
[zigpy.zcl] [0x5bc4:1:0xef00] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=CLUSTER_COMMAND manufacturer_specific=False is_reply=True disable_default_response=False> manufacturer=None tsn=5 command_id=1>
[zigpy.zcl] [0x5bc4:1:0xef00] ZCL request 0x0001: [Command(status=1, tsn=14, command_id=257, function=0, data=[1, 1])]
[zigpy.zcl] [0x5bc4:1:0xef00] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=CLUSTER_COMMAND manufacturer_specific=False is_reply=True disable_default_response=False> manufacturer=None tsn=5 command_id=1>
[zigpy.zcl] [0x5bc4:1:0xef00] ZCL request 0x0001: [Command(status=1, tsn=14, command_id=257, function=0, data=[1, 1])]
[zigpy.zcl] [0x5bc4:1:0xef00] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=CLUSTER_COMMAND manufacturer_specific=False is_reply=True disable_default_response=False> manufacturer=None tsn=5 command_id=1>
[zigpy.zcl] [0x5bc4:1:0xef00] ZCL request 0x0001: [Command(status=1, tsn=14, command_id=257, function=0, data=[1, 1])]
[zigpy.zcl] [0x5bc4:1:0xef00] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=CLUSTER_COMMAND manufacturer_specific=False is_reply=True disable_default_response=False> manufacturer=None tsn=5 command_id=1>
[zigpy.zcl] [0x5bc4:1:0xef00] ZCL request 0x0001: [Command(status=1, tsn=14, command_id=257, function=0, data=[1, 1])]
[zigpy.zcl] [0x5bc4:1:0xef00] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=CLUSTER_COMMAND manufacturer_specific=False is_reply=True disable_default_response=False> manufacturer=None tsn=6 command_id=1>
[zigpy.zcl] [0x5bc4:1:0xef00] ZCL request 0x0001: [Command(status=1, tsn=15, command_id=258, function=0, data=[1, 0])]
[zigpy.zcl] [0x5bc4:1:0xef00] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=CLUSTER_COMMAND manufacturer_specific=False is_reply=True disable_default_response=False> manufacturer=None tsn=7 command_id=1>
[zigpy.zcl] [0x5bc4:1:0xef00] ZCL request 0x0001: [Command(status=1, tsn=16, command_id=259, function=0, data=[1, 1])]
[zigpy.zcl] [0x5bc4:1:0xef00] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=CLUSTER_COMMAND manufacturer_specific=False is_reply=True disable_default_response=False> manufacturer=None tsn=8 command_id=1>
[zigpy.zcl] [0x5bc4:1:0xef00] ZCL request 0x0001: [Command(status=1, tsn=17, command_id=1038, function=0, data=[1, 0])]
[... cut ...]
[zhaquirks.tuya] 60:a4:23:ff:fe:e2:a4:67 - Received switch event message, channel: 1, state: 0
[zhaquirks.tuya] 60:a4:23:ff:fe:e2:a4:67 - Received switch event message, channel: 1, state: 0
[zhaquirks.tuya] 60:a4:23:ff:fe:e2:a4:67 - Received switch event message, channel: 2, state: 0
[zhaquirks.tuya] 60:a4:23:ff:fe:e2:a4:67 - Received switch event message, channel: 782, state: 0
[zhaquirks.tuya] 60:a4:23:ff:fe:e2:a4:67 - Received switch event message, channel: 783, state: 2
```

let me know if I can provide more information.

MattWestb, 2021-11-06:
One user have one experimental quirk that hi is testing but was having some problems but hi is needing more user testing it and also if having more then one switch is being good because its not tested with multiple switches in the same network.
Take one look in this and if you is brave testing how it working with your multi gang switch https://github.com/zigpy/zha-device-handlers/issues/1038.

MPoppinga, 2021-11-07:
Hi, I used the quirk from @javicalle in  #1038 and made three small changes to let it work with 3 gangs. (added 3rd endpoint, added  0x0010 command and added the model info)

It seems to be working now with all 3 gangs for TZE200_tz32mtza. It has a little delay if using the switch until it shows in HA, aside of that ,it seems to be working fine, but I have only one of these in my network.

```

"""Tuya based touch switch."""
import logging
from typing import Any, Callable, Dict, List, Optional, Tuple, Union

from zigpy.profiles import zha
from zigpy.quirks import CustomCluster
import zigpy.types as t
from zigpy.zcl import Cluster, foundation
from zigpy.zcl.clusters.general import Basic, Groups, OnOff, Ota, Scenes, Time

from zhaquirks import EventableCluster
from zhaquirks.const import (
    DEVICE_TYPE,
    ENDPOINTS,
    INPUT_CLUSTERS,
    MODELS_INFO,
    OUTPUT_CLUSTERS,
    PROFILE_ID,
)

from zhaquirks.tuya import (
    Data,
    TuyaManufacturerClusterOnOff,
    TuyaManufacturerWindowCover,
    TuyaManufCluster,
    # TuyaOnOff,
    TuyaSwitch,
    ATTR_ON_OFF,
    SWITCH_EVENT,
    TUYA_CMD_BASE,
    TUYA_SET_DATA,
)

_LOGGER = logging.getLogger(__name__)

class TuyaOnOff_Base(CustomCluster, OnOff):
    """Tuya On/Off base cluster for multibutton On/Off device."""

    _common_endpoint = None

    @classmethod
    def set_endpoint(cls, endpoint):
        """Set the conductor endpoint to use."""
        cls._common_endpoint = endpoint

    @classmethod
    def get_endpoint(cls):
        """Get the conductor endpoint."""
        return cls._common_endpoint

    def __init__(self, *args, **kwargs):
        """Init."""
        super().__init__(*args, **kwargs)
        self.endpoint.device.switch_bus.add_listener(self)

    def switch_event(self, channel, state):
        """Switch event."""
        _LOGGER.debug(
            "%s - Received switch event message --> channel: %d, state: %d",
            self.endpoint.device.ieee,
            channel,
            state,
        )
        # update status only if event == endpoint
        if self.endpoint.endpoint_id == channel:
            self._update_attribute(ATTR_ON_OFF, state)

    def command(
            self,
            command_id: Union[foundation.Command, int, t.uint8_t],
            *args,
            manufacturer: Optional[Union[int, t.uint16_t]] = None,
            expect_reply: bool = True,
            tsn: Optional[Union[int, t.uint8_t]] = None,
    ):
        """Override the default Cluster command."""

        _LOGGER.debug(
            "%s - command --> id: %d, endpoint: %d",
            self.endpoint.device.ieee,
            command_id,
            self.endpoint.endpoint_id,
        )

        if command_id in (0x0000, 0x0001, 0x0010):
            cmd_payload = TuyaManufCluster.Command()
            cmd_payload.status = 0
            cmd_payload.tsn = 0
            cmd_payload.command_id = TUYA_CMD_BASE + self.endpoint.endpoint_id
            cmd_payload.function = 0
            cmd_payload.data = [1, command_id]

            return TuyaOnOff_Base.get_endpoint().tuya_manufacturer.command(
                TUYA_SET_DATA, cmd_payload, expect_reply=True
            )

        return foundation.Status.UNSUP_CLUSTER_COMMAND

class TuyaOnOff_conductor(TuyaOnOff_Base):
    """Tuya OnOff conductor button."""

    name = "TuyaOnOff_conductor"

    def __init__(self, *args, **kwargs):
        """Init."""
        super().__init__(*args, **kwargs)
        TuyaOnOff_Base.set_endpoint(self.endpoint)

class TuyaOnOff_follower(TuyaOnOff_Base):
    """Tuya OnOff fake buttons."""

    name = "TuyaOnOff_follower"

class TuyaTripleSwitch(TuyaSwitch):
    """Tuya  triple switch device."""

    signature = {
        # "node_descriptor": "<NodeDescriptor byte1=1 byte2=64 mac_capability_flags=142 manufacturer_code=4098
        #                       maximum_buffer_size=82 maximum_incoming_transfer_size=82 server_mask=11264
        #                       maximum_outgoing_transfer_size=82 descriptor_capability_field=0>",
        # device_version=1
        # input_clusters=[0x0000, 0x0004, 0x0005, 0xef00]
        # output_clusters=[0x000a, 0x0019]
        # <SimpleDescriptor endpoint=1 profile=260 device_type=51 device_version=1 input_clusters=[0, 4, 5, 61184] output_clusters=[10, 25]>
        MODELS_INFO: [
            ("_TZE200_tz32mtza", "TS0601"),
        ],
        ENDPOINTS: {
            1: {
                PROFILE_ID: zha.PROFILE_ID,
                DEVICE_TYPE: zha.DeviceType.SMART_PLUG,
                INPUT_CLUSTERS: [
                    Basic.cluster_id,
                    Groups.cluster_id,
                    Scenes.cluster_id,
                    TuyaManufCluster.cluster_id,
                ],
                OUTPUT_CLUSTERS: [Time.cluster_id, Ota.cluster_id],
            }
        },
    }

    replacement = {
        ENDPOINTS: {
            1: {
                DEVICE_TYPE: zha.DeviceType.ON_OFF_LIGHT,
                INPUT_CLUSTERS: [
                    Basic.cluster_id,
                    Groups.cluster_id,
                    Scenes.cluster_id,
                    TuyaManufacturerClusterOnOff,
                    TuyaOnOff_conductor,
                ],
                OUTPUT_CLUSTERS: [Time.cluster_id, Ota.cluster_id],
            },
            2: {
                PROFILE_ID: zha.PROFILE_ID,
                DEVICE_TYPE: zha.DeviceType.ON_OFF_LIGHT,
                INPUT_CLUSTERS: [
                    TuyaOnOff_follower,
                ],
                OUTPUT_CLUSTERS: [],
            },
            3: {
                PROFILE_ID: zha.PROFILE_ID,
                DEVICE_TYPE: zha.DeviceType.ON_OFF_LIGHT,
                INPUT_CLUSTERS: [
                    TuyaOnOff_follower,
                ],
                OUTPUT_CLUSTERS: [],
            }
        }
    }
```

MattWestb, 2021-11-07:
Great done !!

Try applying the patch for default response for the cluster that the same user have doing https://github.com/zigpy/zha-device-handlers/pull/1125.
I must being made i the HA docker container or is the custom quirk not getting the changed code.

MPoppinga, 2021-11-07:
I changed the __init__.py, and now the delay is gone. Everything seems to work now like a charm.

javicalle, 2021-11-07:
Hi there,
 I have been testing another (IMHO) better approach. Instead of using the singleton, the commands will be managed through a specific Bus for commands directed to the MCU.

If you would be kind enough to test your device with this implementation:
```python
TUYA_MCU_COMMAND="tuya_mcu_command"

class TuyaMCUSwitch(TuyaSwitch):
    """Tuya device with a MCU."""

    def __init__(self, *args, **kwargs):
        """Init device with a Bus for MCU commands."""
        self.commands_bus = Bus()
        super().__init__(*args, **kwargs)

class TuyaMCUCluster(CustomCluster):
    """Tuya cluster for MCU entities communication."""

    supported_mcu_commands = ()

    def __init__(self, *args, **kwargs):
        """Init."""
        super().__init__(*args, **kwargs)
        self.supported_mcu_commands = self.server_commands.keys() # it should be all the cluster commands

    async def command(
        self,
        command_id: Union[foundation.Command, int, t.uint8_t],
        *args,
        manufacturer: Optional[Union[int, t.uint16_t]] = None,
        expect_reply: bool = True,
        tsn: Optional[Union[int, t.uint8_t]] = None,
    ):
        """Cluster commands are sent to the commands_bus listener."""

        _LOGGER.debug(
            "%s - command --> id: %d, endpoint: %d",
            self.endpoint.device.ieee,
            command_id,
            self.endpoint.endpoint_id,
        )

        if command_id in self.supported_mcu_commands:
            # commands are passed to commands_bus
            self.endpoint.device.commands_bus.listener_event(
                TUYA_MCU_COMMAND,
                self.endpoint.endpoint_id,
                command_id,
            )
            return foundation.Status.SUCCESS

        _LOGGER.warning(
            "[0x%04x:%s:0x%04x] unsupported MCU command_id: %d",
            self.endpoint.device.nwk,
            self.endpoint.endpoint_id,
            self.cluster_id,
            command_id,
        )
        return foundation.Status.UNSUP_CLUSTER_COMMAND

    def tuya_mcu_command(self, channel, command):
        """Tuya MCU command listener. Only endpoint:1 must listen to MCU commands."""
        _LOGGER.debug(
            "[0x%04x:%s:0x%04x] tuya_mcu_command --> channel: %d, command: %d",
            self.endpoint.device.nwk,
            self.endpoint.endpoint_id,
            self.cluster_id,
            channel,
            command,
        )

        if command in self.supported_mcu_commands:
            cmd_payload = TuyaManufCluster.Command()
            cmd_payload.status = 0
            cmd_payload.tsn = self.endpoint.device.application.get_sequence()
            cmd_payload.command_id = TUYA_CMD_BASE + channel
            cmd_payload.function = 0
            cmd_payload.data = [1, command]

            self.create_catching_task(
                self.endpoint.tuya_manufacturer.command(
                    TUYA_SET_DATA, cmd_payload, expect_reply=True
                )
            )
        else:
            _LOGGER.warning(
                "[0x%04x:%s:0x%04x] unsupported command: %d",
                self.endpoint.device.nwk,
                self.endpoint.endpoint_id,
                self.cluster_id,
                command,
            )

class TuyaOnOff_Base(TuyaMCUCluster, OnOff):
    """Tuya On/Off base cluster for MCU multibutton On/Off device."""

    def __init__(self, *args, **kwargs):
        """Init."""
        super().__init__(*args, **kwargs)
        self.endpoint.device.switch_bus.add_listener(self)
        self.supported_mcu_commands = (0x0000, 0x0001) # only OFF / ON are working as expected

    def switch_event(self, channel, state):
        """Switch event."""
        _LOGGER.debug(
            "%s - Received switch event message --> channel: %d, state: %d",
            self.endpoint.device.ieee,
            channel,
            state,
        )
        # update status only if event == endpoint
        if self.endpoint.endpoint_id == channel:
            self._update_attribute(ATTR_ON_OFF, state)

class TuyaOnOff_conductor(TuyaOnOff_Base):
    """Tuya On/Off cluster for endpoint:1 entity."""

    def __init__(self, *args, **kwargs):
        """Init."""
        super().__init__(*args, **kwargs)
        self.endpoint.device.commands_bus.add_listener(self) # Cluster for endpoint: 1 (listen MCU commands)

class TuyaOnOff_follower(TuyaOnOff_Base):
    """Tuya On/Off cluster for entities =! endpoint: 1."""

```

Replace all the `TuyaOnOff_Base`, `TuyaOnOff_conductor` and `TuyaOnOff_follower` with this content and restart deleting the `__pycache__`.

MPoppinga, 2021-11-07:
Do I need to make changes in the TuyaTriple switch?
I get the folloing error

`    self.endpoint.device.commands_bus.add_listener(self) # Cluster for endpoint: 1 (listen MCU commands)
AttributeError: 'TuyaTripleSwitch' object has no attribute 'commands_bus'`

MPoppinga, 2021-11-07:
Ah, I, of cause needed to inherit from TuyaMCUSwitch.

It took a minute to set up in HA, but now it also works without problems.

javicalle, 2021-11-07:
> Ah, I, of cause needed to inherit from TuyaMCUSwitch.

Yes, my fault.

> It took a minute to set up in HA, but now it also works without problems.

Thanks for testing.

Is it properly updating the status when you use the wall device and from HA?
Any trace of error or problem in HA?

MPoppinga, 2021-11-07:
Everything is updating as expected. 
I have a slight delay if I switch fast between on and off (0.5 -1.5 sec) and sometimes a random delay, but that's something I have experienced on some other ZigBee devices.

So far no warnings or errors n the logs.

javicalle, 2021-11-07:
> added 0x0010 command

@MPoppinga can you tell me which command is it?

MPoppinga, 2021-11-07:
Ah, that was my fault, as I misinterpreted the part of the code and the logs. (Forgot once to delete the pycache and misinterpreted the suddenly working code, I thought that I could activate the third gang with this change.)

As far as I see was this nonsense and is **not needed.** In the latest version, I only use the two commands which you originally used.

Aside from the TuyaTripleSwitch https://github.com/zigpy/zha-device-handlers/issues/966#issuecomment-962528460 I'm fully using your code from https://github.com/zigpy/zha-device-handlers/issues/966#issuecomment-962598181 and the patch in https://github.com/zigpy/zha-device-handlers/pull/1125

I currently have sometimes some heavy delays in communication.

MattWestb, 2021-11-07:
https://github.com/zigpy/zha-device-handlers/pull/1125 if fixing the delay with multi commands.

javicalle, 2021-11-07:
> As far as I see was this nonsense and is **not needed.** In the latest version, I only use the two commands which you originally used.

Cool. I was worried that I might have missed something.

> I currently have sometimes some heavy delays in communication.

This is strange.
In my tests, HA -> device communication is almost instantaneous. Device -> HA communication may have a slight delay, but I find it negligible.
You could try activating quirks traces to see what's going on. In my case it helped me to see that the device did several retries, which led to the patch in #1125

javicalle, 2021-11-07:
> #1125 if fixing the delay with multi commands but must being putted in the HA container for working with the local quirk.

Ahhh, that make sense.
Unless you also patch in `custom_zha_quirks` the `TuyaManufacturerClusterOnOff` class. I have overwritten it in the same file where I have my quirks (this is the reason I have the import clause commented).

PR #1166 (2021-11-23): Add quirks for Tuya multigang switches
This PR add quirks for Tuya multigang switches.

It should correct the management of the On/Off cluster for the multiswitch devices that make use of the Tuya cluster (`0xEF00`) for command/status reporting.

The `MODELS_INFO` of those devices that have been reported or an issue has been reported have been added.
For those devices whose number of gangs is not clear, they have been defined as 1-gang devices until someone report correct info.

This could fix the following issues (only on/off switches):
fixes: #813
fixes: #824
fixes: #913
fixes: #972
fixes: # [... cut]
closed 2021-11-23 by dmulcahey