zigpy/zha-device-handlers #967: [BUG] IKEA 2 and 3 gen controllers spamming the network need tsn filtering

MattWestb, 2021-07-17:
**Describe the bug**
The newer IKEA controllers gen 2 and 3 is sending multiple commands to bounded device / groups with the same tsn that is making problems with device automation in HA.
**To Reproduce**
Have made sniffing of all those new devices and they is doing the same nasty thing = every cluster binding and configured reporting is making one more copy of commands being sent with the same transaction number.
So configuring 2 attribute reporting and one group binding you is getting 2 unicast and 2 or 3  bradcast of the same command that is not nice done but is working in the network.
But then users is using device automation in HA they is getting multiple trigger that is one no go.

Devices that doing it: Symfonisk, Shortcut button and Styrbar (E2001/E2002).
**Expected behavior**
If one device is bad behaving like this or is missing getting default response and reseeding one command with the same tsn shall the host system filtering them out so the user is only  getting one command in HA
**Screenshots**

**Additional context**
tuya devices that is not using ZCL  commands is sending more commands if not forcing ZHA sending default response that is implanted in the quirk and also one filtering function that looks working OK.

If its  possible doing the same in `/zhaquirks/ikea/__init__.py`  as is made in the `/tuya/_init_.py` 
 but only the tsn part that can being used in the controllers quirk.

Adminiuga, 2021-07-17:
Don't use sniffer. How many messages coordinator reports in the debug log?

What you are seeing is not spamming, but messages being related by the devices, specially if it is a broadcast packet

MattWestb, 2021-07-17:
I retested the remote and pairing it and its sending on one button pres:
```
[0x8b51:1:0x0006] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=CLUSTER_COMMAND manufacturer_specific=False is_reply=False disable_default_response=False> manufacturer=None tsn=11 command_id=1>
[0x8b51:1:0x0006] ZCL request 0x0001: []
[0x8b51:1:0x0006] No handler for cluster command 1
[0x8b51:1:0x0006] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=CLUSTER_COMMAND manufacturer_specific=False is_reply=False disable_default_response=False> manufacturer=None tsn=11 command_id=1>
[0x8b51:1:0x0006] ZCL request 0x0001: []
[0x8b51:1:0x0006] No handler for cluster command 1
[0x8b51:1:0x0006] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=CLUSTER_COMMAND manufacturer_specific=False is_reply=False disable_default_response=False> manufacturer=None tsn=11 command_id=1>
[0x8b51:1:0x0006] ZCL request 0x0001: []
[0x8b51:1:0x0006] No handler for cluster command 1
[0x8b51:1:0x0006] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=CLUSTER_COMMAND manufacturer_specific=False is_reply=False disable_default_response=False> manufacturer=None tsn=11 command_id=1>
[0x8b51:1:0x0006] ZCL request 0x0001: []
[0x8b51:1:0x0006] No handler for cluster command 1
[0x8b51:1:0x0006] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=CLUSTER_COMMAND manufacturer_specific=False is_reply=False disable_default_response=False> manufacturer=None tsn=11 command_id=1>
[0x8b51:1:0x0006] ZCL request 0x0001: []
[0x8b51:1:0x0006] No handler for cluster command 1
```
If doing one reconfigure and binding one group its being more with the same tsn.

MattWestb, 2021-07-17:
Reconfigure is doing 5 bindings / attribute reporting and then i is getting:
```
[0x8b51:1:0x0006] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=CLUSTER_COMMAND manufacturer_specific=False is_reply=False disable_default_response=False> manufacturer=None tsn=16 command_id=0>
[0x8b51:1:0x0006] ZCL request 0x0000: []
[0x8b51:1:0x0006] No handler for cluster command 0
[0x8b51:1:0x0006] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=CLUSTER_COMMAND manufacturer_specific=False is_reply=False disable_default_response=False> manufacturer=None tsn=16 command_id=0>
[0x8b51:1:0x0006] ZCL request 0x0000: []
[0x8b51:1:0x0006] No handler for cluster command 0
[0x8b51:1:0x0006] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=CLUSTER_COMMAND manufacturer_specific=False is_reply=False disable_default_response=False> manufacturer=None tsn=16 command_id=0>
[0x8b51:1:0x0006] ZCL request 0x0000: []
[0x8b51:1:0x0006] No handler for cluster command 0
[0x8b51:1:0x0006] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=CLUSTER_COMMAND manufacturer_specific=False is_reply=False disable_default_response=False> manufacturer=None tsn=16 command_id=0>
[0x8b51:1:0x0006] ZCL request 0x0000: []
[0x8b51:1:0x0006] No handler for cluster command 0
[0x8b51:1:0x0006] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=CLUSTER_COMMAND manufacturer_specific=False is_reply=False disable_default_response=False> manufacturer=None tsn=16 command_id=0>
[0x8b51:1:0x0006] ZCL request 0x0000: []
[0x8b51:1:0x0006] No handler for cluster command 0
```
= 1 event / binding = very bad behavior device.

Bonded to on group (lights attribute reporting is deleted):
```
[0x8b51:1:0x0006] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=CLUSTER_COMMAND manufacturer_specific=False is_reply=False disable_default_response=False> manufacturer=None tsn=19 command_id=1>
[0x8b51:1:0x0006] ZCL request 0x0001: []
[0x8b51:1:0x0006] No handler for cluster command 1
[0x8b51:1:0x0006] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=CLUSTER_COMMAND manufacturer_specific=False is_reply=False disable_default_response=False> manufacturer=None tsn=19 command_id=1>
[0x8b51:1:0x0006] ZCL request 0x0001: []
[0x8b51:1:0x0006] No handler for cluster command 1
[0x8b51:1:0x0006] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=CLUSTER_COMMAND manufacturer_specific=False is_reply=False disable_default_response=False> manufacturer=None tsn=19 command_id=1>
[0x8b51:1:0x0006] ZCL request 0x0001: []
[0x8b51:1:0x0006] No handler for cluster command 1
[0x8b51:1:0x0006] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=CLUSTER_COMMAND manufacturer_specific=False is_reply=False disable_default_response=False> manufacturer=None tsn=19 command_id=1>
[0x8b51:1:0x0006] ZCL request 0x0001: []
[0x8b51:1:0x0006] No handler for cluster command 1
[0x8b51:1:0x0006] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=CLUSTER_COMMAND manufacturer_specific=False is_reply=False disable_default_response=False> manufacturer=None tsn=19 command_id=1>
[0x8b51:1:0x0006] ZCL request 0x0001: []
[0x8b51:1:0x0006] No handler for cluster command 1
[0x8b51:1:0x0006] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=CLUSTER_COMMAND manufacturer_specific=False is_reply=False disable_default_response=False> manufacturer=None tsn=19 command_id=1>
[0x8b51:1:0x0006] ZCL request 0x0001: []
[0x8b51:1:0x0006] No handler for cluster command 1
[0x8b51:1:0x0006] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=CLUSTER_COMMAND manufacturer_specific=False is_reply=False disable_default_response=False> manufacturer=None tsn=19 command_id=1>
[0x8b51:1:0x0006] ZCL request 0x0001: []
[0x8b51:1:0x0006] No handler for cluster command 1
[0x8b51:1:0x0006] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=CLUSTER_COMMAND manufacturer_specific=False is_reply=False disable_default_response=False> manufacturer=None tsn=19 command_id=1>
[0x8b51:1:0x0006] ZCL request 0x0001: []
[0x8b51:1:0x0006] No handler for cluster command 1
```

Only 8 command received with the same tsn.

And doing on reconfigure after the group binding:
```
[0x8b51:1:0x0006] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=CLUSTER_COMMAND manufacturer_specific=False is_reply=False disable_default_response=False> manufacturer=None tsn=24 command_id=1>
[0x8b51:1:0x0006] ZCL request 0x0001: []
[0x8b51:1:0x0006] No handler for cluster command 1
[0x8b51:1:0x0006] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=CLUSTER_COMMAND manufacturer_specific=False is_reply=False disable_default_response=False> manufacturer=None tsn=24 command_id=1>
[0x8b51:1:0x0006] ZCL request 0x0001: []
[0x8b51:1:0x0006] No handler for cluster command 1
[0x8b51:1:0x0006] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=CLUSTER_COMMAND manufacturer_specific=False is_reply=False disable_default_response=False> manufacturer=None tsn=24 command_id=1>
[0x8b51:1:0x0006] ZCL request 0x0001: []
[0x8b51:1:0x0006] No handler for cluster command 1
[0x8b51:1:0x0006] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=CLUSTER_COMMAND manufacturer_specific=False is_reply=False disable_default_response=False> manufacturer=None tsn=24 command_id=1>
[0x8b51:1:0x0006] ZCL request 0x0001: []
[... cut ...]
[0x8b51:1:0x0006] No handler for cluster command 1
[0x8b51:1:0x0006] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=CLUSTER_COMMAND manufacturer_specific=False is_reply=False disable_default_response=False> manufacturer=None tsn=24 command_id=1>
[0x8b51:1:0x0006] ZCL request 0x0001: []
[0x8b51:1:0x0006] No handler for cluster command 1
```
Its the same as only doing binding so ZHA is doing the setup OK.

I thinkk you can giving E2001/E2002 new name = **Hedda** and **MattWestb**  (= network spammers) !!!

MattWestb, 2022-01-13:
The 2 gen is fixed then they is not sending broadcast but  the 3rd is still having this problem spamming the network.

If the Zigbee stack is receiving one ZCL command and is getting one more with the same tsn is shall not using the last only the first one. Some stacks is doing that but not EZSP and i think some more (only have testing EZSP).

MattWestb, 2022-07-13:
Still one problem if having one large system and spamming the network.

MattWestb, 2023-01-09:
Still one problem then making automatons from this switches :-((

closed 2023-07-15 by github-actions[bot]