zigpy/zha-device-handlers #968: [BUG] zha visualization missing after 2021.7.3

lishan89uc, 2021-07-18:
**Describe the bug**
zha visualization missing after 2021.7.3

**To Reproduce**
Install 2021.7.3 
go to integration
go to zha and press configure
click on the visualization tab on top

**Expected behavior**
see something when you go there

**Screenshots**
[image]

**Additional context**
Add any other context about the problem here. Please include full debug logs as well as the device signature.

almico, 2021-07-19:
It works for me:
[image]

lishan89uc, 2021-07-19:
Sorry I didn't know what the issue was and posted in both zigpy and HA core. https://github.com/home-assistant/core/issues/53132

closed 2021-07-19 by lishan89uc