zigpy/zha-device-handlers #970: [Device Support Request] TS0601 TZE200_nueqqe6k (Another Chain Roller Blinds Pull Bead Curtain Motor)

maxperron, 2021-07-20:
Hi! It looks like there is already a couple of versions of the same device, but I didn't find this one exactly.

**Is your feature request related to a problem? Please describe.**
Device pairs successfully, but no entities are created.

**Describe the solution you'd like**
I would like a quick to handle this device

**Device signature - this can be acquired by removing the device from ZHA and pairing it again from the add devices screen. Be sure to add the entire content of the log panel after pairing the device to a code block below this line.**
`{
  "node_descriptor": "NodeDescriptor(logical_type=<LogicalType.Router: 1>, complex_descriptor_available=0, user_descriptor_available=0, reserved=0, aps_flags=0, frequency_band=<FrequencyBand.Freq2400MHz: 8>, mac_capability_flags=<MACCapabilityFlags.AllocateAddress|RxOnWhenIdle|MainsPowered|FullFunctionDevice: 142>, manufacturer_code=4098, maximum_buffer_size=82, maximum_incoming_transfer_size=82, server_mask=11264, maximum_outgoing_transfer_size=82, descriptor_capability_field=<DescriptorCapability.0: 0>, *allocate_address=True, *is_alternate_pan_coordinator=False, *is_coordinator=False, *is_end_device=False, *is_full_function_device=True, *is_mains_powered=True, *is_receiver_on_when_idle=True, *is_router=True, *is_security_capable=False)",
  "endpoints": {
    "1": {
      "profile_id": 260,
      "device_type": "0x0051",
      "in_clusters": [
        "0x0000",
        "0x0004",
        "0x0005",
        "0xef00"
      ],
      "out_clusters": [
        "0x000a",
        "0x0019"
      ]
    }
  },
  "manufacturer": "_TZE200_nueqqe6k",
  "model": "TS0601",
  "class": "zigpy.device.Device"
}`

**Additional context**
I bought this device: [Aliexpress](https://www.aliexpress.com/item/1005002089679060.html?spm=a2g0s.9042311.0.0.66dd4c4dHdKCSg)
It seems it's already supported by zigbee2mqtt and deconz [Repository](https://zigbee.blakadder.com/Tuya_M515EGZT.html)

Thanks you very much!

MattWestb, 2021-07-20:
Looks being one variant of one already supported tuya roller blinds.

I have adding it that is have the same function in Z2M in the ts0601.py.
Configure local quirk in ZHA #693 (comment) and putting the new (unziped) ts0601.py in the local quirk folder and restarting HA.
[ts0601.zip](https://github.com/zigpy/zha-device-handlers/files/6849077/ts0601.zip)

maxperron, 2021-07-20:
Thanks for you help, the entity was created, but i get service call error on all the 3 buttons on the entity.

```
Logger: homeassistant.components.websocket_api.http.connection
Source: components/zha/core/channels/base.py:56
Integration: Home Assistant WebSocket API (documentation, issues)
First occurred: 12:44:09 PM (6 occurrences)
Last logged: 2:43:47 PM

[140533768511744] '_TZE200_nueqqe6k'
[140533620956800] '_TZE200_nueqqe6k'
[140533653375488] '_TZE200_nueqqe6k'
Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/components/websocket_api/commands.py", line 185, in handle_call_service
    await hass.services.async_call(
  File "/usr/src/homeassistant/homeassistant/core.py", line 1491, in async_call
    task.result()
  File "/usr/src/homeassistant/homeassistant/core.py", line 1526, in _execute_service
    await handler.job.target(service_call)
  File "/usr/src/homeassistant/homeassistant/helpers/entity_component.py", line 213, in handle_service
    await self.hass.helpers.service.entity_service_call(
  File "/usr/src/homeassistant/homeassistant/helpers/service.py", line 658, in entity_service_call
    future.result()  # pop exception if have
  File "/usr/src/homeassistant/homeassistant/helpers/entity.py", line 811, in async_request_call
    await coro
  File "/usr/src/homeassistant/homeassistant/helpers/service.py", line 695, in _handle_entity_call
    await result
  File "/usr/src/homeassistant/homeassistant/components/zha/cover.py", line 126, in async_open_cover
    res = await self._cover_channel.up_open()
  File "/usr/src/homeassistant/homeassistant/components/zha/core/channels/base.py", line 56, in wrapper
    result = await command(*args, **kwds)
  File "/usr/local/lib/python3.9/site-packages/zhaquirks/tuya/__init__.py", line 857, in command
    TUYA_COVER_COMMAND[manufacturer][command_id],
KeyError: '_TZE200_nueqqe6k'
```

MattWestb, 2021-07-20:
Sorry ZHA is also needing one config for the buttons.
I dont knowing the order of then so i was copying the last one (line 91 in the files).
If its working but the order is wrong only editing  the line for your device and deleting the pycash and restarting HA.
[__init__.zip](https://github.com/zigpy/zha-device-handlers/files/6850709/__init__.zip)

maxperron, 2021-07-20:
I'm sorry, I uploaded the new `__init__.py` file in the /config/custom_zha_quirks folder
I also delete `__pycache__` in the same folder.
Rebooted HA
Deleted the device and added it again.

The position slider for the cover make the motor run, but never stop. And none of the buttons (up, stop, down) work.

I get some new errors though:

```
Logger: homeassistant.components.websocket_api.http.connection
Source: components/websocket_api/connection.py:89
Integration: Home Assistant WebSocket API (documentation, issues)
First occurred: 4:46:20 PM (4 occurrences)
Last logged: 4:46:20 PM

[140243225376512] Received invalid command: zha/device
```

```
Logger: zigpy.zcl
Source: /usr/local/lib/python3.9/site-packages/zigpy/zcl/__init__.py:92
First occurred: 4:46:22 PM (2 occurrences)
Last logged: 4:47:31 PM

Unknown cluster 61184
```

```
Logger: homeassistant.components.zha.core.channels.base
Source: components/zha/core/channels/base.py:356
Integration: Zigbee Home Automation (documentation, issues)
First occurred: 4:47:31 PM (1 occurrences)
Last logged: 4:47:31 PM

[0xC17A:1:0x0102]: 'async_configure' stage failed: 'ConfigureReportingResponseRecord' object is not subscriptable
```

```
Logger: bellows.ezsp
Source: /usr/local/lib/python3.9/site-packages/bellows/ezsp/__init__.py:299
First occurred: 4:47:40 PM (1 occurrences)
Last logged: 4:47:40 PM

Exception running handler
Traceback (most recent call last):
  File "/usr/local/lib/python3.9/site-packages/bellows/ezsp/__init__.py", line 297, in handle_callback
    handler(*args)
  File "/usr/local/lib/python3.9/site-packages/bellows/zigbee/application.py", line 220, in ezsp_callback_handler
    self._handle_frame(*args)
  File "/usr/local/lib/python3.9/site-packages/bellows/zigbee/application.py", line 262, in _handle_frame
    self.handle_message(
  File "/usr/local/lib/python3.9/site-packages/zigpy/application.py", line 216, in handle_message
    return sender.handle_message(
  File "/usr/local/lib/python3.9/site-packages/zigpy/device.py", line 358, in handle_message
    return endpoint.handle_message(
  File "/usr/local/lib/python3.9/site-packages/zigpy/endpoint.py", line 210, in handle_message
    handler(hdr, args, dst_addressing=dst_addressing)
  File "/usr/local/lib/python3.9/site-packages/zigpy/zcl/__init__.py", line 202, in handle_message
    self.handle_cluster_request(hdr, args, dst_addressing=dst_addressing)
  File "/usr/local/lib/python3.9/site-packages/zhaquirks/tuya/__init__.py", line 788, in handle_cluster_request
    super().handle_cluster_request(self, hdr, args, dst_addressing)
TypeError: handle_cluster_request() takes 3 positional arguments but 5 were given
```

```
Logger: homeassistant.components.websocket_api.http.connection
Source: components/zha/core/channels/base.py:56
Integration: Home Assistant WebSocket API (documentation, issues)
First occurred: 4:46:53 PM (2 occurrences)
Last logged: 4:47:47 PM

[140243288811168] '_TZE200_nueqqe6k'
Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/components/websocket_api/commands.py", line 185, in handle_call_service
    await hass.services.async_call(
  File "/usr/src/homeassistant/homeassistant/core.py", line 1491, in async_call
    task.result()
  File "/usr/src/homeassistant/homeassistant/core.py", line 1526, in _execute_service
    await handler.job.target(service_call)
  File "/usr/src/homeassistant/homeassistant/helpers/entity_component.py", line 213, in handle_service
    await self.hass.helpers.service.entity_service_call(
  File "/usr/src/homeassistant/homeassistant/helpers/service.py", line 658, in entity_service_call
    future.result()  # pop exception if have
  File "/usr/src/homeassistant/homeassistant/helpers/entity.py", line 811, in async_request_call
    await coro
  File "/usr/src/homeassistant/homeassistant/helpers/service.py", line 695, in _handle_entity_call
    await result
  File "/usr/src/homeassistant/homeassistant/components/zha/cover.py", line 126, in async_open_cover
    res = await self._cover_channel.up_open()
  File "/usr/src/homeassistant/homeassistant/components/zha/core/channels/base.py", line 56, in wrapper
    result = await command(*args, **kwds)
  File "/usr/local/lib/python3.9/site-packages/zhaquirks/tuya/__init__.py", line 857, in command
    TUYA_COVER_COMMAND[manufacturer][command_id],
KeyError: '_TZE200_nueqqe6k'
```

```
Logger: zigpy.zcl
Source: /usr/local/lib/python3.9/site-packages/zigpy/zcl/__init__.py:111
First occurred: 4:48:00 PM (5 occurrences)
Last logged: 4:48:01 PM

[0xc17a:1:0xef00] Unknown cluster-specific command 17
```

maxperron, 2021-07-23:
@MattWestb I have been reviewing the error log and I think it's not using the new `__init__.py` from the local folder?

I have also tried permuting line 91 values, but to no avail. I always get the same error.

MattWestb, 2021-07-23:
Can being. The local quirk is pity new and not so well tested but it looks working OK with "simple quirks".

It can also being that the system is using the `__pycache__` in the docker container.

One way is deleting the `__init__.py` in the local folder (and deleting the __pycache__) and putting the changes in the "original" init that is in the container `/usr/local/lib/python3.9/site-packages/zhaquirks/tuya/__init__.py` and deleting the `/usr/local/lib/python3.9/site-packages/zhaquirks/tuya/__pycache__`.

Instruction  of editing files in the container: https://github.com/zigpy/zha-device-handlers/discussions/693.

You can putting all changes in the container if you like with the same method and deleting the files in the local quirk folder so you is not getting problems with different versions.

MattWestb, 2021-07-24:
I was finding the comments from the devs that its not working using "nested" custom quirks like we are trying then the `ts0601.py` is reading `init `from the "normal" location and not from the local one that we was trying to do.

The good thing its that its very likely is working if putting the changes in the HA container as i was writing in the post above.

maxperron, 2021-07-26:
Yes, you were right, I put everything in the tuya folder and now the new quirk and `__init__.py` are being used. However I can't for the life of me find the right sequence for line 91. I've been trying multiple permutations over the weekend and none were a perfect match. I'm not able to find a pattern. Do you know where I could find help with that?

Also, I assumed that if I update HA, it will overwrite these changes?

MattWestb, 2021-07-26:
The last Q first yes HA is doing it so copy one "set" to /config/ so you can easily copy it back after the update.

I hope you have deleting the `__pycache__` after doing the changes of the key matrix.

I dont have experience of the device but it was much writing in some issues and in the end one user was adding 3 different configs for different devices so i think reading the issues is the right way to finding the right setting.

Look on the historical commits and the  issues linked to them https://github.com/zigpy/zha-device-handlers/commits/dev/zhaquirks/tuya/ts130f.py

MattWestb, 2021-07-26:
Calibration is described here https://github.com/home-assistant/core/issues/46146#issuecomment-871136798.

MattWestb, 2021-07-26:
I was finding one users git that have adding  the device but have not commitment it  to ZHA but is having the keys added https://github.com/rednus/zha-device-handlers/pull/1/commits/b22dc9780373da3d5c5f7d7bf0865ed9d3c8266c.

MattWestb, 2021-07-26:
Looks like the device need being deleted and re added for  working OK https://github.com/zigpy/zha-device-handlers/issues/744#issuecomment-822030768.

You must changing the path for  python then its now 3.9.

Sorry for messing the devices and issues above :-((

maxperron, 2021-08-01:
I have been working on this in the last couple of days. Once added and the calibration done on the device, it's working through automations. If I control the motor through node-red, it's working perfectly. I can set a percentage, open, close. In HA, the buttons don't work, but the slider works. 

The calibration must be done on the device itself, just have to follow the instructions. 

What's the process to get this added to ZHA officially?

Thanks for your help @MattWestb !

MattWestb, 2021-08-01:
Its being some adding to the TS0601 devices so the best is doing  on new PR with your settings for not getting problem with commits.
I can posting one PR for it but i need knowing if you is using the parameters for the remote in the init so i not commitment  the wrong one.
Is you using the `"_TZE200_nueqqe6k": {0x0000: 0x0000, 0x0001: 0x0002, 0x0002: 0x0001},` ?

MattWestb, 2021-08-01:
Added and black was not in fighting mode :-))

maxperron, 2021-08-05:
Sorry for the delay, it has been a crazy few days. Yes I'm using this sequence: `{0x0000: 0x0000, 0x0001: 0x0002, 0x0002: 0x0001}`

MattWestb, 2021-08-05:
Great then the PR is OK for the device as long we can getting it without rewriting the quirk (that is out of my knowledge).

Thanks for confirming the setting !!

And all is working OK but not the buttons in HA ?

maxperron, 2021-08-05:
Yes all the service are working, I can call them through node-red and my cover are now automated. Only the buttons don't work, I've seen people with the same issue in the thread referenced above.

MattWestb, 2021-08-05:
I think it can only being fixed if some devs is getting there hands on one device and can doing some extensive testing and patching the quirk.

But good that all other things is working and you can using it with automatons :-))

Now only waiting that the devs is merging the PR :-))

PR #984 (2021-08-25): Adding "_TZE200_nueqqe6k", "TS0601"
Adding "_TZE200_nueqqe6k", "TS0601" one Chain Roller Blinds Pull Bead Curtain Motor.
Fixes #970.
closed 2021-08-25 by dmulcahey