zigpy/zha-device-handlers #972: [Device Support Request] _TZE200_wunufsil TS0601 (Tuya 2 gang touch switch)

sergeysaley, 2021-07-24:
**Is your feature request related to a problem? Please describe.**
Tuya Zigbee touch switch 2 gang is discoverable without any entities

**Describe the solution you'd like**
It would be great to craft a quirk for this device and integrate it to next release.

**Device signature - this can be acquired by removing the device from ZHA and pairing it again from the add devices screen. Be sure to add the entire content of the log panel after pairing the device to a code block below this line.**
`{
  "node_descriptor": "NodeDescriptor(logical_type=<LogicalType.Router: 1>, complex_descriptor_available=0, user_descriptor_available=0, reserved=0, aps_flags=0, frequency_band=<FrequencyBand.Freq2400MHz: 8>, mac_capability_flags=<MACCapabilityFlags.AllocateAddress|RxOnWhenIdle|MainsPowered|FullFunctionDevice: 142>, manufacturer_code=4098, maximum_buffer_size=82, maximum_incoming_transfer_size=82, server_mask=11264, maximum_outgoing_transfer_size=82, descriptor_capability_field=<DescriptorCapability.0: 0>, *allocate_address=True, *is_alternate_pan_coordinator=False, *is_coordinator=False, *is_end_device=False, *is_full_function_device=True, *is_mains_powered=True, *is_receiver_on_when_idle=True, *is_router=True, *is_security_capable=False)",
  "endpoints": {
    "1": {
      "profile_id": 260,
      "device_type": "0x0051",
      "in_clusters": [
        "0x0000",
        "0x0004",
        "0x0005",
        "0xef00"
      ],
      "out_clusters": [
        "0x000a",
        "0x0019"
      ]
    }
  },
  "manufacturer": "_TZE200_wunufsil",
  "model": "TS0601",
  "class": "zigpy.device.Device"
}`

**ZHA logs during pairing:**
`[0x0000:zdo] ZDO request ZDOCmd.Mgmt_Permit_Joining_req: [60, <Bool.false: 0>]
New device 0xd8ba (84:2e:14:ff:fe:3a:53:9c) joined the network
[0xd8ba] Scheduling initialization
Tries remaining: 3
[0xd8ba] Requesting 'Node Descriptor'
Tries remaining: 2
[0xd8ba] Extending timeout for 0x41 request
Device 0xd8ba (84:2e:14:ff:fe:3a:53:9c) joined the network
[0xd8ba] Scheduling initialization
[0xd8ba] Canceling old initialize call
Received frame on uninitialized device <Device model=None manuf=None nwk=0xD8BA ieee=84:2e:14:ff:fe:3a:53:9c is_initialized=False> from ep 0 to ep 0, cluster 19: b'\xae\xba\xd8\x9cS:\xfe\xff\x14.\x84\x8e'
[0xd8ba:zdo] ZDO request ZDOCmd.Device_annce: [0xD8BA, 84:2e:14:ff:fe:3a:53:9c, 142]
Tries remaining: 3
[0xd8ba] Requesting 'Node Descriptor'
Tries remaining: 2
[0xd8ba] Extending timeout for 0x43 request
Received frame on uninitialized device <Device model=None manuf=None nwk=0xD8BA ieee=84:2e:14:ff:fe:3a:53:9c is_initialized=False> from ep 0 to ep 0, cluster 32770: b'A\x00\xba\xd8\x01@\x8e\x02\x10RR\x00\x00,R\x00\x00'
[0xd8ba:zdo] ZDO request ZDOCmd.Node_Desc_rsp: [<Status.SUCCESS: 0>, 0xD8BA, NodeDescriptor(logical_type=<LogicalType.Router: 1>, complex_descriptor_available=0, user_descriptor_available=0, reserved=0, aps_flags=0, frequency_band=<FrequencyBand.Freq2400MHz: 8>, mac_capability_flags=<MACCapabilityFlags.AllocateAddress|RxOnWhenIdle|MainsPowered|FullFunctionDevice: 142>, manufacturer_code=4098, maximum_buffer_size=82, maximum_incoming_transfer_size=82, server_mask=11264, maximum_outgoing_transfer_size=82, descriptor_capability_field=<DescriptorCapability.0: 0>, *allocate_address=True, *is_alternate_pan_coordinator=False, *is_coordinator=False, *is_end_device=False, *is_full_function_device=True, *is_mains_powered=True, *is_receiver_on_when_idle=True, *is_router=True, *is_security_capable=False)]
[0xd8ba:zdo] Unsupported ZDO request:ZDOCmd.Node_Desc_rsp
Received frame on uninitialized device <Device model=None manuf=None nwk=0xD8BA ieee=84:2e:14:ff:fe:3a:53:9c is_initialized=False> from ep 0 to ep 0, cluster 32770: b'C\x00\xba\xd8\x01@\x8e\x02\x10RR\x00\x00,R\x00\x00'
[0xd8ba] Got Node Descriptor: NodeDescriptor(logical_type=<LogicalType.Router: 1>, complex_descriptor_available=0, user_descriptor_available=0, reserved=0, aps_flags=0, frequency_band=<FrequencyBand.Freq2400MHz: 8>, mac_capability_flags=<MACCapabilityFlags.AllocateAddress|RxOnWhenIdle|MainsPowered|FullFunctionDevice: 142>, manufacturer_code=4098, maximum_buffer_size=82, maximum_incoming_transfer_size=82, server_mask=11264, maximum_outgoing_transfer_size=82, descriptor_capability_field=<DescriptorCapability.0: 0>, *allocate_address=True, *is_alternate_pan_coordinator=False, *is_coordinator=False, *is_end_device=False, *is_full_function_device=True, *is_mains_powered=True, *is_receiver_on_when_idle=True, *is_router=True, *is_security_capable=False)
[0xd8ba] Discovering endpoints
Tries remaining: 3
Received frame on uninitialized device <Device model=None manuf=None nwk=0xD8BA ieee=84:2e:14:ff:fe:3a:53:9c is_initialized=False> from ep 0 to ep 0, cluster 32773: b'E\x00\xba\xd8\x01\x01'
[0xd8ba] Discovered endpoints: [1]
[0xd8ba] Initializing endpoints [<Endpoint id=1 in=[] out=[] status=<Status.NEW: 0>>]
[0xd8ba:1] Discovering endpoint information
Tries remaining: 3
Received frame on uninitialized device <Device model=None manuf=None nwk=0xD8BA ieee=84:2e:14:ff:fe:3a:53:9c is_initialized=False> from ep 0 to ep 0, cluster 32772: b'G\x00\xba\xd8\x14\x01\x04\x01Q\x00\x01\x04\x00\x00\x04\x00\x05\x00\x00\xef\x02\x19\x00\n\x00'
[0xd8ba:1] Discovered endpoint information: SizePrefixedSimpleDescriptor(endpoint=1, profile=260, device_type=81, device_version=1, input_clusters=[0, 4, 5, 61184], output_clusters=[25, 10])
Unknown cluster 61184
[0xd8ba:1:0x0000] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=GLOBAL_COMMAND manufacturer_specific=False is_reply=True disable_default_response=True> manufacturer=None tsn=73 command_id=Command.Read_Attributes_rsp>
[0xd8ba] Read model 'TS0601' and manufacturer '_TZE200_wunufsil' from <Endpoint id=1 in=[basic:0x0000, groups:0x0004, scenes:0x0005, None:0xEF00] out=[ota:0x0019, time:0x000A] status=<Status.ZDO_INIT: 1>>
[0xd8ba] Discovered basic device information for <Device model='TS0601' manuf='_TZE200_wunufsil' nwk=0xD8BA ieee=84:2e:14:ff:fe:3a:53:9c is_initialized=True>
Device is initialized <Device model='TS0601' manuf='_TZE200_wunufsil' nwk=0xD8BA ieee=84:2e:14:ff:fe:3a:53:9c is_initialized=True>
Checking quirks for _TZE200_wunufsil TS0601 (84:2e:14:ff:fe:3a:53:9c)
Considering <class 'zhaquirks.xbee.xbee_io.XBeeSensor'>
Fail because endpoint list mismatch: {232, 230} {1}
Considering <class 'zhaquirks.xbee.xbee3_io.XBee3Sensor'>
[... 147 lines cut ...]
[0xd8ba:1:0xef00] ZCL request 0x0024: b'\x00e'
[0xd8ba:1:0xef00] No handler for cluster command 36
[0xd8ba:1:0xef00] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=CLUSTER_COMMAND manufacturer_specific=False is_reply=True disable_default_response=False> manufacturer=None tsn=74 command_id=17>
[0xd8ba:1:0xef00] Unknown cluster-specific command 17
[0xd8ba:1:0xef00] ZCL request 0x0011: b'\x00f\xc0'
[0xd8ba:1:0xef00] No handler for cluster command 17
[0xd8ba:1:0xef00] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=CLUSTER_COMMAND manufacturer_specific=False is_reply=True disable_default_response=False> manufacturer=None tsn=74 command_id=17>
[0xd8ba:1:0xef00] Unknown cluster-specific command 17
[0xd8ba:1:0xef00] ZCL request 0x0011: b'\x00f\xc0'
[0xd8ba:1:0xef00] No handler for cluster command 17
[0xd8ba:1:0xef00] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=CLUSTER_COMMAND manufacturer_specific=False is_reply=True disable_default_response=False> manufacturer=None tsn=74 command_id=17>
[0xd8ba:1:0xef00] Unknown cluster-specific command 17
[0xd8ba:1:0xef00] ZCL request 0x0011: b'\x00f\xc0'
[0xd8ba:1:0xef00] No handler for cluster command 17
[0xd8ba:1:0xef00] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=CLUSTER_COMMAND manufacturer_specific=False is_reply=True disable_default_response=False> manufacturer=None tsn=74 command_id=17>
[0xd8ba:1:0xef00] Unknown cluster-specific command 17
[0xd8ba:1:0xef00] ZCL request 0x0011: b'\x00f\xc0'
[0xd8ba:1:0xef00] No handler for cluster command 17
[0xd8ba:1:0xef00] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=CLUSTER_COMMAND manufacturer_specific=False is_reply=True disable_default_response=False> manufacturer=None tsn=74 command_id=17>
[0xd8ba:1:0xef00] Unknown cluster-specific command 17
[0xd8ba:1:0xef00] ZCL request 0x0011: b'\x00f\xc0'
[0xd8ba:1:0xef00] No handler for cluster command 17`

**Additional context**
Purchased at https://aliexpress.ru/item/4001096045589.html

MattWestb, 2021-07-24:
I have adding one new switch in the quirk that one user was requesting and was putting in more known working IDs and your was one of them. https://github.com/zigpy/zha-device-handlers/issues/966.
Configure local quirks and putting the unziped ts0601.py file in the directory and restarting HA.

The quirk is supporting only one gang and some must adding support for multi gangs but 
I dont have the device and im not one code worrier so i cant fixing the last problem with the quirk but you is likely getting the first key working on them.

sergeysaley, 2021-07-24:
I put the file ts0601.py in the /usr/local/lib/python3.9/site-packages/zigpy/quirks/ directory into HA container, restarted HA, removed and re-added the device - and nothing changed, alas.

MattWestb, 2021-07-24:
If using local quirks you must putting the cofig for it in HA config (the new method).

You is using the "classical / old method" and was putting the quirk in the quirk root folder in the container so it cant finding its init `/usr/local/lib/python3.9/site-packages/zigpy/quirks/ts0601.py`.
You need putting it like this: `/usr/local/lib/python3.9/site-packages/zigpy/quirks/tuya/ts0501.py`  (in the tuya follder) then it can finding the init from the tuya folder, and then deleting the `__pycache__` in the `/usr/local/lib/python3.9/site-packages/zigpy/quirks/tuya` folder and restarting HA.

Normally you shall not needing repairing the device for the quirk to loading then ZHA is doing it the starting. But if the quirk is loading and the device is still not playing well it can being good resetting and adding it a gen so it dont have bad configuration stored.

sergeysaley, 2021-07-25:
About the new method (put in the HA configuration) - where can I read more? It seems to me more beautiful to work with the HA configuration than to climb inside the container ...

MattWestb, 2021-07-25:
https://github.com/zigpy/zha-device-handlers/discussions/693#discussioncomment-857274 is  how to configuring it in HA.
The local quirks is staying also if you is updating HA so if the quirk is added in HA you shall deleting it in the local quirk folder. 

And you dont need "diving" in the HA container with CLI all the time :-))

Its have one limitation you cant putting more nested quirks in it then its reading all nested quirks from the container and not from the local folder but shall not being any problem for your  device quirk.

MattWestb, 2021-07-25:
If going for the local quirk its best you is deleting the "modded" ts0601.py you was adding in the container (not the original in the tuya folder) so you is not getting dependency problems.

sergeysaley, 2021-07-25:
I reset everything to its original state (inside the container in /usr/local/lib/python3.9/site-packages/zigpy/quirks/ removed tuya / ts0601.py, removed __pycache__), added path to local quirks in HA config, restart HA. As a result, one switch appeared, it even works in both directions, but with a delay of one to several seconds.

MattWestb, 2021-07-25:
Great then the quirk is being loaded :-))

I think the delay is coming from that the quirk have not implanted more gang switches and its not knowing wot to do with the "strange" commands from the switch 2.

Im not one code worrier so i cant fixing the quirk but i hope some with right knowledge can doing it in the near future so all user can using all functions on there device.

I like the design of this switches (the black ones) but they cant being bonded to light groups if having more light that you like searing from the switch or from other remotes but its working "locally" if HA is off line.

Is it working enough for putting the changes in one PR and trying getting the devs accepting it so other user can using it ??

I think its better one gang working then not at all and the device is useless and the other gangs is working "local" but not from ZHA i hope.

sergeysaley, 2021-07-25:
I think yes. A device with one running entity is better than a device with no entities at all :)

MattWestb, 2021-07-25:
I agree for 110% !!

I doing one PR for adding the new device IDs and writing its working OK for the first gang but not for 2 and higher ones and the devs can decide if adding it or not and if not users can using the PR for adding it in local quirk.

sergeysaley, 2021-07-25:
Can I try to add the TuyaDoubleSwitch(TuyaSwitch) class in this file and describe the signature and replacement for two endpoints in it?
I looked at the README of the project, in general, there is nothing particularly complicated there ...

MattWestb, 2021-07-25:
You can trying all you like in your local quirk and see if its working.
Only remember deleting the `__pycache__` in the local quirk folder after doing changes and restarting HA. I think its not nervelessly repairing the device then its only sending and receiving commands from the device and ZHA is applying the new quirk then loading the device on start

If you is getting it working do one new class in the quirk for multiple gang devices so not braking the single ones :-))

One thing if your is bad it can happen ZHA is not starting but you see that  in the log.

sergeysaley, 2021-07-25:
Did I understand correctly that it is necessary to clean up `__pycache__` even if quirk is placed in the HA config?

MattWestb, 2021-07-25:
I think its nervelessly so not old things is being used of the system.
Normally doing one `rm -r __pycache__` is working OK (if being in the right directory).

I have posting  one PR for the new IDs.

sergeysaley, 2021-07-25:
Thanks for your help.
If I succeed, I'll post the result here.

MattWestb, 2021-07-25:
That is great !!

You can looking on how Z2M have doing it but i dont understand the java (for my is for drinking not doing technical things) but its decoding the endpoints from the DP and connecting them to one new cluster on one new end point i think (without knowing as normal).

Hope you is getting it  working !!!

sergeysaley, 2021-07-25:
By the way, I noticed one bug in the existing implementation. If the switch port is physically turned on and HA restart is performed, then the entity of this port starts in the off state, i.e. at startup, HA does not read the current original status of the device port.

MattWestb, 2021-07-25:
"Normal" zigbeee devices is saved the last seen and state in zigbee.db  and ZHA is repopulating the last seen and state from it but for the moment its much work being done then upgrading the zigbee.db and many users (most deCONZ) is rolling back then they is having problems and the devs have doing much things for fixing it but im not 110% sure its stable (last fixes was merged in the night).
But i dont knowing if its the same with the "tuya commands" devices.

I think the status shall being updated in some minutes if the device is reporting OK to the coordinator but i dont knowing if its the same with tuya.

I think its the same with normal zigbee device if changing the status then the coordinator is off line then ZHA is using the "old" status until the device is reporting the new or ZHA is pulling (for not zigbee good behaving devices like Philips  HUE) the new status from the device.

sergeysaley, 2021-07-25:
The status of an entity in HA is updated whenever the device is active.
In my case, the status of the entity of the first port of the device is updated when clicking on any sensor of the device - it doesn't matter the first or the second one.
Probably this is not even a bug, but a feature :)

MattWestb, 2021-07-25:
Last seen shall being updated from all status report from the device so it looks OK then ZHA have only one device for the switch.

"Undocumented futures" is one nice name for bugs ;-)))

sergeysaley, 2021-07-25:
Unfortunately I can't make a working solution.
When trying to add endpoints to signature and replacement quirk stops working. Apparently everything is not as simple as it seemed :)

MattWestb, 2021-07-27:
I was looking in the init for the tuya quirks and its some debug logging in it that  shall showing the events being received from the device.
Try putting little debug logging in your HA config like this:
```
    zhaquirks: info
    zhaquirks.tuya: debug
```
and look if ZHA is receiving some events from the device.

I dont being on programmer but i think one part of the quirk is listening for the events with DP and is sending them to the event bus = HA but i dont understanding how it working.
https://github.com/zigpy/zha-device-handlers/blob/21af0860992021c39b26260b6e2ae4f9c1af75d1/zhaquirks/tuya/__init__.py#L392
and https://github.com/zigpy/zha-device-handlers/blob/21af0860992021c39b26260b6e2ae4f9c1af75d1/zhaquirks/tuya/__init__.py#L435

I have not finding any reference that is limiting the commands received to endpoint 1 or somthing but it can being som other place in the code that i dont knowing.
ZHA have one converter from some commands so they is in there range but i still not knowing how its working.

Perhaps the second gang is using one different index here:
https://github.com/zigpy/zha-device-handlers/blob/21af0860992021c39b26260b6e2ae4f9c1af75d1/zhaquirks/tuya/__init__.py#L93.

sergeysaley, 2021-07-27:
Unfortunately, as a result of multiple pairings / disconnections, the switch stopped entering pairing mode. I don’t know, the fact is that my ZHA installation has deteriorated (although the other 17 devices are connected and working), or the switch itself has gone crazy - but right now I cannot check your thoughts.
I completely cleaned the installation (removed all devices, removed ZHA, reflash the zigbee bridge, cleaned the HA service files, put everything back in - the switch does not want to pair)
I'm sorry...
PS: If I can get it to work properly, I will write here.

MattWestb, 2021-07-27:
Very sad !!!

Is the switch having one reset / pairing button or is it one of the "normal" ones used for reset ?
And is you getting any feedback (LED blinking or somthing) then getting it retested ?

If you have mounted it in one wall box it can being that the antenna is not working so good then its much metal around it.

sergeysaley, 2021-07-27:
The switch lies on the table in front of me, one meter from the zigbee bridge and three meters from the Wi-Fi.
The switch does not have a reset button. Tuya write that to reset, you need to turn on / off the power of the switch several times in a row - and I have done this many times.
Therefore, there are only two options - either the switch has gone crazy, or HA + Zigbee Bridge + ZHA. Moreover, HA + Zigbee Bridge + ZHA works normally in all other respects.

sergeysaley, 2021-07-27:
HA logs during try to pair this switch:
`2021-07-28 00:28:04 DEBUG (MainThread) [bellows.ezsp.protocol] Send command sendBroadcast: (<BroadcastAddress.ALL_ROUTERS_AND_COORDINATOR: 65532>, EmberApsFrame(profileId=0, clusterId=54, sourceEndpoint=0, destinationEndpoint=0, options=<EmberApsOption.APS_OPTION_NONE: 0>, groupId=0, sequence=121), 0, 122, b'y<\x00')
2021-07-28 00:28:04 DEBUG (MainThread) [bellows.ezsp.protocol] Send command addTransientLinkKey: (ff:ff:ff:ff:ff:ff:ff:ff, [90, 105, 103, 66, 101, 101, 65, 108, 108, 105, 97, 110, 99, 101, 48, 57])
2021-07-28 00:28:04 DEBUG (bellows.thread_0) [bellows.uart] Sending: b'562621a9622ae94d59947c25aa5592499c4e5eab97cd1eb7fd42637e'
2021-07-28 00:28:05 DEBUG (bellows.thread_0) [bellows.uart] Data frame: b'6626a1a9622a15adb8727e'
2021-07-28 00:28:05 DEBUG (bellows.thread_0) [bellows.uart] Sending: b'87009f7e'
2021-07-28 00:28:05 DEBUG (MainThread) [bellows.ezsp.protocol] Application frame 54 (sendBroadcast) received: b'001f'
2021-07-28 00:28:05 DEBUG (bellows.thread_0) [bellows.uart] Sending: b'672721a9fb2aea4da66bb5da55aac820fb0c42ceaca20be29ca800eccc47879d7e'
2021-07-28 00:28:05 DEBUG (bellows.thread_0) [bellows.uart] Data frame: b'7626b1a9112a10b259a24a25aa5592499c51d8abedce9874febf5f89294c7e'
2021-07-28 00:28:05 DEBUG (bellows.thread_0) [bellows.uart] Sending: b'8070787e'
2021-07-28 00:28:05 DEBUG (MainThread) [bellows.ezsp.protocol] Application frame 69 (incomingMessageHandler) received: b'05000036000000000000001fff000000ffff03793c00'
2021-07-28 00:28:05 DEBUG (MainThread) [bellows.zigbee.application] Received incomingMessageHandler frame with [<EmberIncomingMessageType.INCOMING_BROADCAST_LOOPBACK: 5>, EmberApsFrame(profileId=0, clusterId=54, sourceEndpoint=0, destinationEndpoint=0, options=<EmberApsOption.APS_OPTION_NONE: 0>, groupId=0, sequence=31), 255, 0, 0x0000, 255, 255, b'y<\x00']
2021-07-28 00:28:05 DEBUG (MainThread) [zigpy.zdo] [0x0000:zdo] ZDO request ZDOCmd.Mgmt_Permit_Joining_req: [60, <Bool.false: 0>]
2021-07-28 00:28:05 DEBUG (bellows.thread_0) [bellows.uart] Data frame: b'0727a1a9fb2a1534ed7e'
2021-07-28 00:28:05 DEBUG (bellows.thread_0) [bellows.uart] Sending: b'8160597e'
2021-07-28 00:28:05 DEBUG (MainThread) [bellows.ezsp.protocol] Application frame 175 (addTransientLinkKey) received: b'00'
2021-07-28 00:28:05 DEBUG (MainThread) [bellows.ezsp.protocol] Send command setPolicy: (<EzspPolicyId.TRUST_CENTER_POLICY: 0>, <EzspDecisionBitmask.ALLOW_UNSECURED_REJOINS|ALLOW_JOINS: 3>)
2021-07-28 00:28:05 DEBUG (bellows.thread_0) [bellows.uart] Sending: b'712421a9012a15b159e8157e'
2021-07-28 00:28:05 DEBUG (bellows.thread_0) [bellows.uart] Data frame: b'1024a1a9012a15f0a17e'
2021-07-28 00:28:05 DEBUG (bellows.thread_0) [bellows.uart] Sending: b'82503a7e'
2021-07-28 00:28:05 DEBUG (MainThread) [bellows.ezsp.protocol] Application frame 85 (setPolicy) received: b'00'
2021-07-28 00:28:05 DEBUG (bellows.thread_0) [bellows.uart] Data frame: b'2024b1a96b2a134ea6944a13aa5592499c4e27b497ce6744d77e'
2021-07-28 00:28:05 DEBUG (bellows.thread_0) [bellows.uart] Sending: b'83401b7e'
2021-07-28 00:28:05 DEBUG (MainThread) [bellows.ezsp.protocol] Application frame 63 (messageSentHandler) received: b'06fcff000036000000000000001f7a0000'
2021-07-28 00:28:05 DEBUG (MainThread) [bellows.zigbee.application] Received messageSentHandler frame with [<EmberOutgoingMessageType.OUTGOING_BROADCAST: 6>, 65532, EmberApsFrame(profileId=0, clusterId=54, sourceEndpoint=0, destinationEndpoint=0, options=<EmberApsOption.APS_OPTION_NONE: 0>, groupId=0, sequence=31), 122, <EmberStatus.SUCCESS: 0>, b'']
2021-07-28 00:28:05 DEBUG (MainThread) [bellows.ezsp.protocol] Send command permitJoining: (60,)
2021-07-28 00:28:05 DEBUG (bellows.thread_0) [bellows.uart] Sending: b'032521a9762a29d26f7e'
2021-07-28 00:28:06 DEBUG (bellows.thread_0) [bellows.uart] Data frame: b'3125a1a9762a153e0e7e'
2021-07-28 00:28:06 DEBUG (bellows.thread_0) [bellows.uart] Sending: b'8430fc7e'
2021-07-28 00:28:06 DEBUG (MainThread) [bellows.ezsp.protocol] Application frame 34 (permitJoining) received: b'00'
2021-07-28 00:28:12 DEBUG (MainThread) [bellows.ezsp.protocol] Send command nop: ()
2021-07-28 00:28:12 DEBUG (bellows.thread_0) [bellows.uart] Sending: b'142a21a9512a867d5e7e'
2021-07-28 00:28:12 DEBUG (bellows.thread_0) [bellows.uart] Data frame: b'422aa1a9512aa6337e'
2021-07-28 00:28:12 DEBUG (bellows.thread_0) [bellows.uart] Sending: b'8520dd7e'
2021-07-28 00:28:12 DEBUG (MainThread) [bellows.ezsp.protocol] Application frame 5 (nop) received: b''
2021-07-28 00:28:22 DEBUG (MainThread) [bellows.ezsp.protocol] Send command nop: ()
2021-07-28 00:28:22 DEBUG (bellows.thread_0) [bellows.uart] Sending: b'252b21a9512a46037e'
2021-07-28 00:28:22 DEBUG (bellows.thread_0) [bellows.uart] Data frame: b'532ba1a9512a53467e'
2021-07-28 00:28:22 DEBUG (bellows.thread_0) [bellows.uart] Sending: b'8610be7e'
2021-07-28 00:28:22 DEBUG (MainThread) [bellows.ezsp.protocol] Application frame 5 (nop) received: b''
2021-07-28 00:28:32 DEBUG (MainThread) [bellows.ezsp.protocol] Send command nop: ()
2021-07-28 00:28:32 DEBUG (bellows.thread_0) [bellows.uart] Sending: b'362821a9512a7cb57e'
2021-07-28 00:28:32 DEBUG (bellows.thread_0) [bellows.uart] Data frame: b'6428a1a9512a5a597e'
2021-07-28 00:28:32 DEBUG (bellows.thread_0) [bellows.uart] Sending: b'87009f7e'
2021-07-28 00:28:32 DEBUG (MainThread) [bellows.ezsp.protocol] Application frame 5 (nop) received: b''
[... cut ...]
2021-07-28 00:28:55 DEBUG (MainThread) [zigpy.zcl] [0x38e7:1:0x0402] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=GLOBAL_COMMAND manufacturer_specific=False is_reply=True disable_default_response=True> manufacturer=None tsn=68 command_id=Command.Report_Attributes>
2021-07-28 00:28:55 DEBUG (MainThread) [zigpy.zcl] [0x38e7:1:0x0402] ZCL request 0x000a: [[Attribute(attrid=0, value=<TypeValue type=int16s, value=2606>)]]
2021-07-28 00:28:55 DEBUG (MainThread) [zigpy.zcl] [0x38e7:1:0x0402] Attribute report received: measured_value=2606`

MattWestb, 2021-07-27:
Only 26.06 °C reported (0.2 °C lesser then my) and no sign of trying joining devices ;-))

Normally its one fuse on the PCB for the power to the Zigbee module it can being one 0 ohm resistor or one real fuse that can have being burned or its the power supply broken.

Can you measuring if you is having 3.3 V to the zigbee module (dont frying your fingers !!!!) ?
If need pinout for the module search for the model and tuya is having manuals for (most) them.

sergeysaley, 2021-07-27:
Probably not today - I already have a deep night, and tomorrow is a hard day ...

sergeysaley, 2021-07-29:
Can't get it to pair. I will try to find Tuya zigbee coordinator to reset it by Tuya app. There will be information - I will write. Thanks for your help. :)

MattWestb, 2021-08-25:
Is the switch still killed ?

I only asking because ZHA is rouing one updated quirks for the next release and i wondering if i shall making one PR for it.

sergeysaley, 2021-09-05:
Hi Matt!
Switch is alive again.
After updating HA to 2021.9 and removing local_quirks, the switch is paired with one entity, i.e. this quirk is already in zha.

**Zigbee info
IEEE: 84:2e:14:ff:fe:3a:53:9c
Nwk: 0x512f
Device Type: Router
LQI: 240
RSSI: -40
Last Seen: 2021-09-05T09:22:53
Power Source: Mains
Quirk: zhaquirks.tuya.ts0601.TuyaSingleSwitch**

MattWestb, 2021-09-05:
Great then you is having one half switch working as expected :-))
That is much better then one not working switch !!

Do you have trying if you is getting some tuya debug prints from the second gang ?

Put this log setting in HA:
```
    zhaquirks: info
    zhaquirks.tuya: debug
```
and restart HA and trigger some commands from the 2 gang and looking in the log.

Also development tools > Event > Listening to event and putting in `zha_event` and clicking start listening you shall getting the events from the first gang.

sergeysaley, 2021-09-05:
logger:
  logs:
    zhaquirks: info
    zhaquirks.tuya: debug

Logs are too big and contain of a lot of not Zigbee related info...
What pattern to grep in logs?

sergeysaley, 2021-09-05:
Nothing in Listening of zha_event. Absolutely.
One button press or another - silence...

MattWestb, 2021-09-05:
My tuya TRV is sending data and its looks like this:
```
2021-09-05 11:00:39 DEBUG (MainThread) [zhaquirks.tuya] [0x7f8b:1:0xef00] Received value [0, 0, 0, 253] for attribute 0x0203 (command 0x0001)
2021-09-05 11:06:35 DEBUG (MainThread) [zhaquirks.tuya] [0x5ce2:1:0xef00] Received value [0, 0, 0, 237] for attribute 0x0203 (command 0x0001)
```
So only logs from `[zhaquirks.tuya]` is interesting for see wot the "tuya magic" is doing or not.

If the switch is sending somthing you shall getting some tuya DP commands / attribute received in the log.

You can also grep on the `Nwk: 0x512f` and you shall see all communication with the switch.

sergeysaley, 2021-09-05:
2021-09-05 12:47:49 DEBUG (MainThread) [zhaquirks.tuya] 84:2e:14:ff:fe:3a:53:9c - Received switch event message, channel: 1, state: 1
2021-09-05 12:47:50 DEBUG (MainThread) [zhaquirks.tuya] 84:2e:14:ff:fe:3a:53:9c - Received switch event message, channel: 1, state: 1
2021-09-05 12:47:50 DEBUG (MainThread) [zhaquirks.tuya] 84:2e:14:ff:fe:3a:53:9c - Received switch event message, channel: 1, state: 1
2021-09-05 12:47:50 DEBUG (MainThread) [zhaquirks.tuya] 84:2e:14:ff:fe:3a:53:9c - Received switch event message, channel: 1, state: 1
2021-09-05 12:47:50 DEBUG (MainThread) [zhaquirks.tuya] 84:2e:14:ff:fe:3a:53:9c - Received switch event message, channel: 1, state: 1
2021-09-05 12:47:50 DEBUG (MainThread) [zhaquirks.tuya] 84:2e:14:ff:fe:3a:53:9c - Received switch event message, channel: 1, state: 1
2021-09-05 12:47:51 DEBUG (MainThread) [zhaquirks.tuya] 84:2e:14:ff:fe:3a:53:9c - Received switch event message, channel: 1, state: 1
2021-09-05 12:47:51 DEBUG (MainThread) [zhaquirks.tuya] 84:2e:14:ff:fe:3a:53:9c - Received switch event message, channel: 1, state: 1
2021-09-05 12:47:51 DEBUG (MainThread) [zhaquirks.tuya] 84:2e:14:ff:fe:3a:53:9c - Received switch event message, channel: 1, state: 1
2021-09-05 12:47:51 DEBUG (MainThread) [zhaquirks.tuya] 84:2e:14:ff:fe:3a:53:9c - Received switch event message, channel: 1, state: 1
2021-09-05 12:47:51 DEBUG (MainThread) [zhaquirks.tuya] 84:2e:14:ff:fe:3a:53:9c - Received switch event message, channel: 1, state: 1
2021-09-05 12:47:52 DEBUG (MainThread) [zhaquirks.tuya] 84:2e:14:ff:fe:3a:53:9c - Received switch event message, channel: 1, state: 1
2021-09-05 12:47:52 DEBUG (MainThread) [zhaquirks.tuya] 84:2e:14:ff:fe:3a:53:9c - Received switch event message, channel: 1, state: 1
2021-09-05 12:47:52 DEBUG (MainThread) [zhaquirks.tuya] 84:2e:14:ff:fe:3a:53:9c - Received switch event message, channel: 1, state: 1
2021-09-05 12:47:52 DEBUG (MainThread) [zhaquirks.tuya] 84:2e:14:ff:fe:3a:53:9c - Received switch event message, channel: 1, state: 1
2021-09-05 12:47:52 DEBUG (MainThread) [zhaquirks.tuya] 84:2e:14:ff:fe:3a:53:9c - Received switch event message, channel: 1, state: 1
2021-09-05 12:47:53 DEBUG (MainThread) [zhaquirks.tuya] 84:2e:14:ff:fe:3a:53:9c - Received switch event message, channel: 1, state: 1
2021-09-05 12:47:53 DEBUG (MainThread) [zhaquirks.tuya] 84:2e:14:ff:fe:3a:53:9c - Received switch event message, channel: 1, state: 1
2021-09-05 12:47:53 DEBUG (MainThread) [zhaquirks.tuya] 84:2e:14:ff:fe:3a:53:9c - Received switch event message, channel: 1, state: 1
2021-09-05 12:47:53 DEBUG (MainThread) [zhaquirks.tuya] 84:2e:14:ff:fe:3a:53:9c - Received switch event message, channel: 1, state: 1
2021-09-05 12:47:53 DEBUG (MainThread) [zhaquirks.tuya] 84:2e:14:ff:fe:3a:53:9c - Received switch event message, channel: 1, state: 1
2021-09-05 12:47:54 DEBUG (MainThread) [zhaquirks.tuya] 84:2e:14:ff:fe:3a:53:9c - Received switch event message, channel: 1, state: 1
2021-09-05 12:47:54 DEBUG (MainThread) [zhaquirks.tuya] 84:2e:14:ff:fe:3a:53:9c - Received switch event message, channel: 1, state: 1
2021-09-05 12:48:12 DEBUG (MainThread) [zhaquirks.tuya] 84:2e:14:ff:fe:3a:53:9c - Received switch event message, channel: 1, state: 0
2021-09-05 12:48:12 DEBUG (MainThread) [zhaquirks.tuya] 84:2e:14:ff:fe:3a:53:9c - Received switch event message, channel: 1, state: 0
2021-09-05 12:48:12 DEBUG (MainThread) [zhaquirks.tuya] 84:2e:14:ff:fe:3a:53:9c - Received switch event message, channel: 1, state: 0
2021-09-05 12:48:13 DEBUG (MainThread) [zhaquirks.tuya] 84:2e:14:ff:fe:3a:53:9c - Received switch event message, channel: 1, state: 0
2021-09-05 12:48:13 DEBUG (MainThread) [zhaquirks.tuya] 84:2e:14:ff:fe:3a:53:9c - Received switch event message, channel: 1, state: 0
2021-09-05 12:48:13 DEBUG (MainThread) [zhaquirks.tuya] 84:2e:14:ff:fe:3a:53:9c - Received switch event message, channel: 1, state: 0
2021-09-05 12:48:13 DEBUG (MainThread) [zhaquirks.tuya] 84:2e:14:ff:fe:3a:53:9c - Received switch event message, channel: 1, state: 0
2021-09-05 12:48:13 DEBUG (MainThread) [zhaquirks.tuya] 84:2e:14:ff:fe:3a:53:9c - Received switch event message, channel: 1, state: 0
2021-09-05 12:48:14 DEBUG (MainThread) [zhaquirks.tuya] 84:2e:14:ff:fe:3a:53:9c - Received switch event message, channel: 1, state: 0
2021-09-05 12:48:14 DEBUG (MainThread) [zhaquirks.tuya] 84:2e:14:ff:fe:3a:53:9c - Received switch event message, channel: 1, state: 0
2021-09-05 12:48:14 DEBUG (MainThread) [zhaquirks.tuya] 84:2e:14:ff:fe:3a:53:9c - Received switch event message, channel: 1, state: 0
2021-09-05 12:48:14 DEBUG (MainThread) [zhaquirks.tuya] 84:2e:14:ff:fe:3a:53:9c - Received switch event message, channel: 1, state: 0
2021-09-05 12:48:14 DEBUG (MainThread) [zhaquirks.tuya] 84:2e:14:ff:fe:3a:53:9c - Received switch event message, channel: 1, state: 0
2021-09-05 12:48:15 DEBUG (MainThread) [zhaquirks.tuya] 84:2e:14:ff:fe:3a:53:9c - Received switch event message, channel: 1, state: 0
2021-09-05 12:48:15 DEBUG (MainThread) [zhaquirks.tuya] 84:2e:14:ff:fe:3a:53:9c - Received switch event message, channel: 1, state: 0
2021-09-05 12:48:15 DEBUG (MainThread) [zhaquirks.tuya] 84:2e:14:ff:fe:3a:53:9c - Received switch event message, channel: 1, state: 0
2021-09-05 12:48:15 DEBUG (MainThread) [zhaquirks.tuya] 84:2e:14:ff:fe:3a:53:9c - Received switch event message, channel: 1, state: 0
2021-09-05 12:48:15 DEBUG (MainThread) [zhaquirks.tuya] 84:2e:14:ff:fe:3a:53:9c - Received switch event message, channel: 1, state: 0
[... cut ...]
2021-09-05 12:48:50 DEBUG (MainThread) [zhaquirks.tuya] 84:2e:14:ff:fe:3a:53:9c - Received switch event message, channel: 2, state: 0
2021-09-05 12:48:50 DEBUG (MainThread) [zhaquirks.tuya] 84:2e:14:ff:fe:3a:53:9c - Received switch event message, channel: 2, state: 0
2021-09-05 12:48:51 DEBUG (MainThread) [zhaquirks.tuya] 84:2e:14:ff:fe:3a:53:9c - Received switch event message, channel: 2, state: 0
2021-09-05 12:48:51 DEBUG (MainThread) [zhaquirks.tuya] 84:2e:14:ff:fe:3a:53:9c - Received switch event message, channel: 2, state: 0
~ $

sergeysaley, 2021-09-05:
Is it normal for a switch to spam the system with status messages? One click comes a couple of dozen identical messages

PS: Sorry for unformatted messages, cannot remind keys :)

sergeysaley, 2021-09-05:
AFK, sorry
Will return tonight or tomorrow morning.

MattWestb, 2021-09-05:
Put ` ``` `before and after the code and its being treated as one code block and can being easier to read.

MattWestb, 2021-09-05:
The good thing you is getting command 0 and 1 from channel 1 and 2 = from both switches :-))

Strange that tuya quirk is getting the commands with IEEEE address and not Nwk address.
Is zigpy also using the IEEE or the Mwk address for this device then receiving and sending commands to this device ?

For multiple command you need look in the ZHA receiving commands if they is having the same tsn (transaction number). ZHA is sending default response for all ZCL commands if its flagged as needed but tuya is using no ZCL cluster so it need one manual action in the quirk for sending it or the device is reseeding the same command with the same tsn then not getting and ack for it.

One example of attribute reporting (from one light):
```
2021-09-05 12:10:15 DEBUG (MainThread) [zigpy.zcl] [0xefd5:1:0x0300] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=GLOBAL_COMMAND manufacturer_specific=False is_reply=True disable_default_response=False> manufacturer=None tsn=70 command_id=Command.Report_Attributes>
2021-09-05 12:10:15 DEBUG (MainThread) [zigpy.zcl] [0xefd5:1:0x0300] ZCL request 0x000a: [[Attribute(attrid=3, value=<TypeValue type=uint16_t, value=32768>), Attribute(attrid=4, value=<TypeValue type=uint16_t, value=15729>)]]
2021-09-05 12:10:15 DEBUG (MainThread) [zigpy.zcl] [0xefd5:1:0x0300] Attribute report received: current_x=32768, current_y=15729
2021-09-05 12:10:15 DEBUG (MainThread) [bellows.zigbee.application] Set source route for 0xEFD5 to []: EmberStatus.SUCCESS
```
The `tsn=70` is the interesting and then its one ZCL command its getting default response from the Zigbee stack but for tuya it can being is not being sent (then its not one ZCL cluster command its one custom command alla tuya) then the devices is reseeding the same command after some milliseconds with the same tsn.

sergeysaley, 2021-09-05:
Logs from ZHA: 
```
[0x0000:zdo] ZDO request ZDOCmd.Mgmt_Permit_Joining_req: [60, <Bool.false: 0>]
[0x512f:1:0xef00] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=CLUSTER_COMMAND manufacturer_specific=False is_reply=True disable_default_response=False> manufacturer=None tsn=119 command_id=2>
[0x512f:1:0xef00] ZCL request 0x0002: [Command(status=0, tsn=114, command_id=257, function=0, data=[1, 1])]
[0x512f:1:0xef00] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=CLUSTER_COMMAND manufacturer_specific=False is_reply=True disable_default_response=False> manufacturer=None tsn=119 command_id=2>
[0x512f:1:0xef00] ZCL request 0x0002: [Command(status=0, tsn=114, command_id=257, function=0, data=[1, 1])]
[0x512f:1:0xef00] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=CLUSTER_COMMAND manufacturer_specific=False is_reply=True disable_default_response=False> manufacturer=None tsn=119 command_id=2>
[0x512f:1:0xef00] ZCL request 0x0002: [Command(status=0, tsn=114, command_id=257, function=0, data=[1, 1])]
[0x512f:1:0xef00] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=CLUSTER_COMMAND manufacturer_specific=False is_reply=True disable_default_response=False> manufacturer=None tsn=119 command_id=2>
[0x512f:1:0xef00] ZCL request 0x0002: [Command(status=0, tsn=114, command_id=257, function=0, data=[1, 1])]
[0x512f:1:0xef00] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=CLUSTER_COMMAND manufacturer_specific=False is_reply=True disable_default_response=False> manufacturer=None tsn=119 command_id=2>
[0x512f:1:0xef00] ZCL request 0x0002: [Command(status=0, tsn=114, command_id=257, function=0, data=[1, 1])]
[0x512f:1:0xef00] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=CLUSTER_COMMAND manufacturer_specific=False is_reply=True disable_default_response=False> manufacturer=None tsn=119 command_id=2>
[0x512f:1:0xef00] ZCL request 0x0002: [Command(status=0, tsn=114, command_id=257, function=0, data=[1, 1])]
[0x512f:1:0xef00] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=CLUSTER_COMMAND manufacturer_specific=False is_reply=True disable_default_response=False> manufacturer=None tsn=119 command_id=2>
[0x512f:1:0xef00] ZCL request 0x0002: [Command(status=0, tsn=114, command_id=257, function=0, data=[1, 1])]
[0x512f:1:0xef00] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=CLUSTER_COMMAND manufacturer_specific=False is_reply=True disable_default_response=False> manufacturer=None tsn=119 command_id=2>
[0x512f:1:0xef00] ZCL request 0x0002: [Command(status=0, tsn=114, command_id=257, function=0, data=[1, 1])]
[0x512f:1:0xef00] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=CLUSTER_COMMAND manufacturer_specific=False is_reply=True disable_default_response=False> manufacturer=None tsn=119 command_id=2>
[0x512f:1:0xef00] ZCL request 0x0002: [Command(status=0, tsn=114, command_id=257, function=0, data=[1, 1])]
[0x512f:1:0xef00] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=CLUSTER_COMMAND manufacturer_specific=False is_reply=True disable_default_response=False> manufacturer=None tsn=119 command_id=2>
[0x512f:1:0xef00] ZCL request 0x0002: [Command(status=0, tsn=114, command_id=257, function=0, data=[1, 1])]
[0x512f:1:0xef00] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=CLUSTER_COMMAND manufacturer_specific=False is_reply=True disable_default_response=False> manufacturer=None tsn=119 command_id=2>
[0x512f:1:0xef00] ZCL request 0x0002: [Command(status=0, tsn=114, command_id=257, function=0, data=[1, 1])]
[0x512f:1:0xef00] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=CLUSTER_COMMAND manufacturer_specific=False is_reply=True disable_default_response=False> manufacturer=None tsn=119 command_id=2>
[0x512f:1:0xef00] ZCL request 0x0002: [Command(status=0, tsn=114, command_id=257, function=0, data=[1, 1])]
[0x512f:1:0xef00] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=CLUSTER_COMMAND manufacturer_specific=False is_reply=True disable_default_response=False> manufacturer=None tsn=119 command_id=2>
[0x512f:1:0xef00] ZCL request 0x0002: [Command(status=0, tsn=114, command_id=257, function=0, data=[1, 1])]
[0x512f:1:0xef00] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=CLUSTER_COMMAND manufacturer_specific=False is_reply=True disable_default_response=False> manufacturer=None tsn=119 command_id=2>
[0x512f:1:0xef00] ZCL request 0x0002: [Command(status=0, tsn=114, command_id=257, function=0, data=[1, 1])]
[0x512f:1:0xef00] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=CLUSTER_COMMAND manufacturer_specific=False is_reply=True disable_default_response=False> manufacturer=None tsn=119 command_id=2>
[0x512f:1:0xef00] ZCL request 0x0002: [Command(status=0, tsn=114, command_id=257, function=0, data=[1, 1])]
[0x512f:1:0xef00] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=CLUSTER_COMMAND manufacturer_specific=False is_reply=True disable_default_response=False> manufacturer=None tsn=119 command_id=2>
[0x512f:1:0xef00] ZCL request 0x0002: [Command(status=0, tsn=114, command_id=257, function=0, data=[1, 1])]
[0x512f:1:0xef00] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=CLUSTER_COMMAND manufacturer_specific=False is_reply=True disable_default_response=False> manufacturer=None tsn=119 command_id=2>
[... cut ...]
[0x512f:1:0xef00] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=CLUSTER_COMMAND manufacturer_specific=False is_reply=True disable_default_response=False> manufacturer=None tsn=122 command_id=2>
[0x512f:1:0xef00] ZCL request 0x0002: [Command(status=0, tsn=117, command_id=258, function=0, data=[1, 0])]
[0x512f:1:0xef00] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=CLUSTER_COMMAND manufacturer_specific=False is_reply=True disable_default_response=False> manufacturer=None tsn=122 command_id=2>
[0x512f:1:0xef00] ZCL request 0x0002: [Command(status=0, tsn=117, command_id=258, function=0, data=[1, 0])]

```

MattWestb, 2021-09-05:
The device is reseeding the same command then its not getting and default response from the coordinator that shall the device do !!

The problem i cant see the default response part its implanted in the on/off part in the tuya quirk only for the Remote OnOff switches :-((

sergeysaley, 2021-09-05:
If ZHA already has quirk for **("_TZE200_wunufsil", "TS0601")** and I make my file and put it in custom_zha_quirks, which of the quirks should be applied?

sergeysaley, 2021-09-05:
As I see, ZHA ignores custom quirk and uses internal one.

MattWestb, 2021-09-05:
I have not testing but the devs have saying that the file in the custom quirk shall being used of the system if the system is having both.

sergeysaley, 2021-09-05:
I left my model in /config/custom_zha_quirks/ts0601.py and added an endpoint for the second port. After rebooting HA and removing / adding the switch, I see that the internal quirk is being used
```
Zigbee info
IEEE: 84:2e:14:ff:fe:3a:53:9c
Nwk: 0x74d3
Device Type: Router
LQI: 240
RSSI: -40
Last Seen: 2021-09-05T17:52:20
Power Source: Mains
Quirk: zhaquirks.tuya.ts0601.TuyaSingleSwitch 
```
My quirk file content is:

```
"""Tuya based touch switch."""
from zigpy.profiles import zha
from zigpy.zcl.clusters.general import Basic, Groups, Ota, Scenes, Time

from zhaquirks.const import (
    DEVICE_TYPE,
    ENDPOINTS,
    INPUT_CLUSTERS,
    MODELS_INFO,
    OUTPUT_CLUSTERS,
    PROFILE_ID,
)
from zhaquirks.tuya import (
    TuyaManufacturerClusterOnOff,
    TuyaManufCluster,
    TuyaOnOff,
    TuyaSwitch
)

class TuyaMySwitch(TuyaSwitch):
    """Tuya touch switch device."""

    signature = {
        # "node_descriptor": "<NodeDescriptor byte1=1 byte2=64 mac_capability_flags=142 manufacturer_code=4098
        #                       maximum_buffer_size=82 maximum_incoming_transfer_size=82 server_mask=11264
        #                       maximum_outgoing_transfer_size=82 descriptor_capability_field=0>",
        # device_version=1
        # input_clusters=[0x0000, 0x0004, 0x0005, 0xef00]
        # output_clusters=[0x000a, 0x0019]
        # <SimpleDescriptor endpoint=1 profile=260 device_type=51 device_version=1 input_clusters=[0, 4, 5, 61184] output_clusters=[10, 25]>
        MODELS_INFO: [
            ("_TZE200_wunufsil", "TS0601")
        ],
        ENDPOINTS: {
            1: {
                PROFILE_ID: zha.PROFILE_ID,
                DEVICE_TYPE: zha.DeviceType.SMART_PLUG,
                INPUT_CLUSTERS: [
                    Basic.cluster_id,
                    Groups.cluster_id,
                    Scenes.cluster_id,
                    TuyaManufCluster.cluster_id,
                ],
                OUTPUT_CLUSTERS: [Time.cluster_id, Ota.cluster_id],
            },
            2: {
                PROFILE_ID: zha.PROFILE_ID,
                DEVICE_TYPE: zha.DeviceType.SMART_PLUG,
                INPUT_CLUSTERS: [
                    Basic.cluster_id,
                    Groups.cluster_id,
                    Scenes.cluster_id,
                    TuyaManufCluster.cluster_id,
                ],
                OUTPUT_CLUSTERS: [Time.cluster_id, Ota.cluster_id],
            }
        },
    }

    replacement = {
        ENDPOINTS: {
            1: {
                DEVICE_TYPE: zha.DeviceType.ON_OFF_LIGHT,
                INPUT_CLUSTERS: [
                    Basic.cluster_id,
                    Groups.cluster_id,
                    Scenes.cluster_id,
                    TuyaManufacturerClusterOnOff,
                    TuyaOnOff,
                ],
                OUTPUT_CLUSTERS: [Time.cluster_id, Ota.cluster_id],
            },
            2: {
                DEVICE_TYPE: zha.DeviceType.ON_OFF_LIGHT,
                INPUT_CLUSTERS: [
                    Basic.cluster_id,
                    Groups.cluster_id,
                    Scenes.cluster_id,
                    TuyaManufacturerClusterOnOff,
                    TuyaOnOff,
                ],
                OUTPUT_CLUSTERS: [Time.cluster_id, Ota.cluster_id],
            }
        }
    }

```

sergeysaley, 2021-09-05:
Maybe TuyaSwitch class contains only one port switch signatures? And for two port switch another parent class must be used?

MattWestb, 2021-09-05:
I think it can not being triggered then the signature is not being the switch is only having endpoint 1.

Look wot the "Zigbee device signature" from the device card if its have only EP1 or if its loaded also EP2.

In the log you was getting tuya DP commands from 2 channels 1 and 2 that is the 2 switches you is having on your device.

sergeysaley, 2021-09-05:
```
{
  "node_descriptor": "NodeDescriptor(logical_type=<LogicalType.Router: 1>, complex_descriptor_available=0, user_descriptor_available=0, reserved=0, aps_flags=0, frequency_band=<FrequencyBand.Freq2400MHz: 8>, mac_capability_flags=<MACCapabilityFlags.AllocateAddress|RxOnWhenIdle|MainsPowered|FullFunctionDevice: 142>, manufacturer_code=4098, maximum_buffer_size=82, maximum_incoming_transfer_size=82, server_mask=11264, maximum_outgoing_transfer_size=82, descriptor_capability_field=<DescriptorCapability.NONE: 0>, *allocate_address=True, *is_alternate_pan_coordinator=False, *is_coordinator=False, *is_end_device=False, *is_full_function_device=True, *is_mains_powered=True, *is_receiver_on_when_idle=True, *is_router=True, *is_security_capable=False)",
  "endpoints": {
    "1": {
      "profile_id": 260,
      "device_type": "0x0100",
      "in_clusters": [
        "0x0000",
        "0x0004",
        "0x0005",
        "0x0006",
        "0xef00"
      ],
      "out_clusters": [
        "0x000a",
        "0x0019"
      ]
    }
  },
  "manufacturer": "_TZE200_wunufsil",
  "model": "TS0601",
  "class": "zhaquirks.tuya.ts0601.TuyaSingleSwitch"
}
```

MattWestb, 2021-09-05:
The single switch quirk  is made for single switches but no one have doing one quirk for multi gang tuya switches so that logic need being added for the second gang to working OK :-((

The device is using the internal quirk in ZHA then its only having EP1 (your quirk shall have showing EP 2 and 2).

sergeysaley, 2021-09-05:
> The device is using the internal quirk in ZHA then its only having EP1 (your quirk shall have showing EP 2 and 2).

I see. But WHY internal quirk is used? There is definition for "manufacturer": "_TZE200_wunufsil",
  "model": "TS0601" in custom_zha_quirks/ts0601.py, and this file must override internal definitions, isn't it?

sergeysaley, 2021-09-05:
In addition to simple overrides, does quirk assume some kind of processing logic in parent classes that cannot be defined in quirk for a particular model, if there is no corresponding processing part above?

MattWestb, 2021-09-05:
First this part must matching the quirk signature for it shall being loaded:
```
{
  "node_descriptor": "NodeDescriptor(logical_type=<LogicalType.Router: 1>, complex_descriptor_available=0, user_descriptor_available=0, reserved=0, aps_flags=0, frequency_band=<FrequencyBand.Freq2400MHz: 8>, mac_capability_flags=<MACCapabilityFlags.AllocateAddress|RxOnWhenIdle|MainsPowered|FullFunctionDevice: 142>, manufacturer_code=4098, maximum_buffer_size=82, maximum_incoming_transfer_size=82, server_mask=11264, maximum_outgoing_transfer_size=82, descriptor_capability_field=<DescriptorCapability.NONE: 0>, *allocate_address=True, *is_alternate_pan_coordinator=False, *is_coordinator=False, *is_end_device=False, *is_full_function_device=True, *is_mains_powered=True, *is_receiver_on_when_idle=True, *is_router=True, *is_security_capable=False)",
  "endpoints": {
    "1": {
      "profile_id": 260,
      "device_type": "0x0100",
      "in_clusters": [
        "0x0000",
        "0x0004",
        "0x0005",
        "0x0006",
        "0xef00"
      ],
      "out_clusters": [
        "0x000a",
        "0x0019"
      ]
    }
  },
  "manufacturer": "_TZE200_wunufsil",
  "model": "TS0601",
  "class": "zhaquirks.tuya.ts0601.TuyaSingleSwitch"
}
```
If its not ZHA is not using the quirk (miss match error in the log).

The logic i cant telling then i dont knowing it in real but the devs have saying custom quirks is taking over internal ones if the signature is OK.

The limiting is all custom quirks is using its import from the internal ones so can having 2 custom quirks that is depending of etch other then one or both need being patched in the HA container.

You can trying editing the quirk in the container as https://github.com/zigpy/zha-device-handlers/discussions/693#discussion-1237451.

sergeysaley, 2021-09-05:
```
021-09-05 18:57:17 DEBUG (MainThread) [zigpy.application] Device is initialized <Device model='TS0601' manuf='_TZE200_wunufsil' nwk=0xD933 ieee=84:2e:14:ff:fe:3a:53:9c is_initialized=True>
2021-09-05 18:57:17 DEBUG (MainThread) [zigpy.quirks.registry] Checking quirks for _TZE200_wunufsil TS0601 (84:2e:14:ff:fe:3a:53:9c)
2021-09-05 18:57:17 DEBUG (MainThread) [zigpy.quirks.registry] Considering <class 'ts0601.TuyaDoubleSwitch'>
2021-09-05 18:57:17 DEBUG (MainThread) [zigpy.quirks.registry] Fail because endpoint list mismatch: {1, 2} {1}
2021-09-05 18:57:17 DEBUG (MainThread) [zigpy.quirks.registry] Considering <class 'zhaquirks.tuya.ts0601.TuyaSingleSwitch'>
```

**Fail because endpoint list mismatch: {1, 2} {1}**

Thus, without access to the logic of the parent classes, we will not solve anything.
Alas

MattWestb, 2021-09-05:
Use the signature from the original 1 button and only changing the "replace" part.
The parent is defined in the https://github.com/zigpy/zha-device-handlers/blob/dev/zhaquirks/tuya/__init__.py.
But if you can doing changes in both in the custom quirk folder then both is "reading" from ther parents in the internal (the HA container) and then must doing the changed in it (at last the `__init__.py`) in the container.

MattWestb, 2021-09-05:
This must being unchaged or the quirk is not being loaded (the "name" is OK to changing). 
```
class TuyaSingleSwitch(TuyaSwitch):
    """Tuya touch switch device."""

    signature = {
        # "node_descriptor": "<NodeDescriptor byte1=1 byte2=64 mac_capability_flags=142 manufacturer_code=4098
        #                       maximum_buffer_size=82 maximum_incoming_transfer_size=82 server_mask=11264
        #                       maximum_outgoing_transfer_size=82 descriptor_capability_field=0>",
        # device_version=1
        # input_clusters=[0x0000, 0x0004, 0x0005, 0xef00]
        # output_clusters=[0x000a, 0x0019]
        # <SimpleDescriptor endpoint=1 profile=260 device_type=51 device_version=1 input_clusters=[0, 4, 5, 61184] output_clusters=[10, 25]>
        MODELS_INFO: [
            ("_TZE200_oisqyl4o", "TS0601"),
            ("_TZE200_wunufsil", "TS0601"),
            ("_TZE200_vhy3iakz", "TS0601"),
            ("_TZ3000_uim07oem", "TS0601"),
            ("_TZE200_tz32mtza", "TS0601"),
        ],
        ENDPOINTS: {
            1: {
                PROFILE_ID: zha.PROFILE_ID,
                DEVICE_TYPE: zha.DeviceType.SMART_PLUG,
                INPUT_CLUSTERS: [
                    Basic.cluster_id,
                    Groups.cluster_id,
                    Scenes.cluster_id,
                    TuyaManufCluster.cluster_id,
                ],
                OUTPUT_CLUSTERS: [Time.cluster_id, Ota.cluster_id],
            }
        },
    }
```` 
Do editing after the replace that is here:
```` 
    replacement = {
        ENDPOINTS: {
            1: {
                DEVICE_TYPE: zha.DeviceType.ON_OFF_LIGHT,
                INPUT_CLUSTERS: [
                    Basic.cluster_id,
                    Groups.cluster_id,
                    Scenes.cluster_id,
                    TuyaManufacturerClusterOnOff,
                    TuyaOnOff,
                ],
                OUTPUT_CLUSTERS: [Time.cluster_id, Ota.cluster_id],
            }
        }
    }
```

sergeysaley, 2021-09-05:
If I leave the signature with one endpoint, and describe two endpoints in replacement, then zha does not start at all

MattWestb, 2021-09-05:
Its one tuya device quirk that is adding EP2 but i dont knowing how its working doing it https://github.com/zigpy/zha-device-handlers/pull/926.

sergeysaley, 2021-09-05:
OK. Will try to read parent classes and understand the logic. Tomorrow...

sergeysaley, 2021-09-05:
I'v got two entities for now.
Added cluster_id and PROFILE_ID in replacement section.
PS: Sometimes whiskey is not only harmful, but also useful)

[30 more replies]

PR #1166 (2021-11-23): Add quirks for Tuya multigang switches
This PR add quirks for Tuya multigang switches.

It should correct the management of the On/Off cluster for the multiswitch devices that make use of the Tuya cluster (`0xEF00`) for command/status reporting.

The `MODELS_INFO` of those devices that have been reported or an issue has been reported have been added.
For those devices whose number of gangs is not clear, they have been defined as 1-gang devices until someone report correct info.

This could fix the following issues (only on/off switches):
fixes: #813
fixes: #824
fixes: #913
fixes: #972
fixes: # [... cut]
closed 2021-11-23 by dmulcahey