zigpy/zha-device-handlers #975: [Device Support Request] IKEA TRADFRI W2045

WeterPeter, 2021-07-24:
Hi,

I have purchased a IKEA light bulb switch bombo, the bulb is supported (LED194G9) but the switch (type W2045 19972) apparently not. 

https://www.ikea.com/nl/nl/p/tradfri-set-met-afstandsbediening-kleur-en-wit-spectrum-60462432/

I could not find any other requests on the type numbers.
I am fairly new to this stuff, is this the right way to ask for a device addition to ZHA? 

Regards,
Peter

MattWestb, 2021-07-24:
IKEA Styrbar alias E2001 and E2002 is working out of the box if using it for sending commands to bounded light groups (as it intend to do).

Device automatons is also working but the quirk is not merged then need fixing the command  spamming from the device. 
You can install the quirk from this pull and adding the last changes that is in the comments for getting it working OK (but it still spamming the automation) #867.

The W2045 is not type number its the production year and week so pleas us the real product number that is E2001 and E2002.

WeterPeter, 2021-07-26:
Thank you! 

Regards,
Peter

PR #867 (2021-08-25): IKEA Styrbar remote quirk
Fixes #863, fixes #975

Adds device automation triggers and doubles the power percentage.
Left/right click are apparently working now.

Like the IKEA shortcut button, this remote seems to generate multiple ``zha_event``s.
(At least for @MattWestb on a Silabs(?) coordinator. AFAIK, some people with a TI coordinator didn't have an issue with multiple events on the shortcut button.)
The "spamming" issue needs to be looked at later (for all newer IKEA remotes), but this isn't caused by the quirk.
(Reference: https://github.com/zigpy/zha-device-handlers/issues/967)

For no [... cut]
closed 2021-08-25 by dmulcahey