zigpy/zha-device-handlers #977: [Device Support Request]

froggyfly, 2021-07-26:
**Is your feature request related to a problem? Please describe.**
OWON Technology Inc. PC321, Energy Monitor support needed please. 
Device can be paired @ZHA, but appeared entity shows only shows "Instantaneous Demand" ( 0x0400，Instantaneous energy consumed summation of the 3 phases)
According to the manufacturer's documents, Please help adding 0x0000 current summary delivered (Current summation delivered of the 3 phases);
This data can be now obtained through the following process:
integration-ZHA-Owon Pc321 device-manage clusters- clusters "metering" (0x0702)- current_ summ_ delivered (id: 0x0000)- get zigbee attribute

more info can be shown would be great, thanks!

**Describe the solution you'd like**

More monitoring data is needed

**Device signature - this can be acquired by removing the device from ZHA and pairing it again from the add devices screen. Be sure to add the entire content of the log panel after pairing the device to a code block below this line.**
{
  "node_descriptor": "NodeDescriptor(logical_type=<LogicalType.Router: 1>, complex_descriptor_available=0, user_descriptor_available=0, reserved=0, aps_flags=0, frequency_band=<FrequencyBand.Freq2400MHz: 8>, mac_capability_flags=<MACCapabilityFlags.AllocateAddress|RxOnWhenIdle|MainsPowered|FullFunctionDevice: 142>, manufacturer_code=4412, maximum_buffer_size=82, maximum_incoming_transfer_size=82, server_mask=11264, maximum_outgoing_transfer_size=82, descriptor_capability_field=<DescriptorCapability.0: 0>, *allocate_address=True, *is_alternate_pan_coordinator=False, *is_coordinator=False, *is_end_device=False, *is_full_function_device=True, *is_mains_powered=True, *is_receiver_on_when_idle=True, *is_router=True, *is_security_capable=False)",
  "endpoints": {
    "1": {
      "profile_id": 260,
      "device_type": "0x000d",
      "in_clusters": [
        "0x0000",
        "0x0003",
        "0x0702"
      ],
      "out_clusters": [
        "0x0003"
      ]
    }
  },
  "manufacturer": "OWON Technology Inc.",
  "model": "PC321",
  "class": "zigpy.device.Device"
}

**Additional context**
Add any other context or screenshots about the feature request here.

Manufacturer's documents can be obtained here:
[https://github.com/froggyfly/pc321-zigbee/blob/main/PC321zigbee.docx](https://github.com/froggyfly/pc321-zigbee/blob/main/PC321zigbee.docx)

mdeweerd, 2021-10-05:
I would also like to get the CurrentSummationDelivered from a device (to have it reported in Home Assistant).

Documentation: http://81.250.221.126//en/telecharger.php?fichier=um_zmeter.pdf

```json
{
  "node_descriptor": "NodeDescriptor(logical_type=<LogicalType.EndDevice: 2>, complex_descriptor_available=0, user_descriptor_available=1, reserved=0, aps_flags=0, frequency_band=<FrequencyBand.Freq2400MHz: 8>, mac_capability_flags=<MACCapabilityFlags.AllocateAddress: 128>, manufacturer_code=4281, maximum_buffer_size=80, maximum_incoming_transfer_size=160, server_mask=0, maximum_outgoing_transfer_size=160, descriptor_capability_field=<DescriptorCapability.ExtendedSimpleDescriptorListAvailable|ExtendedActiveEndpointListAvailable: 3>, *allocate_address=True, *is_alternate_pan_coordinator=False, *is_coordinator=False, *is_end_device=True, *is_full_function_device=False, *is_mains_powered=False, *is_receiver_on_when_idle=False, *is_router=False, *is_security_capable=False)",
  "endpoints": {
    "1": {
      "profile_id": 260,
      "device_type": "0x0501",
      "in_clusters": [
        "0x0000",
        "0x0001",
        "0x0003",
        "0x0009",
        "0x0702"
      ],
      "out_clusters": []
    }
  },
  "manufacturer": "CLEODE",
  "model": "ZMETER/E",
  "class": "zigpy.device.Device"
}
```

Is there some other implementation that we could start from?

I am appending this Request here as I suppose that the implementation is going to be pretty similar.

MattWestb, 2021-10-05:
Its one fix for power measuring in HA beta that is coming in the 2012.10 so testing the beta or waiting for the GA.

Hedda, 2021-10-06:
FYI, it looks like support for more clusters attributes is now being worked on for [ZHA integration](https://www.home-assistant.io/integrations/zha) in Home Assistant core, see:

https://github.com/home-assistant/core/pull/56909

and

https://github.com/home-assistant/core/pull/56666

Also, note the last comment from Adminiuga in PR for ZHA in Home Assistant core -> https://github.com/home-assistant/core/pull/56909

"**_Support for other attributes could be extended later, as I couldn't find too many devices other than Sinopé RM3250ZB Electrical Load Controller – 50A – Zigbee https://www.amazon.com/dp/B07CGJ6XH2/ reporting anything interesting on the electrical measurement cluster._**"

So maybe now you could try to get this working with the latest Home Assistant beta or wait for the next upcoming release:

https://www.home-assistant.io/faq/release/

PS: There is much more discussion about the current ZHA limitation of only supporting 1 sensor per cluster here -> https://github.com/zigpy/zha-device-handlers/issues/605

closed 2022-04-11 by github-actions[bot]

adi90x, 2022-07-17:
Reoppening as it still an issue on zha

fergalish, 2022-09-08:
Hi @froggyfly , I have one of these devices and, like you, it's only reporting a single power & energy value, each the summation of the three phases.
Did you ever manage to reconfigure to obtain each phase independently?
Thanks.