zigpy/zha-device-handlers #981: Tuya TRV TZE200: wrong setpoint temperature

martin3000, 2021-07-29:
I have a TS0602 ,  _TZE200_b6wax7g0
If I set manually set the TRV to 25°C at the valve, home assistant shows 2.5°C. The temperature in HA is always to small by a factor of 10.
If I look at the cluster MoesThermostat -> occupied_heating_setpoint, I see 2.5, but the TRV shows 25.

MattWestb, 2021-07-29:
Its one known issue with the software version `_TZE200_b6wax7g0` and for fixing it it need  one new quirk only for it.
[HA issue](https://community.home-assistant.io/t/zha-fails-after-custom-quirks-added/317056/18)

The good thing the problem is not in ZHA its in the quirk so its easier to paying  with.
You can trying copy the `/tuya/valve.py` to your local quirk directory and editing it if you dont have more moes TRVs in  your system.

I think the converting is taking place in this section https://github.com/zigpy/zha-device-handlers/blob/21af0860992021c39b26260b6e2ae4f9c1af75d1/zhaquirks/tuya/valve.py#L202 or in this https://github.com/zigpy/zha-device-handlers/blob/21af0860992021c39b26260b6e2ae4f9c1af75d1/zhaquirks/tuya/valve.py#L345

If you finding the right place and can changing it the its needed making one new version of  MoesHY368_Type1  and 2 quirk parts (perhaps named "ZTRV-BY-100 Type 1 / 2" and making new moes cluster that is using the new changed and adding the signature from your TRV in the signature but with both cluster version (tuya is using 2 diferent Zigbee modules for there TRVs TZE200 and TYST11 that need having different signatures for being loaded and also shall having the same changes).

I hope you is better on coding then i  am (around zero or minus on my side) and shall being possible implanting but its little tricky for not braking the other 2 types that is using the mors quirk (plus 5 Siterwell TRVs).

martin3000, 2021-07-29:
Yes it is in direct_mapping_attrs, but I wonder why just my TRV version does not need the divisor whereas other models need it. This is unusual.

MattWestb, 2021-07-29:
The ZTRV-BY-100 is one very new TRV and its one updated moes type.

Z2M have adding the ZTRV-BY-100 and have doing some changes from the "normal" moes TRVs and no have updating the ZHA quirks for it and some user have saying it working OK with ZTRV-BY-100 = better then not at all. 

The Moes HY367-ZB HY368-ZB / HY369-ZB is all more or less the same MCU used only external design but the ZTRV-BY-100 looks using one other / newer / updated MCU code that is making the difference.

If you have finding the attribute that needs being updated pleas posting it so other user can patching there system and getting it working better.

martin3000, 2021-07-29:
Does this mean that there are different models with the same manufacturer/model code of "_TZE200_b6wax7g0", "TS0601"

MattWestb, 2021-07-29:
Not with the same model code but very likely one TZE200 and TYST11  version (tuya TRVs is normally coming in both versions) but there is having the same MCU firmware but different zigbee modules.
Look on the [TRV matrix](https://github.com/zigpy/zigpy/discussions/653#discussioncomment-314395).

Your target temp is wrong shall being 250.

My test Siterwell is showing 230 = 23.0 in HA
```
Clusters
SiterwellManufCluster (Endpoint id: 1, Id: 0xef00, Type: in)

Attributes of the selected cluster
target_temperature (id: 0x0202)
 
Value
230
 ```
HA
```
Thermostat Test
Heating (Heat) 23 °C
Currently: 27.2 °C
```
So ZHA is doing the "zigbee" part OK but the device is sending wrong value (/10) for the target_temperature that must being "fixed" in the quirk.

PM: Siterwell TRVa is using the same quirk but one different cluster config then its having lesser functions.

martin3000, 2021-07-29:
Ok I changed the 10 into 100 and that seems to do it:

`        
     MOES_TARGET_TEMP_ATTR: ("occupied_heating_setpoint", lambda value: value * 100),  # jms: 10->100
`        

`        
        "occupied_heating_setpoint": (
            MOES_TARGET_TEMP_ATTR,
            lambda value: round(value / 100),     #jms 10->100
        ),
`

MattWestb, 2021-07-29:
So now its working setting the temperature from HA and also "reading" the temperature from the device ?

If yes can you posting the lines you was changing so other user can finding the lines and doing the change in there local quirks and getting the TRVs working OK.

If some like adding one updated part for "ZTRV-BY-100" with own cluster config is the PR more then welcome  :-))

The 0.5 °C is being done in the ZHA part and is only then setting the temperature not reading it from the device.
It was the same then reading the attribute from the devices but its updated very early to 0.1°C for getting better graph in the GUI.

You can still setting 0.1 °C in cluster commands and HA is making +- of its current reading :-))

MattWestb, 2021-07-29:
I have finding them:
https://github.com/zigpy/zha-device-handlers/blob/21af0860992021c39b26260b6e2ae4f9c1af75d1/zhaquirks/tuya/valve.py#L204
and
https://github.com/zigpy/zha-device-handlers/blob/21af0860992021c39b26260b6e2ae4f9c1af75d1/zhaquirks/tuya/valve.py#L348

Its looks it was one "undocumented future" (= Bug) then all other attributes is 100 and is fixed in the new MCU firmware.

martin3000, 2021-07-29:
I think the problem with the 0.5° is sth. different: the degrees are uint. if you have centidegrees in uints, this is no problem.
But if you store degrees in uint, of course the digit right to the decimal point is lost.

MattWestb, 2021-07-31:
I was trying changing the set point of my test TRV from 23.0 to 22.5 and  then back and its looks little different:
```
2021-07-31 16:04:50 DEBUG (MainThread) [zhaquirks.tuya] [0x0c26:1:0x0201] Mapping standard occupied_heating_setpoint (0x0012) with value 2250 to custom {514: 225}
2021-07-31 16:04:50 DEBUG (MainThread) [zhaquirks.tuya] [0x0c26:1:0xef00] Received value [0, 0, 0, 225] for attribute 0x0202 (command 0x0002)
2021-07-31 16:04:51 DEBUG (MainThread) [zhaquirks.tuya] [0x0c26:1:0xef00] Received value [0, 0, 0, 225] for attribute 0x0202 (command 0x0001)
2021-07-31 16:04:53 DEBUG (MainThread) [zhaquirks.tuya] [0x0c26:1:0x0201] Mapping standard occupied_heating_setpoint (0x0012) with value 2300 to custom {514: 230}
2021-07-31 16:04:53 DEBUG (MainThread) [zhaquirks.tuya] [0x0c26:1:0xef00] Received value [0, 0, 0, 230] for attribute 0x0202 (command 0x0002)
2021-07-31 16:04:53 DEBUG (MainThread) [zhaquirks.tuya] [0x0c26:1:0xef00] Received value [0, 0, 0, 230] for attribute 0x0202 (command 0x0001)
``` 
Its looks like the `zhaquirks.tuya.valve.SiterwellGS361_Type1` is using 10* X °C in the tuya command and your is using X.X °C and then the tuya quirk is rounding the last .5 away :-((

Can you changing the set point temperature on the manufacture cluster with 1/10 ° or is it also "stripping" the decimals away ?

Or is you "converting" in the wrong way then sending the commands and in the right  way then revising the attribute from the device

MattWestb, 2021-07-31:
The original Z2M implantation [2689](https://github.com/Koenkk/zigbee-herdsman-converters/commit/3d53a0154ecdde05238268fbcd26f7ec319e4d00#diff-84f5fe95fe246d93ebc43c90470607799120a06140f1304c4b268fa5a6da69a0) that was not working OK with setpoint and was fixed here [2717](https://github.com/Koenkk/zigbee-herdsman-converters/commit/3d53a0154ecdde05238268fbcd26f7ec319e4d00#diff-84f5fe95fe246d93ebc43c90470607799120a06140f1304c4b268fa5a6da69a0).

martin3000, 2021-07-31:
What is very confusing for me: there is a MoesManufCluster (0xef00) with the target_temperature in degrees celsius.
And there is a MoesThermostat cluster (0x201) with the occupied_heating_setpoint with the target temperature*100.
It is obvious that the MoesManufCluster cannot hold centidegrees, it is only wor integers.
What is the function of these two clusters?

MattWestb, 2021-07-31:
```
Clusters: SiterwellManufCluster (Endpoint id: 1, Id: 0xef00, Type: in)
Attributes of the selected cluster: target_temperature (id: 0x0202)
Value 230

 
Clusters: SiterwellThermostat (Endpoint id: 1, Id: 0x0201, Type: in)
Attributes of the selected cluster: occupied_heating_setpoint (id: 0x0012) 
Value 2300
```
I think (without knowing) that one is working as interface "to tuya" commands and the other is one "to Zigbee" commands.

martin3000, 2021-07-31:
Where you have 230 in your SiterWellManufCluster, I have 23 in my MoesManufCluster, and that is the reason why 0.5 degrees won't work.

MattWestb, 2021-07-31:
Wot is tuya quirk debug saying if you changing the setpoint to somthing like 23.4 local on the TRV and its reporting it to the ZHA ?

Can being that the quirk is using wrong data type for the value in the new TRV.

Its still tuya !!!

martin3000, 2021-07-31:
Setting it to 32.5°C:

[zhaquirks.tuya] [0xda0c:1:0xef00] Received value [0, 0, 0, 32] for attribute 0x0202 (command 0x0001)

Maybe the TRV has a configuration to switch the scaling between 1x and 10x ?

MattWestb, 2021-07-31:
The extra code added  in Z2M `const temp = Math.round(value);` is saying that no decimals is being used that was possible with the old ones.

One thing can being that  the "encoding" the DP is changed / falls but then the new Z2M code was not needed.

Interesting is if tuya ZBGW can setting the temp with decimals or not and if how with "standard" tuya DP or new ones that we dont knowing.

But you have getting ZHA showing the right setpoint temperature i hope but have loosing the decimals in the setting / receiving it from the TRV.

martin3000, 2021-07-31:
I would like to test it with a tuya zigbee hub and smart life app.

martin3000, 2021-07-31:
As of the zigbee cluster library spec, the occupied_heating_setpoint in cluster 201 should be temp*100 (centidegrees)

MattWestb, 2021-07-31:
The cluster 201 = Zigbee side shall working OK (then all TRVs is working OK from ZHA) the problem must being on the 0xef00 cluster with the tuya DP "converter" / decoder or in the MCU firmware.

martin3000, 2021-08-01:
@Adminiuga is it possible to stop HA and read the device cluster attributes (=temperature) with bellows zcl? how do I do that?
I want to read cluster ef00, attribute 202.

Adminiuga, 2021-08-01:
You can do that without stopping the ha. Hint: config panel -> devices -> pick your device -> manage clusters -> pick cluster -> read attributes

A

martin3000, 2021-08-01:
> You can do that without stopping the ha. Hint: config panel -> devices -> pick your device -> manage clusters -> pick cluster -> read attributes
> 
> A

privet alexei, but I want read the original values from the device, without HA manipulating the clusters. as soon as I change the temp. value on the TRV, HA reads and writes values to the clusters.

MattWestb, 2021-09-05:
Martin is you still there ??
Some user like deleting the quirk in Zigpy and ZHA then its not working OK.
Do you like have it staying with limited function or deleting it completely (Its in the normal Z/HA branch) ?

If like having is left pleas leaving one comment in the linked PR above so the devs can desiring !!

By the way the tuya quirk debug:
```
[zhaquirks.tuya] [0xda0c:1:0xef00] Received value [0, 0, 0, 32] for attribute 0x0202 (command 0x0001)
```
is the true reading from the device but it cant being red from ZHA it can only being reported from the device and then you is using the TRV cluster and reading the attribute you is getting the same data but thru the quirk have done with it.

vladi1234, 2021-09-05:
Can I also take a look inside? Darf ich auch mit rein guken? For conrtolle.

martin3000, 2021-09-05:
I did not make a pull request!?
I have my own version and it is working.

vladi1234, 2021-09-06:
I tested Moes BRT-100 with Homessistant + MQTT, works great as it should.

MattWestb, 2021-09-06:
May i kindly asking you doing one PR ?

Its one new and very nice designed TRV that i think many user like and better functionality is always good for our users so not making them disappointed of the device.

Thanks in advance.
LG.

MW

martin3000, 2021-09-06:
> @martin3000 May i kindly asking you doing one PR with your changes ?
> 
> Its one new and very nice designed TRV that i think many user like and better functionality is always good for our users so not making them disappointed of the device.

Yes the TRV is really beautiful. 
With my hack it is working, but it is a hack. 
1) I do not check the model number, so my hack influences all tuya TRV and that is wrong
2) I do not understand why my hack works. I changed a multiplier from 10 to 100 and the result was a division by 10. I don't know if this is the right way to do it.
But if more people buy it and find their way here in this thread, we can discuss it again.

MattWestb, 2021-09-06:
If you have doing it in the quirk and not in ZHA its possible doing one new device type (Moes3 and 4 ?) and adding the "extra setting" in it part and only adding the BRT-100s model ID.

But if no one can doing it its better user can doing the "hacking" in there own system and is working OK if not having more other tuya TRVs in there system.

Thanks for sharing your knowledge !

martin3000, 2021-09-06:
Right it would be necessary to take the model b6wax7g0 out from this class:

```
class Thermostat_TZE200(TuyaThermostat):
    """Moes thermostat valve."""

    signature = {
        #  endpoint=1 profile=260 device_type=81 device_version=0 input_clusters=[0, 4, 5, 61184]
        #  output_clusters=[10, 25]>
        MODELS_INFO: [
            ("_TZE200_c88teujp", "TS0601"),
            ("_TZE200_azqp6ssj", "TS0601"),
            ("_TZE200_yw7cahqs", "TS0601"),
        ],
```
and put it into a separate class.

MattWestb, 2021-09-07:
@Adminiuga Then you have time can you taking one look on this case ?
The TRV is working OK but the setpoint (scaling) is changed in this model.
It need one IF condition in the quirk for normal or the new value but i dont know how to implanting it. I think writing one complete new cluster is little over kill then its only one parameter that need being selected / changed for it.
Then is only moving it too one new device class (Moes_Type3 and 4 for both Zigbee module).

Thanks in advance.

MW

Adminiuga, 2021-09-07:
which multiplier needs the change?

MattWestb, 2021-09-08:
Thanks for helping !!
@martin3000  can you pleas pointing out the changed multiple that need being changed ?

Thanks in advance.

MW

martin3000, 2021-09-08:
In valve.py:

`        
     MOES_TARGET_TEMP_ATTR: ("occupied_heating_setpoint", lambda value: value * 100),  # jms: 10->100
`        

`        
        "occupied_heating_setpoint": (
            MOES_TARGET_TEMP_ATTR,
            lambda value: round(value / 100),     #jms 10->100
        ),
`

There is a rounding issue: the device does 0.5 degree steps, but in HA I see only 1 degree steps.

MattWestb, 2021-10-09:
Hola / Moin / Servus Martin.

OK i was looking on one log you was posting before:
```
2021-07-30 12:39:37 DEBUG (MainThread) [zhaquirks.tuya] [0xda0c:1:0x0201] Mapping standard occupied_heating_setpoint (0x0012) with value 2650 to custom {514: 26.5}
2021-07-30 12:39:38 DEBUG (MainThread) [zhaquirks.tuya] [0xda0c:1:0xef00] Received value [0, 0, 0, 26] for attribute 0x0202 (command 0x0001)
```
You is setting 26,50 and the TRV is replaying it have setting it to 26.
Its the data to and from the TRV so we cant adding more decimals to the received data :-((
tuya shall doing the setting with 0.1 and the reporting with 0.1 and all is working OK and not loosing the 1/10°C of the reported set point temperature.

Its 110% one bug in the firmware and we cant do any thing until tuya is fixing it (not in the Zigbee module but in the tuya MCU firmware).

The the question is is it better doing the "hack" and getting working device with little lesser pressision or letting it like that and having the wrong setpoint in ZHA ??

martin3000, 2021-10-09:
Hehe, good question. 
The device itself does steps in 0.1° or 0.5° (I forgot).
And I think together with the tuya app, the device will work fine. 
So I would like to use the device without HA, maybe with zigpy or bellows.
Then I can check the zigbee clusters.

MattWestb, 2021-10-09:
If you have time and like testing delete all multiplier and diverter and you is getting the "real data" on the cluster attributes then reading and writing the attributes.
Also the quirk debug (tuya DPs) shall showing the real data sent and received to / from the device.

My Siterwell i can setting with 0.1° on the cluster and also the TRV is reporting it OK back to ZHA but i cant only changing with 0.5° from ZHA (also if have sett 21.2 i can changing it in ZHA and local with 0.5° like + => 21.7° and  - => 20.7° both from ZHA and local on the TRV so i think its tuya standard in "stepping" °).

Adminiuga, 2021-10-18:
Close. Should be a tuple of tuples, but test in a terminal
```
if MODELS_INFO in (("_TZE200","dsjdiu"), ("_TZE200","another model")):
    factor=10
else:
    factor=100

martin3000, 2021-10-18:
@MattWestb I see that valve.py is deleted. Can I make my changes into ts0601_trv.py?
Did you change anything or can I overwrite the code in dev?

MattWestb, 2021-10-18:
= Its working :-)))))

MattWestb, 2021-10-18:
If you is doing it OK both shall working OK but i dont have those only Siterwell.
I closing the request in HA to deleting the devise from ZHA extra functions then now its shall staying in ZHA.

martin3000, 2021-10-19:
I connected the TRV to a tuya zigbee gateway and I found the same bug as with HA: the device can make steps in 0.5 degrees, but in the smart life app you only see steps of 1 degree. this is because the target temp cluster holds degrees, but it should hold degrees*10 or degrees*100. I think this is a firmware bug of the TRV.