zigpy/zha-device-handlers #982: [Device Support Request]

FerretMonger, 2021-07-29:
**Is your feature request related to a problem? Please describe.**

I have a bunch of Philips Hue Ambiance Spots, LTW013 to be precise. They are missing the 0x4003 Power On Attribute and cannot be set to power on correctly after toggling mains.

**Describe the solution you'd like**

I would very much want a quirk that supports setting the Power On attribute (0x4003) on the Philips Hue LTW013.

**Device signature - this can be acquired by removing the device from ZHA and pairing it again from the add devices screen. Be sure to add the entire content of the log panel after pairing the device to a code block below this line.**

{
  "node_descriptor": "NodeDescriptor(logical_type=<LogicalType.Router: 1>, complex_descriptor_available=0, user_descriptor_available=0, reserved=0, aps_flags=0, frequency_band=<FrequencyBand.Freq2400MHz: 8>, mac_capability_flags=<MACCapabilityFlags.AllocateAddress|RxOnWhenIdle|MainsPowered|FullFunctionDevice: 142>, manufacturer_code=4107, maximum_buffer_size=71, maximum_incoming_transfer_size=45, server_mask=0, maximum_outgoing_transfer_size=45, descriptor_capability_field=<DescriptorCapability.0: 0>, *allocate_address=True, *is_alternate_pan_coordinator=False, *is_coordinator=False, *is_end_device=False, *is_full_function_device=True, *is_mains_powered=True, *is_receiver_on_when_idle=True, *is_router=True, *is_security_capable=False)",
  "endpoints": {
    "11": {
      "profile_id": 49246,
      "device_type": "0x0220",
      "in_clusters": [
        "0x0000",
        "0x0003",
        "0x0004",
        "0x0005",
        "0x0006",
        "0x0008",
        "0x0300",
        "0x1000"
      ],
      "out_clusters": [
        "0x0019"
      ]
    },
    "242": {
      "profile_id": 41440,
      "device_type": "0x0061",
      "in_clusters": [
        "0x0021"
      ],
      "out_clusters": [
        "0x0021"
      ]
    }
  },
  "manufacturer": "Philips",
  "model": "LTW013",
  "class": "zigpy.device.Device"
}

**Additional context**

The bulbs are running firmware 1.50.2_r30933

Adminiuga, 2021-07-29:
install home assistant beta. 
Fixed by https://github.com/zigpy/zigpy/pull/774

closed 2021-07-29 by Adminiuga

FerretMonger, 2021-07-29:
Oh, wow, thanks! :D

When will this go live?

joshuaspence, 2021-08-09:
Should this be in Home Assistant 2021.8?