zigpy/zha-device-handlers #986: TS0601 by _TZE200_cwnjrr72 and by _TZE200_ywdxldoj autonomously change from AWAY to SCHEDULE preset

almico, 2021-08-04:
**Describe the bug**
I set both TRVs to AWAY preset and, without any further intervention, after some day, I found them set to SCHEDULE.

**To Reproduce**
Steps to reproduce the behavior:
Using the UI, I set the preset to AWAY:
[image]
The UI has some hesitation, but then it shows AWAY. If it doesn't, I set it again to AWAY, until I can confirm I see AWAY reopening that dialog.
The AWAY setting can be verified by the "15 °C" (side-effect) setting.
A few days later, I discover the preset to have changed to "SCHEDULE". I noticed it because I monitor the valve aperture percentage and, given the outside 33 °C temperature, anything but 0% raises a warning 😄

**Expected behavior**
The TRV should keep the original setting.

MattWestb, 2021-08-04:
Interesting undocumented future of the Saswell TRVs !!

I only have  Siterwell TRVs and they is staying the the stetted mode for infinite (only complaining of valve detect then not heating in the summer).

The quirk is only sending commands and encoding attributes sent from the TRV to ZHA so i think its must being one "undocumented future" (= bug) in the device.

Is the display on the TRV changing the working mode then you have changing in HA or is staying in the "old " mode ? 
And then you have getting it in the away mode on the TRV is it later changing to schedule and reporting it to ZHA ?

Both Saswell and Siterwell dont have schedule mode implanted on the ZHA side (only MOES have getting it) but it still possible putting it in the mode local on the TRV.

Sorry cant helping with the code part then its over my knowledge but perhaps other user have more experience.

almico, 2021-08-08:
I investigated a bit and found the reason for this deterministic behaviour. It is due to the following attribute (that I set to 200 to achieve my need):
[image]

MattWestb, 2021-08-08:
Great the Siterwell TRVs dot have away mode with time limit only setting the temperature for the mode and its staying in it but the Moes is having time and temp must being sett for it to working and then the time is running out it shall going back to its default running mode (that looks being schedule),

I My TRVs is the away mode the same as off / antifreeze mode.

Do the Moes have on off mode and is it the same as antifreeze ?

almico, 2021-08-09:
> Do the Moes have on off mode and is it the same as antifreeze ?

Where can I look to give a final answer?

MattWestb, 2021-08-09:
instruction book for the device  ?
But its not importing  then you have all get working.

almico, 2021-08-09:
> instruction book for the device ?

Ah ah ah 😄 I never paired to the official app and threw everything away a few seconds after it successfully paired to my Conbee II 😄

closed 2022-02-12 by github-actions[bot]