zigpy/zha-device-handlers #988: [Device Support Request] Zemismart Temperature & Humidity sensor ( _TZ3000_lfa05ajd TS0201 / ZXZTH)

marciogranzotto, 2021-08-09:
**Is your feature request related to a problem? Please describe.**
https://zigbee.blakadder.com/Zemismart_ZXZTH.html
Temperature and Humidity reports are working, but not battery percentage.

Sniffing the traffic between the sensor and Zemismart's hub, I see that the device reports that it is AC powered (Tuya... right?)
[image]

But then happyly reports battery percentage
[image]

**Describe the solution you'd like**
Add a quirk that listens to power configuration anyways and ignores the AC powered flag

**Device signature - this can be acquired by removing the device from ZHA and pairing it again from the add devices screen. Be sure to add the entire content of the log panel after pairing the device to a code block below this line.**
```
New device 0x126e (00:15:8d:00:05:b0:25:a3) joined the network
[0x126e] Scheduling initialization
Received frame on uninitialized device <Device model=None manuf=None nwk=0x126E ieee=00:15:8d:00:05:b0:25:a3 is_initialized=False> from ep 0 to ep 0, cluster ZDOCmd.Device_annce: b'\xFF\x6E\x12\xA3\x25\xB0\x05\x00\x8D\x15\x00\x84'
[0x126e:zdo] ZDO request ZDOCmd.Device_annce: [0x126E, 00:15:8d:00:05:b0:25:a3, 132]
Tries remaining: 3
[0x126e] Requesting 'Node Descriptor'
Tries remaining: 2
[0x126e] Extending timeout for 0x76 request
Received frame on uninitialized device <Device model=None manuf=None nwk=0x126E ieee=00:15:8d:00:05:b0:25:a3 is_initialized=False> from ep 0 to ep 0, cluster ZDOCmd.Node_Desc_rsp: b'\x76\x00\x6E\x12\x02\x40\x84\x37\x10\x7F\x64\x00\x00\x2C\x64\x00\x00'
[0x126e] Got Node Descriptor: NodeDescriptor(logical_type=<LogicalType.EndDevice: 2>, complex_descriptor_available=0, user_descriptor_available=0, reserved=0, aps_flags=0, frequency_band=<FrequencyBand.Freq2400MHz: 8>, mac_capability_flags=<MACCapabilityFlags.AllocateAddress|MainsPowered: 132>, manufacturer_code=4151, maximum_buffer_size=127, maximum_incoming_transfer_size=100, server_mask=11264, maximum_outgoing_transfer_size=100, descriptor_capability_field=<DescriptorCapability.NONE: 0>, *allocate_address=True, *is_alternate_pan_coordinator=False, *is_coordinator=False, *is_end_device=True, *is_full_function_device=False, *is_mains_powered=True, *is_receiver_on_when_idle=False, *is_router=False, *is_security_capable=False)
[0x126e] Discovering endpoints
Tries remaining: 3
[0x126e] Extending timeout for 0x77 request
Received frame on uninitialized device <Device model=None manuf=None nwk=0x126E ieee=00:15:8d:00:05:b0:25:a3 is_initialized=False> from ep 0 to ep 0, cluster ZDOCmd.Active_EP_rsp: b'\x77\x00\x6E\x12\x01\x01'
[0x126e] Discovered endpoints: [1]
[0x126e] Initializing endpoints [<Endpoint id=1 in=[] out=[] status=<Status.NEW: 0>>]
[0x126e:1] Discovering endpoint information
Tries remaining: 3
[0x126e] Extending timeout for 0x78 request
Received frame on uninitialized device <Device model=None manuf=None nwk=0x126E ieee=00:15:8d:00:05:b0:25:a3 is_initialized=False> from ep 0 to ep 0, cluster ZDOCmd.Simple_Desc_rsp: b'\x78\x00\x6E\x12\x18\x01\x04\x01\x02\x03\x01\x06\x00\x00\x01\x00\x03\x00\x05\x04\x02\x04\xFF\xEE\x02\x03\x00\x19\x00'
[0x126e:1] Discovered endpoint information: SizePrefixedSimpleDescriptor(endpoint=1, profile=260, device_type=770, device_version=1, input_clusters=[0, 1, 3, 1029, 1026, 61183], output_clusters=[3, 25])
Unknown cluster 61183
[0x126e] Extending timeout for 0x79 request
[0x126e:1:0x0000] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=GLOBAL_COMMAND manufacturer_specific=False is_reply=True disable_default_response=True> manufacturer=None tsn=121 command_id=Command.Read_Attributes_rsp>
[0x126e] Read model 'TS0201' and manufacturer '_TZ3000_lfa05ajd' from <Endpoint id=1 in=[basic:0x0000, power:0x0001, identify:0x0003, humidity:0x0405, temperature:0x0402, None:0xEEFF] out=[identify:0x0003, ota:0x0019] status=<Status.ZDO_INIT: 1>>
[0x126e] Discovered basic device information for <Device model='TS0201' manuf='_TZ3000_lfa05ajd' nwk=0x126E ieee=00:15:8d:00:05:b0:25:a3 is_initialized=True>
Device is initialized <Device model='TS0201' manuf='_TZ3000_lfa05ajd' nwk=0x126E ieee=00:15:8d:00:05:b0:25:a3 is_initialized=True>
Checking quirks for _TZ3000_lfa05ajd TS0201 (00:15:8d:00:05:b0:25:a3)
Considering <class 'zhaquirks.xbee.xbee_io.XBeeSensor'>
Fail because endpoint list mismatch: {232, 230} {1}
Considering <class 'zhaquirks.xbee.xbee3_io.XBee3Sensor'>
Fail because endpoint list mismatch: {232, 230} {1}
Considering <class 'zhaquirks.smartthings.tag_v4.SmartThingsTagV4'>
Fail because device_type mismatch on at least one endpoint
Considering <class 'zhaquirks.smartthings.multi.SmartthingsMultiPurposeSensor'>
Fail because device_type mismatch on at least one endpoint
Considering <class 'zhaquirks.netvox.z308e3ed.Z308E3ED'>
Fail because device_type mismatch on at least one endpoint
Considering <class 'zhaquirks.gledopto.soposhgu10.SoposhGU10'>
Fail because endpoint list mismatch: {11, 13} {1}
Considering <class 'bellows.zigbee.application.EZSPCoordinator'>
Fail because device_type mismatch on at least one endpoint
device - 0x126E:00:15:8d:00:05:b0:25:a3 entering async_device_initialized - is_new_join: True
device - 0x126E:00:15:8d:00:05:b0:25:a3 has joined the ZHA zigbee network
[0x126E](TS0201): started configuration
[0x126E:ZDO](TS0201): 'async_configure' stage succeeded
Error handling '_save_attribute' event with (00:15:8d:00:05:b0:25:a3, 1, 0, 4, '_TZ3000_lfa05ajd') params: FOREIGN KEY constraint failed
[0x126e] Extending timeout for 0x7a request
[0x126E:1:0x0000]: finished channel configuration
[0x126e] Extending timeout for 0x7b request
[0x126E:1:0x0019]: finished channel configuration
Error handling '_save_attribute' event with (00:15:8d:00:05:b0:25:a3, 1, 0, 5, 'TS0201') params: FOREIGN KEY constraint failed
[0x126E:1:0x0402]: bound 'temperature' cluster: Status.SUCCESS
[0x126e] Extending timeout for 0x7c request
[0x126E:1:0x0405]: bound 'humidity' cluster: Status.SUCCESS
[... 88 lines cut ...]
[0x126e:1:0x0402] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=GLOBAL_COMMAND manufacturer_specific=False is_reply=True disable_default_response=False> manufacturer=None tsn=17 command_id=Command.Report_Attributes>
[0x126e:1:0x0402] ZCL request 0x000a: [[Attribute(attrid=0, value=<TypeValue type=int16s, value=2402>)]]
[0x126e:1:0x0402] Attribute report received: measured_value=2402
[0x126e:1:0x0405] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=GLOBAL_COMMAND manufacturer_specific=False is_reply=True disable_default_response=False> manufacturer=None tsn=18 command_id=Command.Report_Attributes>
[0x126e:1:0x0405] ZCL request 0x000a: [[Attribute(attrid=0, value=<TypeValue type=int16s, value=6658>)]]
[0x126e:1:0x0405] Attribute report received: measured_value=6658
[0x126e:1:0x0402] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=GLOBAL_COMMAND manufacturer_specific=False is_reply=True disable_default_response=False> manufacturer=None tsn=19 command_id=Command.Report_Attributes>
[0x126e:1:0x0402] ZCL request 0x000a: [[Attribute(attrid=0, value=<TypeValue type=int16s, value=2398>)]]
[0x126e:1:0x0402] Attribute report received: measured_value=2398
[0x126e:1:0x0405] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=GLOBAL_COMMAND manufacturer_specific=False is_reply=True disable_default_response=False> manufacturer=None tsn=20 command_id=Command.Report_Attributes>
[0x126e:1:0x0405] ZCL request 0x000a: [[Attribute(attrid=0, value=<TypeValue type=int16s, value=6652>)]]
[0x126e:1:0x0405] Attribute report received: measured_value=6652
[0x126e:1:0x0402] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=GLOBAL_COMMAND manufacturer_specific=False is_reply=True disable_default_response=False> manufacturer=None tsn=21 command_id=Command.Report_Attributes>
[0x126e:1:0x0402] ZCL request 0x000a: [[Attribute(attrid=0, value=<TypeValue type=int16s, value=2402>)]]
[0x126e:1:0x0402] Attribute report received: measured_value=2402
[0x126e:1:0x0405] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=GLOBAL_COMMAND manufacturer_specific=False is_reply=True disable_default_response=False> manufacturer=None tsn=22 command_id=Command.Report_Attributes>
[0x126e:1:0x0405] ZCL request 0x000a: [[Attribute(attrid=0, value=<TypeValue type=int16s, value=6614>)]]
[0x126e:1:0x0405] Attribute report received: measured_value=6614
[... cut ...]
[0x126e:1:0x0402] ZCL request 0x000a: [[Attribute(attrid=0, value=<TypeValue type=int16s, value=2399>)]]
[0x126e:1:0x0402] Attribute report received: measured_value=2399
[0x126e:1:0x0405] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=GLOBAL_COMMAND manufacturer_specific=False is_reply=True disable_default_response=False> manufacturer=None tsn=24 command_id=Command.Report_Attributes>
[0x126e:1:0x0405] ZCL request 0x000a: [[Attribute(attrid=0, value=<TypeValue type=int16s, value=6588>)]]
[0x126e:1:0x0405] Attribute report received: measured_value=6588
```

MattWestb, 2021-08-09:
I  think i have finding one other device that have getting on patch for this but it was having battery but not working. https://github.com/zigpy/zha-device-handlers/commit/9d01ded6f086438a293430eb43bfe6d16882aa64
The device is having one normal `PowerConfiguration.cluster_id,`
https://github.com/zigpy/zha-device-handlers/blob/8c61c6da4e44906183a4d96e022e67f8c2ba005d/zhaquirks/konke/temp.py#L35  but is getting one with name `PowerConfigurationCluster,` 
https://github.com/zigpy/zha-device-handlers/blob/8c61c6da4e44906183a4d96e022e67f8c2ba005d/zhaquirks/konke/temp.py#L52 that is being "made" in the quirk root init for devices that is having bad battery readings.

The device type looks OK then its one temp and humidity sensor.

Try doing one quirk with the Konke design [konke/temp.py](https://github.com/zigpy/zha-device-handlers/blob/dev/zhaquirks/konke/temp.py) but putting your device signature and cluster and hopefully its being loaded and fixing the battery problem.

MattWestb, 2021-08-09:
Its also possible "cooking" one manual power cluster with different data from the device lik is being made in this commitment https://github.com/zigpy/zha-device-handlers/commit/ddac75b8e407d8e75bf83995e1a0ab1cb63bbcba but i think its better trying the "standard patch".

PR #989 (2021-08-22): TS0201_Zemismart Quirk
Fixes #988 

I tried the solution proposed in the issue, but the problem what that the Device's NodeDescriptor has the MainsPowered flag set.
Clearing the flag whenever the node_descriptor is queried seems to fix the problem!
closed 2021-08-22 by Adminiuga