zigpy/zha-device-handlers #992: [Device Support Request] NEO Tuya ZigBee Water Leak Sensor TS0601

ldrrp, 2021-08-12:
**Describe the solution you'd like**
Adding device support for tuya based water sensor. 

**Device signature **
```
{
  "node_descriptor": "NodeDescriptor(logical_type=<LogicalType.EndDevice: 2>, complex_descriptor_available=0, user_descriptor_available=0, reserved=0, aps_flags=0, frequency_band=<FrequencyBand.Freq2400MHz: 8>, mac_capability_flags=<MACCapabilityFlags.AllocateAddress: 128>, manufacturer_code=4098, maximum_buffer_size=82, maximum_incoming_transfer_size=82, server_mask=11264, maximum_outgoing_transfer_size=82, descriptor_capability_field=<DescriptorCapability.NONE: 0>, *allocate_address=True, *is_alternate_pan_coordinator=False, *is_coordinator=False, *is_end_device=True, *is_full_function_device=False, *is_mains_powered=False, *is_receiver_on_when_idle=False, *is_router=False, *is_security_capable=False)",
  "endpoints": {
    "1": {
      "profile_id": 260,
      "device_type": "0x0051",
      "in_clusters": [
        "0x0000",
        "0x0004",
        "0x0005",
        "0xef00"
      ],
      "out_clusters": [
        "0x000a",
        "0x0019"
      ]
    }
  },
  "manufacturer": "_TZE200_qq9mpfhw",
  "model": "TS0601",
  "class": "zigpy.device.Device"
}
```

**Additional context**
[image]

Logs durring the adding of the device
```
[0x0000:zdo] ZDO request ZDOCmd.Mgmt_Permit_Joining_req: [60, <Bool.false: 0>]
[0x39FC:1:0x0b04]: async_update
[0x8027:1:0x0b04]: async_update
[0x39fc:1:0x0b04] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=GLOBAL_COMMAND manufacturer_specific=False is_reply=True disable_default_response=True> manufacturer=None tsn=85 command_id=Command.Read_Attributes_rsp>
[0x8027:1:0x0b04] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=GLOBAL_COMMAND manufacturer_specific=False is_reply=True disable_default_response=True> manufacturer=None tsn=87 command_id=Command.Read_Attributes_rsp>
[0x0000:zdo] ZDO request ZDOCmd.IEEE_addr_req: [0x8027, 0, 0]
[0x8027:zdo] ZDO request ZDOCmd.IEEE_addr_rsp: [<Status.SUCCESS: 0>, 24:fd:5b:00:01:08:71:98, 0x8027, None, None, []]
[0x8027:zdo] Unsupported ZDO request:ZDOCmd.IEEE_addr_rsp
[0x0000:zdo] ZDO request ZDOCmd.IEEE_addr_req: [0x8027, 0, 0]
[0x8027:zdo] ZDO request ZDOCmd.IEEE_addr_rsp: [<Status.SUCCESS: 0>, 24:fd:5b:00:01:08:71:98, 0x8027, None, None, []]
[0x8027:zdo] Unsupported ZDO request:ZDOCmd.IEEE_addr_rsp
[0x0000:zdo] ZDO request ZDOCmd.IEEE_addr_req: [0x8027, 0, 0]
[0x8027:zdo] ZDO request ZDOCmd.IEEE_addr_rsp: [<Status.SUCCESS: 0>, 24:fd:5b:00:01:08:71:98, 0x8027, None, None, []]
[0x8027:zdo] Unsupported ZDO request:ZDOCmd.IEEE_addr_rsp
[0x0000:zdo] ZDO request ZDOCmd.IEEE_addr_req: [0x8027, 0, 0]
[0x8027:zdo] ZDO request ZDOCmd.IEEE_addr_rsp: [<Status.SUCCESS: 0>, 24:fd:5b:00:01:08:71:98, 0x8027, None, None, []]
[0x8027:zdo] Unsupported ZDO request:ZDOCmd.IEEE_addr_rsp
[0x0000:zdo] ZDO request ZDOCmd.IEEE_addr_req: [0x8027, 0, 0]
[0x39fc:1:0x0006] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=GLOBAL_COMMAND manufacturer_specific=False is_reply=True disable_default_response=True> manufacturer=None tsn=76 command_id=Command.Report_Attributes>
[0x39fc:1:0x0006] ZCL request 0x000a: [[Attribute(attrid=0, value=<TypeValue type=Bool, value=Bool.true>)]]
[0x39fc:1:0x0006] Attribute report received: on_off=1
[0x8027:zdo] ZDO request ZDOCmd.IEEE_addr_rsp: [<Status.SUCCESS: 0>, 24:fd:5b:00:01:08:71:98, 0x8027, None, None, []]
[0x8027:zdo] Unsupported ZDO request:ZDOCmd.IEEE_addr_rsp
[0x0000:zdo] ZDO request ZDOCmd.IEEE_addr_req: [0x8027, 0, 0]
[0x8027:zdo] ZDO request ZDOCmd.IEEE_addr_rsp: [<Status.SUCCESS: 0>, 24:fd:5b:00:01:08:71:98, 0x8027, None, None, []]
[0x8027:zdo] Unsupported ZDO request:ZDOCmd.IEEE_addr_rsp
[0x0000:zdo] ZDO request ZDOCmd.IEEE_addr_req: [0x8027, 0, 0]
[0x8027:zdo] ZDO request ZDOCmd.IEEE_addr_rsp: [<Status.SUCCESS: 0>, 24:fd:5b:00:01:08:71:98, 0x8027, None, None, []]
[0x8027:zdo] Unsupported ZDO request:ZDOCmd.IEEE_addr_rsp
[0x0000:zdo] ZDO request ZDOCmd.IEEE_addr_req: [0x8027, 0, 0]
[0x8027:zdo] ZDO request ZDOCmd.IEEE_addr_rsp: [<Status.SUCCESS: 0>, 24:fd:5b:00:01:08:71:98, 0x8027, None, None, []]
[0x8027:zdo] Unsupported ZDO request:ZDOCmd.IEEE_addr_rsp
[0x39fc:1:0x0b04] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=GLOBAL_COMMAND manufacturer_specific=False is_reply=True disable_default_response=True> manufacturer=None tsn=77 command_id=Command.Report_Attributes>
[0x39fc:1:0x0b04] ZCL request 0x000a: [[Attribute(attrid=1291, value=<TypeValue type=int16s, value=8>)]]
[0x39fc:1:0x0b04] Attribute report received: active_power=8
[0x0000:zdo] ZDO request ZDOCmd.IEEE_addr_req: [0x8027, 0, 0]
[... 286 lines cut ...]
[0xcaf6:zdo] Unsupported ZDO request:ZDOCmd.IEEE_addr_rsp
[0x365d:1:0x0702] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=GLOBAL_COMMAND manufacturer_specific=False is_reply=True disable_default_response=False> manufacturer=None tsn=55 command_id=Command.Report_Attributes>
[0x365d:1:0x0702] ZCL request 0x000a: [[Attribute(attrid=1024, value=<TypeValue type=int24s, value=2>)]]
[0x365d:1:0x0702] Attribute report received: instantaneous_demand=2
[0x39FC:1:0x0b04]: async_update
[0x8027:1:0x0b04]: async_update
[0x39fc:1:0x0b04] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=GLOBAL_COMMAND manufacturer_specific=False is_reply=True disable_default_response=True> manufacturer=None tsn=136 command_id=Command.Read_Attributes_rsp>
[0x8027:1:0x0b04] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=GLOBAL_COMMAND manufacturer_specific=False is_reply=True disable_default_response=True> manufacturer=None tsn=138 command_id=Command.Read_Attributes_rsp>
[0x0000:zdo] ZDO request ZDOCmd.IEEE_addr_req: [0x8027, 0, 0]
[0x8027:zdo] ZDO request ZDOCmd.IEEE_addr_rsp: [<Status.SUCCESS: 0>, 24:fd:5b:00:01:08:71:98, 0x8027, None, None, []]
[0x8027:zdo] Unsupported ZDO request:ZDOCmd.IEEE_addr_rsp
[0x0000:zdo] ZDO request ZDOCmd.IEEE_addr_req: [0x8027, 0, 0]
[0x8027:zdo] ZDO request ZDOCmd.IEEE_addr_rsp: [<Status.SUCCESS: 0>, 24:fd:5b:00:01:08:71:98, 0x8027, None, None, []]
[0x8027:zdo] Unsupported ZDO request:ZDOCmd.IEEE_addr_rsp
[0x1945:1:0xef00] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=CLUSTER_COMMAND manufacturer_specific=False is_reply=True disable_default_response=False> manufacturer=None tsn=6 command_id=17>
[0x1945:1:0xef00] Unknown cluster-specific command 17
[0x1945:1:0xef00] ZCL request 0x0011: b'\x00\x07@'
[0x1945:1:0xef00] No handler for cluster command 17
[0x0000:zdo] ZDO request ZDOCmd.IEEE_addr_req: [0x8027, 0, 0]
[0x8027:zdo] ZDO request ZDOCmd.IEEE_addr_rsp: [<Status.SUCCESS: 0>, 24:fd:5b:00:01:08:71:98, 0x8027, None, None, []]
[0x8027:zdo] Unsupported ZDO request:ZDOCmd.IEEE_addr_rsp
[0x0000:zdo] ZDO request ZDOCmd.IEEE_addr_req: [0x8027, 0, 0]
[0x8027:zdo] ZDO request ZDOCmd.IEEE_addr_rsp: [<Status.SUCCESS: 0>, 24:fd:5b:00:01:08:71:98, 0x8027, None, None, []]
[0x8027:zdo] Unsupported ZDO request:ZDOCmd.IEEE_addr_rsp
```

lepistom, 2022-01-27:
+1 I have the same sensor, and same issue as reported above

ldrrp, 2023-02-12:
Still looking to get this device added, just gathering dust, Looks like [hubitat added it on their end](https://github.com/kkossev/Hubitat/blob/main/Drivers/Tuya%20NEO%20Coolcam%20Zigbee%20Water%20Leak%20Sensor/Tuya%20NEO%20Coolcam%20Zigbee%20Water%20Leak%20Sensor.groovy). Maybe it can be ported over

closed 2023-08-18 by github-actions[bot]