zigpy/zha-device-handlers #994: [Device Support Request]  Tuya Zigbee smart energy meter DDS238-2 Zigbee (_TZE200_byzdayie)

latinlober, 2021-08-15:
Device signature:

_TZE200_byzdayie TS0601

{
  "node_descriptor": "NodeDescriptor(logical_type=<LogicalType.Router: 1>, complex_descriptor_available=0, user_descriptor_available=0, reserved=0, aps_flags=0, frequency_band=<FrequencyBand.Freq2400MHz: 8>, mac_capability_flags=<MACCapabilityFlags.AllocateAddress|RxOnWhenIdle|MainsPowered|FullFunctionDevice: 142>, manufacturer_code=4098, maximum_buffer_size=82, maximum_incoming_transfer_size=82, server_mask=11264, maximum_outgoing_transfer_size=82, descriptor_capability_field=<DescriptorCapability.NONE: 0>, *allocate_address=True, *is_alternate_pan_coordinator=False, *is_coordinator=False, *is_end_device=False, *is_full_function_device=True, *is_mains_powered=True, *is_receiver_on_when_idle=True, *is_router=True, *is_security_capable=False)",
  "endpoints": {
    "1": {
      "profile_id": 260,
      "device_type": "0x0051",
      "in_clusters": [
        "0x0000",
        "0x0004",
        "0x0005",
        "0xef00"
      ],
      "out_clusters": [
        "0x000a",
        "0x0019"
      ]
    }
  },
  "manufacturer": "_TZE200_byzdayie",
  "model": "TS0601",
  "class": "zigpy.device.Device"
}

Other information:

zigbee-herdsman-converters has this device and description is here:

https://github.com/Koenkk/zigbee-herdsman-converters/blob/master/devices/tuya.js

Line 772

code snippet is:

   {
        fingerprint: [{modelID: 'TS0601', manufacturerName: '_TZE200_byzdayie'},
            {modelID: 'TS0601', manufacturerName: '_TZE200_fsb6zw01'}],
        model: 'TS0601_din',
        vendor: 'TuYa',
        description: 'Zigbee smart energy meter DDS238-2 Zigbee',
        fromZigbee: [fz.tuya_dinrail_switch],
        toZigbee: [tz.tuya_switch_state],
        configure: async (device, coordinatorEndpoint, logger) => {
            const endpoint = device.getEndpoint(1);
            await reporting.bind(endpoint, coordinatorEndpoint, ['genOnOff']);
        },
        exposes: [e.switch().setAccess('state', ea.STATE_SET), e.voltage(), e.power(), e.current(), e.energy()],
    }

Home Assistant detect device but no entities are created.

diegomontanaduarte, 2021-09-07:
Hello, I have helped the developers to add support to this device on deCONZ REST-API plugin. Is there anything I can do to help add ZHA support?

https://github.com/dresden-elektronik/deconz-rest-plugin/pull/5287

darckly, 2021-10-03:
Hello, I'm new in the quirck creation, and trying to write my own based on examples from this repository.
At first I want to get the power/voltage/current/total information from device.
Here the code of my _ts0601_din_power.py_ file:

`"""Tuya Din Power Meter."""
from zigpy.profiles import zha
from zigpy.zcl.clusters.general import Basic, Groups, Ota, Scenes, Time 
import zigpy.types as t
from zigpy.quirks import CustomDevice
from zhaquirks import Bus, LocalDataCluster
from zigpy.zcl.clusters.homeautomation import ElectricalMeasurement

from zhaquirks.const import (
    DEVICE_TYPE,
    ENDPOINTS,
    INPUT_CLUSTERS,
    MODELS_INFO,
    OUTPUT_CLUSTERS,
    PROFILE_ID,
)
from zhaquirks.tuya import (
    TuyaManufClusterAttributes,
)

TUYA_TOTAL_ENERGY_ATTR = 0x0211
TUYA_CURRENT_ATTR = 0x0212
TUYA_POWER_ATTR = 0x0213
TUYA_VOLTAGE_ATTR = 0x0214

class TuyaManufClusterDinPower(TuyaManufClusterAttributes):
    """Manufacturer Specific Cluster of the Tuya Power Meter device."""

    manufacturer_attributes = {
        TUYA_TOTAL_ENERGY_ATTR: ('total_energy', t.uint16_t),
        TUYA_CURRENT_ATTR: ('current', t.uint16_t),
        TUYA_POWER_ATTR: ('power', t.uint16_t),
        TUYA_VOLTAGE_ATTR: ('voltage', t.uint16_t),
    }

    def _update_attribute(self, attrid, value):
        super()._update_attribute(attrid, value)
        if attrid == TUYA_TOTAL_ENERGY_ATTR:
            self.endpoint.device.total_energy_bus.listener_event(
                "total_energy_reported", value
            )
        elif attrid == TUYA_CURRENT_ATTR:
            self.endpoint.device.current_bus.listener_event(
                "current_reported", value
            )
        elif attrid == TUYA_POWER_ATTR:
            self.endpoint.device.power_bus.listener_event(
                "power_reported", value
            )
        elif attrid == TUYA_VOLTAGE_ATTR:
            self.endpoint.device.voltage_bus.listener_event(
                "voltage_reported", value / 10
            )    

class TuyaTotalEnergyMeasurement(LocalDataCluster,ElectricalMeasurement):
   
    cluster_id = ElectricalMeasurement.cluster_id    
    TOTAL_ID = 0x0304
    VOLTAGE_ID = 0x0100
    POWER_ID = 0x0106
    CURRENT_ID = 0x0103

    def __init__(self, *args, **kwargs):
        """Init."""
        super().__init__(*args, **kwargs)
        self.endpoint.device.total_energy_bus.add_listener(self)
        self.endpoint.device.voltage_bus.add_listener(self)
        self.endpoint.device.power_bus.add_listener(self)
    def total_energy_reported(self,value):
        """Total Energy reported."""
        self._update_attribute(self.TOTAL_ID, value)
    def voltage_reported(self,value):
        """Voltage reported."""
        self._update_attribute(self.VOLTAGE_ID, value)
    def power_reported(self,value):
        """Power reported."""
        self._update_attribute(self.POWER_ID, value)
    def current_reported(self,value):
        """Current reported."""
        self._update_attribute(self.CURRENT_ID, value)            

class TuyaPowerMeter(CustomDevice):
    """Tuya power meter device."""

    def __init__(self, *args, **kwargs):
        """Init device."""
        self.total_energy_bus = Bus()
        self.current_bus = Bus()
        self.power_bus = Bus()
        self.voltage_bus = Bus()
        self.switch_bus = Bus()
        super().__init__(*args, **kwargs)

    signature = {
        # "node_descriptor": "<NodeDescriptor byte1=1 byte2=64 mac_capability_flags=142 manufacturer_code=4098
        #                       maximum_buffer_size=82 maximum_incoming_transfer_size=82 server_mask=11264
        #                       maximum_outgoing_transfer_size=82 descriptor_capability_field=0>",
        # device_version=1
        # input_clusters=[0x0000, 0x0004, 0x0005, 0xef00]
        # output_clusters=[0x000a, 0x0019]
        # <SimpleDescriptor endpoint=1 profile=260 device_type=51 device_version=1 input_clusters=[0, 4, 5, 61184] output_clusters=[10, 25]>
        MODELS_INFO: [
            ("_TZE200_byzdayie", "TS0601"),
        ],
        ENDPOINTS: {
            1: {
                PROFILE_ID: zha.PROFILE_ID,
                DEVICE_TYPE: zha.DeviceType.SMART_PLUG,
                INPUT_CLUSTERS: [
                    Basic.cluster_id,
                    Groups.cluster_id,
                    Scenes.cluster_id,
                    TuyaManufClusterAttributes.cluster_id,
                ],
                OUTPUT_CLUSTERS: [Time.cluster_id, Ota.cluster_id],
            }
        },
    }

    replacement = {
        ENDPOINTS: {
            1: {
                DEVICE_TYPE: zha.DeviceType.METER_INTERFACE,
                INPUT_CLUSTERS: [
                    Basic.cluster_id,
                    Groups.cluster_id,
                    Scenes.cluster_id,
                    TuyaTotalEnergyMeasurement,
                    TuyaManufClusterDinPower,
                ],
                OUTPUT_CLUSTERS: [Time.cluster_id, Ota.cluster_id],
            }
        }
    }
`
And here I need any help, because I can't see values visible as entities in HA.
After pairing the device has only one entity:
 
[image]

On the cluster page I see values in cluster attributes
[image]

... but I stuck on this step, and have no clue what to do the next, how to expose values to HA.

Any help or suggestions is more than appreciated!

darckly, 2021-10-16:
I attached the current quirck for DDS238-2, still not everything is available.
[ts0601_din_power.zip](https://github.com/zigpy/zha-device-handlers/files/7357602/ts0601_din_power.zip)

**The simplest installation for testing**
1. Create "quircks" directory in your /config folder
2. Extract and copy the ts0601_din_power.py file to the /config/quircks directory
3. Add the following into your configuration.yaml
`zha:
  custom_quirks_path: '/config/quircks'`
4. Restart HA
4. Pair your DDS238-2 device

Currently available as entities:
1. the Power value under *_electrical_measurement 
2. the Total Energy value under *_smartenergy_metering_summation_delivered
3. the Switch under *_on_off

Didn't find how to expose the Voltage and Current yet (((
In fact, they appear in the cluster attributes 0x0500 (Voltage) and 0x0502 (Current), but not as entities.

glcos, 2021-10-16:
I have exactly the same issue here with a DIY metering board. I can see all the values coming in the log but I cannot expose the entities for voltage and current. I could only see power (W) using the ElectricalMeasurement cluster.
I wonder if there is any workaround to visualize such values maybe not using the ElectricalMeasurement cluster, until official support will be provided in the coming updates.

darckly, 2021-10-16:
I found that this is a limitation of ZHA - today is only possible to expose one attribute per cluster. As a workaround, I can write the other values to MQTT server )))))
I also tried to use the PowerConfiguration cluster to expose the Voltage, unfortunately, without success - attributes are filled, but not exposed to UI, no idea why

darckly, 2021-10-16:
No, I'm running the stable. Have only one system for home automation.
If this limitation will be removed soon, better I'll wait

glcos, 2021-10-17:
Just a couple of remarks:

1. it would be nice to add

> 0x0300: ("ac_frequency", t.uint16_t)

> 0x0510: ("power_factor", t.int8s)

as some power related devices such as pzem004t provide those values

2. I would like to have at least one decimal for voltage, power, current. Is there any way to achieve that ?

darckly, 2021-11-06:
Finally, after update HA to version 2021.11, I got all values as sensors.
Quirck for your testing is attached
[ts0601_din_power.zip](https://github.com/zigpy/zha-device-handlers/files/7490394/ts0601_din_power.zip)

[image]

Known issues, maybe someone can advise how to solve them:
1. After the first initialization not all sensors are available, after some period you should restart HA, and all sensors will appear
2. After HA restart the switch is OFF, even the device is ON. But it works later

@glcos I see the power factor and frequency parameters on the device LCD screen, but I don't know is the device sent them, and  what manufacturing_attribute read

MattWestb, 2021-11-06:
@darckly Do you have trying taking the power away from the device and putting it back ?
Normally (but not always) tuya DP devices is sending all the DP commands / attribute then the tuya MCU is being restarted and you can getting them in the log if having debug for `zhaquirks: debug`.
The device must being paired so it can sending all the information to the system or its not working.

glcos, 2021-11-06:
@darckly if you enable logging you can check whether additional attributes such as Frequency and PF are being sent or not.

This is what I have in my configuration.yaml to check ZigBee logs

```
logger:
  default: warning
  logs:
    homeassistant.components.zha: debug
    bellows.zigbee.application: debug
    zigpy.zcl: debug
    zhaquirks: debug
```

Anyway at this time Frequency and PF are not yet supported by ZHA so you are not going to see them even though your device sends them to HA.

I'm still testing my code and I will put it on my repository so you can test it too.

darckly, 2021-11-06:
I have enabled the logging and powered off/on the device. In the log for [zhaquirks.tuya] I see that device is sending the information with using 0x0001, 0x0101, 0x0211, 0x0212, 0x0213 and 0x0214 attributes only.

I suspect that 0x0101 contains the information for switch status (I got [1] in the log, but the switch entity in UI is still OFF...), and the last four attributes are known.
So, looks like no more information from device we can get.

glcos, 2021-11-06:
Unfortunately I don't own this device so I cannot provide additional help

darckly, 2021-11-07:
I solved the issue with the switch state, now it always showed the correct state. Updated quirk is here:
https://github.com/darckly/zha-device-handlers/blob/master/zhaquirks/tuya/ts0601_din_power.py

I checked the log after taking power away and putting back - no information about the commands or attributes was sent.
Summary:
1. Information about voltage, power, current, and switch state the device sends every 30 seconds.
2. Information about total energy device sends after the change

Please test the quirk (who has the device), and if no issues will be found, I'll prepare PR the next weekend

mrkaqz, 2021-11-08:
@darckly I have tested the updated quirk.

It is working just find after paring it is not show sensor yet until restart HA agin (known issue).
switch state seems to work correctly and all sensor reading good except the rms current on my physical device show 0.25 A (I am testing with 60w light bulb). I guess it just the rounding down of the number that why it shows 0 on the HA. If that so, your script working perfectly now.
 

[image]

Celer21, 2021-11-09:
@darckly Thank you! New version works well. 
As @mrkaqz mrkaqz mentioned, OnOff switch is turning on itself (but HW switch on device is turned off).

For me, it is not problem, I want this device only for power monitoring...
But it would be great, if on off switch stays in correct state.

darckly, 2021-11-09:
@mrkaqz I didn't find a way how to show the current with decimals. Maybe @glcos has progress here?

Regarding self-turned-on behaviour: interesting, I'll debug this tomorrow on my device.
Just to double-check the scenario:
1. Device is powered on and switched on
2. You switched off in HA interface, device switched off, red diode lighted up
3. After a few minutes the state of the switch in HA was self-changed to ON but the physical device is still off and the red diode is lighted on the device

Correct?

mrkaqz, 2021-11-09:
> @mrkaqz I didn't find a way how to show the current with decimals. Maybe @glcos has progress here?
>
> Regarding self-turned-on behaviour: interesting, I'll debug this tomorrow on my device. Just to [...]

Yes that correct. When it HA state changed itself to ON, the red LED still stay ON on the hardware.

Celer21, 2021-11-09:
> > @mrkaqz I didn't find a way how to show the current with decimals. Maybe @glcos has progress here?
> > Regarding self-turned-on behaviour: interesting, I'll debug this tomorrow on my device. Just to double-check the scenario:
> > 
> > 1. Device is powered on and switched on
> > 2. You switched off in HA interface, device switched off, red diode lighted up
> > 3. After a few minutes the state of the switch in HA was self-changed to ON but the physical device is still off and the red diode is lighted on the device
> > 
> > Correct?
> 
> Yes that correct. When it HA state changed itself to ON, the red LED still stay ON on the hardware.
> 
> For the decimal, it is not a deal breaker. Since I have test it with just a 1 light bulb. If we install it on the circuit breaker box, the power consumption is much more than a 1 A anyway.

On off switch behave same for me...

I have connected to my AC, so it is pretty big load. But Iam planning buying more DS238 for another appliances, so  consumption with 2 decimals would be better.

glcos, 2021-11-09:
The code that is preventing the decimals from being shown is in:

/srv/homeassistant/lib/python3.9/site-packages/homeassistant/components/zha/sensor.py

I commented out the lines as per below picture and the decimals are shown.

[image]

Honestly I don't see the point in this code, but that should be discussed with the developers.
It would be good at least to be able to manage this from inside the quirk maybe playing with multiplier and divisor... when I have time I will give it a try.

glcos, 2021-11-09:
Did some testing but I could not figure out how to have decimals without touching the original code.
@Adminiuga would you mind checking this ?

darckly, 2021-11-09:
I found a way how to show decimals for current (at least one sign after a point)
[image]

The quirk is attached
[ts0601_din_power.zip](https://github.com/zigpy/zha-device-handlers/files/7507783/ts0601_din_power.zip)

Regarding the switch state  - still debugging

darckly, 2021-11-09:
I'm puzzled a lot because I see that the device sends only four attributes, and attribute 0x0101 corresponds to the switch state. 
If you are switching between states, the device sends the [0] or [1] almost immediately. But if you switch off the device, I see the message with [0], and during the next push, the device sends [1]. 
So, this is how the device sends the data - may be an issue with firmware... Or I don't understand something.

@diegomontanaduarte do you have the same issue with zibgee2mqtt implementation for this device?

diegomontanaduarte, 2021-11-09:
@darckly it's working fine with zigbee2mqtt implementation

mrkaqz, 2021-11-10:
> I found a way how to show decimals for current (at least one sign after a point) [image]
> 
> The quirk is attached [ts0601_din_power.zip](https://github.com/zigpy/zha-device-handlers/files/7507783/ts0601_din_power.zip)
> 
> Regarding the switch state - still debugging

@darckly I have tested this updated quirk. the decimal of the Amp is looking good but somehow the voltage is reading so high. It was correct before on the previous quirk. I am on 220VAC system here.

[image]

darckly, 2021-11-10:
Sorry, with voltage my mistake. Now should be fine
[ts0601_din_power.zip](https://github.com/zigpy/zha-device-handlers/files/7509951/ts0601_din_power.zip)

mrkaqz, 2021-11-10:
> Sorry, with voltage my mistake. Now should be fine [ts0601_din_power.zip](https://github.com/zigpy/zha-device-handlers/files/7509951/ts0601_din_power.zip)

This looks good for all the sensor now :)

darckly, 2021-11-14:
@diegomontanaduarte are you sure that the issue with the switch state is not present on your device?
Today I configured zigbee2mqtt integration just to test this... and after 1 minute the switch was changed to ON, even device was still OFF.

So, for my device, the issue with the switch exists for both, ZHA and Zigbee2Mqtt integrations. I think the issue is with the device firmware.

diegomontanaduarte, 2021-11-14:
@darckly This is an exposed switch that I didn't play so much since I installed this device on my home main breaker box. But I recall to test this switch when I was implement this device and didn't find any further issues

Is there any way to check the device firmware version? Maybe we can compare

diegomontanaduarte, 2021-11-14:
@darckly I'm not sure if you are aware but on the native control application "Tuya Smart App" there is a timer that allows you to turn ON or OFF the device at a specified time. Also there is a countdown function to turn off this device after a specified time interval. Maybe some configuration like this is messing up with your device.

You can find a very detailed review of this device model at https://www.youtube.com/watch?v=taXYgRaleWU

darckly, 2021-11-14:
The device is not turned ON physically, the only switch state in HA is displayed incorrectly. I saw this video before, but this issue is not related to the timer functionality. Unfortunately, I have no ability to check the firmware version also.
Yes, if the device will be used in the main breaker box, it will be always on, so this issue is not so important.

closed 2021-11-19 by dmulcahey

hhh-pixel, 2022-01-20:
@darckly Hi, didn't you work with Hiking tomzn **three phase** power meter? It is recognized the same - '_TZE200_byzdayie'. On the latest HA (2021.12.10) I see the following behavior:
1. on/off doesn't work (it is always off). It is not critical for me, but seems strange, because everyone writes, that the switch is always **on**.  But it's not important.
2. I can see four sensor only from one phase. Do you have mb any ideas, how can I get the sensors from the second and the third phase?

darckly, 2022-01-21:
Hi @hhh-pixel 
I have no such device, could you share the device signature?
The issue with the switch device I fixed and (I hope) it will be available in the next HA version.
You can try the change as a custom quirk, details are here: #1275

hhh-pixel, 2022-01-21:
Yep, I've seen your fix. But as I understood it is for the new manufacturerName? Right? But I've got the old one.

Here is the signature:
```
{
  "node_descriptor": "NodeDescriptor(logical_type=<LogicalType.Router: 1>, complex_descriptor_available=0, user_descriptor_available=0, reserved=0, aps_flags=0, frequency_band=<FrequencyBand.Freq2400MHz: 8>, mac_capability_flags=<MACCapabilityFlags.AllocateAddress|RxOnWhenIdle|MainsPowered|FullFunctionDevice: 142>, manufacturer_code=4098, maximum_buffer_size=82, maximum_incoming_transfer_size=82, server_mask=11264, maximum_outgoing_transfer_size=82, descriptor_capability_field=<DescriptorCapability.NONE: 0>, *allocate_address=True, *is_alternate_pan_coordinator=False, *is_coordinator=False, *is_end_device=False, *is_full_function_device=True, *is_mains_powered=True, *is_receiver_on_when_idle=True, *is_router=True, *is_security_capable=False)",
  "endpoints": {
    "1": {
      "profile_id": 260,
      "device_type": "0x0051",
      "in_clusters": [
        "0x0000",
        "0x0004",
        "0x0005",
        "0x0006",
        "0x0702",
        "0x0b04",
        "0xef00"
      ],
      "out_clusters": [
        "0x000a",
        "0x0019"
      ]
    }
  },
  "manufacturer": "_TZE200_byzdayie",
  "model": "TS0601",
  "class": "zhaquirks.tuya.ts0601_din_power.TuyaPowerMeter"
}
```

darckly, 2022-01-21:
IT also will/should work with the old manufacturerName

Thank you for the description! I'll try to write the quirk for your device during the weekend. So, if you are not against it, I'll use you as my tester ;)
And I have opened a new issue #1298 , let's continue in that tread (because this one is closed already)

PR #1155 (2021-11-19): Add support for Tuya DDS238-2 Din PowerMeter
This is a Hiking DDS238-2 65A DIN Power Meter device, which supports measuring voltage, current, power, and counting total energy. The device also has a switch but seems the issue is present in device firmware, and the device sends the "ON" state even the device is switched off. 
[image]

Resolve #994