zigpy/zha-device-handlers #997: [Device Support Request] Immax Neo Keypad MS-K1AZ

Idskov, 2021-08-17:
Immax Neo Keypad Zigbee 3.0 / MaxSmart MS-K1AZ, is discovered fine by ZHA using a tasmotized SonOff Zigbee bridge, but unfortunately no entities show up.
It seems a Quirk is needed...
System running HA core-2021.8.7 on HA OS 6.2 everything up to date.

**Describe the solution you'd like**
Entities for armed/disarmed/armed-home, battery, SOS and others if possible to show up in HA
Idealy also to have HA be able to arm/disarm the panel (if possible)

**Device signature.**
```
{
  "node_descriptor": "NodeDescriptor(logical_type=<LogicalType.EndDevice: 2>, complex_descriptor_available=0, user_descriptor_available=0, reserved=0, aps_flags=0, frequency_band=<FrequencyBand.Freq2400MHz: 8>, mac_capability_flags=<MACCapabilityFlags.AllocateAddress: 128>, manufacturer_code=4098, maximum_buffer_size=82, maximum_incoming_transfer_size=82, server_mask=11264, maximum_outgoing_transfer_size=82, descriptor_capability_field=<DescriptorCapability.NONE: 0>, *allocate_address=True, *is_alternate_pan_coordinator=False, *is_coordinator=False, *is_end_device=True, *is_full_function_device=False, *is_mains_powered=False, *is_receiver_on_when_idle=False, *is_router=False, *is_security_capable=False)",
  "endpoints": {
    "1": {
      "profile_id": 260,
      "device_type": "0x0051",
      "in_clusters": [
        "0x0000",
        "0x0004",
        "0x0005",
        "0xef00"
      ],
      "out_clusters": [
        "0x000a",
        "0x0019"
      ]
    }
  },
  "manufacturer": "_TZE200_n9clpsht",
  "model": "TS0601",
  "class": "zigpy.device.Device"
}
```

**Additional context**
[image]

I'm happy to test out quirks and give feedback.

Idskov, 2021-08-17:
Logs when adding to ZHA:
```
New device 0x6c01 (84:71:27:ff:fe:05:72:39) joined the network
[0x6c01] Scheduling initialization
Tries remaining: 3
[0x6c01] Requesting 'Node Descriptor'
Tries remaining: 2
[0x6c01] Extending timeout for 0x2d request
Device 0x6c01 (84:71:27:ff:fe:05:72:39) joined the network
[0x6c01] Scheduling initialization
[0x6c01] Canceling old initialize call
Tries remaining: 3
[0x6c01] Requesting 'Node Descriptor'
Tries remaining: 2
[0x6c01] Extending timeout for 0x2f request
[0x489d:1:0xef00] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=CLUSTER_COMMAND manufacturer_specific=False is_reply=True disable_default_response=False> manufacturer=None tsn=67 command_id=1>
[0x489d:1:0xef00] ZCL request 0x0001: [TuyaCommand(status=0, tsn=2, dp=22, data=TuyaData(dp_type=<TuyaDPType.VALUE: 2>, function=0, raw=b'\x0c\x00\x00\x00', *payload=12))]
[0x489d:1:0xef00] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=CLUSTER_COMMAND manufacturer_specific=False is_reply=True disable_default_response=False> manufacturer=None tsn=67 command_id=1>
Device 0x6c01 (84:71:27:ff:fe:05:72:39) joined the network
[0x6c01] Scheduling initialization
[0x6c01] Canceling old initialize call
Received frame on uninitialized device <Device model=None manuf=None nwk=0x6C01 ieee=84:71:27:ff:fe:05:72:39 is_initialized=False> from ep 0 to ep 0, cluster 19: b"\x82\x01l9r\x05\xfe\xff'q\x84\x80"
[0x6c01:zdo] ZDO request ZDOCmd.Device_annce: [0x6C01, 84:71:27:ff:fe:05:72:39, 128]
Tries remaining: 3
[0x6c01] Requesting 'Node Descriptor'
Tries remaining: 2
[0x6c01] Extending timeout for 0x32 request
[0x489d:1:0xef00] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=CLUSTER_COMMAND manufacturer_specific=False is_reply=True disable_default_response=False> manufacturer=None tsn=67 command_id=1>
[0x489d] Invalid state on future for 0x43 seq -- probably duplicate response
[0x489d:1:0xef00] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=CLUSTER_COMMAND manufacturer_specific=False is_reply=True disable_default_response=False> manufacturer=None tsn=68 command_id=1>
[0x489d:1:0xef00] ZCL request 0x0001: [TuyaCommand(status=0, tsn=2, dp=21, data=TuyaData(dp_type=<TuyaDPType.VALUE: 2>, function=0, raw=b'\x1b\x00\x00\x00', *payload=27))]
Received frame on uninitialized device <Device model=None manuf=None nwk=0x6C01 ieee=84:71:27:ff:fe:05:72:39 is_initialized=False> from ep 0 to ep 0, cluster 32770: b'/\x00\x01l\x02@\x80\x02\x10RR\x00\x00,R\x00\x00'
[0x6c01:zdo] ZDO request ZDOCmd.Node_Desc_rsp: [<Status.SUCCESS: 0>, 0x6C01, NodeDescriptor(logical_type=<LogicalType.EndDevice: 2>, complex_descriptor_available=0, user_descriptor_available=0, reserved=0, aps_flags=0, frequency_band=<FrequencyBand.Freq2400MHz: 8>, mac_capability_flags=<MACCapabilityFlags.AllocateAddress: 128>, manufacturer_code=4098, maximum_buffer_size=82, maximum_incoming_transfer_size=82, server_mask=11264, maximum_outgoing_transfer_size=82, descriptor_capability_field=<DescriptorCapability.NONE: 0>, *allocate_address=True, *is_alternate_pan_coordinator=False, *is_coordinator=False, *is_end_device=True, *is_full_function_device=False, *is_mains_powered=False, *is_receiver_on_when_idle=False, *is_router=False, *is_security_capable=False)]
[0x6c01:zdo] Unsupported ZDO request:ZDOCmd.Node_Desc_rsp
[0x489d:1:0xef00] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=CLUSTER_COMMAND manufacturer_specific=False is_reply=True disable_default_response=False> manufacturer=None tsn=68 command_id=1>
[0x489d:1:0xef00] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=CLUSTER_COMMAND manufacturer_specific=False is_reply=True disable_default_response=False> manufacturer=None tsn=69 command_id=1>
[0x489d:1:0xef00] ZCL request 0x0001: [TuyaCommand(status=0, tsn=2, dp=2, data=TuyaData(dp_type=<TuyaDPType.VALUE: 2>, function=0, raw=b'\x8a\x01\x00\x00', *payload=394))]
Received frame on uninitialized device <Device model=None manuf=None nwk=0x6C01 ieee=84:71:27:ff:fe:05:72:39 is_initialized=False> from ep 0 to ep 0, cluster 32770: b'2\x00\x01l\x02@\x80\x02\x10RR\x00\x00,R\x00\x00'
[0x6c01] Got Node Descriptor: NodeDescriptor(logical_type=<LogicalType.EndDevice: 2>, complex_descriptor_available=0, user_descriptor_available=0, reserved=0, aps_flags=0, frequency_band=<FrequencyBand.Freq2400MHz: 8>, mac_capability_flags=<MACCapabilityFlags.AllocateAddress: 128>, manufacturer_code=4098, maximum_buffer_size=82, maximum_incoming_transfer_size=82, server_mask=11264, maximum_outgoing_transfer_size=82, descriptor_capability_field=<DescriptorCapability.NONE: 0>, *allocate_address=True, *is_alternate_pan_coordinator=False, *is_coordinator=False, *is_end_device=True, *is_full_function_device=False, *is_mains_powered=False, *is_receiver_on_when_idle=False, *is_router=False, *is_security_capable=False)
[0x6c01] Discovering endpoints
Tries remaining: 3
[0x6c01] Extending timeout for 0x36 request
[0x489d:1:0xef00] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=CLUSTER_COMMAND manufacturer_specific=False is_reply=True disable_default_response=False> manufacturer=None tsn=70 command_id=1>
[0x489d:1:0xef00] ZCL request 0x0001: [TuyaCommand(status=0, tsn=0, dp=18, data=TuyaData(dp_type=<TuyaDPType.VALUE: 2>, function=0, raw=b'\xe9\x00\x00\x00', *payload=233))]
[... cut ...]
[0x6C01](TS0601): started initialization
[0x6C01:ZDO](TS0601): 'async_initialize' stage succeeded
[0x6C01:1:0x0000]: initializing channel: from_cache: False
[0x6C01:1:0x0000]: finished channel configuration
[0x6C01:1:0x0019]: initializing channel: from_cache: False
[0x6C01:1:0x0019]: finished channel configuration
[0x6C01:1:0x0000]: 'async_initialize' stage succeeded
[0x6C01:1:0x0019]: 'async_initialize' stage succeeded
[0x6C01](TS0601): power source: Battery or Unknown
[0x6C01](TS0601): completed initialization
```

Idskov, 2021-08-18:
From the debug-log:

Arm:
```
2021-08-18 16:50:45 DEBUG (MainThread) [bellows.zigbee.application] Received incomingMessageHandler frame with [<EmberIncomingMessageType.INCOMING_UNICAST: 0>, EmberApsFrame(profileId=260, clusterId=61184, sourceEndpoint=1, destinationEndpoint=1, options=<EmberApsOption.APS_OPTION_ENABLE_ROUTE_DISCOVERY: 256>, groupId=0, sequence=239), 88, -78, 0x2280, 255, 255, b'\t_\x01\x00\x89\x1b\x04\x00\x01\x00']
2021-08-18 16:50:45 DEBUG (MainThread) [zigpy.zcl] [0x2280:1:0xef00] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=CLUSTER_COMMAND manufacturer_specific=False is_reply=True disable_default_response=False> manufacturer=None tsn=95 command_id=1>
2021-08-18 16:50:45 WARNING (MainThread) [zigpy.zcl] [0x2280:1:0xef00] Unknown cluster-specific command 1
2021-08-18 16:50:45 DEBUG (MainThread) [zigpy.zcl] [0x2280:1:0xef00] ZCL request 0x0001: b'\x00\x89\x1b\x04\x00\x01\x00'
2021-08-18 16:50:45 DEBUG (MainThread) [zigpy.zcl] [0x2280:1:0xef00] No handler for cluster command 1
```

Disarm:
```
2021-08-18 16:48:42 DEBUG (MainThread) [bellows.zigbee.application] Received incomingMessageHandler frame with [<EmberIncomingMessageType.INCOMING_UNICAST: 0>, EmberApsFrame(profileId=260, clusterId=61184, sourceEndpoint=1, destinationEndpoint=1, options=<EmberApsOption.APS_OPTION_ENABLE_ROUTE_DISCOVERY: 256>, groupId=0, sequence=232), 92, -77, 0x2280, 255, 255, b'\tX\x01\x00\x82\x17\x01\x00\x01\x00']
2021-08-18 16:48:42 DEBUG (MainThread) [zigpy.zcl] [0x2280:1:0xef00] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=CLUSTER_COMMAND manufacturer_specific=False is_reply=True disable_default_response=False> manufacturer=None tsn=88 command_id=1>
2021-08-18 16:48:42 WARNING (MainThread) [zigpy.zcl] [0x2280:1:0xef00] Unknown cluster-specific command 1
2021-08-18 16:48:42 DEBUG (MainThread) [zigpy.zcl] [0x2280:1:0xef00] ZCL request 0x0001: b'\x00\x82\x17\x01\x00\x01\x00'
2021-08-18 16:48:42 DEBUG (MainThread) [zigpy.zcl] [0x2280:1:0xef00] No handler for cluster command 1
```

Arm-home:
```
2021-08-18 16:56:15 DEBUG (MainThread) [bellows.zigbee.application] Received incomingMessageHandler frame with [<EmberIncomingMessageType.INCOMING_UNICAST: 0>, EmberApsFrame(profileId=260, clusterId=61184, sourceEndpoint=1, destinationEndpoint=1, options=<EmberApsOption.APS_OPTION_ENABLE_ROUTE_DISCOVERY: 256>, groupId=0, sequence=241), 92, -77, 0x2280, 255, 255, b'\ta\x01\x00\x8b\x1c\x04\x00\x01\x00']
2021-08-18 16:56:15 DEBUG (MainThread) [zigpy.zcl] [0x2280:1:0xef00] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=CLUSTER_COMMAND manufacturer_specific=False is_reply=True disable_default_response=False> manufacturer=None tsn=97 command_id=1>
2021-08-18 16:56:15 WARNING (MainThread) [zigpy.zcl] [0x2280:1:0xef00] Unknown cluster-specific command 1
2021-08-18 16:56:15 DEBUG (MainThread) [zigpy.zcl] [0x2280:1:0xef00] ZCL request 0x0001: b'\x00\x8b\x1c\x04\x00\x01\x00'
2021-08-18 16:56:15 DEBUG (MainThread) [zigpy.zcl] [0x2280:1:0xef00] No handler for cluster command 1
```

SOS:
```
2021-08-18 17:00:54 DEBUG (MainThread) [bellows.zigbee.application] Received incomingMessageHandler frame with [<EmberIncomingMessageType.INCOMING_UNICAST: 0>, EmberApsFrame(profileId=260, clusterId=61184, sourceEndpoint=1, destinationEndpoint=1, options=<EmberApsOption.APS_OPTION_ENABLE_ROUTE_DISCOVERY: 256>, groupId=0, sequence=242), 96, -76, 0x2280, 255, 255, b'\tb\x01\x00\x8c\x1d\x04\x00\x01\x00']
2021-08-18 17:00:54 DEBUG (MainThread) [zigpy.zcl] [0x2280:1:0xef00] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=CLUSTER_COMMAND manufacturer_specific=False is_reply=True disable_default_response=False> manufacturer=None tsn=98 command_id=1>
2021-08-18 17:00:54 WARNING (MainThread) [zigpy.zcl] [0x2280:1:0xef00] Unknown cluster-specific command 1
2021-08-18 17:00:54 DEBUG (MainThread) [zigpy.zcl] [0x2280:1:0xef00] ZCL request 0x0001: b'\x00\x8c\x1d\x04\x00\x01\x00'
2021-08-18 17:00:54 DEBUG (MainThread) [zigpy.zcl] [0x2280:1:0xef00] No handler for cluster command 1
```

closed 2022-02-21 by github-actions[bot]

geniesek, 2022-05-03:
Does someone have the external converter for this device? 
thanks

PonyOny, 2022-10-21:
Has anyone done any custom quirk for this device? I have just got one and would love to make it functional.

geniesek, 2022-10-22:
Yes, I have fixed a script I found in different thread (and posted there the fix) to be able to at least read basic operations. See file attached.

[TS0601.js.txt](https://github.com/zigpy/zha-device-handlers/files/9845272/TS0601.js.txt)

PonyOny, 2022-10-22:
> Yes, I have fixed a script I found in different thread (and posted there the fix) to be able to at least read basic operations. See file attached (just rename from .js.txt to .js)
> 
> [TS0601.js.txt](https://github.com/zigpy/zha-device-handlers/files/9845272/TS0601.js.txt)

How can i use this script? is it ZHA? it does not have the quirk format, could you help me out with it please?

Thanks

geniesek, 2022-10-22:
Register it as External converter like here:
[image]

PonyOny, 2022-10-22:
I think you are using Zigbee2MQTT I'm using ZHA

geniesek, 2022-10-22:
right.