zigpy/zha-device-handlers #1001: [Device Support Request] Xiaomi Aqara Opple WXCJKG11LM 2 buttons not working

kalon33, 2021-08-18:
Triggering the two buttons, I got errors:

Button 1:
`[2021-08-07 18:57:06][DEBUG] : Ignoring message (b'180b0a5500210100') on cluster 18: unknown endpoint or cluster id: 'No cluster ID 0x0012 on (04:cf:8c:df:3c:7c:34:9e, 1)'`
   
Button 2:
`[2021-08-07 18:57:03][DEBUG] : Ignoring message (b'180a0a5500210100') on cluster 18: unknown endpoint or cluster id: 'No cluster ID 0x0012 on (04:cf:8c:df:3c:7c:34:9e, 2)'`
   
Here are device raw informations:
```
{
    "ieee": "04:cf:8c:df:3c:7c:34:9e",
    "nwk": 27063,
    "status": 2,
    "lqi": "255",
    "rssi": "-28",
    "last_seen": "1628499450.0291584",
    "node_descriptor": "02:40:84:5f:11:7f:64:00:00:2c:64:00:00",
    "endpoints": [
        {
            "id": 1,
            "status": 1,
            "device_type": 261,
            "profile_id": 260,
            "manufacturer": "LUMI",
            "model": "lumi.remote.b286opcn01",
            "output_clusters": [
                {
                    "id": 3,
                    "name": "Identify",
                    "attributes": []
                },
                {
                    "id": 6,
                    "name": "On\/Off",
                    "attributes": []
                },
                {
                    "id": 8,
                    "name": "Level control",
                    "attributes": []
                },
                {
                    "id": 768,
                    "name": "Color Control",
                    "attributes": []
                }
            ],
            "input_clusters": [
                {
                    "id": 0,
                    "name": "Basic",
                    "attributes": [
                        {
                            "id": 0,
                            "name": "zcl_version",
                            "value": 3
                        },
                        {
                            "id": 1,
                            "name": "app_version",
                            "value": 17
                        },
                        {
                            "id": 2,
                            "name": "stack_version",
                            "value": 2
                        },
                        {
                            "id": 3,
[... 255 lines cut ...]
                    18,
                    3
                ],
                "output_clusters": [
                    6
                ]
            },
            "6": {
                "profile_id": 260,
                "device_type": 259,
                "input_clusters": [
                    18,
                    3
                ],
                "output_clusters": [
                    6
                ]
            }
        }
    },
    "class": "zigpy.device"
}
```
   
Could you please help me to get this working? I have 4 switches, all trigger the same kind of problem.

TheJulianJES, 2021-08-21:
It's likely that creating a new quirk in https://github.com/zigpy/zha-device-handlers/blob/dev/zhaquirks/xiaomi/aqara/opple_remote.py with the signature of your remotes would fix this.
I'll see if I get around to this.

TheJulianJES, 2021-08-21:
@kalon33 Can you download https://raw.githubusercontent.com/TheJulianJES/zha-device-handlers/aqara_opple_two_button/zhaquirks/xiaomi/aqara/opple_remote.py.
Create a folder in your ``/config`` directory called ``custom_quirks`` and drop the ``opple_remote.py`` file in there.

In your Home Assistant ``configuration.yaml``, add the following section:
```
zha:
  custom_quirks_path: /config/custom_quirks
```
(https://www.home-assistant.io/integrations/zha/#custom_quirks_path)

Then restart Home Assistant. Your Aqara Opple 2 button remote should get a quirk applied and send ``zha_event``s.
Device automation triggers should also be created.
If it doesn't work, try re-pairing the device.

If it works, I'll create a PR so the custom quirk is implemented by default.