zigpy/zha-device-handlers #1004: [BUG] TS0601 _TZE200_yvx5lh6k (Smart Air Box) not finding the quirk

gbfromhb, 2021-08-24:
I purchased a Tuya Smart Air Box with a Zigbee ID of: "_TZE200_8ygsuhe1" and it is working as expected. I purchased a second and it came with a Zigbee ID of: "_TZE200_yvx5lh6k" and it is not working as expected. The [Zigbee Device Compatibility Repository](https://zigbee.blakadder.com/Tuya_RSH-AirBox01.html) claims that both of the Zigbee IDs should be working.

**Here is the Zigbee device signature from the Smart Air box that is working correctly (_TZE200_8ygsuhe1).**

```
{
  "node_descriptor": "NodeDescriptor(logical_type=<LogicalType.Router: 1>, complex_descriptor_available=0, user_descriptor_available=0, reserved=0, aps_flags=0, frequency_band=<FrequencyBand.Freq2400MHz: 8>, mac_capability_flags=<MACCapabilityFlags.AllocateAddress|RxOnWhenIdle|MainsPowered|FullFunctionDevice: 142>, manufacturer_code=4098, maximum_buffer_size=82, maximum_incoming_transfer_size=82, server_mask=11264, maximum_outgoing_transfer_size=82, descriptor_capability_field=<DescriptorCapability.NONE: 0>, *allocate_address=True, *is_alternate_pan_coordinator=False, *is_coordinator=False, *is_end_device=False, *is_full_function_device=True, *is_mains_powered=True, *is_receiver_on_when_idle=True, *is_router=True, *is_security_capable=False)",
  "endpoints": {
    "1": {
      "profile_id": 260,
      "device_type": "0x0100",
      "in_clusters": [
        "0x0000",
        "0x0004",
        "0x0005",
        "0x0402",
        "0x0405",
        "0x040d",
        "0x042b",
        "0x042e",
        "0xef00"
      ],
      "out_clusters": [
        "0x000a",
        "0x0019"
      ]
    }
  },
  "manufacturer": "_TZE200_8ygsuhe1",
  "model": "TS0601",
  "class": "zhaquirks.tuya.air.ts0601_air_quality.TuyaCO2Sensor"
}

```
**Here is the Zigbee device signature from the Smart Air Box that is not working (_TZE200_yvx5lh6k)**

```
{
  "node_descriptor": "NodeDescriptor(logical_type=<LogicalType.Router: 1>, complex_descriptor_available=0, user_descriptor_available=0, reserved=0, aps_flags=0, frequency_band=<FrequencyBand.Freq2400MHz: 8>, mac_capability_flags=<MACCapabilityFlags.AllocateAddress|RxOnWhenIdle|MainsPowered|FullFunctionDevice: 142>, manufacturer_code=4098, maximum_buffer_size=82, maximum_incoming_transfer_size=82, server_mask=11264, maximum_outgoing_transfer_size=82, descriptor_capability_field=<DescriptorCapability.NONE: 0>, *allocate_address=True, *is_alternate_pan_coordinator=False, *is_coordinator=False, *is_end_device=False, *is_full_function_device=True, *is_mains_powered=True, *is_receiver_on_when_idle=True, *is_router=True, *is_security_capable=False)",
  "endpoints": {
    "1": {
      "profile_id": 260,
      "device_type": "0x0051",
      "in_clusters": [
        "0x0000",
        "0x0004",
        "0x0005",
        "0xef00"
      ],
      "out_clusters": [
        "0x000a",
        "0x0019"
      ]
    }
  },
  "manufacturer": "_TZE200_yvx5lh6k",
  "model": "TS0601",
  "class": "zigpy.device.Device"
}

```
**Here is an interesting excerpt from the logs:**

```
2021-08-23 10:08:18 DEBUG (MainThread) [zigpy.quirks.registry] Checking quirks for _TZE200_8ygsuhe1 TS0601 (84:fd:27:ff:fe:d6:98:2f)
2021-08-23 10:08:18 DEBUG (MainThread) [zigpy.quirks.registry] Considering <class 'zhaquirks.tuya.air.ts0601_air_quality.TuyaCO2Sensor'>
2021-08-23 10:08:18 DEBUG (MainThread) [zigpy.quirks.registry] Found custom device replacement for 84:fd:27:ff:fe:d6:98:2f: <class 'zhaquirks.tuya.air.ts0601_air_quality.TuyaCO2Sensor'>
2021-08-23 10:08:18 DEBUG (MainThread) [zigpy.quirks.registry] Checking quirks for _TZE200_yvx5lh6k TS0601 (0c:43:14:ff:fe:88:14:b4)
2021-08-23 10:08:18 DEBUG (MainThread) [zigpy.quirks.registry] Considering <class 'zhaquirks.xbee.xbee_io.XBeeSensor'>
2021-08-23 10:08:18 DEBUG (MainThread) [zigpy.quirks.registry] Fail because endpoint list mismatch: {232, 230} {1}
2021-08-23 10:08:18 DEBUG (MainThread) [zigpy.quirks.registry] Considering <class 'zhaquirks.xbee.xbee3_io.XBee3Sensor'>
2021-08-23 10:08:18 DEBUG (MainThread) [zigpy.quirks.registry] Fail because endpoint list mismatch: {232, 230} {1}
2021-08-23 10:08:18 DEBUG (MainThread) [zigpy.quirks.registry] Considering <class 'zhaquirks.smartthings.tag_v4.SmartThingsTagV4'>
2021-08-23 10:08:18 DEBUG (MainThread) [zigpy.quirks.registry] Fail because device_type mismatch on at least one endpoint
2021-08-23 10:08:18 DEBUG (MainThread) [zigpy.quirks.registry] Considering <class 'zhaquirks.smartthings.multi.SmartthingsMultiPurposeSensor'>
2021-08-23 10:08:18 DEBUG (MainThread) [zigpy.quirks.registry] Fail because device_type mismatch on at least one endpoint
2021-08-23 10:08:18 DEBUG (MainThread) [zigpy.quirks.registry] Considering <class 'zhaquirks.netvox.z308e3ed.Z308E3ED'>
2021-08-23 10:08:18 DEBUG (MainThread) [zigpy.quirks.registry] Fail because device_type mismatch on at least one endpoint
2021-08-23 10:08:18 DEBUG (MainThread) [zigpy.quirks.registry] Considering <class 'zhaquirks.gledopto.soposhgu10.SoposhGU10'>
2021-08-23 10:08:18 DEBUG (MainThread) [zigpy.quirks.registry] Fail because endpoint list mismatch: {11, 13} {1}
2021-08-23 10:08:18 DEBUG (MainThread) [zigpy.quirks.registry] Considering <class 'bellows.zigbee.application.EZSPCoordinator'>
2021-08-23 10:08:18 DEBUG (MainThread) [zigpy.quirks.registry] Fail because device_type mismatch on at least one endpoint

```

Adminiuga, 2021-08-24:
actually, looks like the signature for this model was added in https://github.com/zigpy/zha-device-handlers/pull/954 just need to release new version.

Adminiuga, 2021-08-24:
you may add a `my_quirks` folder in your `/config` folder where your configuration  is and add to `configuration.yaml`
```
zha:
  custom_quirks_path: /config/my_quirks
```

and copy https://github.com/zigpy/zha-device-handlers/blob/dev/zhaquirks/tuya/__init__.py and https://github.com/zigpy/zha-device-handlers/tree/dev/zhaquirks/tuya/air to that folder

gbfromhb, 2021-08-25:
Thank you for all the great work.

Regards,
Glenn Byron

From: Alexei Chetroi
Sent: Monday, August 23, 2021 7:12 PM
To: zigpy/zha-device-handlers
Cc: Glenn Byron; Author
Subject: Re: [zigpy/zha-device-handlers] [BUG] TS0601 _TZE200_yvx5lh6k (SmartAir Box) not finding the quirk (#1004)

you may add a my_quirks folder in your /config folder where your configuration is and add to configuration.yaml
zha:
  custom_quirks_path: /config/my_quirks
and copy https://github.com/zigpy/zha-device-handlers/blob/dev/zhaquirks/tuya/__init__.py and https://github.com/zigpy/zha-device-handlers/tree/dev/zhaquirks/tuya/air to that folder

pbvdven, 2021-12-28:
hi @Adminiuga, i have 2 of these devices and one works and the others add with no entities same devices both TS0601 _TZE200_yvx5lh6k. when i look at the log it says

2021-12-28 14:57:21 DEBUG (MainThread) [zigpy.application] Device a4:c1:38:64:4f:63:d7:00 changed id (0xca81 => 0xca81)
2021-12-28 14:57:21 DEBUG (MainThread) [zigpy.device] [0xca81] Skipping initialization, device is fully initialized
2021-12-28 14:57:21 DEBUG (MainThread) [zigpy.application] Device is initialized <Device model='TS0601' manuf='_TZE200_yvx5lh6k' nwk=0xCA81 ieee=a4:c1:38:64:4f:63:d7:00 is_initialized=True>
2021-12-28 14:57:21 DEBUG (MainThread) [zigpy.quirks.registry] Checking quirks for _TZE200_yvx5lh6k TS0601 (a4:c1:38:64:4f:63:d7:00)
2021-12-28 14:57:21 DEBUG (MainThread) [zigpy.quirks.registry] Considering <class 'zhaquirks.tuya.air.ts0601_air_quality.TuyaCO2Sensor'>
2021-12-28 14:57:21 DEBUG (MainThread) [zigpy.quirks.registry] Fail because endpoint list mismatch: {1} {1, 242}

Do you have any idea what did go wrong and how to fix it?

thanks

Kind Regards,

Pim

Adminiuga, 2021-12-28:
These 
"manufacturer": "_TZE200_yvx5lh6k",
"model": "TS0601",

Need to ba added to quirk

pbvdven, 2021-12-28:
Thanks for you answer could you please tell me how i can get it added two the quirk or could i add it as a custom quirk? And how can you see they are different are the not both from the same manufacturer the both  _TZE200_yvx5lh6  one device works fine all 5 sensors but the other one not show any. Thanks for helping

aigarius, 2021-12-31:
Copying over the ts0601_air_quality.py and adding the following section to the ENDPOINTS dictionary of the TuyaCO2Sensor device allowed me to have it operational:
`
            242: {
                  PROFILE_ID: 41440,
                  DEVICE_TYPE: 0x0061,
                  INPUT_CLUSTERS: [ ],
                  OUTPUT_CLUSTERS: [33],
            },
`

MattWestb, 2021-12-31:
ZGPP is added in #1066 so likely only need adding the device ID in: https://github.com/zigpy/zha-device-handlers/blob/d65fd8d4b359b52c689e28542810d8bca678190c/zhaquirks/tuya/air/ts0601_air_quality.py#L83-L85

aigarius, 2021-12-31:
Yes, that works!

commit 03af7b3d (2022-01-12): Fixes #1004 (#1280)

Adds model "_TZE200_yvx5lh6k" to Tuya Air quality device with GPP
closed 2022-01-12 by dmulcahey