zigpy/zha-device-handlers #1015: [Device Support Request] TS0601 _TZE200_znzs7yaw (TuYa thermostat)

Sirse, 2021-08-28:
One of the best of cheap chiniese thermostates (internal and external temp sensors, overload/overheat protections, 0.5C settings step with hysteresis, settings restore after powerfail, almost zero network spam) doesn`t have ZHA support (because proprietary TuYa proto).

```
{
  "node_descriptor": "NodeDescriptor(logical_type=<LogicalType.Router: 1>, complex_descriptor_available=0, user_descriptor_available=0, reserved=0, aps_flags=0, frequency_band=<FrequencyBand.Freq2400MHz: 8>, mac_capability_flags=<MACCapabilityFlags.AllocateAddress|RxOnWhenIdle|MainsPowered|FullFunctionDevice: 142>, manufacturer_code=4098, maximum_buffer_size=82, maximum_incoming_transfer_size=82, server_mask=11264, maximum_outgoing_transfer_size=82, descriptor_capability_field=<DescriptorCapability.NONE: 0>, *allocate_address=True, *is_alternate_pan_coordinator=False, *is_coordinator=False, *is_end_device=False, *is_full_function_device=True, *is_mains_powered=True, *is_receiver_on_when_idle=True, *is_router=True, *is_security_capable=False)",
  "endpoints": {
    "1": {
      "profile_id": 260,
      "device_type": "0x0051",
      "in_clusters": [
        "0x0000",
        "0x0004",
        "0x0005",
        "0xef00"
      ],
      "out_clusters": [
        "0x000a",
        "0x0019"
      ]
    }
  },
  "manufacturer": "_TZE200_znzs7yaw",
  "model": "TS0601",
  "class": "zigpy.device.Device"
}
```

Support was already implemented in Z2M (via https://github.com/Koenkk/zigbee-herdsman-converters/pull/1694)

MattWestb, 2021-08-28:
You can trying if its compatible with the normal tuya electrical heating thermostat that is in https://github.com/zigpy/zha-device-handlers/blob/dev/zhaquirks/tuya/electric_heating.py (the cluster looks the same so that is good).

Add your new manufacturer at https://github.com/zigpy/zha-device-handlers/blob/d43df422d103f2bfa5d4cd37c6721f32cdbde984/zhaquirks/tuya/electric_heating.py#L140 and configuring local quirk and testing the new made quirk.

Sirse, 2021-09-03:
Device uses two different time offsets for UTC and local time payloads:
    UTC timestamp with time offset **2000**
    Local timestamp with time offset **1970**
Current time sync code supports only one universal offset value for both timestamps via _set_time_offset_ https://github.com/zigpy/zha-device-handlers/blob/28bd89a2f34da9a15dd37370734af4e222850316/zhaquirks/tuya/__init__.py#L236

Bug or one more TuYa proto specification?

Sirse, 2021-09-04:
Temporary quirk draft for this device - https://gist.github.com/Sirse/3cf8172935d81c5c00655650854e3160 (almost untested, without schedule support). 
Works only with https://github.com/zigpy/zha-device-handlers/pull/1021 due to own timesync mechanism

MattWestb, 2021-10-03:
The time fix is merged in dev and is coming in HA 2021.10.

Did you have more time testing it ??

I think its more users that like getting the device supported in ZHA if its possible so its being great if you can making it all the way.

Thanks in advance !!

Sirse, 2021-10-03:
All basic features (set/get temp, on/off, get state) works fine.
[image]

Current problems:
- schedule unimplemented
- mode selection unimplemented
- unknown DP for descaling and opened window detection

I can create PR with current quirk (https://gist.github.com/Sirse/3cf8172935d81c5c00655650854e3160) and fix these issues later.

MattWestb, 2021-10-03:
I think its much better then not working at all and the "original" Moes is also not having all parts working like the schedule .

I think is only the Moes TRVs that have getting all possible functions implanted and working but i have not betting and feedback from user of all TRVs that is implanted.

My Revolt (Siterwell) is not having all but they is working well for 10 mounts and other users is using them 2 so i think its better little then not at all.

If you like to knowing what DP the thermostat is one dev have saying that if you is disconnecting the power for some minutes and ting it they is sending all possible DP they is supporting (I have trying and finding that my is supporting valve jammed but i cant implanting it then i can not do the coding ;-(( ).

Also **if** Z2M and deCONZ have implanting it its possible looking what DP they is using.

If you do one PR its being great bit not one must and im sure users is valuing it much then they is getting devices that can working OK in there systems.

MattWestb, 2021-10-03:
Z2M have it but i have not looking how they have doing the DP things in to and from zigbee.

https://github.com/Koenkk/zigbee-herdsman-converters/blob/09b3b3df0c86707921272a775ce6f9d279043714/devices/tuya.js#L1132