zigpy/zha-device-handlers #1017: [Device Support Request] Sonoff Motion Sensor SNZB-03

janeczku, 2021-08-29:
**Is your feature request related to a problem? Please describe.**
Device is not correctly configured upon pairing (only power state is available)

[image]

**Describe the solution you'd like**
Add device quirck handler

**Device signature - this can be acquired by removing the device from ZHA and pairing it again from the add devices screen. Be sure to add the entire content of the log panel after pairing the device to a code block below this line.**

```JSON
{
  "node_descriptor": "NodeDescriptor(logical_type=<LogicalType.EndDevice: 2>, complex_descriptor_available=0, user_descriptor_available=0, reserved=0, aps_flags=0, frequency_band=<FrequencyBand.Freq2400MHz: 8>, mac_capability_flags=<MACCapabilityFlags.AllocateAddress: 128>, manufacturer_code=0, maximum_buffer_size=80, maximum_incoming_transfer_size=160, server_mask=0, maximum_outgoing_transfer_size=160, descriptor_capability_field=<DescriptorCapability.0: 0>, *allocate_address=True, *is_alternate_pan_coordinator=False, *is_coordinator=False, *is_end_device=True, *is_full_function_device=False, *is_mains_powered=False, *is_receiver_on_when_idle=False, *is_router=False, *is_security_capable=False)",
  "endpoints": {
    "1": {
      "profile_id": 260,
      "device_type": "0x0402",
      "in_clusters": [
        "0x0000",
        "0x0001",
        "0x0003",
        "0x0500"
      ],
      "out_clusters": [
        "0x0003"
      ]
    }
  },
  "manufacturer": "eWeLink",
  "model": "MS01",
  "class": "zigpy.device.Device"
}
```

**Additional context**
Add any other context or screenshots about the feature request here.

randomsnack, 2021-08-29:
The device is working fine on my system, if you have a problem. You should remove and rejoin. Sometime it does not join properly.

{
  "node_descriptor": "NodeDescriptor(logical_type=<LogicalType.EndDevice: 2>, complex_descriptor_available=0, user_descriptor_available=0, reserved=0, aps_flags=0, frequency_band=<FrequencyBand.Freq2400MHz: 8>, mac_capability_flags=<MACCapabilityFlags.AllocateAddress: 128>, manufacturer_code=0, maximum_buffer_size=80, maximum_incoming_transfer_size=160, server_mask=0, maximum_outgoing_transfer_size=160, descriptor_capability_field=<DescriptorCapability.NONE: 0>, *allocate_address=True, *is_alternate_pan_coordinator=False, *is_coordinator=False, *is_end_device=True, *is_full_function_device=False, *is_mains_powered=False, *is_receiver_on_when_idle=False, *is_router=False, *is_security_capable=False)",
  "endpoints": {
    "1": {
      "profile_id": 260,
      "device_type": "0x0402",
      "in_clusters": [
        "0x0000",
        "0x0001",
        "0x0003",
        "0x0500"
      ],
      "out_clusters": [
        "0x0003"
      ]
    }
  },
  "manufacturer": "eWeLink",
  "model": "MS01",
  "class": "zigpy.device.Device"

[image]

janeczku, 2021-08-29:
It works

closed 2021-08-29 by janeczku

LeFlairGoD, 2023-08-30:
> Das Gerät funktioniert auf meinem System einwandfrei, falls Sie ein Problem haben. Sie sollten es entfernen und erneut beitreten. Manchmal fügt es sich nicht richtig zusammen.

Could you adjust the sensitivity of the sensor? 
Or set it to maintain the detected state for one minute?