zigpy/zha-device-handlers #1018: [Device Support Request] ROBB smarrt ROB_200-008-0

wouters, 2021-09-01:
**Is your feature request related to a problem? Please describe.**
Adding the ROBB smarrt ROB_200-008-0 switch works, but no entities are created. This switch is a whitelabel version of another brand (not sure but probably Sunricher SR-ZG9001K4-DIM2).

**Describe the solution you'd like**
Would be massive if it could be added :)

**Device signature - this can be acquired by removing the device from ZHA and pairing it again from the add devices screen. {
  "node_descriptor": "NodeDescriptor(logical_type=<LogicalType.EndDevice: 2>, complex_descriptor_available=0, user_descriptor_available=0, reserved=0, aps_flags=0, frequency_band=<FrequencyBand.Freq2400MHz: 8>, mac_capability_flags=<MACCapabilityFlags.AllocateAddress: 128>, manufacturer_code=4644, maximum_buffer_size=82, maximum_incoming_transfer_size=82, server_mask=11264, maximum_outgoing_transfer_size=82, descriptor_capability_field=<DescriptorCapability.NONE: 0>, *allocate_address=True, *is_alternate_pan_coordinator=False, *is_coordinator=False, *is_end_device=True, *is_full_function_device=False, *is_mains_powered=False, *is_receiver_on_when_idle=False, *is_router=False, *is_security_capable=False)",
  "endpoints": {
    "1": {
      "profile_id": 260,
      "device_type": "0x0001",
      "in_clusters": [
        "0x0000",
        "0x0001",
        "0x0003",
        "0x0b05"
      ],
      "out_clusters": [
        "0x0003",
        "0x0004",
        "0x0005",
        "0x0006",
        "0x0008",
        "0x0019",
        "0x0300",
        "0x1000"
      ]
    },
    "2": {
      "profile_id": 260,
      "device_type": "0x0001",
      "in_clusters": [
        "0x0000",
        "0x0001",
        "0x0003",
        "0x0b05"
      ],
      "out_clusters": [
        "0x0003",
        "0x0004",
        "0x0005",
        "0x0006",
        "0x0008",
        "0x0019",
        "0x0300",
        "0x1000"
      ]
    }
  },
  "manufacturer": "ROBB smarrt",
  "model": "ROB_200-008-0",
  "class": "zigpy.device.Device"
}

Thanks!

MattWestb, 2021-09-01:
Its one dual dimmer switch / remote that looks very Zigbee and Z2M is only exposing the zigbee commands for getting entries and battery and no magic converters.

Is you getting events form the switch the listening to events in HA ?

Developer Tools > EVENTS and putting `zha_event` in "Events to subscribe to" and kicking "start listen".

wouters, 2021-09-01:
Hi Matt,

Yes! I'm seeing the events:

`Event 3 fired 20:00:
{
    "event_type": "zha_event",
    "data": {
        "device_ieee": "bc:33:ac:ff:fe:6f:ad:ad",
        "unique_id": "bc:33:ac:ff:fe:6f:ad:ad:2:0x0006",
        "device_id": "367928266b5700330c1a2bfbb0c36895",
        "endpoint_id": 2,
        "cluster_id": 6,
        "command": "off",
        "args": []
    },
    "origin": "LOCAL",
    "time_fired": "2021-09-01T18:00:54.948382+00:00",
    "context": {
        "id": "74cd48c6d502a04c4d2f251e4869db9c",
        "parent_id": null,
        "user_id": null
    }
}`

MattWestb, 2021-09-01:
That great !!
Z2M is hocking on, off, move_with_on_off and stop_with_on_off. for both dimmer "sides".
You can using them in automatons as triggers with events (not device) but you need using more or less all information from the event for getting it working and using endpoint 1 and 2 for getting the right commands from left and right keys. The dimmer commands is very likely long press and release.