zigpy/zha-device-handlers #1019: [Device Support Request] Tuya PIR Sensor - Model zm-35zh-q

rbert001, 2021-09-03:
Looking forward for support of this mini Tuya movement sensor (stated model is zm-35zh-q)
It is recognized by ZHA but no progrees at the interviewing step.

New device 0xaf81 (00:12:4b:00:02:66:29:60) joined the network
[0xaf81] Scheduling initialization
Received frame on uninitialized device <Device model=None manuf=None nwk=0xAF81 ieee=00:12:4b:00:02:66:29:60 is_initialized=False> from ep 0 to ep 0, cluster 19: b'\x00\x81\xaf`)f\x02\x00K\x12\x00\x80'
[0xaf81:zdo] ZDO request ZDOCmd.Device_annce: [0xAF81, 00:12:4b:00:02:66:29:60, 128]
Tries remaining: 3
[0xaf81] Requesting 'Node Descriptor'
Tries remaining: 2
[0xaf81] Extending timeout for 0x09 request
Received frame on uninitialized device <Device model=None manuf=None nwk=0xAF81 ieee=00:12:4b:00:02:66:29:60 is_initialized=False> from ep 1 to ep 1, cluster 1: b'\x18\x03\n \x00 \x1e'
[0xaf81] Received ZCL while uninitialized on endpoint id 1, cluster 0x0001 id, hdr: <ZCLHeader frame_control=<FrameControl frame_type=GLOBAL_COMMAND manufacturer_specific=False is_reply=True disable_default_response=True> manufacturer=None tsn=3 command_id=Command.Report_Attributes>, payload: b' \x00 \x1e'
[0xaf81] Uninitialized device command 'Command.Report_Attributes' args: [[Attribute(attrid=32, value=<TypeValue type=uint8_t, value=30>)]]
Received frame on uninitialized device <Device model=None manuf=None nwk=0xAF81 ieee=00:12:4b:00:02:66:29:60 is_initialized=False> from ep 1 to ep 1, cluster 1: b'\x18\x04\n!\x00 \xc8'
[0xaf81] Received ZCL while uninitialized on endpoint id 1, cluster 0x0001 id, hdr: <ZCLHeader frame_control=<FrameControl frame_type=GLOBAL_COMMAND manufacturer_specific=False is_reply=True disable_default_response=True> manufacturer=None tsn=4 command_id=Command.Report_Attributes>, payload: b'!\x00 \xc8'
[0xaf81] Uninitialized device command 'Command.Report_Attributes' args: [[Attribute(attrid=33, value=<TypeValue type=uint8_t, value=200>)]]
Received frame on uninitialized device <Device model=None manuf=None nwk=0xAF81 ieee=00:12:4b:00:02:66:29:60 is_initialized=False> from ep 1 to ep 1, cluster 1280: b'\x18\x05\n\x02\x00!\x01\x00'

jkottier, 2021-09-22:
I would love to get support for this device as well.

rbert001, 2022-03-21:
Has anyone been able to have it working?

escobarin3, 2022-04-11:
Mine used to work...  I have 2 PIR ZM-RT201 and they stopped working a few weeks ago. Now I had to go back to tuya hub. Anyone checking this issue?

KevinVanthuyne, 2022-10-14:
Would be nice to be able to use all features of this sensor

aalwash, 2023-04-07:
If any one needs debug logs to make it easier to implement this, please let me know

closed 2023-10-11 by github-actions[bot]

jehy, 2023-12-24:
I was able to pair it, but after some time, about 6 hours, it becomes unavailable.

jkykayex, 2024-02-09:
x2 please get something developed :)

aalwash, 2024-02-09:
I used this quirks and it worked perfectly -- https://hochseesegelteam.ch/privatemedia/homeassistantfiles/ts0601_pirmotion.py.txt

bkuhlman4, 2024-10-11:
Hello aalwash,
Can you please share the quirks since the link is private.
I found that the PIR works fine within smartlife app scene PIR ,#1). PIR #2 connected to tuya hub and used in HA does not work as it does not trigger. The automation works for on and off but the PIR does not engage. Does not show any history in HA but does show correct in Smartlife.

From HA automation:
trigger: state
entity_id:
  - binary_sensor.pir_eingang_bewegung
to: "on"
from: null
for:
  hours: 0
  minutes: 0
  seconds: 0

Any progress here? Those little PIR are very neat cheap and helpful. Thanks.

aalwash, 2024-10-11:
Sure, no problem
It's still working for me, I've put it on my Github gists https://gist.github.com/aalwash/7ade6d2e901868e4730f2f64069a437f#file-ts0601_pirmotion-py

bkuhlman4, 2024-10-22:
Thanks again,

I think it needs to go into www buthow do I activate and use
it?

Can give me some pointers.
-- 

Mit freundlichem Gruß
Best Regards

Bernd Kuhlmann
+49-1769-7575757

Request] Tuya PIR Sensor - Model zm-35zh-q (#1019)
Datum: Fri, 11 Oct 2024 13:29:37 -0700

https://gist.github.com/aalwash/7ade6d2e901868e4730f2f64069a437f#file-ts0601_pirmotion-py
unsubscribe.
<zigpy/zha-device-