zigpy/zha-device-handlers #1020: [Device Support Request] Zemismart ZM79E-DT

Faustjohan, 2021-09-03:
**Is your feature request related to a problem? Please describe.**

Request and placehoder for information about the Tuya Curtain motor > specfically:

https://zigbee.blakadder.com/Zemismart_ZM79E-DT.html

z2m device: https://github.com/Koenkk/zigbee-herdsman-converters/blob/9e4c36cd79e15449830f5e09540157c0e8d686a1/devices.js#L1481

**Describe the solution you'd like**

ZHA support, please :)

**Device signature - this can be acquired by removing the device from ZHA and pairing it again from the add devices screen. Be sure to add the entire content of the log panel after pairing the device to a code block below this line.**

> {
  "node_descriptor": "NodeDescriptor(logical_type=<LogicalType.Router: 1>, complex_descriptor_available=0, user_descriptor_available=0, reserved=0, aps_flags=0, frequency_band=<FrequencyBand.Freq2400MHz: 8>, mac_capability_flags=<MACCapabilityFlags.AllocateAddress|RxOnWhenIdle|MainsPowered|FullFunctionDevice: 142>, manufacturer_code=4098, maximum_buffer_size=82, maximum_incoming_transfer_size=82, server_mask=11264, maximum_outgoing_transfer_size=82, descriptor_capability_field=<DescriptorCapability.NONE: 0>, *allocate_address=True, *is_alternate_pan_coordinator=False, *is_coordinator=False, *is_end_device=False, *is_full_function_device=True, *is_mains_powered=True, *is_receiver_on_when_idle=True, *is_router=True, *is_security_capable=False)",
  "endpoints": {
    "1": {
      "profile_id": 260,
      "device_type": "0x0051",
      "in_clusters": [
        "0x0000",
        "0x0004",
        "0x0005",
        "0x000a",
        "0xef00"
      ],
      "out_clusters": [
        "0x0019"
      ]
    }
  },
  "manufacturer": "_TZE200_cowvfni3",
  "model": "TS0601",
  "class": "zigpy.device.Device"
}

MattWestb, 2021-09-03:
Download https://github.com/zigpy/zha-device-handlers/blob/dev/zhaquirks/tuya/ts0601.py the and editing: https://github.com/zigpy/zha-device-handlers/blob/28bd89a2f34da9a15dd37370734af4e222850316/zhaquirks/tuya/ts0601.py#L86
by adding your device model ifo in one new line after the last in the signature.
Then enable local quirks in your HA config and put the ts0601.py in that folder and restarting HA.

Faustjohan, 2021-09-03:
Thanks for your help, I've got an error when I'm using the button that appeared : 

"Failed to call service "cover/close_cover". '_TZE200_cowvfni3'"

MattWestb, 2021-09-03:
Sorry i was forgetting the buttons need being defined in https://github.com/zigpy/zha-device-handlers/blob/28bd89a2f34da9a15dd37370734af4e222850316/zhaquirks/tuya/__init__.py#L93.

The same copy the `__init__` and put on new line with your device.

The problem you must applying this changes in the HA container then local quirk is reading all from it and can importing 2 files for the same device.

Instruction https://github.com/zigpy/zha-device-handlers/discussions/693 but i think its easier making both changes in the container and deleting the file in the local quirk folder.

Dont forgetting deleting the `__pycache__` in the tuya quirk forlder and then restarting HA.

Faustjohan, 2021-09-03:
Ok I've done the modification on : 
https://github.com/zigpy/zha-device-handlers/blob/28bd89a2f34da9a15dd37370734af4e222850316/zhaquirks/tuya/__init__.py#L93

But still doesn't work.

So now, it start beeing complicated :) 
If I understand well, I need to do this : https://github.com/zigpy/zha-device-handlers/discussions/693 because Zha is importing both normal and custom __init__.py and so it's a mess. That's correct ?

Any way to do it easier ? I don't know how to use Pip.
Add it in the code and pull to master So I can have it through an normal update ? :)

MattWestb, 2021-09-03:
Not possible ZHA is importing both but the TS0601.py is **not** using the local `__init__`  its using the one in the container so must adding at least the `__init__` in the HA container.

If installing suplemon (with Pip) as described its easy doing the changes only that changing the `TS0601.py` and not the `valve.py` as in the example :-))

MattWestb, 2021-09-03:
If you using portainer its only open one CLI in the HA container and pasting `pip install suplemon` and is installing and you can start using suplimon.

Faustjohan, 2021-09-07:
Thanks for your help ! 

I've used another methods : with Samba, I put my init.py modified file in a folder that I can access. After that, I use these two commandes on Portainer's console : 

`cp /usr/local/lib/python3.9/site-packages/zhaquirks/tuya/__init__.py /config/custom_components/zhaquirks -v -i
cp /config/custom_components/zhaquirks/tuya/__init__.py /usr/local/lib/python3.9/site-packages/zhaquirks/tuya/ -v -i`

And finally : 
`rm -rf ./__pycache__/`

MattWestb, 2021-09-07:
The SMB  and copy is working 2 !!

Great that you was getting it working !!

I can making one PR with the 2 changes that  so other user can getting there device working well.

Was you using the `{0x0000: 0x0000, 0x0001: 0x0002, 0x0002: 0x0001},` in the `__init__.py` or was you changing it for getting it working OK ?

Faustjohan, 2021-09-07:
I've add this line in the __init__.py : 
`    "_TZE200_cowvfni3": {0x0000: 0x0002, 0x0001: 0x0000, 0x0002: 0x0001},`

MattWestb, 2021-09-07:
OK then i copy your line so its being OK fro the remote :-))

Faustjohan, 2021-09-07:
Perfect, thanks a lot for your help 👍 

I'll close the request.

Have a nice day :)

closed 2021-09-07 by Faustjohan

MattWestb, 2021-09-07:
PR is posted and looks OK (no complains in the tests) so only waiting for getting it merged.
Dont forgetting deleting the files in the local quirk folder then you is upgrading HA with the quirk is in it.