zigpy/zha-device-handlers #1025: [Device Support Request] DiY Customisable 8/12/20 Button Keypad

quannh, 2021-09-07:
**Is your feature request related to a problem? Please describe.**

Request and placehoder for information about the Button Keypad > specfically:

https://zigbee.blakadder.com/DIYRuZ_FreePad.html

z2m device: https://github.com/Koenkk/zigbee-herdsman-converters/blob/9e4c36cd79e15449830f5e09540157c0e8d686a1/devices.js#L4331

**Describe the solution you'd like**
ZHA support, please :)

**Device signature - this can be acquired by removing the device from ZHA and pairing it again from the add devices screen. Be sure to add the entire content of the log panel after pairing the device to a code block below this line.**

`
{
  "node_descriptor": "NodeDescriptor(logical_type=<LogicalType.EndDevice: 2>, complex_descriptor_available=0, user_descriptor_available=0, reserved=0, aps_flags=0, frequency_band=<FrequencyBand.Freq2400MHz: 8>, mac_capability_flags=<MACCapabilityFlags.AllocateAddress: 128>, manufacturer_code=0, maximum_buffer_size=80, maximum_incoming_transfer_size=160, server_mask=0, maximum_outgoing_transfer_size=160, descriptor_capability_field=<DescriptorCapability.0: 0>, *allocate_address=True, *is_alternate_pan_coordinator=False, *is_coordinator=False, *is_end_device=True, *is_full_function_device=False, *is_mains_powered=False, *is_receiver_on_when_idle=False, *is_router=False, *is_security_capable=False)",
  "endpoints": {
    "1": {
      "profile_id": 260,
      "device_type": "0x0006",
      "in_clusters": [
        "0x0000"
      ],
      "out_clusters": [
        "0x0006",
        "0x0008",
        "0x0012",
        "0x0300"
      ]
    },
    "2": {
      "profile_id": 260,
      "device_type": "0x0006",
      "in_clusters": [],
      "out_clusters": [
        "0x0006",
        "0x0012"
      ]
    },
    "3": {
      "profile_id": 260,
      "device_type": "0x0006",
      "in_clusters": [],
      "out_clusters": [
        "0x0006",
        "0x0012"
      ]
    },
    "4": {
      "profile_id": 260,
      "device_type": "0x0006",
      "in_clusters": [],
      "out_clusters": [
        "0x0006",
        "0x0012"
      ]
    },
    "5": {
      "profile_id": 260,
      "device_type": "0x0006",
      "in_clusters": [],
      "out_clusters": [
        "0x0006",
        "0x0008",
        "0x0012",
        "0x0300"
      ]
    },
    "6": {
      "profile_id": 260,
      "device_type": "0x0006",
      "in_clusters": [],
      "out_clusters": [
        "0x0006",
        "0x0012"
      ]
    },
    "7": {
      "profile_id": 260,
      "device_type": "0x0006",
      "in_clusters": [],
      "out_clusters": [
        "0x0006",
        "0x0012"
      ]
    },
    "8": {
      "profile_id": 260,
      "device_type": "0x0006",
      "in_clusters": [],
      "out_clusters": [
        "0x0006",
        "0x0012"
      ]
    },
    "9": {
      "profile_id": 260,
      "device_type": "0x0006",
      "in_clusters": [],
      "out_clusters": [
        "0x0006",
        "0x0008",
        "0x0012",
        "0x0300"
      ]
    },
    "10": {
      "profile_id": 260,
      "device_type": "0x0006",
      "in_clusters": [],
      "out_clusters": [
        "0x0006",
        "0x0012"
      ]
    },
    "11": {
      "profile_id": 260,
      "device_type": "0x0006",
      "in_clusters": [],
      "out_clusters": [
        "0x0006",
        "0x0012"
      ]
    },
    "12": {
      "profile_id": 260,
      "device_type": "0x0006",
      "in_clusters": [],
      "out_clusters": [
        "0x0006",
        "0x0012"
      ]
    },
    "13": {
      "profile_id": 260,
      "device_type": "0x0006",
      "in_clusters": [],
      "out_clusters": [
        "0x0006",
        "0x0008",
        "0x0012",
        "0x0300"
      ]
    },
    "14": {
      "profile_id": 260,
      "device_type": "0x0006",
      "in_clusters": [],
      "out_clusters": [
        "0x0006",
        "0x0012"
      ]
    },
    "15": {
      "profile_id": 260,
      "device_type": "0x0006",
      "in_clusters": [],
      "out_clusters": [
        "0x0006",
        "0x0012"
      ]
    },
    "16": {
      "profile_id": 260,
      "device_type": "0x0006",
      "in_clusters": [],
      "out_clusters": [
        "0x0006",
        "0x0012"
      ]
    },
    "17": {
      "profile_id": 260,
      "device_type": "0x0006",
      "in_clusters": [],
      "out_clusters": [
        "0x0006",
        "0x0008",
        "0x0012",
        "0x0300"
      ]
    },
    "18": {
      "profile_id": 260,
      "device_type": "0x0006",
      "in_clusters": [],
      "out_clusters": [
        "0x0006",
        "0x0012"
      ]
    },
    "19": {
      "profile_id": 260,
      "device_type": "0x0006",
      "in_clusters": [],
      "out_clusters": [
        "0x0006",
        "0x0012"
      ]
    },
    "20": {
      "profile_id": 260,
      "device_type": "0x0006",
      "in_clusters": [],
      "out_clusters": [
        "0x0006",
        "0x0012"
      ]
    }
  },
  "manufacturer": "modkam.ru",
  "model": "DIYRuZ_FreePad",
  "class": "zigpy.device.Device"
}
`

scheusal333, 2021-10-16:
I also wanted to add my diy freepad to my zha, without success. Is there any info missing in the request? Anything I can add to push it?

MattWestb, 2021-10-16:
Was looking little on the device signature and som things looks very good and some not so good:
```
"profile_id": 260 = HA =OK.
"device_type": "0x0006" = REMOTE_CONTROL = OK but not good.

"0x0006" On/Off = OK.
"0x0008" = Level control OK.
"0x0012" = multistate_input = OK if god implanted in the device.
"0x0300" = Color Control = OK but not in the profile.
```
The device type is expecting:
`DeviceType.REMOTE_CONTROL: ([], [0x0004, 0x0005, 0x0006, 0x0008]),`
Missing expected cluster: 0x0004 = groups, 0x0005 = scenes = the device cluster cant being bonded to groups and ZHA is not configuring attribute reporting on 0x0012 and 0x0300.

COLOR_DIMMER_SWITCH shall being better then its binding / configuring reporting of the 0x0300 Color Control cluster but not the 0x0012 that is out of normal profils.

CONTROL_BRIDGE is the best then it can doing all (in)possible things in the network (Its the same as Philips HUE and IKEA is using in there GWs).

ZHA is doing some default attribute reporting configuring and bindings then joining devices but some time its not being OK done and need doing one "reconfigure this device" from device card and waking up the device for it  being in deep sleep for reviving the commands.

After doing that go to the development tools > Events and putting in `zha_event` in the "Listen to events" and clicking "Start listen to events".

If ZHA is reviving somthing from the remote you is getting them in the event window.

PM: The `"0x0012" = multistate_input` shall working OK if the remote is having configuring OK i the firmware but its pity complex and i dont thing its OK implanted.

Also **if** the remote is sending all commands default to 0x0000 (the coordinator in the network) then ZHA shall getting all command it sending as events but its depends on the device.

Test and see how the device is doing it and also putting in debug loging for see how its looks under the "hood".

closed 2022-04-24 by github-actions[bot]

delta010, 2022-10-28:
while reconfiguring i find this,
```
2022-10-28 19:17:54.528 DEBUG (MainThread) [homeassistant.components.zha.core.channels.base] [0xC1BB:2:0x0006]: Failed to bind 'on_off' cluster:
2022-10-28 19:17:54.529 DEBUG (MainThread) [homeassistant.components.zha.core.channels.base] [0xC1BB:2:0x0006]: finished channel configuration
2022-10-28 19:17:54.530 DEBUG (MainThread) [homeassistant.components.zha.core.channels.base] [0xC1BB:2:0x0006]: 'async_configure' stage succeeded
2022-10-28 19:17:54.536 DEBUG (MainThread) [homeassistant.components.zha.core.channels.base] [0xC1BB:3:0x0006]: Failed to bind 'on_off' cluster:
2022-10-28 19:17:54.536 DEBUG (MainThread) [homeassistant.components.zha.core.channels.base] [0xC1BB:3:0x0006]: finished channel configuration
2022-10-28 19:17:54.537 DEBUG (MainThread) [homeassistant.components.zha.core.channels.base] [0xC1BB:3:0x0006]: 'async_configure' stage succeeded
2022-10-28 19:17:54.544 DEBUG (MainThread) [homeassistant.components.zha.core.channels.base] [0xC1BB:4:0x0006]: Failed to bind 'on_off' cluster:
2022-10-28 19:17:54.544 DEBUG (MainThread) [homeassistant.components.zha.core.channels.base] [0xC1BB:4:0x0006]: finished channel configuration
2022-10-28 19:17:54.545 DEBUG (MainThread) [homeassistant.components.zha.core.channels.base] [0xC1BB:4:0x0006]: 'async_configure' stage succeeded
2022-10-28 19:17:54.551 DEBUG (MainThread) [homeassistant.components.zha.core.channels.base] [0xC1BB:5:0x0008]: Failed to bind 'level' cluster:
2022-10-28 19:17:54.551 DEBUG (MainThread) [homeassistant.components.zha.core.channels.base] [0xC1BB:5:0x0008]: finished channel configuration
2022-10-28 19:17:54.560 DEBUG (MainThread) [homeassistant.components.zha.core.channels.base] [0xC1BB:5:0x0006]: Failed to bind 'on_off' cluster:
2022-10-28 19:17:54.560 DEBUG (MainThread) [homeassistant.components.zha.core.channels.base] [0xC1BB:5:0x0006]: finished channel configuration
2022-10-28 19:17:54.598 DEBUG (MainThread) [homeassistant.components.zha.core.channels.base] [0xC1BB:5:0x0300]: Failed to bind 'light_color' cluster: 
```
[image]

dmulcahey, 2022-10-28:
Were you pressing buttons to keep the device awake?

delta010, 2022-10-28:
I was pressing all buttons while the reconfiguring process

dmulcahey, 2022-10-28:
Factory reset the device and pair it again please. Also ensure the battery in the remote is good.

delta010, 2022-10-28:
I reflash the device several times, also I changed the firmware to PTVO configurable firmware create 8 buttons in that case also first 4 buttons work, buttons 5 to 8 did not fire any event data

delta010, 2023-03-26:
its works fine with Zigbee channel 11 but not on channel 15