zigpy/zha-device-handlers #1027: [Device Support Request] TS0601 / TV02 Zigbee TRV

dodekaphilist, 2021-09-08:
**Is your feature request related to a problem? Please describe.**
I bought this thermostat from AliExpress. It's possible to pair with ZHA, but no entities are exposed.

**Describe the solution you'd like**
Ideally, I'd like to control all settings these Tuya based thermostats have. I know that Zigbee2MQTT seems to be able to do that ([Source](https://www.youtube.com/watch?v=iD1U6Q47yLU))

**Device signature**
`{
  "node_descriptor": "NodeDescriptor(logical_type=<LogicalType.EndDevice: 2>, complex_descriptor_available=0, user_descriptor_available=0, reserved=0, aps_flags=0, frequency_band=<FrequencyBand.Freq2400MHz: 8>, mac_capability_flags=<MACCapabilityFlags.AllocateAddress: 128>, manufacturer_code=4098, maximum_buffer_size=82, maximum_incoming_transfer_size=82, server_mask=11264, maximum_outgoing_transfer_size=82, descriptor_capability_field=<DescriptorCapability.NONE: 0>, *allocate_address=True, *is_alternate_pan_coordinator=False, *is_coordinator=False, *is_end_device=True, *is_full_function_device=False, *is_mains_powered=False, *is_receiver_on_when_idle=False, *is_router=False, *is_security_capable=False)",
  "endpoints": {
    "1": {
      "profile_id": 260,
      "device_type": "0x0051",
      "in_clusters": [
        "0x0000",
        "0x0004",
        "0x0005",
        "0xef00"
      ],
      "out_clusters": [
        "0x000a",
        "0x0019"
      ]
    }
  },
  "manufacturer": "_TZE200_hue3yfsn",
  "model": "TS0601",
  "class": "zigpy.device.Device"
}`

**Additional context**
The thermostat has no branding whatsoever (neither the device itself, nor the packaging material), I can't tell who is the actual manufacturer. In the included manual it's refered to as TV02 Zigbee Thermostat Radiator Valve
I bought it from this link: [AliExpress](https://www.aliexpress.com/item/1005003190631469.html)

diti1377, 2021-09-29:
Vielleicht kann das weiter helfen, ich hab meine TRVs angebunden.
10                  frost: 0                           Frost Protection     
101                KSSW: 1 SECOND 
102                KCWD: 5 度                  Open Window
104                SSWD: 21 度                 Comfortable Temp
105                JLWD: 17 度                  ECO Temp
107                Heating_Stopping: 0      Heating Stop
108                mod_program001:         Zeitplan Montag
109                wed_program003: 
110                fri_program005:  
111                sun_program007: 
112                tue_program002: 
113                thur_program004: 
114                sat_program006: 
115                online: 1
16                  temp_set: 18.5 ℃ (SET.Temp)       
2                    mode: auto
24                  temp_current: 21.8 ℃                     
27                  temp_correction: 0.1                         
31                  work_days: all
32                  holiday_temp_set: 17 ℃ Holiday Mode
35                  battery_percentage: 100 %; 
40                  child_lock: 0                                        
45                  fault: 1
46                  holiday_set: 201  
8                    window_check: 0 ;

dodekaphilist, 2021-09-30:
Can you state more precisely where this information needs to go to?

MattWestb, 2021-10-16:
@diti1377 You was writing in [Iobroker](https://forum.iobroker.net/topic/47516/tuya-thermostat-_tze200_hue3yfsn/26?_=1634323613875) that one user have written one quirk for one TRV in ZHA that have the same DPs like this but i cant finding the user name and the quirk.
Can you pleas putting my in the right direction to finding it pleas ?

MattWestb, 2021-10-16:
If you is adding the model_id to more signatures and more then one is matching i think ZHA is using the first match (or the last one i dont knowing the logic). 

Your device is having one on/off switch so i think its one Moes or one Sas that is loaded.
You can see the name of the loaded quirk in the last line on the device profile or on the device card.

Deleting all your added model_ids and only adding it to the `ZonnsmartTV01_ZG` then its looks matching your device DPs and restarting HA.

With the last `ZonnsmartTV01_ZG` you shall **not** do the ZAH fix then its not using the Moes TRV quirk so only adding the model _id or you is getting errors and HA is not starting.

dodekaphilist, 2021-10-16:
Ok, maybe I should clarify that I tested each class separately one at a time and didn't add the model info to all classes simultaneously. Each time with a similar result.

rforro, 2021-11-03:
This TRV is bit strange. It shares like a 95% of DPs with Lidl, but it's completely different device. Actually in one listing you can find the word Zonnsmart. Look here https://de.aliexpress.com/item/1005003190631469.html (Zonnsmart it's not in the description but in comparison image in Energy Saving section)

My assumption is that it needs to be build on top of the current Zonnsmart integration.

@Droggel00 The last two days I was observing mine and it never publishes current temperature. The only option is to restart the device. Does somebody know if it has to be manually requested?

MattWestb, 2021-11-03:
If using Zonnsmart quirk and correcting it for different DP, is one problem with the Zonnsmart implementation its dont updating the local temperature and the same person that  was doing the Zonnsmart quirk is doing  the same in Z2M and they is having the same problem with the device so very likely some strange in the device firmware.
One way is using the "Moes" quirk that is the one with most functions and working well and adjusting the DP and functions for `SHENZHEN RSH` TRVs and see if its working better.

If setting `zhaquirks,tuya` debug in HA you can see the DP commands sent and revived is you getting any with the DP for local temperature or some other that can being that ?

Moes TRVs users is sending `Calibrating local temperature` from HA automation for getting more frequent update then one time in the hour that the Moes (and most tuya) TRVs is doing.

rforro, 2021-11-04:
I was able to extend the zonnsmart quirk with ECO_TEMP, CONFORT_TEMP and TEMP_CORRECTION. I'm pretty sure that this device is ZONNSMART because it uses the same DPs.

As was said by @dodekaphilist [here](https://github.com/zigpy/zha-device-handlers/issues/1027#issuecomment-946715280) by just adding it to current ZONNSMART quirk you'll get current temp, set temp, turn heating on/off. This attributes can be controlled using HA frontend, but by sending values over developer tools you can control child lock, window detection and mode as well. I've extended this quirk with ECO_TEMP, COMFORT_TEMP and TEMP_CORRECTION (aka calibration) which was pretty easy. After that I looked at [zigbee2mqtt](https://github.com/Koenkk/zigbee2mqtt/issues/8522), now I'm sure that this zonnsmart quirk is just missing a lot of attributes.

I can extend this quirk, that's no problem, but I have no idea how to get entities like child_lock, mode, eco_temp, etc. appear on HA frontend. @MattWestb can you navigate me in right direction?

About the current-temp-bug you were right, I could force the publishing by setting the local temperature calibration value. But I'm not the only person who is experiencing this behaviour. [Same device same solution](https://github.com/Koenkk/zigbee-herdsman-converters/issues/3142#issuecomment-958974999) and [here as well, but another solution](https://github.com/Koenkk/zigbee2mqtt/issues/8522). One user offered that he'll try to sniff the communication with tyua hub, we need to be patient.

MattWestb, 2021-11-04:
## Great findings and work !!

You dont need using the development tool if you like only manage cluster from the device card and you is having all functions on the new tuya clusters for your device like window detection and child lock if they is implanted OK.

The original implanting was done early in ZHA and all functions was not discovered but the dev have implanting them in Z2M so shall being possible doing i ZHA.

So this is one Zonnsmart TV01-ZG with LED display on the side :-))
Shall i making on PR for adding it in the quirk so user can start using it in one easy way **?**

The original implantation was not using presets in ZHA clima component so its have very little working modes in ZHA. Adding them is prity easy but  must being made in both the quirk and in ZHA or its braking the function of changing mode in ZHA.

You can looking how the Moes TRVs is implanted and copy it.
I  have making one PR for adding this future in the Siterwell TRV but i cant getting it thru then i braking many tests in Zogpy :-(( 
The problem with Siterwell is that is many in use so not nice braking them  but it can being needed doing for getting all function working in ZHA.
Looks on my Siterwell upgrading project https://github.com/zigpy/zha-device-handlers/discussions/1084.

Switched and reporting alarm and status is coming then https://github.com/zigpy/zha-device-handlers/issues/1061 is getting working OK its possible implanting this functions in the same way but its not tested in production but is coming very soon.

One more time great work done !!

rforro, 2021-11-04:
Wow thanks about the "manage cluster" info, this could save me a lot of time. But better now than never :)

Tomorrow I'll add those missing DPs and do PR. Just to be noted for known reasons I don't plan to integrate scheduler. After that we can work on displaying those entities on HA frontend.

MattWestb, 2021-11-05:
If you like implanting presets in ZHA i have making one `climate.py` for ZS with the `hue3yfs`  ID and the normal ZS ones that you can patching in your HA container and getting the presets working (I have copy the Moes TRV that is having most presets and if you dont need some its only deleting them later but good for testing).

If running HA 2021.11 copy the extracted `climate.py` in to your HA config directory.
Open one terminal **in** your HA docker container and copy the new file to its right place:
```
cd  /usr/src/homeassistant/homeassistant/components/zha/
cp /config/climate.py climate.py
```
Have moving the device to ZS in the tuya TRV matrix and setting status "working with quirk in PR" so its looks better.

rforro, 2021-11-05:
Currently I've only expanded the known DPs and tested them by setting/getting the values manually.

Following DPs works and they behaviour is more o less clear:
```
-MODE        2021-11-04 19:27:40 DEBUG (MainThread) [zhaquirks.tuya] [0xb5eb:1:0xef00] Received value [0] for attribute 0x0402 (command 0x0001)
-WINDOW      2021-11-04 19:27:40 DEBUG (MainThread) [zhaquirks.tuya] [0xb5eb:1:0xef00] Received value [0] for attribute 0x0108 (command 0x0001)
-FROST       2021-11-04 19:27:40 DEBUG (MainThread) [zhaquirks.tuya] [0xb5eb:1:0xef00] Received value [0] for attribute 0x010a (command 0x0001)
-HEATTEMP    2021-11-04 19:27:40 DEBUG (MainThread) [zhaquirks.tuya] [0xb5eb:1:0xef00] Received value [0, 0, 0, 170] for attribute 0x0210 (command 0x0001)
-CURTEMP     2021-11-04 19:27:40 DEBUG (MainThread) [zhaquirks.tuya] [0xb5eb:1:0xef00] Received value [0, 0, 0, 205] for attribute 0x0218 (command 0x0001)
-CORRTEMP    2021-11-04 19:27:40 DEBUG (MainThread) [zhaquirks.tuya] [0xb5eb:1:0xef00] Received value [0, 0, 0, 0] for attribute 0x021b (command 0x0001)
-WORKDAYS    2021-11-04 19:27:40 DEBUG (MainThread) [zhaquirks.tuya] [0xb5eb:1:0xef00] Received value [0] for attribute 0x041f (command 0x0001)
-HOLIDAYTEMP 2021-11-04 19:27:41 DEBUG (MainThread) [zhaquirks.tuya] [0xb5eb:1:0xef00] Received value [0, 0, 0, 170] for attribute 0x0220 (command 0x0001)
-BATT        2021-11-04 19:27:41 DEBUG (MainThread) [zhaquirks.tuya] [0xb5eb:1:0xef00] Received value [0, 0, 0, 100] for attribute 0x0223 (command 0x0001)
-CHILDLOCK   2021-11-04 19:27:41 DEBUG (MainThread) [zhaquirks.tuya] [0xb5eb:1:0xef00] Received value [0] for attribute 0x0128 (command 0x0001)
-BOOST       2021-11-04 19:27:41 DEBUG (MainThread) [zhaquirks.tuya] [0xb5eb:1:0xef00] Received value [0, 0, 0, 0] for attribute 0x0265 (command 0x0001)
-COMFORTTEMP 2021-11-04 19:27:42 DEBUG (MainThread) [zhaquirks.tuya] [0xb5eb:1:0xef00] Received value [0, 0, 0, 210] for attribute 0x0268 (command 0x0001)
-ECOTEMP     2021-11-04 19:27:42 DEBUG (MainThread) [zhaquirks.tuya] [0xb5eb:1:0xef00] Received value [0, 0, 0, 170] for attribute 0x0269 (command 0x0001)
-HEATSTOP    2021-11-04 19:27:42 DEBUG (MainThread) [zhaquirks.tuya] [0xb5eb:1:0xef00] Received value [0] for attribute 0x016b (command 0x0001)
```

In those ones I don't understand the meaning or values.
```
?FAULTY      2021-11-04 19:27:41 DEBUG (MainThread) [zhaquirks.tuya] [0xb5eb:1:0xef00] Received value [0] for attribute 0x052d (command 0x0001)
?HOLIDAY     2021-11-04 19:27:41 DEBUG (MainThread) [zhaquirks.tuya] [0xb5eb:1:0xef00] Received value [50, 48, 50, 49, 48, 49, 48, 49, 48, 49, 48, 49, 50, 48, 50, 49, 48, 49, 48, 49, 48, 49, 48, 49] for attribute 0x032e (command 0x0001)
?WINDOW TEMP 2021-11-04 19:27:42 DEBUG (MainThread) [zhaquirks.tuya] [0xb5eb:1:0xef00] Received value [0, 0, 0, 210] for attribute 0x0266 (command 0x0001)
?ONLINE      2021-11-04 19:27:47 DEBUG (MainThread) [zhaquirks.tuya] [0xb5eb:1:0xef00] Received value [1] for attribute 0x0473 (command 0x0001)
```
- Faulty - what fault is reported?
- Holiday - this is holiday duration not temperature (holiday temp works), but the format is unknown and I don't have Tuya hub to test it.
- Window Temp value is used to recognize opened window. It should represent temp drop, but the parameter 210 is too much.
- Online doesn't make sense, how can device publish online=0 if the connection is down ???

And as I said the scheduler is ignored
```
2021-11-04 19:27:43 DEBUG (MainThread) [zhaquirks.tuya] [0xb5eb:1:0xef00] Received value [36, 0, 170, 72, 0, 210, 84, 0, 170, 102, 0, 210, 144, 0, 170, 144, 0, 210, 144, 0, 170, 144, 0, 210, 144, 0, 170, 144, 0, 210] for attribute 0x006e (command 0x0001)
2021-11-04 19:27:43 DEBUG (MainThread) [zhaquirks.tuya] [0xb5eb:1:0xef00] Received value [36, 0, 170, 72, 0, 210, 84, 0, 170, 102, 0, 210, 144, 0, 170, 144, 0, 210, 144, 0, 170, 144, 0, 210, 144, 0, 170, 144, 0, 210] for attribute 0x0072 (command 0x0001)
2021-11-04 19:27:44 DEBUG (MainThread) [zhaquirks.tuya] [0xb5eb:1:0xef00] Received value [36, 0, 170, 72, 0, 210, 84, 0, 170, 102, 0, 210, 144, 0, 170, 144, 0, 210, 144, 0, 170, 144, 0, 210, 144, 0, 170, 144, 0, 210] for attribute 0x006f (command 0x0001)
2021-11-04 19:27:45 DEBUG (MainThread) [zhaquirks.tuya] [0xb5eb:1:0xef00] Received value [36, 0, 170, 72, 0, 210, 84, 0, 170, 102, 0, 210, 144, 0, 170, 144, 0, 210, 144, 0, 170, 144, 0, 210, 144, 0, 170, 144, 0, 210] for attribute 0x006c (command 0x0001)
2021-11-04 19:27:45 DEBUG (MainThread) [zhaquirks.tuya] [0xb5eb:1:0xef00] Received value [36, 0, 170, 72, 0, 210, 84, 0, 170, 102, 0, 210, 144, 0, 170, 144, 0, 210, 144, 0, 170, 144, 0, 210, 144, 0, 170, 144, 0, 210] for attribute 0x0070 (command 0x0001)
2021-11-04 19:27:46 DEBUG (MainThread) [zhaquirks.tuya] [0xb5eb:1:0xef00] Received value [36, 0, 170, 72, 0, 210, 84, 0, 170, 102, 0, 210, 144, 0, 170, 144, 0, 210, 144, 0, 170, 144, 0, 210, 144, 0, 170, 144, 0, 210] for attribute 0x006d (command 0x0001)
2021-11-04 19:27:46 DEBUG (MainThread) [zhaquirks.tuya] [0xb5eb:1:0xef00] Received value [36, 0, 170, 72, 0, 210, 84, 0, 170, 102, 0, 210, 144, 0, 170, 144, 0, 210, 144, 0, 170, 144, 0, 210, 144, 0, 170, 144, 0, 210] for attribute 0x0071 (command 0x0001)
```

Working quirk [valve.zip](https://github.com/zigpy/zha-device-handlers/files/7483965/valve.zip) can be downloaded. Now I can work on ZHA integration.

MattWestb, 2021-11-05:
Siterwell is having valve detect / anti calk function and if the valve is jammed the valve is sending one alarm. I was testing it by setting the setpoint to high and the valve was trying getting the temperature up but was failing after some time and was having the `!` in the display and and sending the DP for valve jammed.

I think the best is as you have saying all known DP to working OK and try finding the function for the not known function and leave the scheduler until ZHA have getting better support for it.

Look little on Moes and the LIDL implantation and you can see different ways for window and holiday is working and i dont knowing if the `online`  is called ping in some issues but i dont knowing how it working.

rforro, 2021-11-08:
I was able to implement:
    - heating operation off/heat
    - present manual/schedule/holiday
    - opened window detection
    - child lock
    - fault (not able to test)
    - temperature offset

[image]
    
Probably this TRV will be needing its own class because I can't test it with original ZONNSMART. The behaviour like ability to turn it off is missing on most devices like SITERWELL or LIDL.

### Unique behaviour of this TRV:
###Off
You can turn this TRV off, by setting `ZONNSMART_HEATING_STOPPING = 0x016B` to 1. On the TRV will be show HS (heating stopped) and you can't do anything till you deactivate it.

###Present
Manual and Schedule works like expected but holiday mode cannot be started by selecting the mode. It only shows the holiday mode temperature and after 5sec it switches back to schedule mode. (FYI holiday mode is activated automatically by setting begin and end timestamp = not implemented)

###Frost Protection (not implemeted)
I don't know how to implement this one, because it's like one of those Present Modes. After setting `ZONNSMART_FROST_PROTECT_ATTR = 0x010a` attribute, on TRV will be shown AF (antifreeze), it sets temperature to 8 degrees and you cannot do anything until you disable this.

###Don't publishes room temperature
This TRV never publishes room temperature, it has to be manually requested. Currently the only way is to set temperature offset (even with 0), after that the room temperature will be published.

If someone will test it, here is the quirk: [valve.zip](https://github.com/zigpy/zha-device-handlers/files/7494743/valve.zip) but for presets to work, you have to add to `/components/zha/climate.py` following lines:

```
@STRICT_MATCH(
    channel_names=CHANNEL_THERMOSTAT,
    manufacturers={
        "_TZE200_hue3yfsn",
        },
)

class ZONNSMARTThermostat(Thermostat):
    """ZONNSMART Thermostat implementation"""
    def __init__(self, unique_id, zha_device, channels, **kwargs):
        """Initialize ZHA Thermostat instance."""
        super().__init__(unique_id, zha_device, channels, **kwargs)
            self._presets = [
            PRESET_SCHEDULE,
            PRESET_NONE,
            PRESET_AWAY,
        ]
        self._supported_flags |= SUPPORT_PRESET_MODE

    async def async_attribute_updated(self, record):
        """Handle attribute update from device."""
        if record.attr_name == "operation_preset":
            if record.value == 0:
                self._preset = PRESET_SCHEDULE
            if record.value == 1:
                self._preset = PRESET_NONE
            if record.value == 3:
                self._preset = PRESET_AWAY
        await super().async_attribute_updated(record)

    async def async_preset_handler(self, preset: str, enable: bool = False) -> bool:
        """Set the preset mode."""
        mfg_code = self._zha_device.manufacturer_code
        if not enable:
            return await self._thrm.write_attributes(
                {"operation_preset": 1}, manufacturer=mfg_code
            )
        if preset == PRESET_SCHEDULE:
            return await self._thrm.write_attributes(
                {"operation_preset": 0}, manufacturer=mfg_code
            )
        if preset == PRESET_AWAY:
            return await self._thrm.write_attributes(
                {"operation_preset": 3}, manufacturer=mfg_code
            )
        return False
```

jacekk015, 2021-11-08:
"###Present
Manual and Schedule works like expected but holiday mode cannot be started by selecting the mode. It only shows the holiday mode temperature and after 5sec it switches back to schedule mode."

Holiday mode often has date/time of start and time duration. If present time is after the values were set it switches back to Schedule after some time. That 5 sec. for Maxsmart is to start setting the parameters on the TRV.

"###Frost Protection (not implemeted)"
For frost protection you would need to add another switch and a proper code + bus listener

Catch the attribute and set temperature manually if you don't have an attribute for it.
ZONNSMARTManufCluster:
```
        elif attrid == ZONNSMART_FROST_PROTECT_ATTR:
            self.endpoint.device.thermostat_bus.listener_event(
                "temperature_change",
                self.DIRECT_MAPPED_ATTRS[ZONNSMART_TARGET_TEMP_ATTR][0],
                800)
           self.endpoint.device.frost_protection_bus.listener_event(
                "frost_protection_change", mode)
```

Copy&paste child lock class and rename it to frost protection.
Add another endpoint to ZonnsmartTV01_ZG
```
            3: {
                PROFILE_ID: zha.PROFILE_ID,
                DEVICE_TYPE: zha.DeviceType.ON_OFF_SWITCH,
                INPUT_CLUSTERS: [
                    ZONNSMARTFrostProtection,
                ],
                OUTPUT_CLUSTERS: [],
            },
```

Don't forget to INIT bus in ZonnsmartTV01_ZG ->def __init__
```
self.frost_protection_bus= Bus()
```

jacekk015, 2021-11-08:
If above code will not work[probably won't] do three more steps:
Add ZONNSMARTManufClusterSelf variable like below:
```
ZONNSMART_HOLIDAY_DATETIME_ATTR = 0x032e  # holiday mode date and time of begin and end
ZONNSMART_WEEK_FORMAT_ATTR = 0x041f  # # [0] 5+2 days [1] 6+1 days, [2] 7 days
ZONNSMART_FAULT_DETECTION_ATTR = 0x052d  # [0] no fault [1] fault detected
ZONNSMARTManufClusterSelf = None
```

Add INIT to ZONNSMARTManufCluster before manufacturer_attributes:
```
    def __init__(self, *args, **kwargs):
        """Init."""
        super().__init__(*args, **kwargs)
        self.endpoint.device.ZONNSMARTManufCluster .add_listener(self)
        global ZONNSMARTManufClusterSelf 
        ZONNSMARTManufClusterSelf = self
```

Then use that variable in the write_attributes of ZONNSMARTFrostProtection
Change line to:
```
        if has_change:
            return await ZONNSMARTManufClusterSelf .endpoint.tuya_manufacturer.write_attributes(
                {ZONNSMART_FROST_PROTECT_ATTR: value}, manufacturer=manufacturer
            )
```
That's is needed because your ZONNSMARTFrostProtection is on endpoint 3 and ZONNSMART_FROST_PROTECT_ATTR is on endpoint 1.
If you do self.*** it assumes that the self.*** is on same endpoint - but here it isn't.
We just created a reference to endpoint 1 class.

rforro, 2021-11-08:
> Holiday mode often has date/time of start and time duration. If present time is after the values were set it switches back to Schedule after some time. That 5 sec. for Maxsmart is to start setting [...]

Yes I know how it works, I wrote that it starts automatically by setting begin and end time, but here is the format different than the Maxsmart. This TRV uses only one attribute, its value is `[50, 48, 50, 49, 48, 49, 48, 49, 48, 49, 48, 49, 50, 48, 50, 49, 48, 49, 48, 49, 48, 49, 48, 49]` which represents `20210101 01:01 20210101 01:01`. You could in theory use [Input Datetime](https://www.home-assistant.io/integrations/input_datetime/) but it will take too much time to develop. I'd rather use HA automation for this.

> > ###Frost Protection (not implemeted)
> For frost protection you would need to add another switch and a proper code + bus listener.

The implementation into source code is not the problem, what I meant is the behavior. When I activate frost protection the TRV publishes following:

- front protection = on
- target temp = 8
- mode = schedule

which is misinterpreted by HA fronted climate component. This is what I meant. It should display in HA frontend that the heating is off and the input_boolean front protection is active.

MattWestb, 2021-11-08:
If looking in the commit for the first added  tuya TRV presets its looks being easy (if not need adding tests) more presets in ZHA climate component.
https://github.com/home-assistant/core/pull/48178/files

I think one `Holiday` for this and one `FI` for Site can being good but if its possible making them not possible to select but is showing in the GUI and having one witch on the device card and with date and time for the holiday mode .
Then if setting one holiday date on the device card is ZHA shoeing holiday but cant changing it = OK then it must being cancelled in the device card.
The same with the FI cant being sett in ZHA GUI but with commands from cluster or switch on the device card and the user can see that the device is in that mode.

jacekk015, 2021-11-08:
If you have frost protection then heating isn't really off. But if you insist then change my code above to:
```
        elif attrid == ZONNSMART_FROST_PROTECT_ATTR:
            self.endpoint.device.thermostat_bus.listener_event(
                "temperature_change",
                self.DIRECT_MAPPED_ATTRS[ZONNSMART_TARGET_TEMP_ATTR][0],
                800)
            self.endpoint.device.thermostat_bus.listener_event(
                "state_change",
                0)
           self.endpoint.device.frost_protection_bus.listener_event(
                "frost_protection_change", mode)
```
That's all you need.
state_change -> value = 0 OFF
state_change -> value = 1 HEAT

rforro, 2021-11-08:
> That's all you need. state_change -> value = 0 OFF state_change -> value = 1 HEAT

I think that I'll just do the switch. I don't care about this function, if someone does the code can be anytime changed.

> Ah - and you can't use Input Datetime - it's not implemented in ZHA ;-) https://www.home-assistant.io/integrations/zha/
> 
> > There is currently support for the following device types within Home Assistant:
> > ```
> > Alarm Control Panel
> > Binary Sensor
> > Climate (beta)
> > Cover
> > Fan
> > Light
> > Lock
> > Number (i.e. analog output)
> > Sensor
> > Switch
> > ```
> 
> https://github.com/home-assistant/core/tree/dev/homeassistant/components/zha

Thats what I meant with:
> You could in theory use Input Datetime but it will take too much time to develop. I'd rather use HA automation for this.

One could implement Input Datetime into ZHA, but not me :D

BTW. I'd rather invest the time into naming those entities like I mentioned here https://github.com/zigpy/zha-device-handlers/issues/1061#issuecomment-962846745. FYI I've raise an issue on zigpy, I'll overlook integration of description into onoff cluster.

alexd321, 2021-11-09:
I have been testing this and seems to work mostly ok (sometimes gets confused and stuck in "off" mode.

Anyone else noticed the valve changing climate setpoints by itself?

I have an automation set to send a '0' analog_output_temperature_offset every 15 mins to get the current temperature and the set point seems to jump back to 21°C randomly (not every request).

rforro, 2021-11-10:
To be honest, I never installed this TRV, it was just sitting on my desk to development purposes 😄 But I've installed it today.

I'll try to investigate the changing climate setpoint, but can you elaborate what are your settings? Are you using Schedule mode or Manual? What are your automations?

alexd321, 2021-11-10:
I just left the valve in "🏠" (home mode) which I guess is schedule? 

Maybe the device comes with a built-in schedule? 

`alias: 'Radiator: request room temp'
description: Hack to request room temperature every 15 mins
trigger:
  - platform: time_pattern
    minutes: /10
condition: []
action:
  - service: number.set_value
    data:
      value: '0'
    target:
      entity_id:
        - >-
          number.tze200_hue3yfsn_ts0601_5222c3fe_analog_output_temperature_offset
        - >-
          number.tze200_hue3yfsn_ts0601_11307cfe_analog_output_temperature_offset
mode: single
`

And

To adjust setpoint

`alias: 'Radiator: bedroom schedule'
description: ''
trigger:
  - platform: time
    at: input_datetime.morning_heating_on
    id: morning
  - platform: time
    id: evening
    at: '20:15'
  - platform: time
    at: input_datetime.daytime_temp_setback_time
    id: day setback
  - platform: time
    at: input_datetime.weekend_daytime_temp_setback_time
    id: weekend daytime setback
condition:
  - condition: not
    conditions:
      - condition: state
        entity_id: climate.central_heating
        state: 'off'
action:
  - choose:
      - conditions:
          - condition: trigger
            id: morning
        sequence:
          - service: climate.set_temperature
            target:
              entity_id: climate.bedroom_radiator
            data:
              temperature: 16.5
          - service: notify.mobile_app_alex_samsung
            data:
              message: Bedroom Radiator reset morning
              title: Bedroom Radiator
      - conditions:
          - condition: trigger
            id: day setback
          - condition: time
            weekday:
              - mon
              - tue
              - wed
              - thu
              - fri
        sequence:
          - service: climate.set_temperature
            target:
              entity_id: climate.bedroom_radiator
            data:
              temperature: 10
          - service: notify.mobile_app_alex_samsung
            data:
              message: Bedroom Radiator turned off daytime setback
              title: Bedroom Radiator
      - conditions:
          - condition: trigger
            id: evening
        sequence:
          - service: climate.set_temperature
            target:
              entity_id: climate.bedroom_radiator
            data:
              temperature: 16
          - service: notify.mobile_app_alex_samsung
            data:
              message: Bedroom Radiator activated late evening
              title: Bedroom Radiator
      - conditions:
          - condition: trigger
            id: weekend setback
          - condition: time
            weekday:
              - sat
              - sun
        sequence:
          - service: climate.set_temperature
            target:
              entity_id: climate.bedroom_radiator
            data:
              temperature: 10
          - service: notify.mobile_app_alex_samsung
            data:
              message: Bedroom Radiator turned off weekend setback
              title: Bedroom Radiator
    default: []
mode: single
`

[image]

dodekaphilist, 2021-11-10:
@alexd321 I had the same issue, it is indeed related to the pre-programmed schedule. Unfortunately, we cannot customize the schedule yet, but the manual mode is a good workaround

LennardPlay, 2021-11-13:
> Currently working:
> 
>     * ambient temperature is sent to HA
> 
>     * setting a temperature is sent between devices vice-versa (independant of mode)
> 
>     * off-mode and manual mode can be toggled
> 

I think I have the same thermostat. Can you help me to get at least the temperature control working? 
I sadly don't really  understand what device class and so on is.
Would be lovely if you could help me getting it to work :)

LennardPlay, 2021-11-13:
Sorry for not explaining myself very well in the last message.
I'm wondering how I can use the custom ZONNSMART quirk. Which steps should I take to activate it?

ayonix, 2021-11-27:
Is there anything left to do to get this into the next release?

ALERTua, 2021-11-27:
I am also interested in this device support. just bought a few valves. thank you.

jacekk015, 2021-11-27:
@rforro are you able to publish the quirk so the users could use it?

rforro, 2021-11-27:
I was waiting if your PR will be accepted or you'll have to replace the Bus.

alexd321, 2021-12-03:
> I think I have the same thermostat. Can you help me to get at least the temperature control working? I sadly don't really understand what device class and so on is. Would be lovely if you could help me getting it to work :)

@LennardPlay To get the valve partly working, I just:

1. downloaded [the valve.zip](https://github.com/zigpy/zha-device-handlers/files/7494743/valve.zip) file 
2. uploaded the extracted contents of this zip file (valve.py) to /config/custon_zha_quirks/
3. added the following to configuration.yaml:
  ```
  zha:
    custom_quirks_path: /config/custom_zha_quirks/
   ```
4. restarted home assistant

I haven't modified climate.py yet as I was worried about breaking things/ran out of time/it looked a bit more in depth

alexd321, 2021-12-03:
There is a small indentation error in https://github.com/zigpy/zha-device-handlers/issues/1027#issuecomment-962854799 @rforro

```
class ZONNSMARTThermostat(Thermostat):
    """ZONNSMART Thermostat implementation"""
    def __init__(self, unique_id, zha_device, channels, **kwargs):
        """Initialize ZHA Thermostat instance."""
        super().__init__(unique_id, zha_device, channels, **kwargs)
            self._presets = [
            PRESET_SCHEDULE,
```

should instead be:

```
class ZONNSMARTThermostat(Thermostat):
    """ZONNSMART Thermostat implementation"""
    def __init__(self, unique_id, zha_device, channels, **kwargs):
        """Initialize ZHA Thermostat instance."""
        super().__init__(unique_id, zha_device, channels, **kwargs)
        self._presets = [
            PRESET_SCHEDULE,
```

self._presets line changed

rforro, 2022-01-06:
I'm working on this to include boost and anti-freeze mode, but there is one thing which won't let me sleep.

```
2021-11-04 19:27:47 DEBUG (MainThread) [zhaquirks.tuya] [0xb5eb:1:0xef00] Received value [1] for attribute 0x0473 (command 0x0001)
```
There is this ONLINE datapoint which publishes value 1 at pairing only. Now I understand its purpose. Using original Tuya App or Zigbee2mqtt by publishing value 0 (boolean false) to this datapoint the TRV switches itself to "online mode" (antenna icon lights up) and for ca. 1 minute the TRV will be publishing all parameters including current room temperature automatically. The problem is for some reason I cannot publish the correct value (0 or false) which will be accepted by TRV from ZHA. I've tried `t.Bool` and `t.uint8_t` but the TRV just ignores the value and publishes some strange data:

```
2022-01-06 10:59:57 DEBUG (MainThread) [zigpy_znp.api] Received command: AF.IncomingMsg.Callback(GroupId=0x0000, ClusterId=61184, SrcAddr=0x59A9, SrcEndpoint=1, DstEndpoint=1, WasBroadcast=<Bool.false: 0>, LQI=115, SecurityUse=<Bool.false: 0>, TimeStamp=9468090, TSN=0, Data=b'\x0C\x02\x10\x18\x0B\x00\x00', MacSrcAddr=0x59A9, MsgResultRadius=29)
2022-01-06 10:59:57 DEBUG (MainThread) [zigpy.zcl] [0x59a9:1:0xef00] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=GLOBAL_COMMAND manufacturer_specific=True is_reply=True disable_default_response=False> manufacturer=4098 tsn=24 command_id=Command.Default_Response>
```

Correct data for Online DP
```
2022-01-06 10:32:28 DEBUG (MainThread) [zigpy_znp.api] Received command: AF.IncomingMsg.Callback(GroupId=0x0000, ClusterId=61184, SrcAddr=0x59A9, SrcEndpoint=1, DstEndpoint=1, WasBroadcast=<Bool.false: 0>, LQI=110, SecurityUse=<Bool.false: 0>, TimeStamp=4312484, TSN=0, Data=b'\x09\x44\x01\x00\x71\x73\x04\x00\x01\x01', MacSrcAddr=0x59A9, MsgResultRadius=29)
2022-01-06 10:32:28 DEBUG (MainThread) [zigpy.zcl] [0x59a9:1:0xef00] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=CLUSTER_COMMAND manufacturer_specific=False is_reply=True disable_default_response=False> manufacturer=None tsn=68 command_id=1>
```

Can some zigpy-pro dev point me into right direction?

Perrudi, 2022-01-09:
If testing is needed: I have a 'TV02-ZG LED display on the side and rotating knob on the top.' as shown in https://github.com/zigpy/zigpy/discussions/653 and it would be nice if i could help. But i am standard user - no developer or professional in coding and implementing...

rforro, 2022-01-09:
There is a new completely rewritten version, which will transform itself in following days into PR. I removed fault detection, because I cannot test this functionality, but add Frost Protection mode and Boost mode.

This version has:
- heating operation can be off or heat
- presets to choose from manual/schedule/holiday/frost protection
- binary sensor to show when opened window is detected
- analogue input to set temperature when opened window detected
- analogue input to set temperature offset
- switch for child lock
- switch to enable boost mode for 5min

Quirk is here [ts0601_trv.zip](https://github.com/zigpy/zha-device-handlers/files/7835256/ts0601_trv.zip) and don't forget to add into /components/zha/climate.py following lines:

```
@STRICT_MATCH(
    channel_names=CHANNEL_THERMOSTAT,
    manufacturers={
        "_TZE200_hue3yfsn",
        },
)

class ZONNSMARTThermostat(Thermostat):
    """ZONNSMART Thermostat implementation"""

    PRESET_HOLIDAY = "holiday"
    PRESET_FROST = "frost protect"

    def __init__(self, unique_id, zha_device, channels, **kwargs):
        """Initialize ZHA Thermostat instance."""
        super().__init__(unique_id, zha_device, channels, **kwargs)
            self._presets = [
                PRESET_NONE,
                self.PRESET_HOLIDAY,
                PRESET_SCHEDULE,
                self.PRESET_FROST,
        ]
        self._supported_flags |= SUPPORT_PRESET_MODE

    async def async_attribute_updated(self, record):
        """Handle attribute update from device."""
        if record.attr_name == "operation_preset":
            if record.value == 0:
                self._preset = PRESET_SCHEDULE
            if record.value == 1:
                self._preset = PRESET_NONE
            if record.value == 3:
                self._preset = self.PRESET_HOLIDAY
            if record.value == 4:
                self._preset = self.PRESET_FROST
        await super().async_attribute_updated(record)

    async def async_preset_handler(self, preset: str, enable: bool = False) -> bool:
        """Set the preset mode."""
        mfg_code = self._zha_device.manufacturer_code
        if not enable:
            return await self._thrm.write_attributes(
                {"operation_preset": 1}, manufacturer=mfg_code
            )
        if preset == PRESET_SCHEDULE:
            return await self._thrm.write_attributes(
                {"operation_preset": 0}, manufacturer=mfg_code
            )
        if preset == self.PRESET_HOLIDAY:
            return await self._thrm.write_attributes(
                {"operation_preset": 3}, manufacturer=mfg_code
            )
        if preset == self.PRESET_FROST:
            return await self._thrm.write_attributes(
                {"operation_preset": 4}, manufacturer=mfg_code
            )
        return False
```

Perrudi, 2022-01-09:
That was quick. I would like to test, but i couldnt find the file climate.py. I also didnt find the directory /components/zha on my raspi (Home-Assistant core-2021.12.8 supervisor-2021.12.2 Home Assistant OS 7.1)

rforro, 2022-01-09:
> That was quick. I would like to test, but i couldnt find the file climate.py. I also didnt find the directory /components/zha on my raspi (tried to find it via terminal an via HomeAssistant-file-editor: the highest directory there is /config/

I have the same device so it looks like same as mine HA.

The one off_off is child lock and the other one is Boost mode, there is no possibility to pass the name to HA Frontend so you have to test it and rename it by yourself. The binary_input says if an opened window was detected or not.

As for the `climate.py` you cannot find it, because its inside of an docker container. You have to login into a shell inside of the homeassistant docker container and then you can access the `climate.py` file.

Perrudi, 2022-01-09:
Thank you very much. With these informations i was able to identify some functions and rename the equivalent pieces:

[image]

The 'climaty.py'-point must wait (at least until tomorrow), as it seams to work fundamentally until here...

Perrudi, 2022-01-10:
> As for the `climate.py` you cannot find it, because its inside of an docker container. You have to login into a shell inside of the homeassistant docker container and then you can access the `climate.py` file.

I did ssh on my HomeAssistant-Pi (via terminal from my pc) and the only things with 'climate' i found was /root/.oh-my-zsh/plugins/codeclimate/ but no climate.py in there.

The shell inside HomeAssistant (via SSH & Web Terminal -addon version: 10.0.1) also doesnt find the mentioned directory /components/zha/ and the file climate.py.

I am running HomeAssistant OS, not in docker container. Maybe i am too nooby?

alexd321, 2022-01-10:
> I did ssh on my HomeAssistant-Pi (via terminal from my pc) and the only things with 'climate' i found was /root/.oh-my-zsh/plugins/codeclimate/ but no climate.py in there.
>
> The shell inside [...]

you can get the quirk to work by uploading the contents of the zip file to home assistant, and then adding the following to configuration.yaml:

```
zha:
  custom_quirks_path: /config/custom_zha_quirks/
```

Accessing (updating) the files in the docker container is a little more complex (with increased risk of breaking your Home Assistant installation), you need to use the portainer add-on, see https://github.com/zigpy/zha-device-handlers#testing-quirks-in-development-in-docker-based-install

Perrudi, 2022-01-10:
Ok, thanks again. But a "little" to complicated / risky for me. I am using docker + portainer on another machine, but for pi-hole only. So i will wait for the official release to have all this included and for now i let this until here functionally working with your zip-file and the 2 lines in the yaml. :)

JohndeWaal, 2022-01-22:
https://github.com/zigpy/zha-device-handlers/issues/1027#issuecomment-1008433665
> With these informations i was able to identify some functions and rename the equivalent pieces:
> 
> .... it seams to work fundamentally until here...

This looks great, better then the downloadable .py files I tried from this issue. Could you (Perrudi) share your file?

FrKii, 2022-01-22:
@rforro thanks for the work you put into this

For me the "Set Temperature" keeps resetting in HA but the TRV itself keeps the temperature.
I'll look into this a bit more once I have some more time.

[image]

Perrudi, 2022-01-22:
> [#1027 (comment)](https://github.com/zigpy/zha-device-handlers/issues/1027#issuecomment-1008433665)
> 
> > With these informations i was able to identify some functions and rename the equivalent pieces:
> > .... it seams to work fundamentally until here...
> 
> This looks great, better then the downloadable .py files I tried from this issue. Could you (Perrudi) share your file?

This is my file, same as shared a few posts above, unchanged (zipped to be able to post it here, no password):
[valve.py.zip](https://github.com/zigpy/zha-device-handlers/files/7919284/valve.py.zip)

neurove, 2022-01-26:
I have the same TRV from AliExpress and it is working with the files from rforro (with the climate.py ident fix).
Thanks @rforro and @Perrudi.
Hope the PR comes soon and makes it to the master tree :)

ln-12, 2022-02-06:
In case anyone is stuck with `don't forget to add into /components/zha/climate.py following lines` from [this comment](https://github.com/zigpy/zha-device-handlers/issues/1027#issuecomment-1008310052), you can perform the following steps (if you know what you are doing):
1. Install the ssh addon: https://github.com/hassio-addons/addon-ssh
2. Disable protection mode from the addon UI and (re)start the addon
3. Open the ssh terminal and enter the following commands:
- `docker exec  -it homeassistant /bin/bash` to connect to the home assistant docker container
- `find / -name "climate.py" | grep "components/zha/climate.py"` to find the needed file, if you do not have exactly one line as output, something went wrong, e.g. you are not in the docker container
- `vi /usr/src/homeassistant/homeassistant/components/zha/climate.py` to open the file in vim
- `SHIFT+G` to go to the end of the file
- `I` (big i) to switch to insert mode
- `CTRL+V` to paste the following content (**CAUTION**, do not copy directly from [the comment above](https://github.com/zigpy/zha-device-handlers/issues/1027#issuecomment-1008310052) as there are some minor formatting/indent errors which break your config!):
```
@STRICT_MATCH(
    channel_names=CHANNEL_THERMOSTAT,
    manufacturers={
        "_TZE200_hue3yfsn",
    },
)

class ZONNSMARTThermostat(Thermostat):
    """ZONNSMART Thermostat implementation"""

    PRESET_HOLIDAY = "holiday"
    PRESET_FROST = "frost protect"

    def __init__(self, unique_id, zha_device, channels, **kwargs):
        """Initialize ZHA Thermostat instance."""
        super().__init__(unique_id, zha_device, channels, **kwargs)
        self._presets = [
            PRESET_NONE,
            self.PRESET_HOLIDAY,
            PRESET_SCHEDULE,
            self.PRESET_FROST,
        ]
        self._supported_flags |= SUPPORT_PRESET_MODE

    async def async_attribute_updated(self, record):
        """Handle attribute update from device."""
        if record.attr_name == "operation_preset":
            if record.value == 0:
                self._preset = PRESET_SCHEDULE
            if record.value == 1:
                self._preset = PRESET_NONE
            if record.value == 3:
                self._preset = self.PRESET_HOLIDAY
            if record.value == 4:
                self._preset = self.PRESET_FROST
        await super().async_attribute_updated(record)

    async def async_preset_handler(self, preset: str, enable: bool = False) -> bool:
        """Set the preset mode."""
        mfg_code = self._zha_device.manufacturer_code
        if not enable:
            return await self._thrm.write_attributes(
                {"operation_preset": 1}, manufacturer=mfg_code
            )
        if preset == PRESET_SCHEDULE:
            return await self._thrm.write_attributes(
                {"operation_preset": 0}, manufacturer=mfg_code
            )
        if preset == self.PRESET_HOLIDAY:
            return await self._thrm.write_attributes(
                {"operation_preset": 3}, manufacturer=mfg_code
            )
        if preset == self.PRESET_FROST:
            return await self._thrm.write_attributes(
                {"operation_preset": 4}, manufacturer=mfg_code
            )
        return False
```
- `ESC` to switch back to normal mode
- `:wq` to write changes and quit vim
- `exit` to exit the docker container
4. Enable protection mode in the ssh addon UI, (optionally) disable the addon
5. Restart home assistant
6. Profit

dodekaphilist, 2022-02-06:
Some follow-up: If you are running Home Assistant as a container, you need to ssh into your machine and run the docker command with sudo. Everything in the instructions related to the addon can be ignored.

Perrudi, 2022-02-07:
Ok, you warned us: '(if you know what you are doing)'
I am not used to VIM and all these bin-bash-things, but i want to help to bring this forward. So joking aside, gone through step by step and obviously worked. Only at ':wq' I stumbled, but with ESC then single :w (write?) and :q (quit?) I guess it worked. Restarted HA and it is back as before. I will try to see, if anything in lovelace or behavior of the 'thermostat-card' changed.
Thank you.

ln-12, 2022-02-07:
Thanks @rforro for providing the files. I tested the setup today and can confirm that it almost works perfectly. The only thing that seems broken is the current temperature listed in `climate.tze200_hue3yfsn_ts0601_63162ffe_thermostat`. It is set when first connecting the thermostat to HA but never updates after that. The heating regulation is not effected and works as intended, so I think that the updated value is simply not sent to HA. 

Does anyone else also have this issue/is this a known problem?

alexd321, 2022-02-07:
> Thanks @rforro for providing the files. I tested the setup today and can confirm that it almost works perfectly. The only thing that seems broken is the current temperature listed in [...]

known issue, see [here](https://github.com/zigpy/zha-device-handlers/issues/1027#issuecomment-962854799) for explanation

and first example [here](https://github.com/zigpy/zha-device-handlers/issues/1027#issuecomment-964946484) for a work-around in home assistant

neurove, 2022-02-22:
The new code is running well with Homeassistant 2022.02.09.
Thanks @rforro

jirka1022, 2022-02-23:
Hello guys, I have same SMART THERMOSTAT (TV02 Zigbee) mentioned in the beginning. But I am beginner in all this stuff around HA and implementations. Is there anybody who is able to help me step by step with this. I will really appreciate it. THANK YOU SO MUCH FOR YOUR HELP

Perrudi, 2022-02-24:
> 

It's all mentioned above. Short version: Take the last ts0601_trv.zip and put it into homeassistant (in homeassistant included file-editor) and do the steps done in the posts above: https://github.com/zigpy/zha-device-handlers/issues/1027#issuecomment-1008350291 and https://github.com/zigpy/zha-device-handlers/issues/1027#issuecomment-1030847151

Or describe the specific step, you are stuck in.

rforro, 2022-02-24:
> Hello guys, I have same SMART THERMOSTAT (TV02 Zigbee) mentioned in the beginning. But I am beginner in all this stuff around HA and implementations. Is there anybody who is able to help me step by step with this. I will really appreciate it. THANK YOU SO MUCH FOR YOUR HELP

Just wait a month. This code was already merged and soon will be included in one of next releases of HA. Probably not in March but April for sure.

21stce, 2022-03-03:
I have this device https://aliexpress.com/item/1005003172662104.html, the manual says it is a TV02 Zigbee Thermostat Radiator Valve, but I can't get it to pair via zigbee2mqtt 1.23.0 (other devices, i.e. switches do work). Why would it work via this method?  

The unit is sold as if it only works with a Tuya gateway. Is that the case?  If pairing should always work, perhaps the hardware is just broken.

rforro, 2022-03-07:
This quirk has made it into 2022.03 so you don't need to create custom_quirks anymore. But it's not all roses yet, present are still waiting for its release and I think it will be 2022.04

jirka1022, 2022-03-07:
Version 2022.03...Now I Can see my thermostat, actual temperature is ok,but wanted temperatures is 5time bigger...eg. 110 is like 22°C. Thank you for your help.

rforro, 2022-03-07:
> Version 2022.03...Now I Can see my thermostat, actual temperature is ok,but wanted temperatures is 5time bigger...eg. 110 is like 22°C. Thank you for your help.

Can you post some printscreens and as well make a photo of your TRV please.

jirka1022, 2022-03-07:
[image]

jirka1022, 2022-03-07:
[image]

[58 more replies]

closed 2022-04-12 by dmulcahey