zigpy/zha-device-handlers #1032: [Device Support Request] Lockwood Zigbee Digital Deadbolt

Tenwaterview, 2021-09-12:
**Is your feature request related to a problem? Please describe.**
The Lockwood digital deadbolt is the Australian equivalent of the Yale assure deadbolt however zha isn’t recognising it as a lock

**Describe the solution you'd like**
Recognise the device with equivalent sensors to the Yale assure

**Device signature - this can be acquired by removing the device from ZHA and pairing it again from the add devices screen. Be sure to add the entire content of the log panel after pairing the device to a code block below this line.**

1
2
3
4
5
6
7
8
9
10
11
12
13
14
15
16
17
18
19
20
21
22
23
24
25
26
{
  "node_descriptor": "NodeDescriptor(logical_type=<LogicalType.EndDevice: 2>, complex_descriptor_available=0, user_descriptor_available=0, reserved=0, aps_flags=0, frequency_band=<FrequencyBand.Freq2400MHz: 8>, mac_capability_flags=<MACCapabilityFlags.AllocateAddress: 128>, manufacturer_code=4125, maximum_buffer_size=82, maximum_incoming_transfer_size=255, server_mask=0, maximum_outgoing_transfer_size=255, descriptor_capability_field=<DescriptorCapability.NONE: 0>, *allocate_address=True, *is_alternate_pan_coordinator=False, *is_coordinator=False, *is_end_device=True, *is_full_function_device=False, *is_mains_powered=False, *is_receiver_on_when_idle=False, *is_router=False, *is_security_capable=False)",
  "endpoints": {
    "1": {
      "profile_id": 260,
      "device_type": "0x000a",
      "in_clusters": [
        "0x0000",
        "0x0001",
        "0x0003",
        "0x0009",
        "0x000a",
        "0x0020",
        "0x0101",
        "0x0b05"
      ],
      "out_clusters": [
        "0x000a",
        "0x0019"
      ]
    }
  },
  "manufacturer": "Yale",
  "model": "NO LOCK PRESENT",
  "class": "zigpy.device.Device"

**Additional context**
Add any other context or screenshots about the feature request here.

this may be useful background. Where someone got the equivalent device recognised as Yale assure lock on z2m

Yes, I got it working with my HA and cc2531 via Zigbee2mqtt. You will need to add a device profile for it. This is what you will need to do:

I have created a pastebin for the device profile, get it from here: https://pastebin.com/p0Vw4Dak
Save it as YDD-D4F0-TSDB.js under \share\zigbee2mqtt in your HA device
In HA, go to Supervisor > Dashboard - Addons > Zigbee2mqtt > Configuration and add (or update if you already have the entry) the following to point to the above file:
external_converters:
- YDD-D4F0-TSDB.js

Restart Z2M and HA and it should appear in Z2M as supported device and in HA as lock and you should be able to lock/unlock it.

MattWestb, 2021-09-12:
Your device looks like it dont have reading the model info OK but then looking on all Assa/Abloy (and all under names of them) locks its more or less the same and is pity zigbee standard.

You can trying changing this line:https://github.com/zigpy/zha-device-handlers/blob/fbafa19c922854c807861ec37314c8702f3b08c2/zhaquirks/yale/realliving.py#L26
To some think like this:
```
        MODELS_INFO: [
            ("Yale", "YRD210 PB DB"),
            ("Yale", "YRL220 TS LL"),
            ("Yale", "YDD-D4F0 TSDB"),
        ],
```
Configure local quirks in ZHA and saving the new `realliving.py` in the folder you have made for local quirks and restarting HA.
If its loading you is getting the name instead of the general class on the device card.
Now you is having` "class": "zigpy.device.Device"` = no quirk is being loaded for the device.
If ZHA is not loading try deleting it from the device card and waiting 2 minutes and then adding it a gen and look if the quirk have being loaded.

closed 2022-03-18 by github-actions[bot]

mopk47, 2022-05-08:
Hi All, is there any easier way to configure local quirks in ZHA? The following page is just not for a normal user. Wish I won't be forced to ZigbeeMQTT just for this lock.
https://github.com/zigpy/zha-device-handlers#testing-quirks-in-development-in-docker-based-install

MattWestb, 2022-05-08:
https://github.com/zigpy/zha-device-handlers/discussions/693#discussioncomment-857274