zigpy/zha-device-handlers #1033: [Device Support Request] MOES 2 gang  dimmer TS0601 support

Barmatuhin, 2021-09-12:
Have few 2gang dimmers from AliExpress   -https://star.aliexpress.com/share/share.htm?image=Ub93cd2f77da040b8955d0e579ebe4aa9D.jpg&businessType=ProductDetail&title=%E2%82%AC%209%2C44%20%2046%EF%BC%85%20Off%20%7C%20Tuya%20Zigbee%203.0%20Dimmer%20Smart%20Switch%20Module%20Controller%202%20Way%20Remote%20Control%20Smart%20Light%20Switch%20Relay%20Google%20Assistant%20Alexa%20Echo&platform=AE&redirectUrl=https%3A%2F%2Fwww.aliexpress.com%2Fitem%2F1005002662854725.html%3F%26srcSns%3Dsns_More%26tid%3Dwhite_backgroup_101%26mb%3DdPMGv6gO7LDGRlj%26businessType%3DProductDetail%26spreadType%3DsocialShare

**Modify ts0601_dimmer.py for support 2gang dimmer**
Add support for 2 gang dimmer.

**Device signature.**

{
  "node_descriptor": "NodeDescriptor(logical_type=<LogicalType.Router: 1>, complex_descriptor_available=0, user_descriptor_available=0, reserved=0, aps_flags=0, frequency_band=<FrequencyBand.Freq2400MHz: 8>, mac_capability_flags=<MACCapabilityFlags.AllocateAddress|RxOnWhenIdle|MainsPowered|FullFunctionDevice: 142>, manufacturer_code=4098, maximum_buffer_size=82, maximum_incoming_transfer_size=82, server_mask=11264, maximum_outgoing_transfer_size=82, descriptor_capability_field=<DescriptorCapability.NONE: 0>, *allocate_address=True, *is_alternate_pan_coordinator=False, *is_coordinator=False, *is_end_device=False, *is_full_function_device=True, *is_mains_powered=True, *is_receiver_on_when_idle=True, *is_router=True, *is_security_capable=False)",
  "endpoints": {
    "1": {
      "profile_id": 260,
      "device_type": "0x0100",
      "in_clusters": [
        "0x0000",
        "0x0004",
        "0x0005",
        "0x0006",
        "0x0008",
        "0xef00"
      ],
      "out_clusters": [
        "0x000a",
        "0x0019"
      ]
    }
  },
  "manufacturer": "_TZE200_e3oitdyu",
  "model": "TS0601",
  "class": "zhaquirks.tuya.ts0601_dimmer.TuyaSingleSwitchDimmer"
}

Adminiuga, 2021-09-12:
It's already got a quirk.what does not work?

Barmatuhin, 2021-09-13:
It's only one Entity in list for my 2 gang dimmer. How can i control first and second gang?

Adminiuga, 2021-09-13:
You don't control the gangs. The gangs control HA using zha_event or devices triggers. These are remotes, therefore don't have entities (other than the batt) and only send events.

Barmatuhin, 2021-09-13:
It's an end-device, not a remote, which connected to AC and to bulbs 
[image]

Adminiuga, 2021-09-13:
Looks like the other quirk has a matching signature too. Would need some investigation

Barmatuhin, 2021-09-13:
This device not recognized at all from the beginnings, so i locally edit zhaquirks/tuya/ts0601_dimmer.py  and add my _TZE200_e3oitdyu in MODELS_INFO. So at least i got control of one of two entities. But i'm not so experienced, and can't understand how to extend this quirk for second one (entity, gang...)

Adminiuga, 2021-09-13:
This kind of information is worth pointing out in the OP. 
No quirk and incorrectly working quirk are two different things.
Remove you custom quirk and post ZigBee device signature without any quirks.

Barmatuhin, 2021-09-13:
Ok, will do this later at the evening. Please don't close the request.

Barmatuhin, 2021-09-13:
Update first comment

ksga, 2021-10-30:
Zigbee2mqtt have support.
From https://github.com/Koenkk/zigbee-herdsman-converters/blob/master/devices/moes.js
````
    {
        fingerprint: [{modelID: 'TS0601', manufacturerName: '_TZE200_e3oitdyu'}],
        model: 'MS-105B',
        vendor: 'Moes',
        description: 'Smart dimmer module (2 gang)',
        fromZigbee: [fz.moes_105_dimmer, fz.ignore_basic_report],
        toZigbee: [tz.moes_105_dimmer],
        meta: {turnsOffAtBrightness1: true, multiEndpoint: true},
        configure: async (device, coordinatorEndpoint, logger) => {
            await reporting.bind(device.getEndpoint(1), coordinatorEndpoint, ['genOnOff', 'genLevelCtrl']);
            if (device.getEndpoint(2)) await reporting.bind(device.getEndpoint(2), coordinatorEndpoint, ['genOnOff']);
        },
        exposes: [e.light_brightness().withEndpoint('l1').setAccess('state', ea.STATE_SET).setAccess('brightness', ea.STATE_SET),
            e.light_brightness().withEndpoint('l2').setAccess('state', ea.STATE_SET).setAccess('brightness', ea.STATE_SET)],
        endpoint: (device) => {
            return {'l1': 1, 'l2': 1};
        },
    },
````

Barmatuhin, 2021-10-31:
Is it possible to implement it in ZHA? Z2M feels more complex and not so "dummies friendly" :)

ksga, 2021-10-31:
Sure hope so - but I think it's above my pay grade 😔
Was using z2m previously, but made the switch to zha after starting a fresh ha setup

Barmatuhin, 2021-11-08:
@javicalle according our discussion here (https://github.com/zigpy/zha-device-handlers/issues/1038#issuecomment-962601999) 

> I can give a try, but I will need to have the MCU commands.
> Can you activate the logs zigpy.tuya: debug and post your traces and related commands (on, off, dimm_up, dimm_down, > whatever...)?

When i pair this dimmer it don't catch any custom quirk, just pairs without any entities. Do you want me to put this MODEL_INFO (_TZE200_e3oitdyu) into your custom quirk (ts0601.py) and try to pair once again. And after that make some commands (on, off, dimm)?

ksga, 2021-11-15:
Enabled ZHA debugging in logger, created a custom version of the TS0601_dimmer.py with TZE200_e3oitdyu added to models.

The log starts with me removing the device - and after a while it's repaired.
I've tried to clean out anything not related to the Moes device from the logs, but there might be some irrelevant entries still.
https://pastebin.com/AeBkjQ77 

Hope there's some useful info :)

ksga, 2021-11-16:
A bit more info:
When toggling the single light switch that is enabled when adding the device to the quirk, this is the output:
https://pastebin.com/hhVQ5YAB
Toggling does not turn on the switch output (at least not gang no 1)

To folllow - once in a while the device gives a short "blink" on the connected lamp...

lowefraser, 2021-11-17:
I have the same 2 gang dimmer and happy to help with testing.

javicalle, 2021-11-20:
Hi there,
 once PR #1153 has been merged I believe that can be adapted/extended to this case.

But before I can suggest anything, there are a few things I need to clarify or review to get the full picture.

1. How do `level_event` differ between gangs?
In the case of `switch_event`, the `command_id` values are different and the relationship has been established as:
`tuya_payload.command_id - 0x0100 = endpoint_id`
In the case of level events, is it the same? Is there a similar relationship?

2. Are the `INPUT_CLUSTERS` of the` ts0601_dimmer.py` correct?
I may be wrong, but `LevelControl` and `TuyaLevelControl` define the same cluster (having the same `cluster_id`) and the same would happen with `TuyaManufacturerClusterOnOff` and `TuyaManufacturerLevelControl`.

Would someone be so kind as to check if the quirks works anyway with the following definition?:

```python
class TuyaDoubleSwitchDimmer(TuyaDimmerSwitch):
    """Tuya double touch switch device."""

    signature = {
        MODELS_INFO: [
            ("_TZE200_e3oitdyu", "TS0601"),
        ],
        ENDPOINTS: {
            1: {
                PROFILE_ID: zha.PROFILE_ID,
                DEVICE_TYPE: zha.DeviceType.SMART_PLUG,
                INPUT_CLUSTERS: [
                    Basic.cluster_id,
                    Groups.cluster_id,
                    Scenes.cluster_id,
                    TuyaManufCluster.cluster_id,
                ],
                OUTPUT_CLUSTERS: [Time.cluster_id, Ota.cluster_id],
            }
        },
    }

    replacement = {
        ENDPOINTS: {
            1: {
                DEVICE_TYPE: zha.DeviceType.ON_OFF_LIGHT,
                INPUT_CLUSTERS: [
                    Basic.cluster_id,
                    Groups.cluster_id,
                    Scenes.cluster_id,
                    TuyaManufacturerLevelControl,
                    TuyaOnOff,
                    TuyaLevelControl,
                ],
                OUTPUT_CLUSTERS: [Time.cluster_id, Ota.cluster_id],
            },
            2: {
                PROFILE_ID: zha.PROFILE_ID,
                DEVICE_TYPE: zha.DeviceType.ON_OFF_LIGHT,
                INPUT_CLUSTERS: [
                    TuyaOnOff,
                ],
                OUTPUT_CLUSTERS: [],
            }
        }
    }
```

Apart from verifying if the clusters are necessary, with this configuration a second switch should be enabled in HA (only on/off at the moment) that could operate the device.
I'm sure, that changes from the physical device (if that is possible) will not update correctly in HA.

ksga, 2021-11-22:
> 

Replaced __init.py__ in the TUYA folder with the newest in the repo. Cleared cache and replaced the class in my test-py.
Deleted the device from ZHA and readded.
The log output from the pairing is here:
```
[0x4665:1:0x0008] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=GLOBAL_COMMAND manufacturer_specific=False is_reply=True disable_default_response=True> manufacturer=None tsn=41 command_id=Command.Read_Attributes_rsp>
[0x4665:1:0x0008]: received attribute: 0 update with value: 254
[0xc448:1:0x0008] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=GLOBAL_COMMAND manufacturer_specific=False is_reply=True disable_default_response=True> manufacturer=None tsn=42 command_id=Command.Read_Attributes_rsp>
[0xC448:1:0x0008]: received attribute: 0 update with value: 254
[0xc225:1:0x0008] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=GLOBAL_COMMAND manufacturer_specific=False is_reply=True disable_default_response=True> manufacturer=None tsn=43 command_id=Command.Read_Attributes_rsp>
[0xC225:1:0x0008]: received attribute: 0 update with value: 254
[0x0b40:1:0x0008] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=GLOBAL_COMMAND manufacturer_specific=False is_reply=True disable_default_response=True> manufacturer=None tsn=44 command_id=Command.Read_Attributes_rsp>
[0x0B40:1:0x0008]: received attribute: 0 update with value: 254
[0x55df:1:0x0008] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=GLOBAL_COMMAND manufacturer_specific=False is_reply=True disable_default_response=True> manufacturer=None tsn=45 command_id=Command.Read_Attributes_rsp>
[0x55DF:1:0x0008]: received attribute: 0 update with value: 191
[0xf822:1:0x0008] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=GLOBAL_COMMAND manufacturer_specific=False is_reply=True disable_default_response=True> manufacturer=None tsn=40 command_id=Command.Read_Attributes_rsp>
[0xF822:1:0x0008]: received attribute: 0 update with value: 254
[0xc448:1:0x0300] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=GLOBAL_COMMAND manufacturer_specific=False is_reply=True disable_default_response=True> manufacturer=None tsn=48 command_id=Command.Read_Attributes_rsp>
New device 0x5d4a (84:fd:27:ff:fe:60:95:9d) joined the network
[0x5d4a] Scheduling initialization
Received frame on uninitialized device <Device model=None manuf=None nwk=0x5D4A ieee=84:fd:27:ff:fe:60:95:9d is_initialized=False> from ep 0 to ep 0, cluster ZDOCmd.Device_annce: b'\xFF\x4A\x5D\x9D\x95\x60\xFE\xFF\x27\xFD\x84\x8E'
[0x5d4a:zdo] ZDO request ZDOCmd.Device_annce: [0x5D4A, 84:fd:27:ff:fe:60:95:9d, 142]
Tries remaining: 3
[0x5d4a] Requesting 'Node Descriptor'
Tries remaining: 2
[0x5d4a] Extending timeout for 0x31 request
Received frame on uninitialized device <Device model=None manuf=None nwk=0x5D4A ieee=84:fd:27:ff:fe:60:95:9d is_initialized=False> from ep 0 to ep 0, cluster ZDOCmd.Node_Desc_rsp: b'\x31\x00\x4A\x5D\x01\x40\x8E\x02\x10\x52\x52\x00\x00\x2C\x52\x00\x00'
[0x5d4a] Got Node Descriptor: NodeDescriptor(logical_type=<LogicalType.Router: 1>, complex_descriptor_available=0, user_descriptor_available=0, reserved=0, aps_flags=0, frequency_band=<FrequencyBand.Freq2400MHz: 8>, mac_capability_flags=<MACCapabilityFlags.AllocateAddress|RxOnWhenIdle|MainsPowered|FullFunctionDevice: 142>, manufacturer_code=4098, maximum_buffer_size=82, maximum_incoming_transfer_size=82, server_mask=11264, maximum_outgoing_transfer_size=82, descriptor_capability_field=<DescriptorCapability.NONE: 0>, *allocate_address=True, *is_alternate_pan_coordinator=False, *is_coordinator=False, *is_end_device=False, *is_full_function_device=True, *is_mains_powered=True, *is_receiver_on_when_idle=True, *is_router=True, *is_security_capable=False)
[0x5d4a] Discovering endpoints
Tries remaining: 3
Received frame on uninitialized device <Device model=None manuf=None nwk=0x5D4A ieee=84:fd:27:ff:fe:60:95:9d is_initialized=False> from ep 0 to ep 0, cluster ZDOCmd.Active_EP_rsp: b'\x32\x00\x4A\x5D\x01\x01'
[0x5d4a] Discovered endpoints: [1]
[0x5d4a] Initializing endpoints [<Endpoint id=1 in=[] out=[] status=<Status.NEW: 0>>]
[0x5d4a:1] Discovering endpoint information
Tries remaining: 3
Received frame on uninitialized device <Device model=None manuf=None nwk=0x5D4A ieee=84:fd:27:ff:fe:60:95:9d is_initialized=False> from ep 0 to ep 0, cluster ZDOCmd.Simple_Desc_rsp: b'\x33\x00\x4A\x5D\x14\x01\x04\x01\x51\x00\x01\x04\x00\x00\x04\x00\x05\x00\x00\xEF\x02\x19\x00\x0A\x00'
[0x5d4a:1] Discovered endpoint information: SizePrefixedSimpleDescriptor(endpoint=1, profile=260, device_type=81, device_version=1, input_clusters=[0, 4, 5, 61184], output_clusters=[25, 10])
Unknown cluster 61184
[0x5d4a:1:0x0000] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=GLOBAL_COMMAND manufacturer_specific=False is_reply=True disable_default_response=True> manufacturer=None tsn=52 command_id=Command.Read_Attributes_rsp>
[0x5d4a] Read model 'TS0601' and manufacturer '_TZE200_e3oitdyu' from <Endpoint id=1 in=[basic:0x0000, groups:0x0004, scenes:0x0005, None:0xEF00] out=[ota:0x0019, time:0x000A] status=<Status.ZDO_INIT: 1>>
[0x5d4a] Discovered basic device information for <Device model='TS0601' manuf='_TZE200_e3oitdyu' nwk=0x5D4A ieee=84:fd:27:ff:fe:60:95:9d is_initialized=True>
Device is initialized <Device model='TS0601' manuf='_TZE200_e3oitdyu' nwk=0x5D4A ieee=84:fd:27:ff:fe:60:95:9d is_initialized=True>
[... cut ...]
[0x5d4a:1:0x0006] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=GLOBAL_COMMAND manufacturer_specific=False is_reply=True disable_default_response=True> manufacturer=None tsn=65 command_id=Command.Read_Attributes_rsp>
[0x5d4a:1:0x0008] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=GLOBAL_COMMAND manufacturer_specific=False is_reply=True disable_default_response=True> manufacturer=None tsn=67 command_id=Command.Read_Attributes_rsp>
```
The device now has two entities, but none are working:
[image]

javicalle, 2021-11-22:
@ksga Thanks for testing

What type of device is it? I would bet that it is a dimmer with 2 channels, but there is a type of command that I cannot imagine what it can be. Knowing the capabilities of the device can help me to identify the commands.

If you would be so kind as to test the following changes:
in your tuya __init_.py replace the `` class with this one
```
class TuyaManufacturerLevelControl(TuyaManufCluster):
    """Manufacturer Specific Cluster for cover device."""

    def handle_cluster_request(
        self,
        hdr: foundation.ZCLHeader,
        args: Tuple[TuyaManufCluster.Command],
        *,
        dst_addressing: Optional[
            Union[t.Addressing.Group, t.Addressing.IEEE, t.Addressing.NWK]
        ] = None,
    ) -> None:
        """Handle cluster request."""
		tuya_payload = args[0]

        _LOGGER.debug(
            "%s Received Attribute Report. Command is %x, Tuya Paylod values"
            "[Status : %s, TSN: %s, Command: %s, Function: %s, Data: %s]",
            self.endpoint.device.ieee,
            hdr.command_id,
            tuya_payload.status,
            tuya_payload.tsn,
            tuya_payload.command_id,
            tuya_payload.function,
            tuya_payload.data,
        )

        if hdr.command_id in (0x0002, 0x0001):
            # Send default response because the MCU expects it
            if not hdr.frame_control.disable_default_response:
                self.send_default_rsp(hdr, status=foundation.Status.SUCCESS)
			
            if tuya_payload.command_id == TUYA_LEVEL_COMMAND:
                self.endpoint.device.dimmer_bus.listener_event(
                    LEVEL_EVENT,
                    tuya_payload.command_id,
                    tuya_payload.data,
                )
            else:
                self.endpoint.device.switch_bus.listener_event(
                    SWITCH_EVENT,
                    tuya_payload.command_id - TUYA_CMD_BASE,
                    tuya_payload.data[1],
                )
		elif hdr.command_id == TUYA_SET_TIME:
            """Time event call super"""
            _LOGGER.debug("TUYA_SET_TIME --> hdr: %s, args: %s", hdr, args)
            super().handle_cluster_request(hdr, args, dst_addressing=dst_addressing)
        else:
            _LOGGER.warning("Unsupported command: %s", hdr)

```

In your custom quirk `TuyaDoubleSwitchDimmer`:
```
class TuyaDoubleSwitchDimmer(TuyaDimmerSwitch):
    """Tuya double touch switch device."""

    signature = {
        MODELS_INFO: [
            ("_TZE200_e3oitdyu", "TS0601"),
        ],
        ENDPOINTS: {
            1: {
                PROFILE_ID: zha.PROFILE_ID,
                DEVICE_TYPE: zha.DeviceType.SMART_PLUG,
                INPUT_CLUSTERS: [
                    Basic.cluster_id,
                    Groups.cluster_id,
                    Scenes.cluster_id,
                    TuyaManufCluster.cluster_id,
                ],
                OUTPUT_CLUSTERS: [Time.cluster_id, Ota.cluster_id],
            }
        },
    }

    replacement = {
        ENDPOINTS: {
            1: {
                DEVICE_TYPE: zha.DeviceType.ON_OFF_LIGHT,
                INPUT_CLUSTERS: [
                    Basic.cluster_id,
                    Groups.cluster_id,
                    Scenes.cluster_id,
                    TuyaManufacturerLevelControl,
                    TuyaOnOff,
                    TuyaLevelControl,
                ],
                OUTPUT_CLUSTERS: [Time.cluster_id, Ota.cluster_id],
            },
            7: {
                PROFILE_ID: zha.PROFILE_ID,
                DEVICE_TYPE: zha.DeviceType.ON_OFF_LIGHT,
                INPUT_CLUSTERS: [
                    TuyaOnOff,
                ],
                OUTPUT_CLUSTERS: [],
            }
        }
    }
```
I believe that this device publish the gang 2 on/off events with command 7.

It would also be useful if you can activate the traces of your custom quirks. If I'm not wrong it would be:
```yaml
logger:
   logs:
     ts0601_dimmer_test: info
```

In addition to the pairing traces (I don't know if they will be different), it would be very useful for me to identify the traces from different actions (turn on, turn off, dimmer, ...)

ksga, 2021-11-23:
@javicalle Thanks for making the effort 👍

First of, the device should be a 2-way dimmer switch: https://www.moeshouse.com/collections/smart-dimmer-switch/products/mini-diy-tuya-zigbee-3-0-smart-dimmer-switch-relay-module-1-2-way-2-gang-tuya-zigbee3-0-hub-required-smart-life-tuya-app-works-with-alexa-google-home-voice-control

I'll make the edits and post the logs a bit late.

ksga, 2021-11-23:
Made the changes, cleared py cache and now gang 1 is working:
[image]
Gang 2 is greyed out...

Logs (pretty much cleaned from other stuff):
https://pastebin.com/RNGp64eU

The device address is 84:fd:27:ff:fe:60:95:9d if it helps :)

ksga, 2021-11-23:
Sorry - left over device from earlier, so deleted and rejoined.
Control elements now look like this:
[image]
Both are linked to gang 1

Log from pairing:
```
New device 0xb3be (84:fd:27:ff:fe:60:95:9d) joined the network
[0xb3be] Scheduling initialization
Received frame on uninitialized device <Device model=None manuf=None nwk=0xB3BE ieee=84:fd:27:ff:fe:60:95:9d is_initialized=False> from ep 0 to ep 0, cluster ZDOCmd.Device_annce: b'\xFF\xBE\xB3\x9D\x95\x60\xFE\xFF\x27\xFD\x84\x8E'
[0xb3be:zdo] ZDO request ZDOCmd.Device_annce: [0xB3BE, 84:fd:27:ff:fe:60:95:9d, 142]
Tries remaining: 3
[0xb3be] Requesting 'Node Descriptor'
Tries remaining: 2
[0xb3be] Extending timeout for 0x3d request
Received frame on uninitialized device <Device model=None manuf=None nwk=0xB3BE ieee=84:fd:27:ff:fe:60:95:9d is_initialized=False> from ep 0 to ep 0, cluster ZDOCmd.Node_Desc_rsp: b'\x3D\x00\xBE\xB3\x01\x40\x8E\x02\x10\x52\x52\x00\x00\x2C\x52\x00\x00'
[0xb3be] Got Node Descriptor: NodeDescriptor(logical_type=<LogicalType.Router: 1>, complex_descriptor_available=0, user_descriptor_available=0, reserved=0, aps_flags=0, frequency_band=<FrequencyBand.Freq2400MHz: 8>, mac_capability_flags=<MACCapabilityFlags.AllocateAddress|RxOnWhenIdle|MainsPowered|FullFunctionDevice: 142>, manufacturer_code=4098, maximum_buffer_size=82, maximum_incoming_transfer_size=82, server_mask=11264, maximum_outgoing_transfer_size=82, descriptor_capability_field=<DescriptorCapability.NONE: 0>, *allocate_address=True, *is_alternate_pan_coordinator=False, *is_coordinator=False, *is_end_device=False, *is_full_function_device=True, *is_mains_powered=True, *is_receiver_on_when_idle=True, *is_router=True, *is_security_capable=False)
[0xb3be] Discovering endpoints
Tries remaining: 3
Received frame on uninitialized device <Device model=None manuf=None nwk=0xB3BE ieee=84:fd:27:ff:fe:60:95:9d is_initialized=False> from ep 0 to ep 0, cluster ZDOCmd.Active_EP_rsp: b'\x3E\x00\xBE\xB3\x01\x01'
[0xb3be] Discovered endpoints: [1]
[0xb3be] Initializing endpoints [<Endpoint id=1 in=[] out=[] status=<Status.NEW: 0>>]
[0xb3be:1] Discovering endpoint information
Tries remaining: 3
Received frame on uninitialized device <Device model=None manuf=None nwk=0xB3BE ieee=84:fd:27:ff:fe:60:95:9d is_initialized=False> from ep 0 to ep 0, cluster ZDOCmd.Simple_Desc_rsp: b'\x3F\x00\xBE\xB3\x14\x01\x04\x01\x51\x00\x01\x04\x00\x00\x04\x00\x05\x00\x00\xEF\x02\x19\x00\x0A\x00'
[0xb3be:1] Discovered endpoint information: SizePrefixedSimpleDescriptor(endpoint=1, profile=260, device_type=81, device_version=1, input_clusters=[0, 4, 5, 61184], output_clusters=[25, 10])
Unknown cluster 61184
[0xb3be:1:0x0000] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=GLOBAL_COMMAND manufacturer_specific=False is_reply=True disable_default_response=True> manufacturer=None tsn=64 command_id=Command.Read_Attributes_rsp>
[0xb3be] Read model 'TS0601' and manufacturer '_TZE200_e3oitdyu' from <Endpoint id=1 in=[basic:0x0000, groups:0x0004, scenes:0x0005, None:0xEF00] out=[ota:0x0019, time:0x000A] status=<Status.ZDO_INIT: 1>>
[0xb3be] Discovered basic device information for <Device model='TS0601' manuf='_TZE200_e3oitdyu' nwk=0xB3BE ieee=84:fd:27:ff:fe:60:95:9d is_initialized=True>
Device is initialized <Device model='TS0601' manuf='_TZE200_e3oitdyu' nwk=0xB3BE ieee=84:fd:27:ff:fe:60:95:9d is_initialized=True>
Checking quirks for _TZE200_e3oitdyu TS0601 (84:fd:27:ff:fe:60:95:9d)
Considering <class 'ts0601_dimmer_test.TuyaDoubleSwitchDimmer'>
Found custom device replacement for 84:fd:27:ff:fe:60:95:9d: <class 'ts0601_dimmer_test.TuyaDoubleSwitchDimmer'>
device - 0xB3BE:84:fd:27:ff:fe:60:95:9d entering async_device_initialized - is_new_join: True
device - 0xB3BE:84:fd:27:ff:fe:60:95:9d has joined the ZHA zigbee network
[0xB3BE](TS0601): started configuration
[0xB3BE:ZDO](TS0601): 'async_configure' stage succeeded
[0xB3BE:1:0x0000]: finished channel configuration
[0xB3BE:1:0x0019]: finished channel configuration
Error handling '_save_attribute' event with (84:fd:27:ff:fe:60:95:9d, 1, 0, 4, '_TZE200_e3oitdyu') params: FOREIGN KEY constraint failed
Error handling '_save_attribute' event with (84:fd:27:ff:fe:60:95:9d, 1, 0, 5, 'TS0601') params: FOREIGN KEY constraint failed
[0xB3BE:1:0x0006]: bound 'on_off' cluster: Status.SUCCESS
[0xB3BE:1:0x0008]: bound 'level' cluster: Status.SUCCESS
[0xB3BE:7:0x0006]: bound 'on_off' cluster: Status.SUCCESS
[0xb3be:1:0x0006] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=GLOBAL_COMMAND manufacturer_specific=False is_reply=True disable_default_response=True> manufacturer=None tsn=68 command_id=Command.Configure_Reporting_rsp>
[0xB3BE:1:0x0006]: Successfully configured reporting for 'set()' on 'on_off' cluster
[0xB3BE:1:0x0006]: Failed to configure reporting for '['on_off']' on 'on_off' cluster: [ConfigureReportingResponseRecord(status=134, direction=0, attrid=0)]
[0xB3BE:1:0x0006]: finished channel configuration
Error handling '_unsupported_attribute_added' event with (84:fd:27:ff:fe:60:95:9d, 1, 6, 0) params: FOREIGN KEY constraint failed
[0xb3be:1:0x0008] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=GLOBAL_COMMAND manufacturer_specific=False is_reply=True disable_default_response=True> manufacturer=None tsn=69 command_id=Command.Configure_Reporting_rsp>
[0xB3BE:1:0x0008]: Successfully configured reporting for 'set()' on 'level' cluster
[... cut ...]
[0x55df:1:0x0019] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=CLUSTER_COMMAND manufacturer_specific=False is_reply=False disable_default_response=False> manufacturer=None tsn=106 command_id=1>
[0x55df:1:0x0019] ZCL request 0x0001: [1, 4107, 2067, 49, 732]
[0x55df:1:0x0019] OTA query_next_image handler for 'Nordtronic A/S BoxDIMZ 98425031': field_control=1, manufacture_id=4107, image_type=2067, current_file_version=49, hardware_version=732
[0x55df:1:0x0019] No OTA image is available
```

javicalle, 2021-11-23:
> Both are linked to gang 1. 

Yes, that's expected and can be addressed.

> The "level, on_off" entity is in control, and the "on, off" seems to be broken.

Is the dimmer function working for device 1? In both directions (updating from HA and from the remote?

ksga, 2021-11-23:
> 

Dimmer is working yes. Only seems to actually do anything in the 25-75% range.
I don't have a remote for the device, but dimming using a physical switch (shorting L and S1 on the device) is reported in HA using roughly the same percentage range as mentioned above.

javicalle, 2021-11-23:
I really need you to activate the traces for your local quirk.
If your local file is named `ts0601_dimmer_test.py` set the log level to debug as `ts0601_dimmer_test: debug`. This can be done in your `configuration.yaml` or with the `logger.set_level` service.

If your tuya `__init__.py` file has been updates with the latest version from repo, the on/off events must be consistent in HA. Maybe the second switch would not work at all, but events would be isolates between switches.
This is the code that ensures that `switch_event`s aren't merged:
https://github.com/zigpy/zha-device-handlers/blob/fbee5ba7a378b42c0196b21ef42ebfc361af526a/zhaquirks/tuya/__init__.py#L466-L468

Also, there should be no repetitions in the commands either. In this case, the person responsible is this code:
https://github.com/zigpy/zha-device-handlers/blob/fbee5ba7a378b42c0196b21ef42ebfc361af526a/zhaquirks/tuya/__init__.py#L512-L515

This was replicated in the `TuyaManufacturerLevelControl`  above.

I have fixed the code example to take care also from `set_time` commands. Please update your local quirk with te code from https://github.com/zigpy/zha-device-handlers/issues/1033#issuecomment-975937379

I would appreciate it if after activating the traces you could send me the logs of the events:
* on/off of each switch (identifying event and switch)
* dim each switch indicating approximate values (brightest, medium, min)

This will help me distinguish the commands.

Until now I have identify:
| Command (dec) | Hex | Command |
| ------------- | ------------- | --- |
| 257 | 0x0101 | On/Off |
| 514 | 0x0202 |
| 515 | 0x0203 |
| 263 | 0x0107 | On/Off |
| 520 | 0x0208 |
| 521 | 0x0209 |

javicalle, 2021-11-23:
And, whats next?
There is a lot of work to do. 
* [ ] Figure what each command is. I bellieve that for each switch there are 3 commands:
   * on/off
   * dim
   * dim range (set the min&max dim values)
* [ ] Use the `command_bus` to communicate with the `tuya_manufacturer` (this must be easy)
* [ ] IMO it will be great to keep switch commands enclosed in endpoints, so we have to figure a way to be able to relate `command_id` with endpoint. In the `TuyaOnOff` I simplified it to `command_id = endpoint`, but it is clear that it will not be that simple here

Barmatuhin, 2021-11-25:
hello once again. i'm also implement changes in init.py and in custom quirk. So i've got two entites, first one with dimming options and second just on/off
[image]

here pairing log 
https://pastebin.com/i3aFCTPs

javicalle, 2021-11-26:
On your system, the device keeps sending retries because it doesn't get a response, and that shouldn't be the case:
```
[0xa52f:1:0xef00] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=CLUSTER_COMMAND manufacturer_specific=False is_reply=True disable_default_response=False> manufacturer=None tsn=83 command_id=2>
[0xa52f:1:0xef00] ZCL request 0x0002: [Command(status=0, tsn=143, command_id=257, function=0, data=[1, 0])]
[0xa52f:1:0xef00] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=CLUSTER_COMMAND manufacturer_specific=False is_reply=True disable_default_response=False> manufacturer=None tsn=83 command_id=2>
[0xa52f:1:0xef00] ZCL request 0x0002: [Command(status=0, tsn=143, command_id=257, function=0, data=[1, 0])]
[0xa52f:1:0xef00] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=CLUSTER_COMMAND manufacturer_specific=False is_reply=True disable_default_response=False> manufacturer=None tsn=83 command_id=2>
[0xa52f:1:0xef00] ZCL request 0x0002: [Command(status=0, tsn=143, command_id=257, function=0, data=[1, 0])]
[0xa52f:1:0xef00] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=CLUSTER_COMMAND manufacturer_specific=False is_reply=True disable_default_response=False> manufacturer=None tsn=83 command_id=2>
```

Please, can you verify that in the class `TuyaManufacturerLevelControl` you have this change (from [comment#975937379](https://github.com/zigpy/zha-device-handlers/issues/1033#issuecomment-975937379)):
```python
        if hdr.command_id in (0x0002, 0x0001):
            # Send default response because the MCU expects it
            if not hdr.frame_control.disable_default_response:
                self.send_default_rsp(hdr, status=foundation.Status.SUCCESS)
```

It would also be useful if you activated the traces for the packages:
* custom_zha_quirks
* TS0601_Moes2GangDimmer

I only see traces of zha and I am missing information

Barmatuhin, 2021-11-26:
Here this class from __init__.py

```python
class TuyaManufacturerLevelControl(TuyaManufCluster):
    """Manufacturer Specific Cluster for cover device."""

    def handle_cluster_request(
        self,
        hdr: foundation.ZCLHeader,
        args: Tuple[TuyaManufCluster.Command],
        *,
        dst_addressing: Optional[
            Union[t.Addressing.Group, t.Addressing.IEEE, t.Addressing.NWK]
        ] = None,
    ) -> None:
        """Handle cluster request."""
        tuya_payload = args[0]

        _LOGGER.debug(
            "%s Received Attribute Report. Command is %x, Tuya Paylod values"
            "[Status : %s, TSN: %s, Command: %s, Function: %s, Data: %s]",
            self.endpoint.device.ieee,
            hdr.command_id,
            tuya_payload.status,
            tuya_payload.tsn,
            tuya_payload.command_id,
            tuya_payload.function,
            tuya_payload.data,
        )

        if hdr.command_id in (0x0002, 0x0001):
            if tuya_payload.command_id == TUYA_LEVEL_COMMAND:
                self.endpoint.device.dimmer_bus.listener_event(
                    LEVEL_EVENT,
                    tuya_payload.command_id,
                    tuya_payload.data,
                )
            else:
                self.endpoint.device.switch_bus.listener_event(
                    SWITCH_EVENT,
                    tuya_payload.command_id - TUYA_CMD_BASE,
                    tuya_payload.data[1],
                )
```
i add this extra IF, but have the same second on/off entity without level
here log (sorry, htere can be extra records) - https://pastebin.com/9VZLfETj

javicalle, 2021-11-26:
Yes that's correct, I haven't fixed the `TuyaLevelControl` for the second switch yet (I am working on it), but for now we have fixed the retries.
Please, change the whole class to the one from the previous comment, this will fix also the command for `TUYA_SET_TIME`.

From your logs I can see a few `move_to_level_with_on_off`. Could you tell me that they correspond to approximate values?

Barmatuhin, 2021-11-26:
Now my class looks like that, is it correct? 
Can you explaint in other words, what it mean "fix also the command for TUYA_SET_TIME"? All command and response looking good, when i switch on in dashboard all switches good, light turn on reasonably and quickly, without any lag.

```python
class TuyaManufacturerLevelControl(TuyaManufCluster):
    """Manufacturer Specific Cluster for cover device."""

    def handle_cluster_request(
        self,
        hdr: foundation.ZCLHeader,
        args: Tuple[TuyaManufCluster.Command],
        *,
        dst_addressing: Optional[
            Union[t.Addressing.Group, t.Addressing.IEEE, t.Addressing.NWK]
        ] = None,
    ) -> None:
        """Handle cluster request."""
        tuya_payload = args[0]

        _LOGGER.debug(
            "%s Received Attribute Report. Command is %x, Tuya Paylod values"
            "[Status : %s, TSN: %s, Command: %s, Function: %s, Data: %s]",
            self.endpoint.device.ieee,
            hdr.command_id,
            tuya_payload.status,
            tuya_payload.tsn,
            tuya_payload.command_id,
            tuya_payload.function,
            tuya_payload.data,
        )

        if hdr.command_id in (0x0002, 0x0001):
			# Send default response because the MCU expects it
            if not hdr.frame_control.disable_default_response:
                self.send_default_rsp(hdr, status=foundation.Status.SUCCESS)
            if tuya_payload.command_id == TUYA_LEVEL_COMMAND:
                self.endpoint.device.dimmer_bus.listener_event(
                    LEVEL_EVENT,
                    tuya_payload.command_id,
                    tuya_payload.data,
                )
            else:
                self.endpoint.device.switch_bus.listener_event(
                    SWITCH_EVENT,
                    tuya_payload.command_id - TUYA_CMD_BASE,
                    tuya_payload.data[1],
                )
```
Yes, i make few level changes specially, and itš looks normal, when i set 10% the light turns for approxymately for 10% and when i set level for example for 90% the light sets for 90% (i think so).

For me, at this moment, is  just missing level control for second endpoint. If you working on it, that's great. We will patiently wait, no problem.

javicalle, 2021-11-26:
Well, there's a very alpha version for the `TuyaLevelControl`. This part focuses on getting dim events to be transmitted to `tuya_manufacturer`.
It is implemented from the logs provided and without the possibility of testing it, so it would normally not work at all.

There are a few changes that must be done at several points, I hope i'm clear enough.

1. Change de `TUYA_LEVEL_COMMAND` value:
```python
# TUYA_LEVEL_COMMAND = 514
TUYA_LEVEL_COMMAND = 513  ## Magic number
```

2. New `TuyaManufacturerLevelControl` version (replace all the class):
```python
class TuyaManufacturerLevelControl(TuyaManufCluster):
    """Manufacturer Specific Cluster for cover device."""

    def handle_cluster_request(
        self,
        hdr: foundation.ZCLHeader,
        args: Tuple[TuyaManufCluster.Command],
        *,
        dst_addressing: Optional[
            Union[t.Addressing.Group, t.Addressing.IEEE, t.Addressing.NWK]
        ] = None,
    ) -> None:
        """Handle cluster request."""
        tuya_payload = args[0]

        _LOGGER.debug(
            "%s Received Attribute Report. Command is %x, Tuya Paylod values"
            "[Status : %s, TSN: %s, Command: %s, Function: %s, Data: %s]",
            self.endpoint.device.ieee,
            hdr.command_id,
            tuya_payload.status,
            tuya_payload.tsn,
            tuya_payload.command_id,
            tuya_payload.function,
            tuya_payload.data,
        )

        if hdr.command_id in (0x0002, 0x0001):
            # Send default response because the MCU expects it
            if not hdr.frame_control.disable_default_response:
                self.send_default_rsp(hdr, status=foundation.Status.SUCCESS)

            if tuya_payload.command_id == TUYA_LEVEL_COMMAND:
                self.endpoint.device.dimmer_bus.listener_event(
                    LEVEL_EVENT,
                    tuya_payload.command_id - TUYA_LEVEL_COMMAND,
                    tuya_payload.data,
                )
            else:
                self.endpoint.device.switch_bus.listener_event(
                    SWITCH_EVENT,
                    tuya_payload.command_id - TUYA_CMD_BASE,
                    tuya_payload.data[1],
                )
        elif hdr.command_id == TUYA_SET_TIME:
            """Time event call super"""
            _LOGGER.debug("TUYA_SET_TIME --> hdr: %s, args: %s", hdr, args)
            super().handle_cluster_request(hdr, args, dst_addressing=dst_addressing)
        else:
            _LOGGER.warning("Unsupported command: %s", hdr)

```

3. New `TuyaLevelControl` version (replace all the class):
```python
class TuyaLevelControl(CustomCluster, LevelControl):
    """Tuya Level cluster for dimmable device."""
[... 89 lines cut ...]
                INPUT_CLUSTERS: [
                    Basic.cluster_id,
                    Groups.cluster_id,
                    Scenes.cluster_id,
                    TuyaManufacturerLevelControl,
                    TuyaOnOff,
                    TuyaLevelControl,
                ],
                OUTPUT_CLUSTERS: [Time.cluster_id, Ota.cluster_id],
            },
            7: {
                PROFILE_ID: zha.PROFILE_ID,
                DEVICE_TYPE: zha.DeviceType.ON_OFF_LIGHT,
                INPUT_CLUSTERS: [
                    TuyaOnOff,
                    TuyaLevelControl,
                ],
                OUTPUT_CLUSTERS: [],
            }
        }
    }

```

Clean cache, restart and test.

javicalle, 2021-11-26:
> Now my class looks like that, is it correct?
Yes it is, but it is different from mine and can lead to errors as the traces you provide do not match with what I see in my code. Please understand that I am programming blindly and cannot test to verify.

> Can you explaint in other words, what it mean "fix also the command for TUYA_SET_TIME"? 
Device is sending an `TUYA_SET_TIME` from time to time. This no affects to your 'normal' switch use, but I don't know how it can affect the device in the long term. If we do not respond to the commands of the device, it can have negative effects on the operation so it is better to implement everything we can (even if you are not going to use it ;-) ).

Barmatuhin, 2021-11-26:
@javicalle - you are my hero once again! second endpoint also became dimmable. Have some status notifications in dashboard when change light level. I hope you can find it in logs
https://pastebin.com/fBJzWKcf
This is my device NWK (0x3b4f)
and looks like this is notification text in statusbar of HA dashboard - [homeassistant.components.websocket_api.http.connection] [140236431398800] object Status can't be used in 'await' expression

javicalle, 2021-11-26:
Ummmm, there must be an error elsewhere, but for now, please change this line at `TuyaManufacturerLevelControl`:
```python
            # if tuya_payload.command_id == TUYA_LEVEL_COMMAND:
            if (tuya_payload.command_id >> 4) == 0x20:

```

I'm going to look to see if I find the error

Barmatuhin, 2021-11-26:
great, last change fix entity card bulb icon behavior  (when i dim light, bulb icon also change intensity), before this last change bulb icon always stay fully on or fully off color.
Also test dimming from external controller, all works fine and card in dashboard also reflect position of dimmer. And when i change intensity of light from controller the alert in status don't appear, only when i change intensity in web page dashboard
[image]

javicalle, 2021-11-26:
I caught it!

Replace at `TuyaLevelControl` the `command` method definition (look at `async` qualifier):
```python
    async def command(

```

Barmatuhin, 2021-11-27:
@javicalle perfect! now it's work without alerts and notifications. Is it possible to make small tune? When i set dimmer to 100% and set turn on or turn off, the transition time is about 4-5 sec. Looks a little bit lazy. Or this behavior is hardcoded in dimmer?

javicalle, 2021-11-27:
I will need traces to see if there's anything that can justify that behaviour.
Perform some tests keeping a delay between actions.
I will need also information from which action belongs traces.

Barmatuhin, 2021-11-28:
Ummm traces... can you help me with advice how to make them, traces...

MattWestb, 2021-11-28:
Then tuya DP devices is normally using the same MCU but different radio so it can being interesting what ESPHome have finding with configuring tuya dimmers https://esphome.io/components/light/tuya.html and also the code they is using https://github.com/esphome/esphome/tree/dev/esphome/components/tuya.
I can not understanding how the code is working but for better knowing devs can perhaps helping.

javicalle, 2021-11-28:
> Ummm traces... can you help me with advice how to make them, traces...

I mean logs from the actions. It is just the same logs you have provided until now. Timing about what's done it's needed in order to see where the delay can be. So with the logs, a little explanation must be helpful. Something like:
* With the light off, switch on from HA
* Dimm light from N to M
* Swith off 
* etc

With this information I will try to match what is done with every log and to guess what can be doing the code

Barmatuhin, 2021-12-07:
@javicalle here logs from operations with dimmer from HA dashboard
make all changes with interval about 15sec.

1. on (100%) - transition time about 4-5 sec
2. reduce light from 100 to 50%
3. reduce light from 50 to 10
4. off
5. on (100%)
6. off

https://pastebin.com/f4Ak2DtA

MattWestb, 2021-12-07:
I was reading ESPHome tuya implementation and i think tuya is using 5 seconds as default transition time and is little tricky changing then its not working as normal Zigbee transition time commands / options but it is coming later then have getting the basic function working OK.

### Great work @Barmatuhin !!

Barmatuhin, 2021-12-09:
@MattWestb  - thank you to for your time and help. All thanks goes to @javicalle  - without him we can't reach that success!
Any perspective to implement this in master repo, to have possibility to get this in fresh HA installation?

MattWestb, 2021-12-09:
Im not in sync with the progress then i have spending much time with tuya TRV that is one large mess and trying getting same but different versions of devices working without braking the other.

I think javicalle have the basic function working in the code and getting it / endpoint but hi like getting the extra function working and need test that only one user is doing (Barmatuhin) but i think its going well then hi have time looking and implanting the last pieces.

The great thing is that the tuya multi gang PD switches is in HA beta and is coming in 2021.12 :-))))))

Hoping 2022.01 for dimmers if all is going well but you never knowing.

Barmatuhin, 2021-12-09:
Great, thank you. 
I'm ready for making extra tests, if needed. Please tag me here if any test assistance necessary.

javicalle, 2021-12-09:
> @javicalle here logs from operations with dimmer from HA dashboard make all changes with interval about 15sec.
> 
> 1. on (100%) - transition time about 4-5 sec
> 2. reduce light from 100 to 50%
> 3. reduce light from 50 to 10
> 4. off
> 5. on (100%)
> 6. off

Are all events triggered from HA (not from physical decive/remote)?

More or less is the same I can see in your logs:
1. executed 'on' command
2. executed 'move_to_level_with_on_off' command with args: '(128, 1)'
3. executed 'move_to_level_with_on_off' command with args: '(26, 1)'
4. executed 'off' command
5. executed 'move_to_level_with_on_off' command with args: '(254, 1)'
?. executed 'on' command
?. executed 'move_to_level_with_on_off' command with args: '(254, 1)'
6. executed 'off' command

I don't see any delay at first call (`1. on (100%) - transition time about 4-5 sec`). All events are triggered at the same second, despite the fact that 2 commands are triggered here:
* `on` value to the On/Off cluster
* `dimm=254` to the Level control cluster

It seems that this is an internal ZHA implementation to ensure dimmed lights always turn on/off. It also can be seen at `off` command.
But as I said, this happens at the same second, so no delay here.

I have doubts about some commands, because I did not think that they could be generated in that sequence:
```
5. executed 'move_to_level_with_on_off' command with args: '(254, 1)'
?. executed 'on' command
?. executed 'move_to_level_with_on_off' command with args: '(254, 1)'
```

My guess is that when dim=100% (point 5.) the light changes to `on` state, so you shouldn't be able to send the `on` command from HA again (at least fromthe UI)
And then a dim=100% is resent. Possible but strange to me.

Although it does not match the sequence you have indicated, it seems to me that all are interactions made from the UI, but it is an assumption.
And although there are a few seconds difference between commands, everything indicates that in the first execution, the light would have already turned on. So, once again, no delay in the code.

But this is all about assumptions of what the traces show and what you indicate in your comments.

> I was reading ESPHome tuya implementation and i think tuya is using 5 seconds as default transition time

This makes sense and (for now) is the most plausible explanation.

javicalle, 2021-12-09:
> I think javicalle have the basic function working in the code and getting it / endpoint but hi like getting the extra function working and need test that only one user is doing (Barmatuhin) but i think its going well then hi have time looking and implanting the last pieces.

It's clear to me that the way to go is like the `TuyaNewManufCluster` with a declarative way to map datapoints to cluster, but i haven't had time to design and test a solution. I hope I can do it by 2022.01 :sweat_smile:

MattWestb, 2021-12-09:
Good brave users and some dream coding all is possible also if you dont have the device but i dont saying its easy :-))
(for my impossible !!)

Have you looking how Z2M is doing and witch commands / DP they is using.
Perhaps Tasmota can giving one clue then i have seen many issues in the git with tuya dimmers (WiFi variant but they is using the same MCU and commands).

nworbneb, 2021-12-12:
> Then tuya DP devices is normally using the same MCU but different radio so it can being interesting what ESPHome have finding with configuring tuya dimmers [...]

What looks like a WiFi version of this dimmer is described for Tasmota here: https://templates.blakadder.com/moes_MS-105B.html.  Some background on the Tasmota TuyaMCU dimming is here: https://tasmota.github.io/docs/TuyaMCU/#dimmer

From what I understand of this as well as the normal dpId of 0x02 and 0x08 for the two dimmer levels you can set 0x06 and 0x0C as night light levels.

deconz implementation has "transitiontime" but it does not do anything.  https://github.com/dresden-elektronik/deconz-rest-plugin/issues/5320

nworbneb, 2021-12-14:
Thank you @javicalle for your work on this.

I've added your code into a custom quirk and seems to be working mostly ok for me.  Not really an issue but in debug message from TuyaManufacturerLevelControl there is a typo "Paylod" should be "Payload" and same logger causes an exception when the payload is not the "tuya_payload" structure, ie TUYA_SET_TIME.

I occassionally get:

`2021-12-14 16:42:35 WARNING (MainThread) [ts0601_dualdimmer] Unsupported command: <ZCLHeader frame_control=<FrameControl frame_type=CLUSTER_COMMAND manufacturer_specific=False is_reply=True disable_default_response=False> manufacturer=None tsn=20 command_id=17>`

TUYA_MCU_VERSION_RSP = 0x11 (Zigbee->Gw MCU return version or actively report version)

I've also just got it into the situation where it is working using external switches, but remote commands do not get actuated, looks like command is sent but not response received.   Not sure if this is because we didn't respond to the TUYA_MCU_VERSION_RSP?

```
2021-12-14 16:49:26 DEBUG (MainThread) [ts0601_dualdimmer] 84:fd:27:ff:fe:60:01:ae Sending Tuya Cluster Command.. Cluster Command is 4, Arguments are (247, 1)
2021-12-14 16:49:26 DEBUG (MainThread) [ts0601_dualdimmer] 84:fd:27:ff:fe:60:01:ae Sending Tuya Payload Command is 4, Tuya Payload [Status : 0, TSN: 0, Command: 520, Function: 0, Data: [4, 0, 0, 3, 207]]
2021-12-14 16:49:32 DEBUG (MainThread) [ts0601_dualdimmer] 84:fd:27:ff:fe:60:01:ae Sending Tuya Cluster Command.. Cluster Command is 4, Arguments are (89, 1)
2021-12-14 16:49:32 DEBUG (MainThread) [ts0601_dualdimmer] 84:fd:27:ff:fe:60:01:ae Sending Tuya Payload Command is 4, Tuya Payload [Status : 0, TSN: 0, Command: 520, Function: 0, Data: [4, 0, 0, 1, 95]]
```

I would normally see:

```
2021-12-14 07:44:29 DEBUG (MainThread) [ts0601_dualdimmer] 84:fd:27:ff:fe:60:01:ae Sending Tuya Cluster Command.. Cluster Command is 4, Arguments are (254, 1)
2021-12-14 07:44:29 DEBUG (MainThread) [ts0601_dualdimmer] 84:fd:27:ff:fe:60:01:ae Sending Tuya Payload Command is 4, Tuya Payload [Status : 0, TSN: 0, Command: 514, Function: 0, Data: [4, 0, 0, 3, 235]]
2021-12-14 07:44:29 DEBUG (MainThread) [ts0601_dualdimmer] 84:fd:27:ff:fe:60:01:ae Received Attribute Report. Command is 2, Tuya Payload values[Status : 4, TSN: 85, Command: 514, Function: 0, Data: [4, 0, 0, 3, 235]]
```

MattWestb, 2021-12-14:
Both request and response is in INIT and i think clovers is using it for not getting problem with command sent to and  receiving commands from some devices.
https://github.com/zigpy/zha-device-handlers/blob/5228c54d70d5eee0be09ff6ca2eeb81a538eefa1/zhaquirks/tuya/__init__.py#L35-L36

javicalle, 2021-12-14:
> TuyaManufacturerLevelControl there is a typo "Paylod" should be "Payload"

Thanks, fixed.

> I occassionally get:
> 
> `2021-12-14 16:42:35 WARNING (MainThread) [ts0601_dualdimmer] Unsupported command: <ZCLHeader frame_control=<FrameControl frame_type=CLUSTER_COMMAND manufacturer_specific=False is_reply=True disable_default_response=False> manufacturer=None tsn=20 command_id=17>`

This message must be catched here:
```python
        elif hdr.command_id == TUYA_SET_TIME:
            """Time event call super"""
            _LOGGER.debug("TUYA_SET_TIME --> hdr: %s, args: %s", hdr, args)
            super().handle_cluster_request(hdr, args, dst_addressing=dst_addressing)
```

Can you check that you also have this part in your quirk?

I think your problem is related to the command (or dp) used by your device. In order to help you, it would be useful to know (and identify) the dp of each switch for each action (on / off + dimm)

For example, this command must be a dimm function of switch 1 ¿correct?:
```
2021-12-14 07:44:29 DEBUG (MainThread) [ts0601_dualdimmer] 84:fd:27:ff:fe:60:01:ae Sending Tuya Cluster Command.. Cluster Command is 4, Arguments are (254, 1)
2021-12-14 07:44:29 DEBUG (MainThread) [ts0601_dualdimmer] 84:fd:27:ff:fe:60:01:ae Sending Tuya Payload Command is 4, Tuya Payload [Status : 0, TSN: 0, Command: 514, Function: 0, Data: [4, 0, 0, 3, 235]]
2021-12-14 07:44:29 DEBUG (MainThread) [ts0601_dualdimmer] 84:fd:27:ff:fe:60:01:ae Received Attribute Report. Command is 2, Tuya Payload values[Status : 4, TSN: 85, Command: 514, Function: 0, Data: [4, 0, 0, 3, 235]]
```

But this one seems to be a dimm for switch 2:
```
2021-12-14 16:49:26 DEBUG (MainThread) [ts0601_dualdimmer] 84:fd:27:ff:fe:60:01:ae Sending Tuya Cluster Command.. Cluster Command is 4, Arguments are (247, 1)
2021-12-14 16:49:26 DEBUG (MainThread) [ts0601_dualdimmer] 84:fd:27:ff:fe:60:01:ae Sending Tuya Payload Command is 4, Tuya Payload [Status : 0, TSN: 0, Command: 520, Function: 0, Data: [4, 0, 0, 3, 207]]
```

I don't know which commad is using your device. It should not be **520** as it does not work. 
With the traces of all the events / switches I can get an idea of how to workaround it.

nworbneb, 2021-12-14:
> This message must be catched here:
>
> ```python
>         elif hdr.command_id == TUYA_SET_TIME:
>             """Time event call super"""
>             _LOGGER.debug("TUYA_SET_TIME --> hdr: %s, args: %s", [...]

I do have this, but TUYA_SET_TIME is command 0x24 (decimal 36).  The command I am receiving is TUYA_MCA_VERSION_RSP (0x11, decimal 17)

> I think your problem is related to the command (or dp) used by your device. In order to help you, it would be useful to know (and identify) the dp of each switch for each action (on / off + dimm) [...]

Correct this is dimmer 1.  Addressed as 514 (0x202) = TUYA_LEVEL_COMMAND (0x201) + endpoint (0x1)

> But this one seems to be a dimm for switch 2:
>
> ```
> 2021-12-14 16:49:26 DEBUG (MainThread) [ts0601_dualdimmer] 84:fd:27:ff:fe:60:01:ae Sending Tuya Cluster Command.. Cluster Command is 4, Arguments [...]

This is generally working ok, address 520 = 0x208 = TUYA_LEVEL_COMMAND (0x201) + endpoint (0x7)

I think my previous issues with lack of response are down to zigbee signal - someone turned off a switch used to bridge the signal to this device!  So the only issue I really see is the lack of support of the TUYA_MCA_VERSION_RSP command.

Thank you again for this :)

javicalle, 2021-12-14:
> I do have this, but TUYA_SET_TIME is command 0x24 (decimal 36). The command I am receiving is TUYA_MCA_VERSION_RSP (0x11, decimal 17)

Yes, you are right :sweat_smile:

I have started working on another solution approach (no ETA), so I don't think I will review it for the current implementation.
I will check if the error happens also in the new implementation.

nworbneb, 2021-12-15:
Thanks @javicalle.  What you've provided gets me going with these, so should be ok until this all gets fed down into release home assistant, though I'm happy to test whatever you come up with next.

Do you know if these modules can configure minimum dimmer level or bulb type?  The Tuya docs seem to indicate this is set through the 0xFC0x input clusters, but I don't know how to check what features are/are not supported via zigbee.  Is there a way to poll zigbee devices via zigpy/ZHA to figure out what commands/settings are supported?  I assume if you send random commands it will respond with a "not supported" type response?

MattWestb, 2021-12-15:
The 0xfc00 is used on the Zigbee 0x0008 Level Control Cluster and is not working on "normal"  DP devices if not being one mixes version that  is having both normal Zigbee cluster and the tuya DP cluster (0xEF00).

But i think tuya MCUs can having DP commands for doing the same only need sketching and finding them.

Some Zigbee cluster attribute that is used by tuxa and others that i have finding https://github.com/zigpy/zigpy/discussions/823

nworbneb, 2021-12-15:
Thanks, I added:

```
    attributes = TuyaManufCluster.attributes.copy()
    attributes.update({0xfc00: ("min_max_dimming", t.uint16_t)})
    attributes.update({0xfc02: ("bulb_type", t.uint8_t)})
    attributes.update({0xfc03: ("scr_state", t.uint8_t)})
    attributes.update({0xfc05: ("min_percentage", t.uint8_t)})
```
to both TuyaManufacturerLevelControl and TuyaLevelControl classes, but these return "None" values with a <Status.UNSUPPORTED_ATTRIBUTE: 134> in log file.

MattWestb, 2021-12-15:
The tuya DP devices (TS0601) dont  have "real" Zigbee cluster and must being done with DP commands so it was expected as long we not have finding the DP commands for doing the same.

[39 more replies]

closed 2022-01-24 by dmulcahey