zigpy/zha-device-handlers #1037: [Device Support Request] - Linkind ZS23002-IE-V1A02 - 5 BUTTON REMOTE

jerrit, 2021-09-15:
Shows in home assistant as:
ZBT-RGBWSwitch-D0801
by lk
Zigbee Coordinator

Device has paired with home assistant ZHA, however no zha_events show when buttons are clicked on the remote. 

Would like for device to produce zha_events in home assistant. 

**Device signature - this can be acquired by removing the device from ZHA and pairing it again from the add devices screen. Be sure to add the entire content of the log panel after pairing the device to a code block below this line.**

{
  "node_descriptor": "NodeDescriptor(logical_type=<LogicalType.EndDevice: 2>, complex_descriptor_available=0, user_descriptor_available=0, reserved=0, aps_flags=0, frequency_band=<FrequencyBand.Freq2400MHz: 8>, mac_capability_flags=<MACCapabilityFlags.AllocateAddress: 128>, manufacturer_code=4456, maximum_buffer_size=82, maximum_incoming_transfer_size=82, server_mask=11264, maximum_outgoing_transfer_size=82, descriptor_capability_field=<DescriptorCapability.NONE: 0>, *allocate_address=True, *is_alternate_pan_coordinator=False, *is_coordinator=False, *is_end_device=True, *is_full_function_device=False, *is_mains_powered=False, *is_receiver_on_when_idle=False, *is_router=False, *is_security_capable=False)",
  "endpoints": {
    "1": {
      "profile_id": 260,
      "device_type": "0x0800",
      "in_clusters": [
        "0x0000",
        "0x0001",
        "0x0003",
        "0x0b05",
        "0x1000",
        "0xfd01"
      ],
      "out_clusters": [
        "0x0003",
        "0x0004",
        "0x0006",
        "0x0008",
        "0x0019",
        "0x0300",
        "0x1000"
      ]
    }
  },
  "manufacturer": "lk",
  "model": "ZBT-RGBWSwitch-D0801",
  "class": "zigpy.device.Device"
}

tube0013, 2021-09-15:
Which coordinator are you using? If it's like the linkind single button it's likely sending only to a group. Some coordinators (SiLabs) don't get that traffic, and the group would need to be found and the coordinator bonded to it.

jerrit, 2021-09-16:
I have the SiLabs HUSBZB-1, interesting. I will need to look up how to do that.

jerrit, 2021-09-16:
> Which coordinator are you using? If it's like the linkind single button it's likely sending only to a group. Some coordinators (SiLabs) don't get that traffic, and the group would need to be found and the coordinator bonded to it.

I have the SiLabs HUSBZB-1, interesting. I will need to look up how to do that.

closed 2021-09-16 by jerrit

Apflkuacha, 2022-01-04:
I do want to reopen this request. In Zigbee2MQTT this remote is sending loads of actions as taken from the herdsman-converters: 
` exposes: [e.battery(), e.battery_low(), e.action(['on', 'off', 'brightness_step_up',
            'brightness_step_down', 'color_temperature_move', 'color_move', 'brightness_move_up', 'brightness_move_down', 'brightness_stop',
            'brightness_move_to_level', 'color_temperature_move_up', 'color_temperature_move_down'])]`

but in ZHA the device only shows a Power-Entity and in the ZHA_Events there are only following events: on, off, move_to_color_temp, move_to_color and move_to_level. 
Others moved from ZHA to Zigbee2MQTT but it would be nice to support this one also with ZHA.