zigpy/zha-device-handlers #1041: [Device Support Request] Stop command level control does not specify the button

Emrvb, 2021-09-21:
I'm working on the device handler for several iCasa remotes and I'm running into an issue with the level control. https://github.com/zigpy/zha-device-handlers/pull/1039

With these remotes you can dim up or down with on and off buttons. I want to add LONG_RELEASE for the on and off buttons, however the stop command does not include the button that was released. It seems zigpy does not keep track of it either.

How should I proceed?

1. Is there a workaround for this situation?
2. Is there a generic trigger is should use?
3. Should I just ignore this trigger?

MattWestb, 2021-09-21:
I was doing the tuya TS0404F that is one 4 button dimmer switch but i was not able doing the nice stop command with from up or down dimming.
It shall being possible doing if looking on IKEA Open/Close bottom have implanting logic for it:
https://github.com/zigpy/zha-device-handlers/blob/1c2b7ece00f494d16e38bb41093a525b1ed64f42/zhaquirks/ikea/opencloseremote.py#L48
If you is implanting it for the dimmer i can using it for the tuya dimmer switch :-))

Emrvb, 2021-09-21:
Thank you for pointing me in that direction. After a quick glance this indeed seems a solution to a similar problem. It should be no problem for me to implement that. 

However, if multiple remotes need this particular workaround, perhaps a more generalized (custom) cluster should be implemented, instead of copy and pasting the same fix around.

MattWestb, 2021-09-23:
I like to do but i and my laptop is on Holiday in Spain but coming "home" on Monday and rejoining with my Zigbee networks, then i can trying if i can getting it working with my new IKEA Styrbar i have baying in France (the stainless steel version for 10€) that have the same commands but little more for making some magic with scenes in IKEA lights.

O I see you is storing the options from last move and using it for the stop commands is being receiver and can getting the from up or down :-)).

The devs have more knowledge and is knowing if its one better way to do it or if its can making  problems in the higher level (ZHA).

Great work done Eric !!!

PING @dmulcahey
Sorry for addressing you directly Dave but is this one OK way for getting the "long press release" as 2 different automation triggers (stop from moving up and stop from moving down) from dimmer switches that many user like to having as device trigger from there devices or is it making problems in ZHA ?
Or is it one better way implanting it in the quirk root so all quirks can using it ?

I like implanting it for the IKEA dimmer swishes and also tuya TS004F if is working OK and not making problems.

MattWestb, 2021-09-30:
Im at home but was not have the time testing your "hack" and i will waiting little more so i can updating the tests and production systems to 2021.10 and then i patching one test systems and adding one IKEA Styrbar and see how its working.

I think one side effect of this is that ZHA is complain that the data is not OK like it doing with the IKEA "scene remotes" that is having some extra data attached for sending next and privies scene to IKEA lights.

Perhaps its possible "branding" the commands with manufacture ID in frames and ZHA is ignoring the bad data.

In the end its up to the ZHA / Zigpy devs saying how they like (if the do) to have it and if its can being implanted in the system.

Emrvb, 2021-09-30:
No rush.

I think the main problem here is that we now emit raw events that are out of spec. It also raises the question whether this behavior (event key release af dimming) should be handled here or downstream (inside the home assistant integration for example). But that's getting rather philosophical.

Anyway, I would love some input from a maintainer.

closed 2022-04-05 by github-actions[bot]