zigpy/zha-device-handlers #1042: [Device Support Request] Zigbee motion sensor - Model ZM-RT201

jkottier, 2021-09-22:
Could you also add support for a Zigbee motion sensor model ZM-RT201. From what I understand this model is similar to ZM-35ZH-Q. There is a support request for this device registered under #1019

This is the output when adding this device. It is stuck on "Starting Interview" when adding it via ZHA in Home Assistant:

New device 0xa5db (00:12:4b:00:1b:ab:a7:f5) joined the network
[0xa5db] Scheduling initialization
Received frame on uninitialized device <Device model=None manuf=None nwk=0xA5DB ieee=00:12:4b:00:1b:ab:a7:f5 is_initialized=False> from ep 0 to ep 0, cluster ZDOCmd.Device_annce: b'\xFF\xDB\xA5\xF5\xA7\xAB\x1B\x00\x4B\x12\x00\x80'
[0xa5db:zdo] ZDO request ZDOCmd.Device_annce: [0xA5DB, 00:12:4b:00:1b:ab:a7:f5, 128]
Tries remaining: 3
[0xa5db] Requesting 'Node Descriptor'
Tries remaining: 2
[0xa5db] Extending timeout for 0x40 request
Received frame on uninitialized device <Device model=None manuf=None nwk=0xA5DB ieee=00:12:4b:00:1b:ab:a7:f5 is_initialized=False> from ep 1 to ep 1, cluster 1: b'\x18\x02\x0A\x20\x00\x20\x1E'
[0xa5db] Received ZCL while uninitialized on endpoint id 1, cluster 0x0001 id, hdr: <ZCLHeader frame_control=<FrameControl frame_type=GLOBAL_COMMAND manufacturer_specific=False is_reply=True disable_default_response=True> manufacturer=None tsn=2 command_id=Command.Report_Attributes>, payload: b' \x00 \x1e'
[0xa5db] Uninitialized device command 'Command.Report_Attributes' args: [[Attribute(attrid=32, value=<TypeValue type=uint8_t, value=30>)]]
Received frame on uninitialized device <Device model=None manuf=None nwk=0xA5DB ieee=00:12:4b:00:1b:ab:a7:f5 is_initialized=False> from ep 1 to ep 1, cluster 1: b'\x18\x03\x0A\x21\x00\x20\xC8'
[0xa5db] Received ZCL while uninitialized on endpoint id 1, cluster 0x0001 id, hdr: <ZCLHeader frame_control=<FrameControl frame_type=GLOBAL_COMMAND manufacturer_specific=False is_reply=True disable_default_response=True> manufacturer=None tsn=3 command_id=Command.Report_Attributes>, payload: b'!\x00 \xc8'
[0xa5db] Uninitialized device command 'Command.Report_Attributes' args: [[Attribute(attrid=33, value=<TypeValue type=uint8_t, value=200>)]]
Received frame on uninitialized device <Device model=None manuf=None nwk=0xA5DB ieee=00:12:4b:00:1b:ab:a7:f5 is_initialized=False> from ep 1 to ep 1, cluster 1: b'\x18\x04\x0A\x20\x00\x20\x1E'
[0xa5db] Received ZCL while uninitialized on endpoint id 1, cluster 0x0001 id, hdr: <ZCLHeader frame_control=<FrameControl frame_type=GLOBAL_COMMAND manufacturer_specific=False is_reply=True disable_default_response=True> manufacturer=None tsn=4 command_id=Command.Report_Attributes>, payload: b' \x00 \x1e'
[0xa5db] Uninitialized device command 'Command.Report_Attributes' args: [[Attribute(attrid=32, value=<TypeValue type=uint8_t, value=30>)]]
Received frame on uninitialized device <Device model=None manuf=None nwk=0xA5DB ieee=00:12:4b:00:1b:ab:a7:f5 is_initialized=False> from ep 1 to ep 1, cluster 1: b'\x18\x05\x0A\x21\x00\x20\xC8'
[0xa5db] Received ZCL while uninitialized on endpoint id 1, cluster 0x0001 id, hdr: <ZCLHeader frame_control=<FrameControl frame_type=GLOBAL_COMMAND manufacturer_specific=False is_reply=True disable_default_response=True> manufacturer=None tsn=5 command_id=Command.Report_Attributes>, payload: b'!\x00 \xc8'
[0xa5db] Uninitialized device command 'Command.Report_Attributes' args: [[Attribute(attrid=33, value=<TypeValue type=uint8_t, value=200>)]]
Received frame on uninitialized device <Device model=None manuf=None nwk=0xA5DB ieee=00:12:4b:00:1b:ab:a7:f5 is_initialized=False> from ep 0 to ep 0, cluster ZDOCmd.Node_Desc_rsp: b'\x40\x00\xDB\xA5\x02\x40\x80\x00\x00\x50\xA0\x00\x00\x2A\xA0\x00\x00'
[0xa5db] Got Node Descriptor: NodeDescriptor(logical_type=<LogicalType.EndDevice: 2>, complex_descriptor_available=0, user_descriptor_available=0, reserved=0, aps_flags=0, frequency_band=<FrequencyBand.Freq2400MHz: 8>, mac_capability_flags=<MACCapabilityFlags.AllocateAddress: 128>, manufacturer_code=0, maximum_buffer_size=80, maximum_incoming_transfer_size=160, server_mask=10752, maximum_outgoing_transfer_size=160, descriptor_capability_field=<DescriptorCapability.NONE: 0>, *allocate_address=True, *is_alternate_pan_coordinator=False, *is_coordinator=False, *is_end_device=True, *is_full_function_device=False, *is_mains_powered=False, *is_receiver_on_when_idle=False, *is_router=False, *is_security_capable=False)
[0xa5db] Discovering endpoints
Tries remaining: 3
[0xa5db] Extending timeout for 0x41 request
Received frame on uninitialized device <Device model=None manuf=None nwk=0xA5DB ieee=00:12:4b:00:1b:ab:a7:f5 is_initialized=False> from ep 1 to ep 1, cluster 3: b'\x11\x01\x01'
[0xa5db] Received ZCL while uninitialized on endpoint id 1, cluster 0x0003 id, hdr: <ZCLHeader frame_control=<FrameControl frame_type=CLUSTER_COMMAND manufacturer_specific=False is_reply=False disable_default_response=True> manufacturer=None tsn=1 command_id=1>, payload: b''
Received frame on uninitialized device <Device model=None manuf=None nwk=0xA5DB ieee=00:12:4b:00:1b:ab:a7:f5 is_initialized=False> from ep 1 to ep 1, cluster 1280: b'\x18\x06\x0A\x02\x00\x21\x01\x00'
[0xa5db] Received ZCL while uninitialized on endpoint id 1, cluster 0x0500 id, hdr: <ZCLHeader frame_control=<FrameControl frame_type=GLOBAL_COMMAND manufacturer_specific=False is_reply=True disable_default_response=True> manufacturer=None tsn=6 command_id=Command.Report_Attributes>, payload: b'\x02\x00!\x01\x00'
[0xa5db] Uninitialized device command 'Command.Report_Attributes' args: [[Attribute(attrid=2, value=<TypeValue type=uint16_t, value=1>)]]
Received frame on uninitialized device <Device model=None manuf=None nwk=0xA5DB ieee=00:12:4b:00:1b:ab:a7:f5 is_initialized=False> from ep 1 to ep 1, cluster 1280: b'\x18\x07\x0A\x02\x00\x21\x01\x00'
[0xa5db] Received ZCL while uninitialized on endpoint id 1, cluster 0x0500 id, hdr: <ZCLHeader frame_control=<FrameControl frame_type=GLOBAL_COMMAND manufacturer_specific=False is_reply=True disable_default_response=True> manufacturer=None tsn=7 command_id=Command.Report_Attributes>, payload: b'\x02\x00!\x01\x00'
[0xa5db] Uninitialized device command 'Command.Report_Attributes' args: [[Attribute(attrid=2, value=<TypeValue type=uint16_t, value=1>)]]
Received frame on uninitialized device <Device model=None manuf=None nwk=0xA5DB ieee=00:12:4b:00:1b:ab:a7:f5 is_initialized=False> from ep 1 to ep 1, cluster 1280: b'\x18\x08\x0A\x02\x00\x21\x01\x00'
[0xa5db] Received ZCL while uninitialized on endpoint id 1, cluster 0x0500 id, hdr: <ZCLHeader frame_control=<FrameControl frame_type=GLOBAL_COMMAND manufacturer_specific=False is_reply=True disable_default_response=True> manufacturer=None tsn=8 command_id=Command.Report_Attributes>, payload: b'\x02\x00!\x01\x00'
[0xa5db] Uninitialized device command 'Command.Report_Attributes' args: [[Attribute(attrid=2, value=<TypeValue type=uint16_t, value=1>)]]
Received frame on uninitialized device <Device model=None manuf=None nwk=0xA5DB ieee=00:12:4b:00:1b:ab:a7:f5 is_initialized=False> from ep 1 to ep 1, cluster 1280: b'\x18\x09\x0A\x02\x00\x21\x01\x00'
[0xa5db] Received ZCL while uninitialized on endpoint id 1, cluster 0x0500 id, hdr: <ZCLHeader frame_control=<FrameControl frame_type=GLOBAL_COMMAND manufacturer_specific=False is_reply=True disable_default_response=True> manufacturer=None tsn=9 command_id=Command.Report_Attributes>, payload: b'\x02\x00!\x01\x00'
[0xa5db] Uninitialized device command 'Command.Report_Attributes' args: [[Attribute(attrid=2, value=<TypeValue type=uint16_t, value=1>)]]
Tries remaining: 2
[0xa5db] Extending timeout for 0x42 request
Received frame on uninitialized device <Device model=None manuf=None nwk=0xA5DB ieee=00:12:4b:00:1b:ab:a7:f5 is_initialized=False> from ep 1 to ep 1, cluster 3: b'\x11\x02\x01'
[0xa5db] Received ZCL while uninitialized on endpoint id 1, cluster 0x0003 id, hdr: <ZCLHeader frame_control=<FrameControl frame_type=CLUSTER_COMMAND manufacturer_specific=False is_reply=False disable_default_response=True> manufacturer=None tsn=2 command_id=1>, payload: b''
Received frame on uninitialized device <Device model=None manuf=None nwk=0xA5DB ieee=00:12:4b:00:1b:ab:a7:f5 is_initialized=False> from ep 1 to ep 1, cluster 1280: b'\x18\x0A\x0A\x02\x00\x21\x01\x00'
[0xa5db] Received ZCL while uninitialized on endpoint id 1, cluster 0x0500 id, hdr: <ZCLHeader frame_control=<FrameControl frame_type=GLOBAL_COMMAND manufacturer_specific=False is_reply=True disable_default_response=True> manufacturer=Non [... cut ...]
Received frame on uninitialized device <Device model=None manuf=None nwk=0xA5DB ieee=00:12:4b:00:1b:ab:a7:f5 is_initialized=False> from ep 1 to ep 1, cluster 1280: b'\x18\x0B\x0A\x02\x00\x21\x01\x00'
[0xa5db] Received ZCL while uninitialized on endpoint id 1, cluster 0x0500 id, hdr: <ZCLHeader frame_control=<FrameControl frame_type=GLOBAL_COMMAND manufacturer_specific=False is_reply=True disable_default_response=True> manufacturer=None tsn=11 command_id=Command.Report_Attributes>, payload: b'\x02\x00!\x01\x00'
[0xa5db] Uninitialized device command 'Command.Report_Attributes' args: [[Attribute(attrid=2, value=<TypeValue type=uint16_t, value=1>)]]

MattWestb, 2021-11-15:
Do you still not getting the device paring OK ?
Some device can being little tricky not getting in sleep so ZHA can requesting all data from them. 
Your device looks working then its sending data to the coordinator but ZHA is not knowing the device and cant applying the quirk for it.

jkottier, 2021-11-19:
I'll give it another shot this weekend, I'll keep you posted!

jkottier, 2021-11-23:
Unfortunately no succes, the device is still stuck on "Start interviewing". In the mean time I've been able to test the Motion Sensors with a Tuya Zigbee hub. This does work. Is there some additional information I could send you to help me resolve this issue?

MattWestb, 2021-11-23:
The good is that your device is working OK with tuya ZBGW !!!

Some device is falling i sleep very fast and ZHA cant getting all information from them but your is very communicative by sending attribute reports from power configuration cluster, but is not listening so well to commands.

Its looks like ZHA is not having enough time for getting all information and need more  time.

You can using service from developer tools and `Zigbee Home Automation: Permit`
 and putting in longer times (standard is 60 seconds) for joining devices and clicking call service.
You can have 2 windows of HA open so you can see how its working.
Also sending more permit commands and having the window open shall working OK.

jkottier, 2021-11-29:
After some playing around I think I've made some good progress:

**Steps performed**:
1. send permit call with duration 240 seconds && IEEE set to the USB-macaddress
2. add new device (not the first attempt):
`Received frame on uninitialized device <Device model=None manuf=None nwk=0x3286 ieee=00:12:4b:00:11:c2:54:9b is_initialized=False> from ep 1 to ep 1, cluster 1280: b'\x18\x15\x0A\x02\x00\x21\x01\x00'
[0x3286] Received ZCL while uninitialized on endpoint id 1, cluster 0x0500 id, hdr: <ZCLHeader frame_control=<FrameControl frame_type=GLOBAL_COMMAND manufacturer_specific=False is_reply=True disable_default_response=True> manufacturer=None tsn=21 command_id=Command.Report_Attributes>, payload: b'\x02\x00!\x01\x00'
[0x3286] Uninitialized device command 'Command.Report_Attributes' args: [[Attribute(attrid=2, value=<TypeValue type=uint16_t, value=1>)]]
Device 0x3286 (00:12:4b:00:11:c2:54:9b) left the network
Device 0xdc6f (00:12:4b:00:11:c2:54:9b) joined the network
Device 00:12:4b:00:11:c2:54:9b changed id (0xdc6f => 0xdc6f)
[0xdc6f] Scheduling initialization
Tries remaining: 3
[0xdc6f] Already have endpoints: {0: <zigpy.zdo.ZDO object at 0x7f7b6872e0>, 1: <Endpoint id=1 in=[] out=[] status=<Status.NEW: 0>>}
[0xdc6f] Initializing endpoints [<Endpoint id=1 in=[] out=[] status=<Status.NEW: 0>>]
[0xdc6f:1] Discovering endpoint information
Tries remaining: 3
[0xdc6f] Extending timeout for 0x8e request
Device 0xdc6f (00:12:4b:00:11:c2:54:9b) joined the network
[0xdc6f] Scheduling initialization
[0xdc6f] Canceling old initialize call
Received frame on uninitialized device <Device model=None manuf=None nwk=0xDC6F ieee=00:12:4b:00:11:c2:54:9b is_initialized=False> from ep 0 to ep 0, cluster ZDOCmd.Device_annce: b'\xFF\x6F\xDC\x9B\x54\xC2\x11\x00\x4B\x12\x00\x80'
[0xdc6f:zdo] ZDO request ZDOCmd.Device_annce: [0xDC6F, 00:12:4b:00:11:c2:54:9b, 128]
Tries remaining: 3
[0xdc6f] Already have endpoints: {0: <zigpy.zdo.ZDO object at 0x7f7b6872e0>, 1: <Endpoint id=1 in=[] out=[] status=<Status.NEW: 0>>}
[0xdc6f] Initializing endpoints [<Endpoint id=1 in=[] out=[] status=<Status.NEW: 0>>]
[0xdc6f:1] Discovering endpoint information
Tries remaining: 3
[0xdc6f] Extending timeout for 0x8f request
Received frame on uninitialized device <Device model=None manuf=None nwk=0xDC6F ieee=00:12:4b:00:11:c2:54:9b is_initialized=False> from ep 1 to ep 1, cluster 1: b'\x18\x02\x0A\x20\x00\x20\x1E'
[0xdc6f] Received ZCL while uninitialized on endpoint id 1, cluster 0x0001 id, hdr: <ZCLHeader frame_control=<FrameControl frame_type=GLOBAL_COMMAND manufacturer_specific=False is_reply=True disable_default_response=True> manufacturer=None tsn=2 command_id=Command.Report_Attributes>, payload: b' \x00 \x1e'
[0xdc6f] Uninitialized device command 'Command.Report_Attributes' args: [[Attribute(attrid=32, value=<TypeValue type=uint8_t, value=30>)]]
Received frame on uninitialized device <Device model=None manuf=None nwk=0xDC6F ieee=00:12:4b:00:11:c2:54:9b is_initialized=False> from ep 1 to ep 1, cluster 1: b'\x18\x03\x0A\x21\x00\x20\xC8'
[0xdc6f] Received ZCL while uninitialized on endpoint id 1, cluster 0x0001 id, hdr: <ZCLHeader frame_control=<FrameControl frame_type=GLOBAL_COMMAND manufacturer_specific=False is_reply=True disable_default_response=True> manufacturer=None tsn=3 command_id=Command.Report_Attributes>, payload: b'!\x00 \xc8'
[0xdc6f] Uninitialized device command 'Command.Report_Attributes' args: [[Attribute(attrid=33, value=<TypeValue type=uint8_t, value=200>)]]
Received frame on uninitialized device <Device model=None manuf=None nwk=0xDC6F ieee=00:12:4b:00:11:c2:54:9b is_initialized=False> from ep 1 to ep 1, cluster 1: b'\x18\x04\x0A\x20\x00\x20\x1E'
[0xdc6f] Received ZCL while uninitialized on endpoint id 1, cluster 0x0001 id, hdr: <ZCLHeader frame_control=<FrameControl frame_type=GLOBAL_COMMAND manufacturer_specific=False is_reply=True disable_default_response=True> manufacturer=None tsn=4 command_id=Command.Report_Attributes>, payload: b' \x00 \x1e'
[0xdc6f] Uninitialized device command 'Command.Report_Attributes' args: [[Attribute(attrid=32, value=<TypeValue type=uint8_t, value=30>)]]
Received frame on uninitialized device <Device model=None manuf=None nwk=0xDC6F ieee=00:12:4b:00:11:c2:54:9b is_initialized=False> from ep 1 to ep 1, cluster 1: b'\x18\x05\x0A\x21\x00\x20\xC8'
[0xdc6f] Received ZCL while uninitialized on endpoint id 1, cluster 0x0001 id, hdr: <ZCLHeader frame_control=<FrameControl frame_type=GLOBAL_COMMAND manufacturer_specific=False is_reply=True disable_default_response=True> manufacturer=None tsn=5 command_id=Command.Report_Attributes>, payload: b'!\x00 \xc8'
[0xdc6f] Uninitialized device command 'Command.Report_Attributes' args: [[Attribute(attrid=33, value=<TypeValue type=uint8_t, value=200>)]]
Received frame on uninitialized device <Device model=None manuf=None nwk=0xDC6F ieee=00:12:4b:00:11:c2:54:9b is_initialized=False> from ep 0 to ep 0, cluster ZDOCmd.Simple_Desc_rsp: b'\x8F\x00\x6F\xDC\x18\x01\x04\x01\x02\x04\x01\x04\x00\x00\x03\x00\x01\x00\x00\x05\x04\x00\x00\x03\x00\x01\x00\x00\x05'
[0xdc6f:1] Discovered endpoint information: SizePrefixedSimpleDescriptor(endpoint=1, profile=260, device_type=1026, device_version=1, input_clusters=[0, 3, 1, 1280], output_clusters=[0, 3, 1, 1280])
[0xdc6f] Extending timeout for 0x90 request
[... cut ...]
      "profile_id": 260,
      "device_type": "0x0402",
      "in_clusters": [
        "0x0000",
        "0x0001",
        "0x0003",
        "0x0500"
      ],
      "out_clusters": [
        "0x0000",
        "0x0001",
        "0x0003",
        "0x0500"
      ]
    }
  },
  "manufacturer": "unk_manufacturer",
  "model": "unk_model",
  "class": "zigpy.device.Device"
}`

Although the manufacturer and model are "unknonwn" the motion sensor seems to have been added now.
It also detectecs motion and battery % state .

MattWestb, 2021-11-29:
Have you trying reading both attributes on the basic cluster then its pared OK ?
Perhaps you can getting the information and ZHA is saving the attribute and can using quirks.

I have problem with one TS004F switch that most of the time is not getting manufacture and model and the quirk is not loading.
Its looks restarting HA and open the network for joining without deleting the device and working but its little roll the dice.

rbert001, 2022-01-18:
I'm still struggling to get this sensor working in ZHA.. I did the same as jkottier but I can't move out of the interviewing step..  I already increase the permit duaration and tried pairing it serveral times...

escobarin3, 2022-04-11:
> 2\. ived frame on uninitialized device  from ep 1 to ep 1, cluster 1280: b'\x18\x15\x0A\x02\x00\x21\x01\x00'
>     [0x3286] Received ZCL while uninitialized on endpoint id 1, cluster 0x0

Did you solve it? I didn't

escobarin3, 2022-04-11:
> I'm still struggling to get this sensor working in ZHA.. I did the same as jkottier but I can't move out of the interviewing step.. I already increase the permit duaration and tried pairing it serveral times...

Did you solve it? I couldn't

closed 2022-10-26 by github-actions[bot]