zigpy/zha-device-handlers #1045: [Device Support Request] - TS011F by _TZ3000_ew3ldmgx

miit86, 2021-09-23:
Hi All,

Bought a Smartplug from ALI TS011F by _TZ3000_ew3ldmgx

HA added 1 switch and 3 sensors:

switch.tz3000_ew3ldmgx_ts011f_bb1466fe_on_off                               "SWITCH"
sensor.tz3000_ew3ldmgx_ts011f_bb1466fe_electrical_measurement    "192 W"
sensor.tz3000_ew3ldmgx_ts011f_bb1466fe_smartenergy_metering    "UNKNOWN"

**Describe the solution you'd like**
The "smartenergy_metering" is not working. It should show the KwH
On other forums there talking it could be a TS0121 thats advertises wrong.
Also Zigbee2MQTT supports this Plug with metering.

**Device signature - this can be acquired by removing the device from ZHA and pairing it again from the add devices screen. Be sure to add the entire content of the log panel after pairing the device to a code block below this line.**

`
{
  "node_descriptor": "NodeDescriptor(logical_type=<LogicalType.Router: 1>, complex_descriptor_available=0, user_descriptor_available=0, reserved=0, aps_flags=0, frequency_band=<FrequencyBand.Freq2400MHz: 8>, mac_capability_flags=<MACCapabilityFlags.AllocateAddress|RxOnWhenIdle|MainsPowered|FullFunctionDevice: 142>, manufacturer_code=4098, maximum_buffer_size=82, maximum_incoming_transfer_size=82, server_mask=11264, maximum_outgoing_transfer_size=82, descriptor_capability_field=<DescriptorCapability.NONE: 0>, *allocate_address=True, *is_alternate_pan_coordinator=False, *is_coordinator=False, *is_end_device=False, *is_full_function_device=True, *is_mains_powered=True, *is_receiver_on_when_idle=True, *is_router=True, *is_security_capable=False)",
  "endpoints": {
    "1": {
      "profile_id": 260,
      "device_type": "0x010a",
      "in_clusters": [
        "0x0000",
        "0x0003",
        "0x0004",
        "0x0005",
        "0x0006",
        "0x0702",
        "0x0b04",
        "0xe000",
        "0xe001"
      ],
      "out_clusters": [
        "0x000a",
        "0x0019"
      ]
    },
    "242": {
      "profile_id": 41440,
      "device_type": "0x0061",
      "in_clusters": [],
      "out_clusters": [
        "0x0021"
      ]
    }
  },
  "manufacturer": "_TZ3000_ew3ldmgx",
  "model": "TS011F",
  "class": "zigpy.device.Device"
}
`

LOG
`
[0x0000:zdo] ZDO request ZDOCmd.Mgmt_Permit_Joining_req: [60, <Bool.false: 0>]
New device 0x0432 (84:fd:27:ff:fe:68:de:64) joined the network
[0x0432] Scheduling initialization
Tries remaining: 3
[0x0432] Requesting 'Node Descriptor'
Tries remaining: 2
[0x0432] Extending timeout for 0xce request
Device 0x0432 (84:fd:27:ff:fe:68:de:64) joined the network
[0x0432] Scheduling initialization
[0x0432] Canceling old initialize call
Tries remaining: 3
[0x0432] Requesting 'Node Descriptor'
Tries remaining: 2
[... 101 lines cut ...]
[0x0432:1:0x0702]: 'async_initialize' stage succeeded
[0x0432:1:0x0b04]: 'async_initialize' stage succeeded
[0x0432:1:0x0019]: 'async_initialize' stage succeeded
[0x0432](TS011F): power source: Mains
[0x0432](TS011F): completed initialization
[0x0432:1:0x0006]: attempting to update onoff state - from cache: False
[0x0432:1:0x0006] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=GLOBAL_COMMAND manufacturer_specific=False is_reply=True disable_default_response=True> manufacturer=None tsn=247 command_id=Command.Read_Attributes_rsp>
[0x278f:1:0x0400] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=GLOBAL_COMMAND manufacturer_specific=False is_reply=True disable_default_response=True> manufacturer=None tsn=9 command_id=Command.Report_Attributes>
[0x278f:1:0x0400] ZCL request 0x000a: [[Attribute(attrid=0, value=<TypeValue type=uint16_t, value=0>)]]
[0x278f:1:0x0400] Attribute report received: measured_value=0
[0x2F84:1:0x0b04]: async_update
[0xD44E:1:0x0b04]: async_update
[0xB4C2:1:0x0b04]: async_update
[0x0432:1:0x0b04]: async_update
[0x2f84:1:0x0b04] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=GLOBAL_COMMAND manufacturer_specific=False is_reply=True disable_default_response=True> manufacturer=None tsn=249 command_id=Command.Read_Attributes_rsp>
[0xd44e:1:0x0b04] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=GLOBAL_COMMAND manufacturer_specific=False is_reply=True disable_default_response=True> manufacturer=None tsn=251 command_id=Command.Read_Attributes_rsp>
[0xb4c2:1:0x0b04] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=GLOBAL_COMMAND manufacturer_specific=False is_reply=True disable_default_response=True> manufacturer=None tsn=253 command_id=Command.Read_Attributes_rsp>
[0x0432:1:0x0b04] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=GLOBAL_COMMAND manufacturer_specific=False is_reply=True disable_default_response=True> manufacturer=None tsn=255 command_id=Command.Read_Attributes_rsp>
[0xd44e:1:0x0b04] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=GLOBAL_COMMAND manufacturer_specific=False is_reply=True disable_default_response=True> manufacturer=None tsn=109 command_id=Command.Report_Attributes>
[0xd44e:1:0x0b04] ZCL request 0x000a: [[Attribute(attrid=1291, value=<TypeValue type=int16s, value=203>)]]
[0xd44e:1:0x0b04] Attribute report received: active_power=203
[0xb4c2:1:0x0b04] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=GLOBAL_COMMAND manufacturer_specific=False is_reply=True disable_default_response=True> manufacturer=None tsn=106 command_id=Command.Report_Attributes>
[0xb4c2:1:0x0b04] ZCL request 0x000a: [[Attribute(attrid=1291, value=<TypeValue type=int16s, value=1689>)]]
[0xb4c2:1:0x0b04] Attribute report received: active_power=1689
`

PlusPlus-ua, 2021-09-27:
Have same problems with device _TZ3000_cphmq0q7. It seems the same device:
```
{
  "node_descriptor": "NodeDescriptor(logical_type=<LogicalType.Router: 1>, complex_descriptor_available=0, user_descriptor_available=0, reserved=0, aps_flags=0, frequency_band=<FrequencyBand.Freq2400MHz: 8>, mac_capability_flags=<MACCapabilityFlags.AllocateAddress|RxOnWhenIdle|MainsPowered|FullFunctionDevice: 142>, manufacturer_code=4098, maximum_buffer_size=82, maximum_incoming_transfer_size=82, server_mask=11264, maximum_outgoing_transfer_size=82, descriptor_capability_field=<DescriptorCapability.NONE: 0>, *allocate_address=True, *is_alternate_pan_coordinator=False, *is_coordinator=False, *is_end_device=False, *is_full_function_device=True, *is_mains_powered=True, *is_receiver_on_when_idle=True, *is_router=True, *is_security_capable=False)",
  "endpoints": {
    "1": {
      "profile_id": 260,
      "device_type": "0x010a",
      "in_clusters": [
        "0x0000",
        "0x0003",
        "0x0004",
        "0x0005",
        "0x0006",
        "0x0702",
        "0x0b04",
        "0xe000",
        "0xe001"
      ],
      "out_clusters": [
        "0x000a",
        "0x0019"
      ]
    },
    "242": {
      "profile_id": 41440,
      "device_type": "0x0061",
      "in_clusters": [],
      "out_clusters": [
        "0x0021"
      ]
    }
  },
  "manufacturer": "_TZ3000_cphmq0q7",
  "model": "TS011F",
  "class": "zigpy.device.Device"
}
```

miit86, 2021-09-27:
> Have same problems with device _TZ3000_cphmq0q7. It seems the same device:
>
> ```
> {
>   "node_descriptor": "NodeDescriptor(logical_type=<LogicalType.Router: 1>, complex_descriptor_available=0, [...]

HI,

Yes it's the same PLUG. But different manufacturer. mine is from Girier
Hope they can fix it.
I know there is another one, but can't find the exact model anymore.

PlusPlus-ua, 2021-09-28:
According to this file from Zigbee2mqtt: https://github.com/Koenkk/zigbee-herdsman-converters/blob/master/devices/tuya.js , same devices have next manufacturer names: _TZ3000_cphmq0q7, _TZ3000_ew3ldmgx, _TZ3000_ps3dmato, _TZ3000_mraovvmm, _TZ3000_jvzvulen, _TZ3000_dpo1ysak.

mitaka, 2021-09-30:
I have similar device:

```
{
  "node_descriptor": "NodeDescriptor(logical_type=<LogicalType.Router: 1>, complex_descriptor_available=0, user_descriptor_available=0, reserved=0, aps_flags=0, frequency_band=<FrequencyBand.Freq2400MHz: 8>, mac_capability_flags=<MACCapabilityFlags.AllocateAddress|RxOnWhenIdle|MainsPowered|FullFunctionDevice: 142>, manufacturer_code=4098, maximum_buffer_size=82, maximum_incoming_transfer_size=82, server_mask=11264, maximum_outgoing_transfer_size=82, descriptor_capability_field=<DescriptorCapability.NONE: 0>, *allocate_address=True, *is_alternate_pan_coordinator=False, *is_coordinator=False, *is_end_device=False, *is_full_function_device=True, *is_mains_powered=True, *is_receiver_on_when_idle=True, *is_router=True, *is_security_capable=False)",
  "endpoints": {
    "1": {
      "profile_id": 260,
      "device_type": "0x010a",
      "in_clusters": [
        "0x0000",
        "0x0003",
        "0x0004",
        "0x0005",
        "0x0006",
        "0x0702",
        "0x0b04",
        "0xe000",
        "0xe001"
      ],
      "out_clusters": [
        "0x000a",
        "0x0019"
      ]
    },
    "242": {
      "profile_id": 41440,
      "device_type": "0x0061",
      "in_clusters": [],
      "out_clusters": [
        "0x0021"
      ]
    }
  },
  "manufacturer": "_TZ3000_mraovvmm",
  "model": "TS011F",
  "class": "zigpy.device.Device"
}
```

It is BlitzWolf BW-SHP15 and it the "smartenergy_metering" does not work.

Adminiuga, 2021-10-01:
Try latest beta

MattWestb, 2021-10-01:
It shall being in the first beat of HA and all newer betas and releases.
https://rc.home-assistant.io/blog/2021/09/29/release-202110/#all-changes
Search for zha in the _all-changes_ section.

miit86, 2021-10-01:
Hi, Found it and used the Beta. Summary is working now. But metering summary has weird values. 

Used the integration calculation within HA, before using beta. Works almost perfectly. (its a little bit off)

Plug_oven_integration_kwh_calc_Oven:  2.203 Kwh
PLug_Oven_smart_energy_summary: 302.0 KwH

Plug_Server_integration_kwh_calc_Oven:  27.812 Kwh
PLug_Server_smart_energy_summary: 4,774.0 KwH

MattWestb, 2021-10-01:
If the first is you oven and is heating i think 2203 w (2,203 Kw) is realistic but its depend of your voltage and the cased power of it and if 240V is using 9.18 A that is realistic. And the sum is for 137 hours if using 9.18 A all the time.

The next is little more trickery then 27812 w is drawing 115 A at 240 V and i think your plug is not rated so high and the sum is for 176 hours if using 115 A all the time (sound expensive).

miit86, 2021-10-01:
> If the first is you oven and is heating i think 2203 w (2,203 Kw) is realistic but its depend of your voltage and the cased power of it and if 240V is using 9.18 A that is realistic. And the sum is [...]

The kwh integration just calculates the KwH total usage. If I turn on my oven 1800W for 1 Hour it ad's 1,8 KwH. 
I did not  use the oven that much the last 8 days. so the total KwH is 2,203 KwH
but the smartplug smart energy summary says 302.0 KwH. 

My server (always on) uses between 190 and 210 W and it used 27.812 Kwh since the day I configured the integration. (8days)
but the smartplug smart energy summary says 4,774.0 KwH. so thats not a match. also 4,774.0 it has a , and a .
thats also strange.

I'm trying to figure out how to reset the summary calculation on the plug. But cant find the option in my cluster.
Also the after power loss cluster can't be set. the value stays NONE. I Tried 0,1,2 or on,off, restore

So even in the BETA the plug does not work as expected.

MattWestb, 2021-10-01:
The question is how long the plug have running then its saving  the summery over the time and if it was retested or have being used before and have "old" data in its memory.

miit86, 2021-10-01:
> The question is how long the plug have running then its saving the summery over the time and if it was retested or have being used before and have "old" data in its memory.

I removed the device from ZHA and rejoined it.
the KwH is 0 now.
I will measure it from now on.

MattWestb, 2021-10-01:
Great !!

Interesting if its doing all OK in the end.

I still think your server looks little strange then its not realistic with 115 A or its one comma wrong in the readings (its still tuya device).

U * I = P (Voltage (V) * Current (A) = Power (W)).
Or P /  U = I or P / I = U.

Adminiuga, 2021-10-01:
Reconfigure zha device so it reports new values

miit86, 2021-10-01:
I gave the Device a Factory reset and a Reconfigure.
Now it's on 0.0 Kwh.

Server: 200W
KwH in hour: 0.2 

25 minutes later:
the sensor for summary reports. 9.0 Kwh. that's interesting

60 minutes later:
the sensor for summary reports 20.0KwH. so I think the comma is off.
Should be 0.2KwH.  20 / 100 = 0.2

Let me check later this day again with some calculations.

MattWestb, 2021-10-01:
Do you have the same with the first plug ?
It can being wrong in ZHA or its tuya that have cooking little wrong and then we need making one change in ZHA or Zigpy / Quirk for fixing that.

Still very interesting !!

miit86, 2021-10-01:
Hi,
Same problem with this plug. (both are TZ3000_ew3ldmgx)

Heater: 1650W
KwH in hour: 1.650

20 minutes later:
the sensor for summary reports. 56.0 Kwh.
1 hour: 56 x 3 = 168 KwH

Should be 1,650 KwH.   168 / **100** = 1,680 KwH

One more thing,
1 Plug is for my "ups" and connected servers. so it's important that the power state restores on power failure.
I used Manage Clusters,
Cluster: OnOff (Endpoint id: 1, Id: 0x0006, Type: in)
Attribute: start_up_on_off (id: 0x4003)
get attribute: Value: None
I tried to set attribute: 0, 1, 2, on, off, restore.
but none off them seems to work.

Thanxx.

MattWestb, 2021-10-01:
I think its better the devs is looking on the wrong diverter and let it running for some more hours so you is getting better data for comparing. and the interesting if other devices is doing  the same or if its only tuya devices.

I was today baying 2 LIDL outdoor plugs for my balcony (Christmas lights) and its also have the same attribute and i have trying reading it and always get `none` but is saying OK if setting it to 0,1 and 2 so i think its one undocumented tuya future (= bug in the firmware). But i have not testing if its working the setting it and then repower the plug.

MattWestb, 2021-10-01:
One explaining witch setting is possible for the cluster https://github.com/zigpy/zigpy/discussions/660#discussioncomment-418182 but i think its one bug in tuya device firmware that is looks not working.

miit86, 2021-10-01:
Hi,

found this in another post:
endpoint.saveClusterAttributeKeyValue('seMetering', {divisor: 100, multiplier: 1});

MattWestb, 2021-10-01:
For all normal zigbee devices or only tuya ?
I think is only for tuya TS011F devices then its need being added in one quirk for the device.

By the way the response then setting power on attribute and reading it back is:
```
0 = StartUpOnOff.Off
1 = StartUpOnOff.On
2 = StartUpOnOff.Toggle
255 = StartUpOnOff.PreviousValue
```
and `none` shall being not supported.

Z2M is adding all possible functions but its likely all is not working then not all functions is not tested with all devices.

miit86, 2021-10-01:
I think only Tuya.
TS011F and TS0121 

this plug does not have Toggle or PreviousValue

exposes.enum('power_outage_memory', ea.STATE_SET, ['on', 'off', 'restore'])

MattWestb, 2021-10-01:
If other smart plugs with power monitoring is OK the tuya is have doing it wrong (a gen) but its better then not having it working at all as some other manufactures have doing in the past.

I think both the tuya custom diverter and the no standard power on mode must being implanted in one quirk for all TS011F (and the TS0121 then user is having them) for getting all things working OK.

miit86, 2021-10-03:
Hi Matt,

found this in plug.py (quirk)
```
class PowerOnState(t.enum8):
    """Tuya power on state enum."""

    Off = 0x00
    On = 0x01
    LastState = 0x02

class OnOffRestorePowerCluster(CustomCluster, OnOff):
    """Tuya on off cluster with restore state."""

    attributes = OnOff.attributes.copy()
    attributes.update({0x8002: ("power_on_state", PowerOnState)})
```
I tried to make my own quirk see below: copied TS011F.py to "config\custom_zha_quirks"
But I wont Load.

```
"""TS011F"""
from zigpy.profiles import zha
from zigpy.quirks import CustomCluster, CustomDevice
import zigpy.types as t
from zigpy.zcl.clusters.general import Basic, Groups, OnOff, Ota, Scenes, Time
from zigpy.zcl.clusters.homeautomation import ElectricalMeasurement
from zigpy.zcl.clusters.smartenergy import Metering

from zhaquirks.const import (
    DEVICE_TYPE,
    ENDPOINTS,
    INPUT_CLUSTERS,
    MODELS_INFO,
    OUTPUT_CLUSTERS,
    PROFILE_ID,
)

class PowerOnState(t.enum8):
    """Tuya power on state enum."""

    Off = 0x00
    On = 0x01
    LastState = 0x02

class OnOffRestorePowerCluster(CustomCluster, OnOff):
    """Tuya on off cluster with restore state."""

    attributes = OnOff.attributes.copy()
    attributes.update({0x8002: ("power_on_state", PowerOnState)})

class Plug(CustomDevice):
    """Tuya plug with restore power state support."""

    signature = {
        MODELS_INFO: [
			("_TZ3000_ew3ldmgx", "TS011F"),
			
        ],
        ENDPOINTS: {
            # <SimpleDescriptor endpoint=1 profile=260 device_type=81
            # device_version=1
            # input_clusters=[0, 3, 4, 5, 6, 9, 1794, 2820, 57344, 57345]
            # output_clusters=[10, 25]>
            1: {
                PROFILE_ID: zha.PROFILE_ID,
                DEVICE_TYPE: zha.DeviceType.SMART_PLUG,
                INPUT_CLUSTERS: [
                    Basic.cluster_id,
                    Groups.cluster_id,
                    Scenes.cluster_id,
                    OnOff.cluster_id,
                    Metering.cluster_id,
                    ElectricalMeasurement.cluster_id,
                ],
                OUTPUT_CLUSTERS: [Time.cluster_id, Ota.cluster_id],
            },
			
	    # <SimpleDescriptor endpoint=242 profile=41440 device_type=97
            # device_version=1
            # input_clusters=[ ]
            # output_clusters=[33]>
			242: {
				PROFILE_ID: zha.PROFILE_ID,
                DEVICE_TYPE: zha.DeviceType.SMART_PLUG,
                INPUT_CLUSTERS: [ ],
                OUTPUT_CLUSTERS: [33],			
			},
			
			
        },
    }
    replacement = {
        ENDPOINTS: {
            1: {
                PROFILE_ID: zha.PROFILE_ID,
                DEVICE_TYPE: zha.DeviceType.SMART_PLUG,
                INPUT_CLUSTERS: [
                    Basic.cluster_id,
                    Groups.cluster_id,
                    Scenes.cluster_id,
                    OnOffRestorePowerCluster,
                    Metering.cluster_id,
                    ElectricalMeasurement.cluster_id,
                ],
                OUTPUT_CLUSTERS: [Time.cluster_id, Ota.cluster_id],
            },
        },
    }
```

MattWestb, 2021-10-03:
I have seen it and i think it shall working for your TS011F device and all TS011F devices.
In the quirk use `        MODEL: "TS011F",` instead then all TS011F devices with the same endpoints and cluster then ZHA is loading the quirk for them and not only your ID.

If the quirk is not loading is normally some endpoint or cluster that is not 110% the same on the device and the quirk. Take on more look and you is finding the wrong setting in the quirk (if you have doing the config for local quirk OK in HA settings).

MattWestb, 2021-10-03:
The second EP looks being wrong formatted:
```
			
	    # <SimpleDescriptor endpoint=242 profile=41440 device_type=97
            # device_version=1
            # input_clusters=[ ]
            # output_clusters=[33]>
			242: {
				PROFILE_ID: zha.PROFILE_ID,
                DEVICE_TYPE: zha.DeviceType.SMART_PLUG,
                INPUT_CLUSTERS: [ ],
                OUTPUT_CLUSTERS: [33],			
			},
			
			
        },
```

miit86, 2021-10-03:
just changed it.

```
			# <SimpleDescriptor endpoint=242 profile=41440 device_type=97
            # device_version=1
            # input_clusters=[]
            # output_clusters=[33]>
			242: {
				PROFILE_ID: 41440,
                DEVICE_TYPE: 97,
                INPUT_CLUSTERS: [],
                OUTPUT_CLUSTERS: [GreenPowerProxy.cluster_id],			
			},
```

MattWestb, 2021-10-03:
Good and EP 1 have 9 cluster in your signature and the quirk is only having 6 = no match.

MattWestb, 2021-10-03:
For the 2 unknown cluster you can do like this:
https://github.com/zigpy/zha-device-handlers/blob/83624ba2dea5bfb8af2c489211e26c877ca01c35/zhaquirks/tuya/ts0201_zemismart.py#L39

miit86, 2021-10-03:
Ok, how do I now the names of the missing clusters?
57344, 57345

3 is easy: 
Identify.cluster_id,

MattWestb, 2021-10-03:
I think the first 2 is energy (need checking the name of them) things and the last 2 is manufacture cluster that you very likely not needed so only using the method from the ts0201_zemismart.py for them.

MattWestb, 2021-10-03:
0x0702 = 1794 = Metering.cluster_id = OK
0X0b04 = 2820 = ElectricalMeasurement.cluster_id = OK

miit86, 2021-10-03:
I will Try to do that.

maybe stupid question?
could the unknown cluster be Voltage and Current.
or should these values be under measurement?

QUIRK:
```
"""Tuya plug."""
from zigpy.profiles import zha
from zigpy.quirks import CustomCluster, CustomDevice
import zigpy.types as t
from zigpy.zcl.clusters.general import Basic, Groups, OnOff, Ota, Scenes, Time
from zigpy.zcl.clusters.homeautomation import ElectricalMeasurement
from zigpy.zcl.clusters.smartenergy import Metering

from zhaquirks.const import (
    DEVICE_TYPE,
    ENDPOINTS,
    INPUT_CLUSTERS,
    MODELS_INFO,
    OUTPUT_CLUSTERS,
    PROFILE_ID,
)

class PowerOnState(t.enum8):
    """Tuya power on state enum."""

    Off = 0x00
    On = 0x01
    LastState = 0x02

class OnOffRestorePowerCluster(CustomCluster, OnOff):
    """Tuya on off cluster with restore state."""

    attributes = OnOff.attributes.copy()
    attributes.update({0x8002: ("power_on_state", PowerOnState)})

class Plug(CustomDevice):
    """Tuya plug with restore power state support."""

    signature = {
        MODELS_INFO: [
			("TS011F"),
			
        ],
        ENDPOINTS: {
            # <SimpleDescriptor endpoint=1 profile=260 device_type=81
            # device_version=1
            # input_clusters=[0, 3, 4, 5, 6, 9, 1794, 2820, 57344, 57345]
            # output_clusters=[10, 25]>
            1: {
                PROFILE_ID: zha.PROFILE_ID,
                DEVICE_TYPE: zha.DeviceType.SMART_PLUG,
                INPUT_CLUSTERS: [
                    Basic.cluster_id,
					Identify.cluster_id,
                    Groups.cluster_id,
                    Scenes.cluster_id,
                    OnOff.cluster_id,
                    Metering.cluster_id,
                    ElectricalMeasurement.cluster_id,
					0xe000, # Unknown
					0xe001, # Unknown
                ],
                OUTPUT_CLUSTERS: [Time.cluster_id, Ota.cluster_id],
            },
			
			# <SimpleDescriptor endpoint=242 profile=41440 device_type=97
            # device_version=1
[... 200 lines cut ...]
[0xD7E8:1:0x0006]: attempting to update onoff state - from cache: False
[0xd7e8:1:0x0006] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=GLOBAL_COMMAND manufacturer_specific=False is_reply=True disable_default_response=True> manufacturer=None tsn=47 command_id=Command.Read_Attributes_rsp>
[0xd7e8:zdo] ZDO request ZDOCmd.Node_Desc_rsp: [<Status.SUCCESS: 0>, 0xD7E8, NodeDescriptor(logical_type=<LogicalType.Router: 1>, complex_descriptor_available=0, user_descriptor_available=0, reserved=0, aps_flags=0, frequency_band=<FrequencyBand.Freq2400MHz: 8>, mac_capability_flags=<MACCapabilityFlags.AllocateAddress|RxOnWhenIdle|MainsPowered|FullFunctionDevice: 142>, manufacturer_code=4098, maximum_buffer_size=82, maximum_incoming_transfer_size=82, server_mask=11264, maximum_outgoing_transfer_size=82, descriptor_capability_field=<DescriptorCapability.NONE: 0>, *allocate_address=True, *is_alternate_pan_coordinator=False, *is_coordinator=False, *is_end_device=False, *is_full_function_device=True, *is_mains_powered=True, *is_receiver_on_when_idle=True, *is_router=True, *is_security_capable=False)]
[0xd7e8:zdo] Unsupported ZDO request:ZDOCmd.Node_Desc_rsp
[0xd7e8:zdo] ZDO request ZDOCmd.Node_Desc_rsp: [<Status.SUCCESS: 0>, 0xD7E8, NodeDescriptor(logical_type=<LogicalType.Router: 1>, complex_descriptor_available=0, user_descriptor_available=0, reserved=0, aps_flags=0, frequency_band=<FrequencyBand.Freq2400MHz: 8>, mac_capability_flags=<MACCapabilityFlags.AllocateAddress|RxOnWhenIdle|MainsPowered|FullFunctionDevice: 142>, manufacturer_code=4098, maximum_buffer_size=82, maximum_incoming_transfer_size=82, server_mask=11264, maximum_outgoing_transfer_size=82, descriptor_capability_field=<DescriptorCapability.NONE: 0>, *allocate_address=True, *is_alternate_pan_coordinator=False, *is_coordinator=False, *is_end_device=False, *is_full_function_device=True, *is_mains_powered=True, *is_receiver_on_when_idle=True, *is_router=True, *is_security_capable=False)]
[0xd7e8:zdo] Unsupported ZDO request:ZDOCmd.Node_Desc_rsp
[0x0c77:1:0x0000] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=GLOBAL_COMMAND manufacturer_specific=False is_reply=True disable_default_response=True> manufacturer=None tsn=15 command_id=Command.Report_Attributes>
[0x0c77:1:0x0000] ZCL request 0x000a: [[Attribute(attrid=1, value=<TypeValue type=uint8_t, value=67>)]]
[0x0c77:1:0x0000] Attribute report received: app_version=67
[0xd44e:1:0x0b04] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=GLOBAL_COMMAND manufacturer_specific=False is_reply=True disable_default_response=True> manufacturer=None tsn=108 command_id=Command.Report_Attributes>
[0xd44e:1:0x0b04] ZCL request 0x000a: [[Attribute(attrid=1291, value=<TypeValue type=int16s, value=196>)]]
[0xd44e:1:0x0b04] Attribute report received: active_power=196
[0x0432:1:0x0b04]: async_update
[0x8DA2:1:0x0b04]: async_update
[0xD44E:1:0x0b04]: async_update
[0xD7E8:1:0x0b04]: async_update
[0x0432:1:0x0b04] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=GLOBAL_COMMAND manufacturer_specific=False is_reply=True disable_default_response=True> manufacturer=None tsn=49 command_id=Command.Read_Attributes_rsp>
[0x8da2:1:0x0b04] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=GLOBAL_COMMAND manufacturer_specific=False is_reply=True disable_default_response=True> manufacturer=None tsn=51 command_id=Command.Read_Attributes_rsp>
[... cut ...]
[0xd7e8:1:0x0b04] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=GLOBAL_COMMAND manufacturer_specific=False is_reply=True disable_default_response=True> manufacturer=None tsn=55 command_id=Command.Read_Attributes_rsp>
[0xd44e:1:0x0702] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=GLOBAL_COMMAND manufacturer_specific=False is_reply=True disable_default_response=True> manufacturer=None tsn=109 command_id=Command.Report_Attributes>
[0xd44e:1:0x0702] ZCL request 0x000a: [[Attribute(attrid=0, value=<TypeValue type=uint48_t, value=936>)]]
[0xd44e:1:0x0702] Attribute report received: current_summ_delivered=936

```

MattWestb, 2021-10-03:
Your formatting is is worse then my but you can using https://black.vercel.app/ for fixing it (past your code in the left side and you is getting the fixed in the left side).

The quirks is not loaded / or ZHA have not loading it then its have to bad formatting but that you can see in HA log.

Try this its ficed with back.
```
"""Tuya plug."""
from zigpy.profiles import zha
from zigpy.quirks import CustomCluster, CustomDevice
import zigpy.types as t
from zigpy.zcl.clusters.general import Basic, Groups, OnOff, Ota, Scenes, Time
from zigpy.zcl.clusters.homeautomation import ElectricalMeasurement
from zigpy.zcl.clusters.smartenergy import Metering

from zhaquirks.const import (
    DEVICE_TYPE,
    ENDPOINTS,
    INPUT_CLUSTERS,
    MODELS_INFO,
    OUTPUT_CLUSTERS,
    PROFILE_ID,
)

class PowerOnState(t.enum8):
    """Tuya power on state enum."""

    Off = 0x00
    On = 0x01
    LastState = 0x02

class OnOffRestorePowerCluster(CustomCluster, OnOff):
    """Tuya on off cluster with restore state."""

    attributes = OnOff.attributes.copy()
    attributes.update({0x8002: ("power_on_state", PowerOnState)})

class Plug(CustomDevice):
    """Tuya plug with restore power state support."""

    signature = {
        MODELS_INFO: [
            ("TS011F"),
        ],
        ENDPOINTS: {
            # <SimpleDescriptor endpoint=1 profile=260 device_type=81
            # device_version=1
            # input_clusters=[0, 3, 4, 5, 6, 9, 1794, 2820, 57344, 57345]
            # output_clusters=[10, 25]>
            1: {
                PROFILE_ID: zha.PROFILE_ID,
                DEVICE_TYPE: zha.DeviceType.SMART_PLUG,
                INPUT_CLUSTERS: [
                    Basic.cluster_id,
                    Identify.cluster_id,
                    Groups.cluster_id,
                    Scenes.cluster_id,
                    OnOff.cluster_id,
                    Metering.cluster_id,
                    ElectricalMeasurement.cluster_id,
                    0xE000,  # Unknown
                    0xE001,  # Unknown
                ],
                OUTPUT_CLUSTERS: [Time.cluster_id, Ota.cluster_id],
            },
            # <SimpleDescriptor endpoint=242 profile=41440 device_type=97
            # device_version=1
            # input_clusters=[]
            # output_clusters=[33]>
            242: {
                PROFILE_ID: 41440,
                DEVICE_TYPE: 97,
                INPUT_CLUSTERS: [],
                OUTPUT_CLUSTERS: [GreenPowerProxy.cluster_id],
            },
        },
    }
    replacement = {
        ENDPOINTS: {
            1: {
                PROFILE_ID: zha.PROFILE_ID,
                DEVICE_TYPE: zha.DeviceType.SMART_PLUG,
                INPUT_CLUSTERS: [
                    Basic.cluster_id,
                    Identify.cluster_id,
                    Groups.cluster_id,
                    Scenes.cluster_id,
                    OnOffRestorePowerCluster,
                    Metering.cluster_id,
                    ElectricalMeasurement.cluster_id,
                    0xE000,  # Unknown
                    0xE001,  # Unknown
                ],
                OUTPUT_CLUSTERS: [Time.cluster_id, Ota.cluster_id],
            },
        },
    }
```

miit86, 2021-10-03:
Now I get error in HA about identify.

```
Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/config_entries.py", line 304, in async_setup
    result = await component.async_setup_entry(hass, self)  # type: ignore
  File "/usr/src/homeassistant/homeassistant/components/zha/__init__.py", line 99, in async_setup_entry
    setup_quirks(config)
  File "/usr/local/lib/python3.9/site-packages/zhaquirks/__init__.py", line 403, in setup
    importer.find_module(modname).load_module(modname)
  File "<frozen importlib._bootstrap_external>", line 529, in _check_name_wrapper
  File "<frozen importlib._bootstrap_external>", line 1029, in load_module
  File "<frozen importlib._bootstrap_external>", line 854, in load_module
  File "<frozen importlib._bootstrap>", line 274, in _load_module_shim
  File "<frozen importlib._bootstrap>", line 711, in _load
  File "<frozen importlib._bootstrap>", line 680, in _load_unlocked
  File "<frozen importlib._bootstrap_external>", line 850, in exec_module
  File "<frozen importlib._bootstrap>", line 228, in _call_with_frames_removed
  File "/config/custom_zha_quirks/TS011F.py", line 34, in <module>
    class Plug(CustomDevice):
  File "/config/custom_zha_quirks/TS011F.py", line 51, in Plug
    Identify.cluster_id,
NameError: name 'Identify' is not defined
```

MattWestb, 2021-10-03:
```
  File "/config/custom_zha_quirks/TS011F.py", line 51, in Plug
    Identify.cluster_id,
NameError: name 'Identify' is not defined
``` 
Shall being added in the `import: from zigpy.zcl.clusters.general import Basic, Groups, OnOff, Ota, Scenes, Time`

miit86, 2021-10-03:
I changed this:
``` from zigpy.zcl.clusters.general import Basic, Groups, OnOff, Ota, Scenes, Time, GreenPowerProxy, Identify ```
rebooting now

MattWestb, 2021-10-03:
You need putting it in sorted way then HA is using isort but it working for testing (between Groups and OnOff).
The lines:
```        MODELS_INFO: [
            ("TS011F"),
        ],
```
shall being:
```
        MODEL: "TS011F",
```
And run black after all changes so you have the formatting is OK or HA is not loading it.

MattWestb, 2021-10-03:
This is working for testing:
```
from zigpy.zcl.clusters.general import Basic, Groups, OnOff, Ota, Scenes, Time, GreenPowerProxy, Identify
```
Must being sorted like this:
```
from zigpy.zcl.clusters.general import Basic, GreenPowerProxy, Groups, Identify, OnOff, Ota, Scenes, Time
```
Or you cant adding it HA (test is blocking bad sorting of variables).

MattWestb, 2021-10-03:
If you have debug logging for zhaquirks in your config you is getting somthing like this then starting in HA.
```
2021-10-02 20:42:20 DEBUG (MainThread) [zhaquirks] Loading quirks module zhaquirks.ikea.fourbtnremote 
```
(from IKEA 4 button remote) with your new quirk name in stead of the IKEA one ;-))

miit86, 2021-10-03:
Hi Matt,

Didn't do that but the error is gone.
But the Quirk is still not loaded for TS011F

I will add this to make sure quirk is loaded.
```
logger:
  default: info
  logs:
    zhaquirks: debug
```

result:
```2021-10-03 14:07:57 DEBUG (MainThread) [zhaquirks] Loading custom quirks module TS011F```

So the Quirk is loaded. 
When I remove the device and add it again. it should match:

NO LUCK :(
```
[0xc962] Read model 'TS011F' and manufacturer '_TZ3000_ew3ldmgx' from <Endpoint id=1 in=[basic:0x0000, identify:0x0003, groups:0x0004, scenes:0x0005, on_off:0x0006, smartenergy_metering:0x0702, electrical_measurement:0x0B04, None:0xE000, None:0xE001] out=[ota:0x0019, time:0x000A] status=<Status.ZDO_INIT: 1>>
[0xc962] Discovered basic device information for <Device model='TS011F' manuf='_TZ3000_ew3ldmgx' nwk=0xC962 ieee=84:fd:27:ff:fe:66:14:bb is_initialized=True>
Device is initialized <Device model='TS011F' manuf='_TZ3000_ew3ldmgx' nwk=0xC962 ieee=84:fd:27:ff:fe:66:14:bb is_initialized=True>
Checking quirks for _TZ3000_ew3ldmgx TS011F (84:fd:27:ff:fe:66:14:bb)
Considering <class 'TS011F.Plug'>
Fail because device_type mismatch on at least one endpoint
Considering <class 'zhaquirks.xbee.xbee_io.XBeeSensor'>
Fail because endpoint list mismatch: {232, 230} {1, 242}
Considering <class 'zhaquirks.xbee.xbee3_io.XBee3Sensor'>
Fail because endpoint list mismatch: {232, 230} {1, 242}
Considering <class 'zhaquirks.smartthings.tag_v4.SmartThingsTagV4'>
Fail because endpoint list mismatch: {1} {1, 242}
Considering <class 'zhaquirks.smartthings.multi.SmartthingsMultiPurposeSensor'>
Fail because endpoint list mismatch: {1} {1, 242}
Considering <class 'zhaquirks.netvox.z308e3ed.Z308E3ED'>
Fail because endpoint list mismatch: {1} {1, 242}
Considering <class 'zhaquirks.gledopto.soposhgu10.SoposhGU10'>
Fail because endpoint list mismatch: {11, 13} {1, 242}
Considering <class 'bellows.zigbee.application.EZSPCoordinator'>
Fail because endpoint list mismatch: {1} {1, 242}
device - 0xC962:84:fd:27:ff:fe:66:14:bb entering async_device_initialized - is_new_join: True
device - 0xC962:84:fd:27:ff:fe:66:14:bb has joined the ZHA zigbee network
[0xC962](TS011F): started configuration
[0xC962:ZDO](TS011F): 'async_configure' stage succeeded
Error handling '_save_attribute' event with (84:fd:27:ff:fe:66:14:bb, 1, 0, 4, '_TZ3000_ew3ldmgx') params: FOREIGN KEY constraint failed
[0xC962:1:0x0000]: finished channel configuration
Error handling '_save_attribute' event with (84:fd:27:ff:fe:66:14:bb, 1, 0, 5, 'TS011F') params: FOREIGN KEY constraint failed
```

MattWestb, 2021-10-03:
ZHA have loading your quirk OK :-)))))))))))))

But somthing is wrong with the signature in your quirk so its not being one match:
```
Checking quirks for _TZ3000_ew3ldmgx TS011F (84:fd:27:ff:fe:66:14:bb)
Considering <class 'TS011F.Plug'>
Fail because device_type mismatch on at least one endpoint
```
Post your last version and i can taking on look on it.

miit86, 2021-10-03:
```
"""Tuya plug."""
from zigpy.profiles import zha
from zigpy.quirks import CustomCluster, CustomDevice
import zigpy.types as t
from zigpy.zcl.clusters.general import Basic, GreenPowerProxy, Groups, Identify, OnOff, Ota, Scenes, Time
from zigpy.zcl.clusters.homeautomation import ElectricalMeasurement
from zigpy.zcl.clusters.smartenergy import Metering

from zhaquirks.const import (
    DEVICE_TYPE,
    ENDPOINTS,
    INPUT_CLUSTERS,
    MODEL,
    OUTPUT_CLUSTERS,
    PROFILE_ID,
)

class PowerOnState(t.enum8):
    """Tuya power on state enum."""

    Off = 0x00
    On = 0x01
    LastState = 0x02

class OnOffRestorePowerCluster(CustomCluster, OnOff):
    """Tuya on off cluster with restore state."""

    attributes = OnOff.attributes.copy()
    attributes.update({0x8002: ("power_on_state", PowerOnState)})

class Plug(CustomDevice):
    """Tuya plug with restore power state support."""

    signature = {
        MODEL: "TS011F",
        ENDPOINTS: {
            # <SimpleDescriptor endpoint=1 profile=260 device_type=81
            # device_version=1
            # input_clusters=[0, 3, 4, 5, 6, 9, 1794, 2820, 57344, 57345]
            # output_clusters=[10, 25]>
            1: {
                PROFILE_ID: zha.PROFILE_ID,
                DEVICE_TYPE: zha.DeviceType.SMART_PLUG,
                INPUT_CLUSTERS: [
                    Basic.cluster_id,
                    Identify.cluster_id,
                    Groups.cluster_id,
                    Scenes.cluster_id,
                    OnOff.cluster_id,
                    Metering.cluster_id,
                    ElectricalMeasurement.cluster_id,
                    0xE000,  # Unknown
                    0xE001,  # Unknown
                ],
                OUTPUT_CLUSTERS: [Time.cluster_id, Ota.cluster_id],
            },
            # <SimpleDescriptor endpoint=242 profile=41440 device_type=97
            # device_version=1
            # input_clusters=[]
            # output_clusters=[33]>
            242: {
                PROFILE_ID: 41440,
                DEVICE_TYPE: 97,
                INPUT_CLUSTERS: [],
                OUTPUT_CLUSTERS: [GreenPowerProxy.cluster_id],
            },
        },
    }
    replacement = {
        ENDPOINTS: {
            1: {
                PROFILE_ID: zha.PROFILE_ID,
                DEVICE_TYPE: zha.DeviceType.SMART_PLUG,
                INPUT_CLUSTERS: [
                    Basic.cluster_id,
                    Identify.cluster_id,
                    Groups.cluster_id,
                    Scenes.cluster_id,
                    OnOffRestorePowerCluster,
                    Metering.cluster_id,
                    ElectricalMeasurement.cluster_id,
                    0xE000,  # Unknown
                    0xE001,  # Unknown
                ],
                OUTPUT_CLUSTERS: [Time.cluster_id, Ota.cluster_id],
            },
        },
    }
```

MattWestb, 2021-10-03:
By the way dont forgetting deleting the `__pycache__` folder (in your local quirk forder) after making changes in the quirk of the system can using some old parameters saved in it.

miit86, 2021-10-03:
I know the ```__pycache__```

what does: ```Considering <class 'TS011F.Plug'>``` do?

MattWestb, 2021-10-03:
ZHA is testing if the quirk "TS011F.Plug" can being matched with the device.
If the device signature is matching that in the quirk ZHA is loading the quirk for the device.
So for the moment is only about the device signature for getting it loaded for your device.

MattWestb, 2021-10-03:
Your device is `"device_type": "0x010a"` = 266 is not matching `DEVICE_TYPE: zha.DeviceType.SMART_PLUG,` (= device_type=81) 
I must finding the right type.

MattWestb, 2021-10-03:
I have finding it in IKEA plug :-))
```
                "device_type": zha.DeviceType.ON_OFF_PLUG_IN_UNIT,
```

MattWestb, 2021-10-03:
And then  this line is false:
```
            # <SimpleDescriptor endpoint=1 profile=260 device_type=81
```
Shall being:
```
            # <SimpleDescriptor endpoint=1 profile=260 device_type=266
```

miit86, 2021-10-03:
JEEH :)

Youre amazing, I find it difficult to find al the name/numbers/hex that have to match.
Also the Power Restore is working.

I would like to go on:
- With the metering. Let me try to find a example somewhere to divide the value by 100.
- Also it would be nice if there is a sensor for Voltage and Current.

MattWestb, 2021-10-03:
# Great done !!
For the multiplier you need finding one quirk that have implanting it and normally its easiest with tuya but it can being implanted in other so look around and you is finding some.

Then the problem to implanting it i think its easiest making one "new" cluster with the / 100 and adding it in the replacement.

I dont knowing if the device is doing Voltage and Current but if its being made by Zigbee standard it shall working out of the box or if not you need finding how to hacking it in the quirk.

MattWestb, 2021-10-03:
By the way can you posing the quirk that is working so other user can testing it and getting there devices getting little _taste_ of the power Metering (little high but still working ;-)) ).

MattWestb, 2021-10-03:
Adminiuga have doing it for tuya AIR but is using tuya PD and not normal Zigbee clusters like your device but perhaps its one clue how to do it.
https://github.com/zigpy/zha-device-handlers/blob/dev/zhaquirks/tuya/air/__init__.py
I looking little more around for hacks.

MattWestb, 2021-10-03:
Bitron/SMaBiT thermostats is "patching" the battery and adding it in one new cluster so i thing its one good example how to do the "new" patched cluster. https://github.com/zigpy/zha-device-handlers/blob/dev/zhaquirks/bitron/thermostat.py.
Then its "only" putting in the multiplexer with "normal" way like `"new" attribute = old * 100` i think (= not knowing)

MattWestb, 2021-10-03:
One more IKEA thing !!

IKEA controllers is reporting battery then full as 100% and Zigbe shall doing it as 200%.
In this is the 2* being done and also the cluster https://github.com/zigpy/zha-device-handlers/blob/83624ba2dea5bfb8af2c489211e26c877ca01c35/zhaquirks/__init__.py#L172 .

I think its easiest making the same and using the "new" cluster in the replacement but implanting it in tuya `__init__.py`or in your quirk if no more devices is needing it.

Its looks very easy and short implanting only changing the name of the cluster and attribute and the multiplier :-)))

MattWestb, 2021-10-03:
I have making one PR #1056 that is implanting the power on state in the tuya INIT and also removing it in the plug.py and adding the new on/off cluster in it and renaming plug.py to ts0121_plug.py. #1055
So your quirk can being named ts011f_plug.py and also using the power on state from the INIT.

I think i shall doing the same for TS0101 then i have getting 2 LIDL outdoor plugs but need writing one quirk for it :-(((

miit86, 2021-10-03:
Hi Matt,

Using the power on state from the INIT is cool. But plug.py only has lines for power on state. is it absolute then?
I will name my quirk ts011f_plug.py

I'm not used to code, so it's difficult to understand how to build it.

If I add this below I will get a new output. also see replacements
But I have no clue on how to change the numbers from metering and change it before HA gets the numbers.

```
class TS011FMetering(Metering):
    """Reported Value from current_summ_delivered dived by 100. """

    current_summ_delivered = 0x00
    current_summ_delivered = current_summ_delivered / 100 
```
Replacements:
```
                INPUT_CLUSTERS: [
                    Basic.cluster_id,
                    Identify.cluster_id,
                    Groups.cluster_id,
                    Scenes.cluster_id,
                    OnOffRestorePowerCluster,
                    TS011FMetering,
                    ElectricalMeasurement.cluster_id,
                    0xE000,  # Unknown
                    0xE001,  # Unknown
```

MattWestb, 2021-10-03:
I think using this from the quirk root INIT:
```
class DoublingPowerConfigurationCluster(CustomCluster, PowerConfiguration):
    """PowerConfiguration cluster implementation.
    This implementation doubles battery pct remaining for non standard devices
    that don't follow the reporting spec.
    """

    cluster_id = PowerConfiguration.cluster_id
    BATTERY_PERCENTAGE_REMAINING = 0x0021

    def _update_attribute(self, attrid, value):
        if attrid == self.BATTERY_PERCENTAGE_REMAINING:
            value = value * 2
        super()._update_attribute(attrid, value)
```
and changing the cluster name `PowerConfiguration` to wot you will changing.
Then is the attribute you like changing `BATTERY_PERCENTAGE_REMAINING = 0x0021` to your attibute.

And the recalculating is made with `value = value * 2` so making your / 100 instead.

miit86, 2021-10-03:
Something like this:
```
class DividingMeteringCluster(CustomCluster, Metering):
    """Metering cluster implementation.
    This implementation divides the KwH for tuya devices
    that don't follow the reporting spec.
    """

    cluster_id = Metering.cluster_id
    current_summ_delivered = 0x0000

    def _update_attribute(self, attrid, value):
        if attrid == self.current_summ_delivered:
            value = value / 100
        super()._update_attribute(attrid, value)
```

how to implement this in my quirk?

MattWestb, 2021-10-03:
I think its looks OK but im no coding worrier :-((

If its only used for your device in then in your quirk and if its being used for more devices in the tuya INIT or if global in the quirk INIT = in your device quirk.

I dont knowing if its need more imports but try putting it in.

Do it like the power on state state class is made (same line spaces before and after).
And run black after before you is putting in your system then if its bad formatted it can being that HA is not starting

miit86, 2021-10-03:
Hi, 

WauW, That worked :P ..! Very Nice :)

[16 more replies]

closed 2021-10-16 by miit86