zigpy/zha-device-handlers #1047: [Device Support Request] Tuya/MOES "Smart Brightness Thermometer" TS0222 _TYZB01_kvwjujy9

vvuk, 2021-09-27:
This device seems to be sort-of supported -- the measurements do come through ZHA. However, something in the communication goes wrong, and the device keeps resetting itself every few minutes as soon as it's part of the zigbee network.

Here's the device signature:

```
{
  "node_descriptor": "NodeDescriptor(logical_type=<LogicalType.EndDevice: 2>, complex_descriptor_available=0, user_descriptor_available=0, reserved=0, aps_flags=0, frequency_band=<FrequencyBand.Freq2400MHz: 8>, mac_capability_flags=<MACCapabilityFlags.AllocateAddress: 128>, manufacturer_code=4098, maximum_buffer_size=82, maximum_incoming_transfer_size=82, server_mask=11264, maximum_outgoing_transfer_size=82, descriptor_capability_field=<DescriptorCapability.NONE: 0>, *allocate_address=True, *is_alternate_pan_coordinator=False, *is_coordinator=False, *is_end_device=True, *is_full_function_device=False, *is_mains_powered=False, *is_receiver_on_when_idle=False, *is_router=False, *is_security_capable=False)",
  "endpoints": {
    "1": {
      "profile_id": 260,
      "device_type": "0x0106",
      "in_clusters": [
        "0x0000",
        "0x0001",
        "0x0400"
      ],
      "out_clusters": [
        "0x000a",
        "0x0019"
      ]
    },
    "2": {
      "profile_id": 260,
      "device_type": "0x0302",
      "in_clusters": [
        "0x0402",
        "0x0405"
      ],
      "out_clusters": []
    }
  },
  "manufacturer": "_TYZB01_kvwjujy9",
  "model": "TS0222",
  "class": "zigpy.device.Device"
}
```

Which all seems fine (I think? weird that things like measurements are "in" clusters but zigbee always confuses me).  I can try to provide some logs of the comms right before the device resets.

vvuk, 2021-09-27:
Here's  a log file roughly from when the device (0x0279, cc:86:ec:ff:fe:f3:ea:c7) joins, and then through when it resets and re-joins again.  There's some other chatter with some of my other devices in here as well but not much.

It looks like there's regular comms to the device until 15:14:47 when a MAC_INDIRECT_TIMEOUT shows up, but then looks like a few more normal measured values show up, and then a final sendUnicast to the device, with a 0x00fc reply at 15:14:51.  And then nothing until 15:17:15 when the rejoin happens.

[LOG2.txt](https://github.com/zigpy/zha-device-handlers/files/7239665/LOG2.txt)

vvuk, 2021-09-27:
I should add -- I added a custom device handler to add 0x0400 to the in_clusters of endpoint 2 as well.  The device seems to send 0x0400 (illuminance) on both endpoints; with its self-reported config, ZHA was reporting "no cluster on endpoint" errors.

MattWestb, 2021-09-28:
You can taking one look on #961 there some users is trying getting it working and its looks like the time cluster is making the device leaving the network if its not getting OK response on it.

Great longs you have posted !!

vvuk, 2021-09-28:
Arg, I swear I did a bunch of searching, I somehow completely missed the existing issue -- especially since it has a bunch of steps I've already taken (quirk, 0x0400 on both endpoints etc).  Thank you, sorry for the spam!

closed 2021-09-28 by vvuk

MattWestb, 2021-09-28:
No problems but i think its better working together on the same device then its being easier getting it working and not need reinventing the wheel one more time or 2 ;-))

I have doing the same some time and the making one PR for one device and some one have doing the same some days before and getting it added so the work was for no.