zigpy/zha-device-handlers #1049: [Device Support Request] new variant of lumi.plug (ZNCZ02LM)

jonwaland, 2021-09-29:
**Is your feature request related to a problem? Please describe.**
There is a 2nd variant of the llumi.plug which doens't match the current lumi.plug quirk

**Describe the solution you'd like**
can anyone assist with a new quirk?

**Device signature - this can be acquired by removing the device from ZHA and pairing it again from the add devices screen. Be sure to add the entire content of the log panel after pairing the device to a code block below this line.**
```
{
  "node_descriptor": "NodeDescriptor(byte1=1, byte2=64, mac_capability_flags=142, manufacturer_code=4447, maximum_buffer_size=127, maximum_incoming_transfer_size=100, server_mask=0, maximum_outgoing_transfer_size=100, descriptor_capability_field=0)",
  "endpoints": {
    "1": {
      "profile_id": 260,
      "device_type": "0x0051",
      "in_clusters": [
        "0x0000",
        "0x0001",
        "0x0002",
        "0x0003",
        "0x0004",
        "0x0005",
        "0x0006",
        "0x000a",
        "0x0010"
      ],
      "out_clusters": [
        "0x000a",
        "0x0019"
      ]
    },
    "2": {
      "profile_id": 260,
      "device_type": "0x0009",
      "in_clusters": [
        "0x000c"
      ],
      "out_clusters": [
        "0x0004",
        "0x000c"
      ]
    },
    "3": {
      "profile_id": 260,
      "device_type": "0x0053",
      "in_clusters": [
        "0x000c"
      ],
      "out_clusters": [
        "0x000c"
      ]
    },
    "100": {
      "profile_id": 260,
      "device_type": "0x0107",
      "in_clusters": [
        "0x000f"
      ],
      "out_clusters": [
        "0x0004",
        "0x000f"
      ]
    }
  },
  "manufacturer": "LUMI",
  "model": "lumi.plug",
  "class": "zigpy.device.Device"
```

**Additional context**
see https://github.com/home-assistant/core/issues/47650 for differences between this and the current "lumi.plug"
also see https://www.zigbee2mqtt.io/devices/ZNCZ02LM.html

hacshacdgacs, 2022-05-30:
another bump

hacshacdgacs, 2022-11-26:
Still unresolved with latest update

TheJulianJES, 2023-01-05:
Other ``lumi.plug`` issues (issues might be solved together when someone gets to it):
- https://github.com/zigpy/zha-device-handlers/issues/1135
- https://github.com/zigpy/zha-device-handlers/issues/1880

Also related:
- https://github.com/home-assistant/core/issues/47650

TheJulianJES, 2023-01-14:
See https://github.com/zigpy/zha-device-handlers/issues/1135#issuecomment-1382627491 for testing a custom quirk that *could* fix this.

jonwaland, 2023-01-16:
> See [#1135 (comment)](https://github.com/zigpy/zha-device-handlers/issues/1135#issuecomment-1382627491) for testing a custom quirk that _could_ fix this.

testing it now. thanks.

TheJulianJES, 2023-01-16:
There's an updated "v3" (still very much in testing though) in this comment: https://github.com/zigpy/zha-device-handlers/issues/1135#issuecomment-1383695421

That version should now create a total consumption sensor (need to re-pair) and should give "warning" logs with messages that I need for debugging.

closed 2023-01-23 by dmulcahey