zigpy/zha-device-handlers #1050: LDS ZBT-CCTSwitch-D0001 direct binding

puzz3d, 2021-09-29:
I have several kits (switch + bulb) from HomeDepot and many additional light bulbs from them. So I want to control different light groups by these remotes.

Problem: can't set up direct binding through Conbee II and ZHA.
Binding with bulb or group by managing clusters doesn't work (despite "success" messages in the log).
Binding by adding a bulb to a group "No name group 0x4003" actually works, BUT it is a solution for just one remote because there is only one 0x4003 group no matter how many remotes I had paired. 

Is there any solution to create several groups to get direct binding work right?

[image]
[image]

MattWestb, 2021-09-30:
The first part then binding light to one remote is making one group of lights and then going to the remotes device card and open "manage cluster".
In the lower part you have device and group binding.
In the group binding select your group for your light and check the on/off and light level and more clusters if you need them being bounded.
Click on bind and waking the remote up by clicking on one button so its waking up and can receive the configuration you have made.

Some remotes can having problem with reporting after changing binding the do one "reconfigure this device" from the device card and waking the remote up before sending the new configuration.

puzz3d, 2021-09-30:
> The first part then binding light to one remote is making one group of lights and then going to the remotes device card and open "manage cluster". In the lower part you have device and group [...]

Thank you, I have tried this. Binding goes well but nothing will happen when I press buttons after that, no response from light group.
I'll post some logs later.

puzz3d, 2021-09-30:
Here we go:
Resetting the remote then adding it to ZHA.
Going to "manage cluster", selecting the light group, and all the clusters. Binding and pressing a remote button to awake it.
**Got this:**

2021-09-30 23:45:44 DEBUG (MainThread) [zigpy_deconz.api] Device state changed response: [<DeviceState.128|APSDE_DATA_REQUEST_SLOTS_AVAILABLE|APSDE_DATA_INDICATION|2: 170>, 0]
2021-09-30 23:45:44 DEBUG (MainThread) [zigpy_deconz.api] Command Command.aps_data_indication (1, 1)
2021-09-30 23:45:44 DEBUG (MainThread) [zigpy_deconz.api] APS data indication response: [30, <DeviceState.APSDE_DATA_REQUEST_SLOTS_AVAILABLE|APSDE_DATA_INDICATION|2: 42>, <DeconzAddress address_mode=ADDRESS_MODE.NWK address=0x0000>, 1, <DeconzAddress address_mode=ADDRESS_MODE.NWK address=0xf268>, 1, 260, 6, b'\x08\x13\n\x00\x00\x10\x00', 0, 175, 255, 252, 0, 131, 0, -61]
2021-09-30 23:45:44 DEBUG (MainThread) [zigpy.zcl] [0xf268:1:0x0006] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=GLOBAL_COMMAND manufacturer_specific=False is_reply=True disable_default_response=False> manufacturer=None tsn=19 command_id=Command.Report_Attributes>
2021-09-30 23:45:44 DEBUG (MainThread) [zigpy.zcl] [0xf268:1:0x0006] ZCL request 0x000a: [[Attribute(attrid=0, value=<TypeValue type=Bool, value=Bool.false>)]]
2021-09-30 23:45:44 DEBUG (MainThread) [zigpy.zcl] [0xf268:1:0x0006] Attribute report received: on_off=0
2021-09-30 23:45:44 DEBUG (MainThread) [zigpy_deconz.zigbee.application] Sending Zigbee request with tsn 19 under 179 request id, data: b'18130b0a00'
2021-09-30 23:45:44 DEBUG (MainThread) [zigpy_deconz.api] 'aps_data_indication' response from <DeconzAddress address_mode=ADDRESS_MODE.NWK address=0xf268>, ep: 1, profile: 0x0104, cluster_id: 0x0006, data: b'08130a00001000'
2021-09-30 23:45:44 DEBUG (MainThread) [zigpy_deconz.api] Command Command.aps_data_indication (1, 1)
2021-09-30 23:45:44 DEBUG (MainThread) [zigpy_deconz.api] APS data indication response: [30, <DeviceState.APSDE_DATA_REQUEST_SLOTS_AVAILABLE|2: 34>, <DeconzAddress address_mode=ADDRESS_MODE.NWK address=0x0000>, 1, <DeconzAddress address_mode=ADDRESS_MODE.NWK address=0xf268>, 1, 260, 8, b'\x08\x14\n\x00\x00 \x03', 0, 175, 247, 211, 0, 131, 0, -63]
2021-09-30 23:45:44 DEBUG (MainThread) [zigpy.zcl] [0xf268:1:0x0008] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=GLOBAL_COMMAND manufacturer_specific=False is_reply=True disable_default_response=False> manufacturer=None tsn=20 command_id=Command.Report_Attributes>
2021-09-30 23:45:44 DEBUG (MainThread) [zigpy.zcl] [0xf268:1:0x0008] ZCL request 0x000a: [[Attribute(attrid=0, value=<TypeValue type=uint8_t, value=3>)]]
2021-09-30 23:45:44 DEBUG (MainThread) [zigpy.zcl] [0xf268:1:0x0008] Attribute report received: current_level=3
2021-09-30 23:45:44 DEBUG (MainThread) [zigpy_deconz.zigbee.application] Sending Zigbee request with tsn 20 under 180 request id, data: b'18140b0a00'
2021-09-30 23:45:44 DEBUG (MainThread) [zigpy_deconz.api] 'aps_data_indication' response from <DeconzAddress address_mode=ADDRESS_MODE.NWK address=0xf268>, ep: 1, profile: 0x0104, cluster_id: 0x0008, data: b'08140a00002003'
2021-09-30 23:45:44 DEBUG (MainThread) [zigpy_deconz.api] Command Command.aps_data_request (20, 179, 0, <DeconzAddressEndpoint address_mode=2 address=0xF268 endpoint=1>, 260, 6, 1, b'\x18\x13\x0b\n\x00', 2, 0)
2021-09-30 23:45:44 DEBUG (MainThread) [zigpy_deconz.api] APS data request response: [2, <DeviceState.APSDE_DATA_REQUEST_SLOTS_AVAILABLE|2: 34>, 179]
2021-09-30 23:45:44 DEBUG (MainThread) [zigpy_deconz.api] Command Command.aps_data_request (20, 180, 0, <DeconzAddressEndpoint address_mode=2 address=0xF268 endpoint=1>, 260, 8, 1, b'\x18\x14\x0b\n\x00', 2, 0)
2021-09-30 23:45:44 DEBUG (MainThread) [zigpy_deconz.api] Device state changed response: [<DeviceState.128|APSDE_DATA_REQUEST_SLOTS_AVAILABLE|APSDE_DATA_CONFIRM|2: 166>, 0]
2021-09-30 23:45:44 DEBUG (MainThread) [zigpy_deconz.api] APS data request response: [2, <DeviceState.APSDE_DATA_REQUEST_SLOTS_AVAILABLE|APSDE_DATA_CONFIRM|2: 38>, 180]
2021-09-30 23:45:44 DEBUG (MainThread) [zigpy_deconz.api] Command Command.aps_data_confirm (0,)
2021-09-30 23:45:44 DEBUG (MainThread) [zigpy_deconz.api] Device state changed response: [<DeviceState.128|APSDE_DATA_REQUEST_SLOTS_AVAILABLE|APSDE_DATA_CONFIRM|2: 166>, 0]
2021-09-30 23:45:44 DEBUG (MainThread) [zigpy_deconz.api] APS data confirm response for request with id 179: 00
2021-09-30 23:45:44 DEBUG (MainThread) [zigpy_deconz.api] Request id: 0xb3 'aps_data_confirm' for <DeconzAddressEndpoint address_mode=ADDRESS_MODE.NWK address=0xf268 endpoint=1>, status: 0x00
2021-09-30 23:45:44 DEBUG (MainThread) [zigpy_deconz.api] Command Command.aps_data_confirm (0,)
2021-09-30 23:45:44 DEBUG (MainThread) [zigpy_deconz.api] APS data confirm response for request with id 180: 00
2021-09-30 23:45:44 DEBUG (MainThread) [zigpy_deconz.api] Request id: 0xb4 'aps_data_confirm' for <DeconzAddressEndpoint address_mode=ADDRESS_MODE.NWK address=0xf268 endpoint=1>, status: 0x00
2021-09-30 23:45:49 DEBUG (MainThread) [zigpy_deconz.api] Device state changed response: [<DeviceState.128|APSDE_DATA_REQUEST_SLOTS_AVAILABLE|APSDE_DATA_INDICATION|2: 170>, 0]
2021-09-30 23:45:49 DEBUG (MainThread) [zigpy_deconz.api] Command Command.aps_data_indication (1, 1)
[... cut ...]
2021-10-01 00:03:50 WARNING (MainThread) [zigpy.zcl] [0xeb5d:1:0x0008] Data remains after deserializing ZCL frame
2021-10-01 00:03:50 DEBUG (MainThread) [zigpy.zcl] [0xeb5d:1:0x0008] ZCL request 0x0000: [64, 3]
2021-10-01 00:03:50 DEBUG (MainThread) [zigpy.zcl] [0xeb5d:1:0x0008] No handler for cluster command 0
2021-10-01 00:03:50 DEBUG (MainThread) [zigpy_deconz.api] 'aps_data_indication' response from <DeconzAddress address_mode=ADDRESS_MODE.NWK address=0xeb5d>, ep: 1, profile: 0x0104, cluster_id: 0x0008, data: b'010b004003000000'

No response from the lights.

MattWestb, 2021-10-01:
The remote is sending commands OK but deCONZ is not showing witch group the commands was sent to (I using EZSP and the coordinator is writing the group the command was issued for).

But switch is made as one light link switch (and not supporting Zigbee 3) and can controlling light direly the paring direly to them (IKEA is doing the same) but then pairing with one coordinator they is always using the same group 0x4003 then IKEA remotes is using one random group then resetting the remote so they dont have this problem (and with updated zigbee 3 firmware they can changing group with binding).

More info: https://github.com/zigpy/zha-device-handlers/issues/209 https://github.com/zigpy/zha-device-handlers/issues/235.

Very likely is possible changing the group with touch link commands but no open system have implanting sending touch link commands (IKEA 5 button remote cant change the "scene cluster" binding with normal Zigbee 3 commands so its also using one default group that is not random but can being changed with touch link commands adding the new group).

I think its no way getting this switches working with other groups then the default 0x4003 as long the manufacture is fixing that in the device firmware.

Using them and doing automatons in HA is possible but if your system is down your light is not working then ;.((

Sorry but i cant helping so much more :-((

puzz3d, 2021-10-01:
> I think its no way getting this switches working with other groups then the default 0x4003 as long the manufacture is fixing that in the device firmware.

Ok, I will use them via automations then. Thank you for this clarification.

closed 2021-10-01 by puzz3d