zigpy/zha-device-handlers #1052: [Device Support Request] TS0215A by _TZ3000_fsiepnrh

rgcouto, 2021-10-01:
Hi everyone,

I've recently bought two remote controls [Nedis ZBRC10WT](https://zigbee.blakadder.com/Nedis_ZBRC10WT.html), and I'm having a hard time configuring them as they're detected as alarm panel controls but they're really simply just a remote control with four buttons, which is different from some similar products which have buttons that lead to a alarm panel control, like the [Woox Smart Remote Control](https://zigbee.blakadder.com/Woox_R7054.html).

**Is your feature request related to a problem? Please describe.**
On Home Assistant I've figured out that I can change the device type to a sensor, but then when I'm listening to zha_events I can  only find buttons I (or 1), A and B, the button O (or 0) doesn't seem to be working. This happens on both controls, which indicates its probably not a faulty device/button. 

So I can read buttons I, A and B, but not O.

**Describe the solution you'd like**
In this particular case, since this device is aesthetically closer to a generic remote control, It would be nicer if it would be possible to issue commands like O, I, A and B instead of "Disarm", "Arm_Day_Home_Only", etc. And also readings from the four buttons instead of only three.

**Device signature - this can be acquired by removing the device from ZHA and pairing it again from the add devices screen. Be sure to add the entire content of the log panel after pairing the device to a code block below this line.**
Device Signature:
`{
  "node_descriptor": "NodeDescriptor(logical_type=<LogicalType.EndDevice: 2>, complex_descriptor_available=0, user_descriptor_available=0, reserved=0, aps_flags=0, frequency_band=<FrequencyBand.Freq2400MHz: 8>, mac_capability_flags=<MACCapabilityFlags.AllocateAddress: 128>, manufacturer_code=4098, maximum_buffer_size=82, maximum_incoming_transfer_size=82, server_mask=11264, maximum_outgoing_transfer_size=82, descriptor_capability_field=<DescriptorCapability.NONE: 0>, *allocate_address=True, *is_alternate_pan_coordinator=False, *is_coordinator=False, *is_end_device=True, *is_full_function_device=False, *is_mains_powered=False, *is_receiver_on_when_idle=False, *is_router=False, *is_security_capable=False)",
  "endpoints": {
    "1": {
      "profile_id": 260,
      "device_type": "0x0401",
      "in_clusters": [
        "0x0000",
        "0x0001",
        "0x0500",
        "0x0501"
      ],
      "out_clusters": [
        "0x000a",
        "0x0019"
      ]
    }
  },
  "manufacturer": "_TZ3000_fsiepnrh",
  "model": "TS0215A",
  "class": "zigpy.device.Device"
}`

Log:
`Sending 'zdo_leave_req' failed: 
light.led_sala: polling current state
[0x05b7:1:0x0006] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=GLOBAL_COMMAND manufacturer_specific=False is_reply=True disable_default_response=True> manufacturer=None tsn=135 command_id=Command.Read_Attributes_rsp>
[0x05b7:1:0x0008] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=GLOBAL_COMMAND manufacturer_specific=False is_reply=True disable_default_response=True> manufacturer=None tsn=136 command_id=Command.Read_Attributes_rsp>
[0x05B7:1:0x0008]: received attribute: 0 update with value: 61
[0x05b7:1:0x0300] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=GLOBAL_COMMAND manufacturer_specific=False is_reply=True disable_default_response=True> manufacturer=None tsn=137 command_id=Command.Read_Attributes_rsp>
New device 0xb4d5 (bc:33:ac:ff:fe:5d:26:d8) joined the network
[0xb4d5] Scheduling initialization
Received frame on uninitialized device <Device model=None manuf=None nwk=0xB4D5 ieee=bc:33:ac:ff:fe:5d:26:d8 is_initialized=False> from ep 0 to ep 0, cluster ZDOCmd.Device_annce: b'\xFF\xD5\xB4\xD8\x26\x5D\xFE\xFF\xAC\x33\xBC\x80'
[0xb4d5:zdo] ZDO request ZDOCmd.Device_annce: [0xB4D5, bc:33:ac:ff:fe:5d:26:d8, 128]
Tries remaining: 3
[0xb4d5] Requesting 'Node Descriptor'
Tries remaining: 2
[0xb4d5] Extending timeout for 0x8a request
Received frame on uninitialized device <Device model=None manuf=None nwk=0xB4D5 ieee=bc:33:ac:ff:fe:5d:26:d8 is_initialized=False> from ep 0 to ep 0, cluster ZDOCmd.Node_Desc_rsp: b'\x8A\x00\xD5\xB4\x02\x40\x80\x02\x10\x52\x52\x00\x00\x2C\x52\x00\x00'
[0xb4d5] Got Node Descriptor: NodeDescriptor(logical_type=<LogicalType.EndDevice: 2>, complex_descriptor_available=0, user_descriptor_available=0, reserved=0, aps_flags=0, frequency_band=<FrequencyBand.Freq2400MHz: 8>, mac_capability_flags=<MACCapabilityFlags.AllocateAddress: 128>, manufacturer_code=4098, maximum_buffer_size=82, maximum_incoming_transfer_size=82, server_mask=11264, maximum_outgoing_transfer_size=82, descriptor_capability_field=<DescriptorCapability.NONE: 0>, *allocate_address=True, *is_alternate_pan_coordinator=False, *is_coordinator=False, *is_end_device=True, *is_full_function_device=False, *is_mains_powered=False, *is_receiver_on_when_idle=False, *is_router=False, *is_security_capable=False)
[0xb4d5] Discovering endpoints
Tries remaining: 3
[0xb4d5] Extending timeout for 0x8b request
Received frame on uninitialized device <Device model=None manuf=None nwk=0xB4D5 ieee=bc:33:ac:ff:fe:5d:26:d8 is_initialized=False> from ep 0 to ep 0, cluster ZDOCmd.Active_EP_rsp: b'\x8B\x00\xD5\xB4\x01\x01'
[0xb4d5] Discovered endpoints: [1]
[0xb4d5] Initializing endpoints [<Endpoint id=1 in=[] out=[] status=<Status.NEW: 0>>]
[0xb4d5:1] Discovering endpoint information
Tries remaining: 3
[0xb4d5] Extending timeout for 0x8c request
Received frame on uninitialized device <Device model=None manuf=None nwk=0xB4D5 ieee=bc:33:ac:ff:fe:5d:26:d8 is_initialized=False> from ep 0 to ep 0, cluster ZDOCmd.Simple_Desc_rsp: b'\x8C\x00\xD5\xB4\x14\x01\x04\x01\x01\x04\x01\x04\x00\x00\x01\x00\x00\x05\x01\x05\x02\x19\x00\x0A\x00'
[0xb4d5:1] Discovered endpoint information: SizePrefixedSimpleDescriptor(endpoint=1, profile=260, device_type=1025, device_version=1, input_clusters=[0, 1, 1280, 1281], output_clusters=[25, 10])
[0xb4d5] Extending timeout for 0x8d request
[0xb4d5:1:0x0000] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=GLOBAL_COMMAND manufacturer_specific=False is_reply=True disable_default_response=True> manufacturer=None tsn=141 command_id=Command.Read_Attributes_rsp>
[0xb4d5] Read model 'TS0215A' and manufacturer '_TZ3000_fsiepnrh' from <Endpoint id=1 in=[basic:0x0000, power:0x0001, ias_zone:0x0500, ias_ace:0x0501] out=[ota:0x0019, time:0x000A] status=<Status.ZDO_INIT: 1>>
[0xb4d5] Discovered basic device information for <Device model='TS0215A' manuf='_TZ3000_fsiepnrh' nwk=0xB4D5 ieee=bc:33:ac:ff:fe:5d:26:d8 is_initialized=True>
Device is initialized <Device model='TS0215A' manuf='_TZ3000_fsiepnrh' nwk=0xB4D5 ieee=bc:33:ac:ff:fe:5d:26:d8 is_initialized=True>
[... 66 lines cut ...]
[0xb4d5:1:0x0500] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=GLOBAL_COMMAND manufacturer_specific=False is_reply=True disable_default_response=False> manufacturer=None tsn=148 command_id=Command.Default_Response>
[0xb4d5:1:0x0500] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=GLOBAL_COMMAND manufacturer_specific=False is_reply=True disable_default_response=False> manufacturer=None tsn=149 command_id=Command.Default_Response>
[0xb4d5:1:0x0001] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=GLOBAL_COMMAND manufacturer_specific=False is_reply=True disable_default_response=True> manufacturer=None tsn=150 command_id=Command.Read_Attributes_rsp>
[0xb4d5] Extending timeout for 0x98 request
[0xb4d5:1:0x0500] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=GLOBAL_COMMAND manufacturer_specific=False is_reply=True disable_default_response=True> manufacturer=None tsn=151 command_id=Command.Read_Attributes_rsp>
[0xB4D5:1:0x0500]: finished channel configuration
[0xb4d5:1:0x0001] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=GLOBAL_COMMAND manufacturer_specific=False is_reply=True disable_default_response=True> manufacturer=None tsn=152 command_id=Command.Read_Attributes_rsp>
[0xB4D5:1:0x0001]: finished channel configuration
[0xB4D5:1:0x0501]: 'async_initialize' stage succeeded
[0xB4D5:1:0x0001]: 'async_initialize' stage succeeded
[0xB4D5:1:0x0500]: 'async_initialize' stage succeeded
[0xB4D5:1:0x0000]: 'async_initialize' stage succeeded
[0xB4D5:1:0x0019]: 'async_initialize' stage succeeded
[0xB4D5](TS0215A): power source: Battery or Unknown
[0xB4D5](TS0215A): completed initialization
[... cut ...]
[0xb4d5:1:0x0500] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=GLOBAL_COMMAND manufacturer_specific=False is_reply=True disable_default_response=False> manufacturer=None tsn=153 command_id=Command.Default_Response>
[0xb4d5:1:0x000a] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=GLOBAL_COMMAND manufacturer_specific=False is_reply=False disable_default_response=False> manufacturer=None tsn=50 command_id=Command.Read_Attributes>
[0xb4d5:1:0x000a] ZCL request 0x0000: [[7]]`

Thanks for your precious work and for reviewing my issue.

rgcouto, 2022-04-06:
Is there any chance someone might pick this up?

drfuture, 2022-06-10:
Hi, i bought this too and have the same Problem, if i press the 0 Button i recieve a "Alarm activated" because its recognised as "ias_ace" maybe its detected as this one? 
https://zigbee.blakadder.com/Tuya_TS0215A.html
but its this one:
https://zigbee.blakadder.com/Nedis_ZBRC10WT.html

rgcouto, 2022-06-10:
Hi! Did you managed to find a solution?

Sent from mobile

https://zigbee.blakadder.com/Tuya_TS0215A.html
https://zigbee.blakadder.com/Nedis_ZBRC10WT.html

drfuture, 2022-06-10:
no, not yet

jpa77, 2022-10-18:
Same problem here, thought this would work as easily as other Nedis devices that I have tried/used.

mar1u50, 2022-11-01:
Same problem here. Any update?

No0Vad, 2022-11-15:
I also have this device and same issue (no solution).

The only thing the [O] button does is trigger the "alarm" (you can disable it with 1234 in the UI)
[image]

The [O] event is only fired when the alarm is off (a state_changed).
The [I], [A] and [B] is sent on every press (a state_changed and a zha_event).

Also when pressing [O] no zha_event is sent.
From what I can see when I listen to all events in developer tools.

rgcouto, 2022-11-16:
Hi thanks for your input. In my case I also always receive 3 times the same signal, which messes up any automation I have. Do you know any workaround? Or have you experienced the same issue?

mar1u50, 2022-11-16:
Yes, 3 times here too.

rgcouto, 2022-11-16:
Currently on Home Assistant I'm making automations to run only single (not in parallel) and in the end I just add a sleep so that the other two inputs are, hopefully, ignored, but this doesn't work everytime which sucks!

No0Vad, 2022-11-17:
> Hi thanks for your input. In my case I also always receive 3 times the same signal, which messes up any automation I have. Do you know any workaround? Or have you experienced the same issue?

I just got the device and noticed it was registered as an alarm and I found this github issue while googling 😅

mycanaletto, 2023-01-16:
This remote control, like the one from Heiman, is seen as an AlarmControlPanel. This is a choice of the integrators.

The problem is that if you want to monitor the events you will find that it sends several in a row and **it is impossible to differentiate them.**

The temporary solution is to process only the first one in single mode and pause it. But this is not a totally reliable solution, especially if you operate several of them...

I tried to change the type :

```
zha:
  device_config:
     a4:c1:38:96:0b:cf:xx:xx-1:
       type: "remote"  # or button
```

It doesn't work

I noticed that under Z2MQTT it works fine as a remote, the behavior is normal and we can process the information correctly.
```

> arm_day_zones
> arm_all_zones
> disarm
> emergency

```

So under ZHA we should be able to choose the operating mode.

I tried to play with clusters or ZhaToolkit without success...

Information and updates welcome...

etpedro, 2023-03-03:
I'm having the same issue with this device. As anyone found a solution yet?

@mycanaletto under Z2MQTT you can use this device as a remote instead of an alarm device?

rgcouto, 2023-03-03:
At this point I basically just gave up waiting for a solution. 

I basically just switched to this https://www.ikea.com/pt/pt/p/tradfri-regulador-intensidade-s-fios-smart-branco-70408595/ works like a charm!

timiman, 2023-03-03:
Neither I've found a working solution with this Woox 4-button controller.
I'll be using remote switches like the IKEA's one or Aqara Mini Switch (if its short, double and long click is working with ZHA/SkyConnect).

jpa77, 2023-03-03:
Zigbee2MQTT solved my problem, configured this remote with lights on/of and brightness +/-

mycanaletto, 2023-03-03:
> Zigbee2MQTT a résolu mon problème

We all know that some objects are better managed under Z2M. But here the objective is to make it work properly in ZHA...

mar1u50, 2023-03-06:
> [ikea.com/pt/pt/p/tradfri-regulador-intensidade-s-fios-smart-branco-70408595](https://www.ikea.com/pt/pt/p/tradfri-regulador-intensidade-s-fios-smart-branco-70408595/) works like a charm

Yes, it works, but the battery lasts only a few weeks :(

rgcouto, 2023-03-06:
I've got mine for more than half a year now and it is still rocking it's
first battery, maybe you should activate the warranty on the device because
there might be something wrong with it.

sylv1697, 2023-08-25:
Bonjour, peut-on espérer une évolution de ZHA, pour une prise en compte direct des boutons de cette télécommande  ?

apn201, 2023-10-22:
Oh no. I just bough this device before reading these comments. It is an universal remote. Why in the world would anyone hardcode it as an alarm control panel? I had a bit similar problem with Tuya scene controller, but luckily there was a magic key combination which changed the behavior. Nothing like that here? Would it even help to have the Nedis hub and program it from there?

teffcz, 2023-12-19:
Same problem, any update?

JoeyDieleman78, 2024-01-18:
Also looking for a solution 😅

teeten, 2024-02-10:
Same here. I guess an alternative custom integration is needed to make this appear as a simple 4-button remote. Does someone know of a step-by-step tutorial on how to create a custom ZHA integration? I might try it if it is not too complicated.

mycanaletto, 2024-02-11:
What I did at the end of my article
https://www.canaletto.fr/post/home-assistant-and-alarm-encore

charlierh, 2024-03-11:
2024 same problem, no solution... i have a Philips Hue dimer switch, and this one works very well. I think I'll buy another one of these. Tuya only have good open door sensors, but, no motion sensor, no buttons, these devices are a pieces of junk

rgcouto, 2024-09-08:
No this is still not an issue that has been solved yet.

mikquadrat, 2025-10-18:
Same here, even a manually created custom quirk did not work. That's a HA native support for this device would be appreciated.

Freezerz, 2026-10-09:
I created this quirk which adds ZHA events upon pressing a button.

This means that an automation trigger can be added like this (be sure to use "By type" as described, because "By target" doesn't expose the triggers of the device):
Add trigger -> By type -> Generic _section_ -> Device -> Triggers _section_ -> Device
Choose your remote, and the button in question like **"First button" pressed**.

You can also disable the alarm control panel like this (so that a button press will not trigger the alarm functionality):
Navigate to the remote control device in the Zigbee integration -> Controls _section_ -> Alarm control panel -> Settings (gear icon)
Then toggle "Enabled" to off.

I'm sure there's plenty of room for improvement, but this is good enough for me in HA 2026.9.4.

To apply the quirk it **may** be necessary to remove the device, reboot HA and re-add the device in order to load the quirk properly.

```from __future__ import annotations

from typing import Any

from zigpy.profiles import zha
from zigpy.typing import AddressingMode
from zigpy.zcl import foundation
from zigpy.zcl.clusters.security import IasAce

from zhaquirks.clusters import CustomCluster
from zhaquirks.const import (
    BUTTON_1,
    BUTTON_2,
    BUTTON_3,
    CLUSTER_ID,
    COMMAND,
    DEVICE_TYPE,
    ENDPOINT_ID,
    ENDPOINTS,
    INPUT_CLUSTERS,
    MODELS_INFO,
    OUTPUT_CLUSTERS,
    PROFILE_ID,
    SHORT_PRESS,
    ZHA_SEND_EVENT,
)
from zhaquirks.legacy import CustomDevice

class ZBRC10WTCluster(CustomCluster, IasAce):
    # Convert IAS ACE arm commands into button events.

    cluster_id = 0x0501

    BUTTONS = {
        3: BUTTON_1,
        1: BUTTON_2,
        0: BUTTON_3,
    }

    def handle_cluster_request(
        self,
        hdr: foundation.ZCLHeader,
        args: list[Any],
        *,
        dst_addressing: AddressingMode | None = None,
    ) -> None:

        if hdr.command_id == 0:
            arm_mode = self._get_arm_mode(args)
            button = self.BUTTONS.get(arm_mode)

            if button is not None:
                self.listener_event(
                    ZHA_SEND_EVENT,
                    button,
                    [],
                )

            # Do not pass the Arm command to IasAce.
            return

        super().handle_cluster_request(
            hdr,
            args,
            dst_addressing=dst_addressing,
        )

    @staticmethod
    def _get_arm_mode(args: Any) -> int | None:
        # Extract arm_mode from zigpy command arguments.

        if isinstance(args, dict):
            value = args.get("arm_mode")
            return int(value) if value is not None else None

        value = getattr(args, "arm_mode", None)
        if value is not None:
            return int(value)

        if isinstance(args, (list, tuple)) and args:
            value = args[0]

            if isinstance(value, dict):
                value = value.get("arm_mode")
                return int(value) if value is not None else None

            arm_mode = getattr(value, "arm_mode", None)
            if arm_mode is not None:
                return int(arm_mode)

            if isinstance(value, int):
                return int(value)

        return None

class ZBRC10WT(CustomDevice):
    # Below we identify the original signature, in order to be able to apply the quirk by using a replacement signature

    # Original signature
    signature = {
        MODELS_INFO: [
            ("_TZ3000_eo3dttwe", "TS0215A"),
        ],
        ENDPOINTS: {
            1: {
                PROFILE_ID: 0x0104,
                DEVICE_TYPE: 0x0401,
                INPUT_CLUSTERS: [
                    0x0000,
                    0x0001,
                    0x0003,
                    0x0500,
                ],
                OUTPUT_CLUSTERS: [
                    0x000A,
                    0x0019,
                    0x0501,
                ],
            },
        },
    }

    # Replacement signature
    replacement = {
        ENDPOINTS: {
            1: {
                PROFILE_ID: zha.PROFILE_ID,
                DEVICE_TYPE: zha.DeviceType.REMOTE_CONTROL,
                INPUT_CLUSTERS: [
                    0x0000,
                    0x0001,
                    0x0003,
                    0x0500,
                ],
                OUTPUT_CLUSTERS: [
                    0x000A,
                    0x0019,
                    ZBRC10WTCluster,
                ],
            },
        },
    }

    # Makes the buttons I (1), A (2), B (3) available as automation triggers. Button O isn't handled
    device_automation_triggers = {
        (SHORT_PRESS, BUTTON_1): {
            ENDPOINT_ID: 1,
            CLUSTER_ID: 0x0501,
            COMMAND: BUTTON_1,
        },
        (SHORT_PRESS, BUTTON_2): {
            ENDPOINT_ID: 1,
            CLUSTER_ID: 0x0501,
            COMMAND: BUTTON_2,
        },
        (SHORT_PRESS, BUTTON_3): {
            ENDPOINT_ID: 1,
            CLUSTER_ID: 0x0501,
            COMMAND: BUTTON_3,
        },
    }```