zigpy/zha-device-handlers #1054: LUX KONOZ Thermostat Custom Quirk

sKalach, 2021-10-03:
Hi,

It looks like I need to do custom adjustment to this thermostat. The Heating, Cooling, Idling and Fan "hvac_action" properties look incorrect.
The device has "running_state" attribute in "Thermostat" cluster which shows the needed values like:
RunningState.Idle
RunningState.Fan_State_On|Heat_State_On
RunningState.Fan_State_On|Cool_State_On
RunningState.Fan_State_On

The current model uses "system_mode" attribute with values {"SystemMode.Heat", 'SystemMode.Cool'}, so the ZHA attribute "hvac_mode" as "heating" or "cooling" only.

This is from Signature:
*is_receiver_on_when_idle=False
which does not look right and the device is fully functional (by itself to do the job) at least from my point of view (is_full_function_device=False).
Inside I can see many other attributes like temperature range for Cooling and Heating separately (four numbers) but they are not critical for basic proper functioning.

Also time to time I have "TimeOut" error when I switch Fan from "On" to "Auto" (it actually means "Fan Off" when HVAC in Idling).
_Error executing service: <ServiceCall climate.set_fan_mode (c:a06a89a5160944fccf078594e1a2b710): fan_mode=auto, entity_id=['climate.lux_konoz_thermostat_fan_thermostat']>
Traceback (most recent call last):
  File "/usr/local/lib/python3.9/asyncio/tasks.py", line 492, in wait_for
    fut.result()
asyncio.exceptions.CancelledError_

What would be the right "quirk.py" template / example  I can start customizing with?

Appreciate any help and thank you in advance.

RPi 4 (all current updates installed)
deCONZ II (did last firmware update a month ago)

{
  "node_descriptor": "NodeDescriptor(logical_type=<LogicalType.EndDevice: 2>, complex_descriptor_available=0, user_descriptor_available=0, reserved=0, aps_flags=0, frequency_band=<FrequencyBand.Freq2400MHz: 8>, mac_capability_flags=<MACCapabilityFlags.AllocateAddress: 128>, manufacturer_code=4610, maximum_buffer_size=82, maximum_incoming_transfer_size=82, server_mask=11264, maximum_outgoing_transfer_size=82, descriptor_capability_field=<DescriptorCapability.NONE: 0>, *allocate_address=True, *is_alternate_pan_coordinator=False, *is_coordinator=False, *is_end_device=True, *is_full_function_device=False, *is_mains_powered=False, *is_receiver_on_when_idle=False, *is_router=False, *is_security_capable=False)",
  "endpoints": {
    "1": {
      "profile_id": 260,
      "device_type": "0x0301",
      "in_clusters": [
        "0x0000",
        "0x0001",
        "0x0003",
        "0x0004",
        "0x0005",
        "0x0020",
        "0x0201",
        "0x0202",
        "0x0204",
        "0x0b05"
      ],
      "out_clusters": [
        "0x000a",
        "0x0019"
      ]
    }
  },
  "manufacturer": "LUX",
  "model": "KONOZ",
  "class": "zigpy.device.Device"
}

hvac_modes: off, heat_cool, cool, heat
min_temp: 7.2
max_temp: 35
fan_modes: auto, on
current_temperature: 23.3
temperature: 23.7
target_temp_high: null
target_temp_low: null
fan_mode: on
hvac_action: heating
system_mode: [4]/heat
occupied_cooling_setpoint: 2480
occupied_heating_setpoint: 2370
friendly_name: LUX KONOZ THERMOSTAT fan, thermostat
supported_features: 11

Adminiuga, 2021-10-04:
> It looks like I need to do custom adjustment to this thermostat. The Heating, Cooling, Idling and Fan "hvac_action" properties look incorrect.
The device has "running_state" attribute in "Thermostat" cluster which shows the needed values like:
RunningState.Idle
RunningState.Fan_State_On|Heat_State_On
RunningState.Fan_State_On|Cool_State_On
RunningState.Fan_State_On

Does it show these running states all the time? Can you post the running states when when thermostat is set into each mode manually and running_state is updated in HA? Also post the value of the "running_mode" attribute for each mode set manually

> The current model uses "system_mode" attribute with values {"SystemMode.Heat", 'SystemMode.Cool'}, so the ZHA attribute "hvac_mode" as "heating" or "cooling" only.

Why is this incorrect?

> *is_receiver_on_when_idle=False 
which does not look right and the device is fully functional (by itself to do the job) at least from my point of view

This is from node descriptor and tells that this is a battery operated devices which does not have the radio all time on, which is typical for battery operated devices. 

> Also time to time I have "TimeOut" error when I switch Fan from "On" to "Auto" (it actually means "Fan Off" when HVAC in Idling).
Error executing service: <ServiceCall climate.set_fan_mode (c:a06a89a5160944fccf078594e1a2b710): fan_mode=auto, entity_id=['climate.lux_konoz_thermostat_fan_thermostat']>
Traceback (most recent call last):
File "/usr/local/lib/python3.9/asyncio/tasks.py", line 492, in wait_for
fut.result()
asyncio.exceptions.CancelledError

FAN_MODE_AUTO -- does mean the fan is controlled by the thermostat. in other words it's OFF when thermostat is idle and ON when thermostat is calling for HEAT or COOL. 
The cancelled error is not right, but enable debug and post debug logs.

At this point there isn't enough information to call for a quirk. May need a custom implementation like Zen thermostat in ZHA, but so far it is not enough information

sKalach, 2021-10-04:
I really appreciate your explanations and help, thank you.

All below information is from HA ZHA. I have not set up any Zegbee sniffers. This LUX thermostat I've been using for two years now controlling it from Hubitat C-5 and then from C-7. By it self the thermostat works without any issues and Hubitat managed to write the good driver for it.

All information are being refreshed accordingly to the device state / mode and I can read it from HA ZHA. The average delay is about 1-2 sec to do the actual command (turn On or Off something) and the status in HA is ready right away after that.

I setup ControlSequenceOfOperation.Cooling_and_Heating from thermostat directly. It can be Cooling, Heating, Off, Cooling_and_Heating. The "Cooling and Heating"  allows to operate in both modes - that is the restriction from the thermostat. There are also temperature limits for both modes (Min and Max) - these are also safety restrictions on the hardware level.

The thermostat can be turn on for Heating or Cooling by changing temperature only. I would call it as Automated Device  - no manual settings to turn these modes On / Off directly. The FAN can be operated manually if the thermostat is not in an active Cooling or Heating mode.
-------------------------------------------
Idle: Fan is OFF, No Heating or Cooling
Cluster "Fan (Endpoint 1, id: 0x0202 in)"
   Fan Mode (0x0000) - FanMode.Auto
   fan_mode_sequence (0x0001) - FanModeSequence.On_Auto

Cluster Thermostat (1,0x0201,in)
running_state (0x0029) - RunningState.Idle
running_mode (0x001e) - RunningMode.Heat     It does not matter! It keeps the last / current mode.
ctrl_seqe_of_oper (0x001b) - ControlSequenceOfOperation.Cooling_and_Heating  - Setup on the thermostat dashboard itself by me
system_mode (0x001c) - SystemMode.Heat    It does not matter! It keeps the last / current mode.

-------------------------------------------
Fan is ON, No Heating or Cooling
Cluster "Fan (Endpoint 1, id: 0x0202 in)"
   fan_mode (0x0000) - FanMode.On
   fan_mode_sequence (0x0001) - FanModeSequence.On_Auto

Cluster Thermostat (1,0x0201,in)
running_state (0x0029) - RunningState.Fan_State_On
running_mode (0x001e) - RunningMode.Heat     It does not matter! It keeps the last / current mode.
ctrl_seqe_of_oper (0x001b) - ControlSequenceOfOperation.Cooling_and_Heating
system_mode (0x001c) - SystemMode.Heat

----------------------------------------------
Heating Mode (the FAN is always turned ON physically by the thermostat - it must be ON). We have TWO cases!
First case when it turned ON automatically:
Cluster "Fan (Endpoint 1, id: 0x0202 in)"
   fan_mode (0x0000) - FanMode.Auto                 (The FAN is turned ON by the hardware - it is blowing)
   fan_mode_sequence (0x0001) - FanModeSequence.On_Auto

Cluster Thermostat (1,0x0201,in)
running_state (0x0029) - RunningState.Heat_State_On
running_mode (0x001e) - RunningMode.Heat   
ctrl_seqe_of_oper (0x001b) - ControlSequenceOfOperation.Cooling_and_Heating
system_mode (0x001c) - SystemMode.Heat

Second case IF I TURN FAN ON BY SET_FAN_MODE in HA when Heating is running then we have 
fan_mode (0x0000) - FanMode.On
running_state (0x0029) - RunningState.Fan_State_On|Heat_State_On

---------------------------------------------------
Now Fan is OFF and I switched to Cooling Mode from Standard HA Front End (it does three right buttons - Cooling, Heating and Off). It is not COOLING yet.
Cluster "Fan (Endpoint 1, id: 0x0202 in)"
   Fan Mode (0x0000) - FanMode.Auto
   fan_mode_sequence (0x0001) - FanModeSequence.On_Auto

Cluster Thermostat (1,0x0201,in)
running_state (0x0029) - RunningState.Idle
running_mode (0x001e) - RunningMode.Cool   
ctrl_seqe_of_oper (0x001b) - ControlSequenceOfOperation.Cooling_and_Heating
system_mode (0x001c) - SystemMode.Cool

------------------------------
I am changing the temperature to force device into active COOLING (it turned FAN Mode Automatically in this case)
Cluster "Fan (Endpoint 1, id: 0x0202 in)"
   Fan Mode (0x0000) - FanMode.On
   fan_mode_sequence (0x0001) - FanModeSequence.On_Auto

Cluster Thermostat (1,0x0201,in)
running_state (0x0029) - RunningState.Fan_State_On|Cool_State_On  
running_mode (0x001e) - RunningMode.Cool   
ctrl_seqe_of_oper (0x001b) - ControlSequenceOfOperation.Cooling_and_Heating
system_mode (0x001c) - SystemMode.Cool

We have slightly different behavior when going to Cooling or Heating mode with FAN settings. It might be the bug in thermostat firmware. Operationally everything works but I would expect to see these two attributes like this:
fan_mode (0x0000) - FanMode.On
running_state (0x0029) - RunningState.Fan_State_On|Heat_State_On

right after the Heating started.

I was not able to find the case when we have running_state (0x0029) - RunningState.Cool_State_On without Fan_State_On. But maybe I missed something.
 
About the error I mentioned before: 
It is only had happened for me when I have active Heating Mode and do SET_FAN_MODE to "Auto". The thermostat might not give the answer to this command and we might see Time Out error. If I had "Idle" state info I would avoid such combination. It could be an irrelevant issue now. 

An additional info from0x0201 cluster (if needed) with my personal values as an example:

abs_max_cool_setpoint_limit (id: 0x0006) Value: 3500 - it's 35.00 C
abs_max_heat_setpoint_limit (id: 0x0004) - 3222 in my settings from thermostat itself

abs_min_cool_setpoint_limit (id: 0x0005) - 1000
abs_min_heat_setpoint_limit (id: 0x0003) - 722 - 7.22 C

local_temp (id: 0x0000) - 2389 - it is supported already in ZHA

max_cool_setpoint_limit (id: 0x0018) - 3500
max_heat_setpoint_limit (id: 0x0016) - 2611
min_cool_setpoint_limit (id: 0x0017) - 1778
min_heat_setpoint_limit (id: 0x0015) - 1222
min_setpoint_dead_band (id: 0x0019) - 11        (I am not sure what it is ??? Channel 11 as a reserve. It might not matter for the driver)

Many others with value None

Please let me know if there is anything else I can check.

Slawa

sKalach, 2021-10-04:
My apologies about the BOLD characters in the last response. I was my mistake, did not check preview.
Just couple of an additional observations. I had provided the attributes from HA, here is what I have now:

hvac_modes: off, heat_cool, cool, heat
min_temp: 12.2
max_temp: 35
fan_modes: auto, on
current_temperature: 23.9
temperature: 24
target_temp_high: null
target_temp_low: null
fan_mode: auto
hvac_action: cooling
system_mode: [3]/cool
occupied_cooling_setpoint: 2400
occupied_heating_setpoint: 2190
friendly_name: LUX KONOZ THERMOSTAT
supported_features: 11

The 
min_temp: 12.2
max_temp: 35

This values depends on Thermostat mode and might be taken from   
max_cool_setpoint_limit (id: 0x0018) - 3500
min_cool_setpoint_limit (id: 0x0017) - 1778
or
max_heat_setpoint_limit (id: 0x0016) - 2611
min_heat_setpoint_limit (id: 0x0015) - 1222

These three are supported already in HA ZHA:
occupied_cooling_setpoint: 2400
occupied_heating_setpoint: 2190
temperature: 24
also the values of "Occupied" ones are being controlled by Thermostat firmware. There is a dependency on temperature settings plus the delta (safe temperature gap) between them which can be changed from the thermostat itself. 

This one is good too:
current_temperature: 23.9
and it can be regulated from frontend thermostat card.
[image]

I hope I did not miss anything.

Slawa

Adminiuga, 2021-10-04:
From what you are describing, it look very similarly to ZenWithin thermostat. The 1-2 second (up to 6s) delay is normal for the battery operated devices.

Try changing the code in ZHA in `homeassistant/components/zha/climate.py` there should be a line https://github.com/home-assistant/core/blob/2f9943fe7a63889a0eeba4693e76cd01eaac2f78/homeassistant/components/zha/climate.py#L576 containing 
```
    manufacturers="Zen Within",
```

change that line to 
```
    manufacturers={"Zen Within", "LUX"},
```

and restart home assistant.

The Fan behavior seems to be correct. I don't think it is supposed to change  the `fan_mode = FanMode.On` when operated for Heating/Cooling. But should change to `FanMode.On` if you turn on FAN from Thermostat itself.

Adminiuga, 2021-10-04:
also, can you try reading `pi_heating_demand` and `pi_cooling_demand` attributes?

sKalach, 2021-10-04:
Both pi_heating_demand and pi_cooling_demand attributes are None (NULL)

The programing_oper_mode (id: 0x0025) is set to ProgrammingOperationMode.Simple.

Maybe it is really simplified. It is enough to use it though.

I'll set up SSH to get to Components subfolder and I'll keep you informed.

Cheers.

sKalach, 2021-10-04:
It did not work. It was not recognized or settings were different I added  "LUX" in this string manufacturers={"Zen Within", "LUX"}, to Climate.py file from the link you provided and put the file under "custom_zha_quirks" folder. I can see __pycashe__ folder and the file inside (Climate.cpython-39.pyc). 

Unless I had to replace Climate.py file under this path homeassistant/components/zha/ it is not working. There are no Manage Clusters or other available properties, and there is no device itself.

Adminiuga, 2021-10-05:
you do need to replace/modify file `climate.py` in `homeassistant/components/zha/` folder

Adminiuga, 2021-10-05:
alternatively copy the entire `zha` folder from github to `/config/custom_components` so it looks like `/config/custom_components/zha/`
This would run the ZHA as a custom component and could be easier to make changes, **BUT** you have to modify `/config/custom_components/zha/manifest.json` and add a `"version": "2021.9.0"` key in the json file. Be careful with the formatting and all the commas. Without the version key, HA will ignore the custom component.

sKalach, 2021-10-05:
Copying whole ZHA folder from GitHub sounds good to me. Could you give me the link to zha project? To be sure I work with the right one. I'll download and unpack the archived file and then modify JSON config.

Adminiuga, 2021-10-05:
Zha is not a separate project, but a Home Assistant Core integration -- https://github.com/home-assistant/core/tree/dev/homeassistant/components/zha

sKalach, 2021-10-05:
Thank you! It took an additional couple of minutes during reboot. I'll do testing.

sKalach, 2021-10-05:
Thank you again. We have "Idle" state now.
[image]

It will help me to create proper for it. Should I test something else? I mean for the customization if needed.

sKalach, 2021-10-05:
... create proper logic for it .... missing words sometime.

Adminiuga, 2021-10-05:
Does it not change when you operated? Mate take 30s for the thermostat to send an update

sKalach, 2021-10-05:
Yes it does update the status (5-10 sec):
[image]

My device has main power as 24 V and plus I keep batteries in it,

Adminiuga, 2021-10-05:
So what logic is not working as expected that you would like to update?

sKalach, 2021-10-05:
For me it already fully workable. I might question min_temp and max_temp - these numbers are as MIN and MAX through both states {Heating & Cooling}. If this is the way it should be then it is.

Also target_temp_high (low) - if this is for FULL AUTO mode then it is good too. My device does not switch automatically between them and in such case this is good too. 

I would say it is your call now, I am good.

sKalach, 2021-12-29:
Hi,

I hope you are doing well and keep yourself healthy.

Can we promote the changes you provided to production? Is there anything else I should do for this, like opening a new ticket or providing an additional testing?

As of now I do update "    manufacturers={"Zen Within", "LUX"}," string in climate.py under "custom_components" folder.

closed 2022-01-11 by sKalach