zigpy/zha-device-handlers #1060: [Device Support Request] Aqara Smart Plug model SP-EUC01

APayerl, 2021-10-06:
**Is your feature request related to a problem? Please describe.**
My device, a Aqara Smart Plug model SP-EUC01 is sold as a device having energy metering information.
When added to home assistant (with the name basement_pc_corner) the device displays as:
[image]
And as you can see, the smartenergy_metering is unknown for some reason...
My guess is that this device needs a quirk?

**Describe the solution you'd like**
I would like to be able to use the device as an energy meter.

**Device signature - this can be acquired by removing the device from ZHA and pairing it again from the add devices screen. Be sure to add the entire content of the log panel after pairing the device to a code block below this line.**
```
{
  "node_descriptor": "NodeDescriptor(logical_type=<LogicalType.Router: 1>, complex_descriptor_available=0, user_descriptor_available=0, reserved=0, aps_flags=0, frequency_band=<FrequencyBand.Freq2400MHz: 8>, mac_capability_flags=<MACCapabilityFlags.AllocateAddress|RxOnWhenIdle|MainsPowered|FullFunctionDevice: 142>, manufacturer_code=4447, maximum_buffer_size=127, maximum_incoming_transfer_size=100, server_mask=11264, maximum_outgoing_transfer_size=100, descriptor_capability_field=<DescriptorCapability.NONE: 0>, *allocate_address=True, *is_alternate_pan_coordinator=False, *is_coordinator=False, *is_end_device=False, *is_full_function_device=True, *is_mains_powered=True, *is_receiver_on_when_idle=True, *is_router=True, *is_security_capable=False)",
  "endpoints": {
    "1": {
      "profile_id": 260,
      "device_type": "0x0051",
      "in_clusters": [
        "0x0000",
        "0x0002",
        "0x0003",
        "0x0004",
        "0x0005",
        "0x0006",
        "0x0009",
        "0x0702",
        "0x0b04"
      ],
      "out_clusters": [
        "0x000a",
        "0x0019"
      ]
    },
    "242": {
      "profile_id": 41440,
      "device_type": "0x0061",
      "in_clusters": [],
      "out_clusters": [
        "0x0021"
      ]
    }
  },
  "manufacturer": "LUMI",
  "model": "lumi.plug.maeu01",
  "class": "zigpy.device.Device"
}
```

tetele, 2021-12-17:
That's weird, I have the same model and mine doesn't even show any power reading (it's stuck to 0), although the device signature is the same. [Details here](https://community.home-assistant.io/t/aqara-lumi-eu-smart-plug-in-zha-not-reading-power-consumption/367570).

Using HA 2021.12.1.

MattWestb, 2021-12-17:
@tetele Your device dont have one Metering cluster so it can reporting that data.

https://github.com/zigpy/zigpy/blob/0e901b013da074d24de18aca1e579287e756cd84/zigpy/zcl/clusters/smartenergy.py#L21-L22

tetele, 2021-12-17:
Yeah, initially it didn't "see" that cluster, then, a few pairings later, it did.

I have no idea why that's happening, but when it finally paired as it should, my power reading is always zero. I don't know whether @APayerl did anything special to see the instant power consumption or if my plug is simply broken, but I'm willing to test if you can provide any instructions or suggestions.

APayerl, 2021-12-17:
For me this issue (kinda) resolved itself since after a while another entity "popped up" automatically...
Now it looks like this for me... [image]

tetele, 2021-12-17:
Not the case for me.

[image]

But it seems my problem is off-topic in regard to this issue, so I'll stop now.

closed 2022-06-22 by github-actions[bot]