zigpy/zha-device-handlers #1075: [Device Support Request] _TZE200_yenbr4om Tuya curtain wall switch

joaoasilva, 2021-10-17:
**Is your feature request related to a problem? Please describe.**
At the moment it just adds it as a "router" device, doesn't detect as a curtain

**Describe the solution you'd like**
Use it to open/close curtains

**Device signature - this can be acquired by removing the device from ZHA and pairing it again from the add devices screen. Be sure to add the entire content of the log panel after pairing the device to a code block below this line.**
```
{
  "node_descriptor": "NodeDescriptor(logical_type=<LogicalType.Router: 1>, complex_descriptor_available=0, user_descriptor_available=0, reserved=0, aps_flags=0, frequency_band=<FrequencyBand.Freq2400MHz: 8>, mac_capability_flags=<MACCapabilityFlags.AllocateAddress|RxOnWhenIdle|MainsPowered|FullFunctionDevice: 142>, manufacturer_code=4098, maximum_buffer_size=82, maximum_incoming_transfer_size=82, server_mask=11264, maximum_outgoing_transfer_size=82, descriptor_capability_field=<DescriptorCapability.NONE: 0>, *allocate_address=True, *is_alternate_pan_coordinator=False, *is_coordinator=False, *is_end_device=False, *is_full_function_device=True, *is_mains_powered=True, *is_receiver_on_when_idle=True, *is_router=True, *is_security_capable=False)",
  "endpoints": {
    "1": {
      "profile_id": 260,
      "device_type": "0x0051",
      "in_clusters": [
        "0x0000",
        "0x0004",
        "0x0005",
        "0xef00"
      ],
      "out_clusters": [
        "0x000a",
        "0x0019"
      ]
    }
  },
  "manufacturer": "_TZE200_yenbr4om",
  "model": "TS0601",
  "class": "zigpy.device.Device"
}
```

Logs:
```
2021-10-17 01:44:51 DEBUG (MainThread) [zigpy_deconz.zigbee.application] Sending Zigbee broadcast with tsn 5 under 6 request id, data: b'053c00'
2021-10-17 01:44:51 DEBUG (MainThread) [zigpy_deconz.api] Command Command.aps_data_request (17, 6, 0, <DeconzAddressEndpoint address_mode=1 address=65532 endpoint=None>, 0, 54, 0, b'\x05<\x00', 2, 0)
2021-10-17 01:44:51 DEBUG (MainThread) [zigpy_deconz.uart] Send: 0x121b0018001100060001fcff00003600000300053c000200
2021-10-17 01:44:51 DEBUG (MainThread) [zigpy_deconz.uart] Frame received: 0x121b00090002002206
2021-10-17 01:44:51 DEBUG (MainThread) [zigpy_deconz.api] APS data request response: [2, <DeviceState.APSDE_DATA_REQUEST_SLOTS_AVAILABLE|2: 34>, 6]
2021-10-17 01:44:51 DEBUG (MainThread) [zigpy_deconz.uart] Frame received: 0x0e1c000700ae00
2021-10-17 01:44:51 DEBUG (MainThread) [zigpy_deconz.api] Device state changed response: [<DeviceState.128|APSDE_DATA_REQUEST_SLOTS_AVAILABLE|APSDE_DATA_INDICATION|APSDE_DATA_CONFIRM|2: 174>, 0]
2021-10-17 01:44:51 DEBUG (MainThread) [zigpy_deconz.api] Command Command.aps_data_indication (1, 1)
2021-10-17 01:44:51 DEBUG (MainThread) [zigpy_deconz.uart] Send: 0x171c000800010001
2021-10-17 01:44:51 DEBUG (MainThread) [zigpy_deconz.uart] Frame received: 0x171c0021001a002601fcff0102000000000036000300053c0000afd95334040004
2021-10-17 01:44:51 DEBUG (MainThread) [zigpy_deconz.api] APS data indication response: [26, <DeviceState.APSDE_DATA_REQUEST_SLOTS_AVAILABLE|APSDE_DATA_CONFIRM|2: 38>, <DeconzAddress address_mode=ADDRESS_MODE.GROUP address=0xfffc>, 1, <DeconzAddress address_mode=ADDRESS_MODE.NWK address=0x0000>, 0, 0, 54, b'\x05<\x00', 0, 175, 217, 83, 52, 4, 0, 4]
2021-10-17 01:44:51 DEBUG (MainThread) [zigpy.zdo] [0x0000:zdo] ZDO request ZDOCmd.Mgmt_Permit_Joining_req: [60, <Bool.false: 0>]
2021-10-17 01:44:51 DEBUG (MainThread) [zigpy_deconz.api] 'aps_data_indication' response from <DeconzAddress address_mode=ADDRESS_MODE.NWK address=0x0000>, ep: 0, profile: 0x0000, cluster_id: 0x0036, data: b'053c00'
2021-10-17 01:44:51 DEBUG (MainThread) [zigpy_deconz.api] Command Command.aps_data_confirm (0,)
2021-10-17 01:44:51 DEBUG (MainThread) [zigpy_deconz.uart] Send: 0x041d0007000000
2021-10-17 01:44:51 DEBUG (MainThread) [zigpy_deconz.uart] Frame received: 0x041d0012000b00220601fcff000000000000
2021-10-17 01:44:51 DEBUG (MainThread) [zigpy_deconz.api] APS data confirm response for request with id 6: 00
2021-10-17 01:44:51 DEBUG (MainThread) [zigpy_deconz.api] Request id: 0x06 'aps_data_confirm' for <DeconzAddressEndpoint address_mode=ADDRESS_MODE.GROUP address=0xfffc endpoint=None>, status: 0x00
2021-10-17 01:44:51 DEBUG (MainThread) [zigpy_deconz.api] Command Command.write_parameter (2, <NetworkParameter.permit_join: 33>, b'<')
2021-10-17 01:44:51 DEBUG (MainThread) [zigpy_deconz.uart] Send: 0x0b1e0009000200213c
2021-10-17 01:44:51 DEBUG (MainThread) [zigpy_deconz.uart] Frame received: 0x0b1e000800010021
2021-10-17 01:44:51 DEBUG (MainThread) [zigpy_deconz.api] Write parameter permit_join: SUCCESS
2021-10-17 01:44:52 DEBUG (MainThread) [zigpy_deconz.uart] Frame received: 0x1c1f000c000500026fc6ffc7
2021-10-17 01:44:53 DEBUG (MainThread) [zigpy_deconz.uart] Frame received: 0x1c20000c000500026fc6ffcf
2021-10-17 01:44:54 DEBUG (MainThread) [zigpy_deconz.uart] Frame received: 0x1c21000c000500026fc6ffcd
2021-10-17 01:44:55 DEBUG (MainThread) [zigpy_deconz.uart] Frame received: 0x1c22000c000500026fc6ffce
2021-10-17 01:44:56 DEBUG (MainThread) [zigpy_deconz.uart] Frame received: 0x1c23000c000500026fc6ffce
2021-10-17 01:44:57 DEBUG (MainThread) [zigpy_deconz.uart] Frame received: 0x1c24000c000500026fc6ffd2
2021-10-17 01:44:58 DEBUG (MainThread) [zigpy_deconz.uart] Frame received: 0x1c25000c000500026fc6ffc6
2021-10-17 01:44:59 DEBUG (MainThread) [zigpy_deconz.uart] Frame received: 0x1c26000c000500026fc6ffc9
2021-10-17 01:45:00 DEBUG (MainThread) [zigpy_deconz.uart] Frame received: 0x1c27000c000500026fc6ffc0
2021-10-17 01:45:01 DEBUG (MainThread) [zigpy_deconz.uart] Frame received: 0x1c28000c000500026fc6ffc6
2021-10-17 01:45:02 DEBUG (MainThread) [zigpy_deconz.uart] Frame received: 0x1c29000c000500026fc6ffbc
2021-10-17 01:45:03 DEBUG (MainThread) [zigpy_deconz.uart] Frame received: 0x1c2a000c000500026fc6ffcd
2021-10-17 01:45:04 DEBUG (MainThread) [zigpy_deconz.uart] Frame received: 0x1c2b000c000500026fc6ffcb
2021-10-17 01:45:05 DEBUG (MainThread) [zigpy_deconz.uart] Frame received: 0x1c2c0012000b000300797ffeff15cd04ffc1
[... 289 lines cut ...]
2021-10-17 01:45:08 DEBUG (MainThread) [zigpy.zcl] [0xfc17:1:0xef00] ZCL request 0x0001: b'\x00\x00\n\x02\x00\x04\x00\x00t0'
2021-10-17 01:45:08 DEBUG (MainThread) [zigpy.zcl] [0xfc17:1:0xef00] No handler for cluster command 1
2021-10-17 01:45:08 DEBUG (MainThread) [zigpy_deconz.api] 'aps_data_indication' response from <DeconzAddress address_mode=ADDRESS_MODE.NWK address=0xfc17>, ep: 1, profile: 0x0104, cluster_id: 0xef00, data: b'09000100000a02000400007430'
2021-10-17 01:45:08 DEBUG (MainThread) [zigpy_deconz.uart] Frame received: 0x0e39000700aa00
2021-10-17 01:45:08 DEBUG (MainThread) [zigpy_deconz.api] Device state changed response: [<DeviceState.128|APSDE_DATA_REQUEST_SLOTS_AVAILABLE|APSDE_DATA_INDICATION|2: 170>, 0]
2021-10-17 01:45:08 DEBUG (MainThread) [zigpy_deconz.api] Command Command.aps_data_indication (1, 1)
2021-10-17 01:45:08 DEBUG (MainThread) [zigpy_deconz.uart] Send: 0x1739000800010001
2021-10-17 01:45:08 DEBUG (MainThread) [zigpy_deconz.uart] Frame received: 0x1739002b00240022020000010217fc01040100ef0d0009000100000a0200040000743000afff55340400ca
2021-10-17 01:45:08 DEBUG (MainThread) [zigpy_deconz.api] APS data indication response: [36, <DeviceState.APSDE_DATA_REQUEST_SLOTS_AVAILABLE|2: 34>, <DeconzAddress address_mode=ADDRESS_MODE.NWK address=0x0000>, 1, <DeconzAddress address_mode=ADDRESS_MODE.NWK address=0xfc17>, 1, 260, 61184, b'\t\x00\x01\x00\x00\n\x02\x00\x04\x00\x00t0', 0, 175, 255, 85, 52, 4, 0, -54]
2021-10-17 01:45:08 DEBUG (MainThread) [zigpy.zcl] [0xfc17:1:0xef00] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=CLUSTER_COMMAND manufacturer_specific=False is_reply=True disable_default_response=False> manufacturer=None tsn=0 command_id=1>
2021-10-17 01:45:08 WARNING (MainThread) [zigpy.zcl] [0xfc17:1:0xef00] Unknown cluster-specific command 1
2021-10-17 01:45:08 DEBUG (MainThread) [zigpy.zcl] [0xfc17:1:0xef00] ZCL request 0x0001: b'\x00\x00\n\x02\x00\x04\x00\x00t0'
2021-10-17 01:45:08 DEBUG (MainThread) [zigpy.zcl] [0xfc17:1:0xef00] No handler for cluster command 1
2021-10-17 01:45:08 DEBUG (MainThread) [zigpy_deconz.api] 'aps_data_indication' response from <DeconzAddress address_mode=ADDRESS_MODE.NWK address=0xfc17>, ep: 1, profile: 0x0104, cluster_id: 0xef00, data: b'09000100000a02000400007430'
2021-10-17 01:45:09 DEBUG (MainThread) [zigpy_deconz.uart] Frame received: 0x0e3a000700aa00
[... cut ...]  <DeconzAddress address_mode=ADDRESS_MODE.NWK address=0xfc17>, 1, 260, 61184, b'\t\x00\x01\x00\x00\n\x02\x00\x04\x00\x00t0', 0, 175, 255, 85, 52, 4, 0, -56]
2021-10-17 01:45:09 DEBUG (MainThread) [zigpy.zcl] [0xfc17:1:0xef00] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=CLUSTER_COMMAND manufacturer_specific=False is_reply=True disable_default_response=False> manufacturer=None tsn=0 command_id=1>
```

**Additional context**
Thanks in advance for all your help and hard work 🙇

joaoasilva, 2021-10-17:
Update:
Added the model to this quirk:
```
class TuyaZemismartSmartCover0601_3(TuyaWindowCover):
    """Tuya Zemismart blind cover motor."""

    signature = {
        # "node_descriptor": "<NodeDescriptor byte1=1 byte2=64 mac_capability_flags=142 manufacturer_code=4098
        #                       maximum_buffer_size=82 maximum_incoming_transfer_size=82 server_mask=11264
        #                       maximum_outgoing_transfer_size=82 descriptor_capability_field=0>",
        # input_clusters=[0x0000, 0x0004, 0x0005, 0x000a, 0xef00]
        # output_clusters=[0x0019]
        # <SimpleDescriptor endpoint=1 profile=260 device_type=51 input_clusters=[0, 4, 5, 61184] output_clusters=[25]>
        MODELS_INFO: [
            ("_TZE200_fzo2pocs", "TS0601"),
            ("_TZE200_yenbr4om", "TS0601"),
        ],
        ENDPOINTS: {
            1: {
                PROFILE_ID: zha.PROFILE_ID,
                DEVICE_TYPE: zha.DeviceType.SMART_PLUG,
                INPUT_CLUSTERS: [
                    Basic.cluster_id,
                    Groups.cluster_id,
                    Scenes.cluster_id,
                    TuyaManufCluster.cluster_id,
                ],
                OUTPUT_CLUSTERS: [Time.cluster_id, Ota.cluster_id],
            },
        },
    }
    replacement = {
        ENDPOINTS: {
            1: {
                PROFILE_ID: zha.PROFILE_ID,
                DEVICE_TYPE: zha.DeviceType.WINDOW_COVERING_DEVICE,
                INPUT_CLUSTERS: [
                    Basic.cluster_id,
                    Groups.cluster_id,
                    Scenes.cluster_id,
                    TuyaManufacturerWindowCover,
                    TuyaWindowCoverControl,
                ],
                OUTPUT_CLUSTERS: [Ota.cluster_id],
            },
        },
    }
```

And now it seems to get an entity but gives this error when trying to open or close:
```
2021-10-17 03:27:32 ERROR (MainThread) [homeassistant.components.websocket_api.http.connection] [547829836144] '_TZE200_yenbr4om'
Traceback (most recent call last):
File "/usr/src/homeassistant/homeassistant/components/websocket_api/commands.py", line 185, in handle_call_service
await hass.services.async_call(
File "/usr/src/homeassistant/homeassistant/core.py", line 1491, in async_call
task.result()
File "/usr/src/homeassistant/homeassistant/core.py", line 1526, in _execute_service
await handler.job.target(service_call)
File "/usr/src/homeassistant/homeassistant/helpers/entity_component.py", line 213, in handle_service
await self.hass.helpers.service.entity_service_call(
File "/usr/src/homeassistant/homeassistant/helpers/service.py", line 658, in entity_service_call
future.result() # pop exception if have
File "/usr/src/homeassistant/homeassistant/helpers/entity.py", line 830, in async_request_call
await coro
File "/usr/src/homeassistant/homeassistant/helpers/service.py", line 695, in _handle_entity_call
await result
File "/usr/src/homeassistant/homeassistant/components/zha/cover.py", line 147, in async_stop_cover
res = await self._cover_channel.stop()
File "/usr/src/homeassistant/homeassistant/components/zha/core/channels/base.py", line 59, in wrapper
result = await command(*args, **kwds)
File "/usr/local/lib/python3.9/site-packages/zhaquirks/tuya/__init__.py", line 882, in command
TUYA_COVER_COMMAND[manufacturer][command_id],
KeyError: '_TZE200_yenbr4om'
```

Does anyone have any idea of the next steps?

Thanks in advance

MattWestb, 2021-10-17:
I think you have finding all right places for putting your device id but local quirk have one limiting, they is reading all import from the HA container. So your updated `ts0601.py` is getting the data / code from `__init__.py` from the container and not from `zha_custom_quirks/tuya/__init__.py`.

For getting it working you need putting the device id in` /usr/local/lib/python3.9/site-packages/zhaquirks/tuya/__init__.py`. 
Instruction to editing files in the container https://github.com/zigpy/zha-device-handlers/discussions/693.

You can doing the same with `ts0601.py` but not needed and is easier changing witch class you need to have for it working OK if letting in local quirk folder.

joaoasilva, 2021-10-17:
Thanks but I haven't found a way to install any editor to edit files in the container. pip gives an error ModuleNotFoundError: No module named 'distutils.cmd'
I think that's the only thing missing as I get the position reported fine, it's just the commands that need to be in that property and tests but I can't test it because I can't edit my container files... Any more ideas?

MattWestb, 2021-10-17:
Good and bad :-)(,

After making changes you must deleting `__pycache__` folders in both local quirk and the the container folder there the change was made or it can the the system is reusing "old" code / data.

The last error is pointing to the "/usr/src/homeassistant/homeassistant/components/zha/cover.py" and its depends so take one more look if the changes you have made in the `/usr/local/lib/python3.9/site-packages/zhaquirks/tuya/__init__.py` so not one extra spaces or other caters have jumping in the code.

MattWestb, 2021-10-18:
Not so bad :-)))

May i doing one PR and adding your device so other user can using it out of the box ?
For ding  that i need the string you was putting in the `__init__.py` and the rest was only the ID in the `class TuyaZemismartSmartCover0601_3(TuyaWindowCover):`.

In the original implanting thread #744 is some user and devs on the hut for the problem with missing position reporting and its looks like they have founding the problem but not implanting it.

MattWestb, 2021-10-19:
I think its need being fixed in the code used for the quirk and hopefully they is fixing it and your position reporting is getting working.

Shall i making one PR as i was asked in https://github.com/zigpy/zha-device-handlers/issues/1075#issuecomment-945193329 ??

blakadder, 2021-10-24:
Do you have a link to the device?

joaoasilva, 2021-11-08:
https://github.com/zigpy/zha-device-handlers/pull/1133
Added the device to the inverted position list, it's working perfectly now