zigpy/zha-device-handlers #1113: Ikea 5 button remote (E1810) no longer matches the existing quirks when the firmware is upgraded to 2.3.080

nmeachen, 2021-10-30:
**Is your feature request related to a problem? Please describe.**
With the release of the 2.3.080 firmware for the Ikea E1810 5 button remote the device signature has changed so the existing quirks no longer match. Support for group binding no longer seems to work but multiple devices can be bound to the remote through Manager Clusters, so the same end result can be achieved.

**Describe the solution you'd like**
I do not know what functionality the fivebtnremotezha quirks provided, but the device seems to function without it. It certainly would not be benefiting from the battery fixes @MattWestb recently worked on.

**Device signature - this can be acquired by removing the device from ZHA and pairing it again from the add devices screen. Be sure to add the entire content of the log panel after pairing the device to a code block below this line.**
{
  "node_descriptor": "NodeDescriptor(logical_type=<LogicalType.EndDevice: 2>, complex_descriptor_available=0, user_descriptor_available=0, reserved=0, aps_flags=0, frequency_band=<FrequencyBand.Freq2400MHz: 8>, mac_capability_flags=<MACCapabilityFlags.AllocateAddress: 128>, manufacturer_code=4476, maximum_buffer_size=82, maximum_incoming_transfer_size=82, server_mask=11264, maximum_outgoing_transfer_size=82, descriptor_capability_field=<DescriptorCapability.NONE: 0>, *allocate_address=True, *is_alternate_pan_coordinator=False, *is_coordinator=False, *is_end_device=True, *is_full_function_device=False, *is_mains_powered=False, *is_receiver_on_when_idle=False, *is_router=False, *is_security_capable=False)",
  "endpoints": {
    "1": {
      "profile_id": 260,
      "device_type": "0x0820",
      "in_clusters": [
        "0x0000",
        "0x0001",
        "0x0003",
        "0x0020",
        "0x1000",
        "0xfc7c"
      ],
      "out_clusters": [
        "0x0003",
        "0x0004",
        "0x0005",
        "0x0006",
        "0x0008",
        "0x0019",
        "0x1000"
      ]
    }
  },
  "manufacturer": "IKEA of Sweden",
  "model": "TRADFRI remote control",
  "class": "zigpy.device.Device"
}

**Additional context**
Add any other context or screenshots about the feature request here.

MattWestb, 2021-10-31:
Hey thanks that i have missing !!

All my 5 buttons is on the bad firmware and is not getting the new PullControlCluster so i need moving one from my IKEA GW that is still on the old working FW and upgrading it in ZHA (my IKEA GW have upgrading one E1743 to the new but one is not done and the 5 button remote after 3 days).

Perhaps its helping user with battery draining problem then getting the pull control working on the remotes (i was missing that the 5 button was not having it then hunting the battery draining problems).

All IKEA is light controllers and shall working out of the box and being bond to one light group (that looks being gone for the moment but i think IKEA must fixing that) so you is only missing the battery fix (showing half battery % and no extras) and also the device automatons is not working if you are using them.

Also need sniffing the upgraded then Z2M have reporting the scene switching is not working but perhaps  they have doing it different like in Styrbar but is not so urgent then ZHA its not possible setting up Zigbee Scenes ;-(

I making on PR with the new device signature and taking the other things later then we can see how it working with other changes IKEA have doing.

MattWestb, 2021-10-31:
Ops one of the old version have pull control but no scene cluster and the other have scene and no pull control so its one combination of both but perhaps is the scene cluster possible to binding that was not possible before then it was not existing.

MattWestb, 2021-10-31:
@nmeachen I have making one version 3 of the 5 button remote but have not pushed it then i have not tested it.

Can you downloading the updated quirk and installing it as local quirk i ZHA and testing if its loading OK ?

https://github.com/MattWestb/zha-device-handlers/blob/IKEA-5BT-V3/zhaquirks/ikea/fivebtnremotezha.py

Guide for installing local quirk: https://github.com/zigpy/zha-device-handlers/discussions/693#discussioncomment-857274.
Dont forgetting deleting the local quirk then ZHA is having it inside for future updated working OK

ksga, 2021-10-31:
Got it working with my E1810.
Added output_cluster "5", and ScenesCluster.cluster_id to signature like this:
````
class IkeaTradfriRemote(CustomDevice):
    """Custom device representing IKEA of Sweden TRADFRI remote control."""

    signature = {
        # <SimpleDescriptor endpoint=1 profile=260 device_type=2080
        # device_version=1
        # input_clusters=[0, 1, 3, 32, 4096, 64636]
        # output_clusters=[3, 4, 5, 6, 8, 25, 4096]>
        MODELS_INFO: [
            (IKEA, "TRADFRI remote control"),
            ("IKEA of Sweden", "TRADFRI remote control"),
        ],
        ENDPOINTS: {
            1: {
                PROFILE_ID: zha.PROFILE_ID,
                DEVICE_TYPE: zha.DeviceType.NON_COLOR_CONTROLLER,
                INPUT_CLUSTERS: [
                    Basic.cluster_id,
                    PowerConfiguration.cluster_id,
                    Identify.cluster_id,
                    PollControl.cluster_id,
                    LightLink.cluster_id,
                    IKEA_CLUSTER_ID,
                ],
                OUTPUT_CLUSTERS: [
                    Identify.cluster_id,
                    Groups.cluster_id,
                    ScenesCluster.cluster_id,
                    OnOff.cluster_id,
                    LevelControl.cluster_id,
                    Ota.cluster_id,
                    LightLink.cluster_id,
                ],
            }
        },
    }

````

MattWestb, 2021-10-31:
Updated the quirk so the linking to the "main" IKEA remote is working after renaming it and fixing the wrong mane of one cluster.

Now loading OK and getting the "IKEA scene commands" as events, but is not working then being bounded with one CWS3 GU10 spot ;-((

nmeachen, 2021-10-31:
@MattWestb: As I have 2010.10.6, I had to switch the power configuration back to the DoublingPowerConfigurationCluster as I didn't have PowerConfiguration1CRCluster, but after doing that the custom quirk loaded and the device info for the remote showed the quirk.

MattWestb, 2021-10-31:
Sorry i am also using HA 2021.10 but have patching the IKEA "INIT" with the battery PR for getting it working but must being made in the HA container or is not working with the local quirk.
The easiest is doing like you have done and is working until you is updating HA.

nmeachen, 2021-10-31:
@MattWestb thanks for addressing this so quickly.

MattWestb, 2021-10-31:
You are very welcome !!

Its one updates for Shortcut button but is not so urgent but good.
If you finding what is added in the new firmware ;-)

PR #1114 (2021-10-31): IKEA 5 BTM Remote  V3
Adding version 3 of the 5 button remote for working with the new firmware 2.3.080 that have broken group binding but can binding to devices.
It must being one bug IKEA have made or they dont like the community using there products.

Device automatons looks OK also from the "IKEA scene cluster".

Also renamed the remote classes so can see witch is being loaded.
Updated firmware is using `TradfriRemote3` and "old production" is using `TradfriRemote1` so nuber 2 must being one older version that is not in use this days if the device is on current  firmware.

Fixes #1113
closed 2021-10-31 by dmulcahey