zigpy/zha-device-handlers #1119: error in line 91

pervolianinen, 2021-11-01:
https://github.com/zigpy/zha-device-handlers/blob/02aeb2910f2fce12aa0499ab0b8184d39f8b7cf3/zhaquirks/tuya/ts0601_trv.py#L91

no need operation division to 10 in line 91, becauser after it not correct set temperature

MattWestb, 2021-11-01:
Can you pleas explaining !
My TRVs is being sett OK and is reporting back to ZHA OK.

pervolianinen, 2021-11-01:
When i set 30 degrees on moes _TZE200_b6wax7g0 by using it buttons, i get 3 degrees in card thermostat in hpmeassistent. 35 - 3.5 und so on. When i set minimum in component thermostat in home assistent moes opens for full.

pervolianinen, 2021-11-01:
[image]
[image]

MattWestb, 2021-11-01:
Your devicse is **NOT** one SITERWELL its one BRT-100 Made of BecaSmart and its one know problem and was using https://github.com/zigpy/zha-device-handlers/blob/02aeb2910f2fce12aa0499ab0b8184d39f8b7cf3/zhaquirks/tuya/ts0601_trv.py#L1047
but one user have making one fixed type for it https://github.com/zigpy/zha-device-handlers/pull/1082 so it shall working in dev / HA beta and shall being in HA 2012.11 GA

For more info of tuya TRVs https://github.com/zigpy/zigpy/discussions/653#discussioncomment-314395

MattWestb, 2021-11-01:
You can downloading the updated quirk https://github.com/zigpy/zha-device-handlers/blob/02aeb2910f2fce12aa0499ab0b8184d39f8b7cf3/zhaquirks/tuya/ts0601_trv.py and configuring local quirk in ZHA https://github.com/zigpy/zha-device-handlers/discussions/693#discussioncomment-857274 and putting it in the local  quirk folder and restarting HA and it shall working OK or waiting some days for the new HA release.

pervolianinen, 2021-11-02:
> You can downloading the updated quirk https://github.com/zigpy/zha-device-handlers/blob/02aeb2910f2fce12aa0499ab0b8184d39f8b7cf3/zhaquirks/tuya/ts0601_trv.py and configuring local quirk in ZHA [#693 (comment)](https://github.com/zigpy/zha-device-handlers/discussions/693#discussioncomment-857274) and putting it in the local quirk folder and restarting HA and it shall working OK or waiting some days for the new HA release.

Thanks!

closed 2021-11-02 by pervolianinen