zigpy/zha-device-handlers #1123: [Device Support Request] Improving "TS0601" "_TZE200_b6wax7g0" TRV thermostatic radiator valve

Youpimatin, 2021-11-03:
Hello,
(I didn't know where I should post, so feel free to merge with the relevant subject if there is one in particular, maybe in the discussions, I don't know.)
I bought a dozen of these TRV (on Amazon.fr, Qiumi brand) to install in my house : https://zigbee.blakadder.com/Beca_BRT-100.html
I choosed these because I looked at the blackladder page and saw it was supported in ZHA (and Zigbee2MQTT)
In reality it's not the case for all parameters (battery, child lock, window detection)

I used custom_zha_quirk ts0601_trv.py (dated 2021-09-22) "ts0601_trv.MoesHY368_Type1new"

There are 3 entities :
- Thermostat (Temperature set and sensed in 1°C increments (0.5°C increments on the valve... even if 0.1°C would be more useful for the temperature sensor))
- on/off switch (I did play with it but don't understand what is it for)
- power (unkwnown value, maybe updated only when battery voltage is below 1.8V)

I didn't want to install Tuya app on my phone to check if these others parameters are supported, but if the user manual is correct, on the image shown, you can see :
- battery percentage
- countdown for the duration of boost heating
- the mode (Auto / Manual / TempHand(?) / Holidays)
- weekly programming
- ...

On the valve, you can also see if it's open or closed (just a LED, it's binary),
in reality I counted a minimum of 5 positions of the motor (maybe 0%, 25%, 50%, 75%, 100%),
this would be really cool if the position was published ! (even more if we could force a position)

Other settings exists in the documentation, you can change or check them in the valve menus :
- Temperature compensation (-9 to +9°C)
- Window open detection temperature drop / time
- Min setting temp (5°C to 15°C)
- Max setting temp (15°C to 35°C)
- Boost heating coutdown (100s to 900s)
- Dead zone (0.5°C to 5°C)
- Energy saving mode / temp
- Low temp protection (0°C to 5°C)
- Standby screen brightness
- Factory reset
- Version number

**Device signature :**
`
{
  "node_descriptor": "NodeDescriptor(byte1=2, byte2=64, mac_capability_flags=128, manufacturer_code=4098, maximum_buffer_size=82, maximum_incoming_transfer_size=82, server_mask=11264, maximum_outgoing_transfer_size=82, descriptor_capability_field=0, *allocate_address=True, *complex_descriptor_available=False, *is_alternate_pan_coordinator=False, *is_coordinator=False, *is_end_device=True, *is_full_function_device=False, *is_mains_powered=False, *is_receiver_on_when_idle=False, *is_router=False, *is_security_capable=False, *is_valid=True, *logical_type=<LogicalType.EndDevice: 2>, *user_descriptor_available=False)",
  "endpoints": {
    "1": {
      "profile_id": 260,
      "device_type": "0x0301",
      "in_clusters": [
        "0x0000",
        "0x0001",
        "0x0004",
        "0x0005",
        "0x0006",
        "0x0201",
        "0x0204",
        "0xef00"
      ],
      "out_clusters": [
        "0x000a",
        "0x0019"
      ]
    }
  },
  "manufacturer": "_TZE200_b6wax7g0",
  "model": "TS0601",
  "class": "ts0601_trv.MoesHY368_Type1new"
}
`

**Pairing log :**
`
[0x0000:zdo] ZDO request ZDOCmd.Mgmt_Permit_Joining_req: [60, <Bool.false: 0>]
[... 155 lines cut ...]
[0xc7f4:1:0xef00] Unknown cluster-specific command 17
[0xc7f4:1:0xef00] ZCL request 0x0011: b'\x00\x9c@'
[0xc7f4:1:0xef00] No handler for cluster command 17
[0xc7f4:1:0xef00] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=CLUSTER_COMMAND manufacturer_specific=False is_reply=True disable_default_response=False> manufacturer=None tsn=60 command_id=17>
[0xc7f4:1:0xef00] Unknown cluster-specific command 17
[0xc7f4:1:0xef00] ZCL request 0x0011: b'\x00\x9c@'
[0xc7f4:1:0xef00] No handler for cluster command 17
[0x8e5a:3:0x0006] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=GLOBAL_COMMAND manufacturer_specific=False is_reply=True disable_default_response=True> manufacturer=None tsn=23 command_id=Command.Report_Attributes>
[0x8e5a:3:0x0006] ZCL request 0x000a: [[Attribute(attrid=0, value=<TypeValue type=Bool, value=Bool.false>)]]
[0x8e5a:3:0x0006] Attribute report received: on_off=0
[0xeb95:11:0x0006] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=GLOBAL_COMMAND manufacturer_specific=False is_reply=True disable_default_response=True> manufacturer=None tsn=80 command_id=Command.Report_Attributes>
[0xeb95:11:0x0006] ZCL request 0x000a: [[Attribute(attrid=0, value=<TypeValue type=Bool, value=Bool.false>)]]
[0xeb95:11:0x0006] Attribute report received: on_off=0
[0xb2d6:1:0x0000] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=GLOBAL_COMMAND manufacturer_specific=False is_reply=True disable_default_response=True> manufacturer=None tsn=120 command_id=Command.Report_Attributes>
[0xb2d6:1:0x0000] ZCL request 0x000a: [[Attribute(attrid=1, value=<TypeValue type=uint8_t, value=69>)]]
[0xb2d6:1:0x0000] Attribute report received: app_version=69
[0xff60:11:0x0000] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=GLOBAL_COMMAND manufacturer_specific=False is_reply=True disable_default_response=True> manufacturer=None tsn=126 command_id=Command.Report_Attributes>
[0xff60:11:0x0000] ZCL request 0x000a: [[Attribute(attrid=1, value=<TypeValue type=uint8_t, value=66>)]]
[0xff60:11:0x0000] Attribute report received: app_version=66
`

I don't kow zigbee, and don't know how I can help, but here are the information I can get.
I can test anything to get more information, or modified quirks ...

**I don't mind if it breaks temporarily my house heating (I have only two of these valves installed for now)**

MattWestb, 2021-11-03:
I can not answering all your question then i dont have the TRV but some i can.
This TRV is only supporting setting the temperature and reporting it in 1.0° C and its one fix for getting it right in ZHA (it was 10 off) Its one open request for it https://github.com/zigpy/zha-device-handlers/issues/1088 merged on 22 October.
The TRV is not reporting battery % only battery low.
Its one TRV and and dont have one "off" mode then its always working and have antifreeze protection but i thing the switch is trying simulating turning it off.
If you like turning the TRV on and off you is loosing algorithm for keeping the temperature in the room (its taking over one day for the TRV to calculating it) and and is better having one manual central valve for you heating and truing it on and off then you like it war or cold.
The TRV mode changes shall being possible doing with presets on the GUI (the . . .) as presets and also Schedule  shall being implanted but you must do the setting of it on the cluster and the other things you is missing is implanted on the device clusters so you can using them in automatons.
You need looking on the Moes clusters and finding your functions and how to using them in your automation and implanting them in one custom card you is making for getting all working all the way.

Youpimatin, 2021-11-03:
Thanks for the answers.

There is another problem in the thermostat min-max values, they seem to be off by a factor of 10 (at least the max is 300 when it should be 30 maybe.... for the min I don't know if it should be 7°C or 0.7°C)

[image]

I'm surprised for the 1°C increment (maybe the translation of the values is not optimal, does the values come from the valve in tenth of a degree ?)
I must find how to spy on the ZHA communication channel, to see how are the raw values transmitted by the TRV.

I also found this page from zigbee2mqtt : [BRT-100-TRV](https://github.com/Koenkk/zigbee2mqtt.io/blob/master/docs/devices/BRT-100-TRV.md)
It is said that the battery values and the valve opening are published in %

MattWestb, 2021-11-03:
The min and max setpoint you is setting and reading on the thermostat cluster (0x0201) with attribute 0x0015 and 0x0016.

For see what the TRV is sending and ZHA is sending to it you shall configuring in HA:
```
    zhaquirks.tuya: debug
```
The set the min and max on the cluster and see what is possible setting on the TRV and what ZHA is is putting in for limits in the GUI.

With the 1°C is somthing that is made in the TRVs firmware and is out of our control and cant being fixed in our side.

I think your 300.0 °C is because you have setting somthing out of range for the TRV and 7.0°C is standard under limit in ZHA if you not have setting it on the cluster.

PR for fixing not good working functions is more then welcome !!

jacekk015, 2021-11-07:
@Youpimatin if you wanna to move this forward please enable debug logging in configuration.yaml
```
logger:
  default: info
  logs:
    homeassistant.components.zha: debug
    zigpy: debug
    zhaquirks: debug
```
and the resend pairing log

MattWestb, 2021-11-07:
I think the fix is working OK but the implementation is little not so elegant and can being better made then making more or less one new device instead for one If / else and changing the setpoing in the quirk.
https://github.com/zigpy/zha-device-handlers/pull/1082

jacekk015, 2021-11-07:
The problem is[with temperature] that someone is using uint, which is unsigned int type and try to put a float type in it
```
2021-07-30 12:39:37 DEBUG (MainThread) [zhaquirks.tuya] [0xda0c:1:0x0201] Mapping standard occupied_heating_setpoint (0x0012) with value 2650 to custom {514: **26.5**}
2021-07-30 12:39:38 DEBUG (MainThread) [zhaquirks.tuya] [0xda0c:1:0xef00] Received value [0, 0, 0, 26] for attribute 0x0202 (command 0x0001)
```
Besides I would like to see a full proper DEBUG log, to exclude any mistakes and misunderstandings.

Youpimatin, 2021-11-08:
OK, I thought i was in debug mode for the log, but here is a new attempt :

```
[0x0000:zdo] ZDO request ZDOCmd.Mgmt_Permit_Joining_req: [60, <Bool.false: 0>]
Device 0xef69 (84:fd:27:ff:fe:a5:a9:fb) joined the network
[0xef69] Requesting 'Node Descriptor'
Tries remaining: 2
[0xef69] Extending timeout for 0xe1 request
Device 0xef69 (84:fd:27:ff:fe:a5:a9:fb) joined the network
Skip initialization for existing device 84:fd:27:ff:fe:a5:a9:fb
Device 0xef69 (84:fd:27:ff:fe:a5:a9:fb) joined the network
Skip initialization for existing device 84:fd:27:ff:fe:a5:a9:fb
[0xef69:zdo] ZDO request ZDOCmd.Device_annce: [0xEF69, 84:fd:27:ff:fe:a5:a9:fb, 128]
[0xef69] Node Descriptor: NodeDescriptor(byte1=2, byte2=64, mac_capability_flags=128, manufacturer_code=4098, maximum_buffer_size=82, maximum_incoming_transfer_size=82, server_mask=11264, maximum_outgoing_transfer_size=82, descriptor_capability_field=0, *allocate_address=True, *complex_descriptor_available=False, *is_alternate_pan_coordinator=False, *is_coordinator=False, *is_end_device=True, *is_full_function_device=False, *is_mains_powered=False, *is_receiver_on_when_idle=False, *is_router=False, *is_security_capable=False, *is_valid=True, *logical_type=<LogicalType.EndDevice: 2>, *user_descriptor_available=False)
[0xef69] Discovering endpoints
Tries remaining: 3
[0xef69] Extending timeout for 0xe3 request
[0xef69] Discovered endpoints: [1]
[0xef69:1] Discovering endpoint information
Tries remaining: 3
[0xef69] Extending timeout for 0xe5 request
[0xef69:1] Discovered endpoint information: SizePrefixedSimpleDescriptor(endpoint=1, profile=260, device_type=81, device_version=1, input_clusters=[0, 4, 5, 61184], output_clusters=[25, 10])
Unknown cluster 61184
[0xef69] Extending timeout for 0xe7 request
[0xef69:1:0x0000] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=GLOBAL_COMMAND manufacturer_specific=False is_reply=True disable_default_response=True> manufacturer=None tsn=231 command_id=Command.Read_Attributes_rsp>
[0xef69:1] Manufacturer: _TZE200_b6wax7g0
[0xef69:1] Model: TS0601
Checking quirks for _TZE200_b6wax7g0 TS0601 (84:fd:27:ff:fe:a5:a9:fb)
Considering <class 'ts0601_trv.MoesHY368_Type1new'>
Found custom device replacement for 84:fd:27:ff:fe:a5:a9:fb: <class 'ts0601_trv.MoesHY368_Type1new'>
device - 0xEF69:84:fd:27:ff:fe:a5:a9:fb entering async_device_initialized - is_new_join: True
device - 0xEF69:84:fd:27:ff:fe:a5:a9:fb has joined the ZHA zigbee network
[0xEF69](TS0601): started configuration
[0xEF69:ZDO](TS0601): 'async_configure' stage succeeded
[0xEF69:1:0x0201]: bound 'thermostat' cluster: Status.SUCCESS
[0xEF69:1:0x0201]: attr reporting for '{'local_temp': (30, 900, 25), 'occupied_cooling_setpoint': (30, 900, 25), 'occupied_heating_setpoint': (30, 900, 25), 'unoccupied_cooling_setpoint': (30, 900, 25)}' on 'thermostat': ConfigureReportingResponseRecord(status=0)
[0xEF69:1:0x0201]: attr reporting for '{'unoccupied_heating_setpoint': (30, 900, 25), 'running_mode': (30, 900, 25), 'running_state': (30, 900, 5), 'system_mode': (30, 900, 25)}' on 'thermostat': ConfigureReportingResponseRecord(status=0)
[0xEF69:1:0x0201]: attr reporting for '{'occupancy': (30, 900, 1), 'pi_cooling_demand': (30, 900, 5), 'pi_heating_demand': (30, 900, 5)}' on 'thermostat': ConfigureReportingResponseRecord(status=0)
[0xEF69:1:0x0201]: finished channel configuration
[0xEF69:1:0x0006]: bound 'on_off' cluster: Status.SUCCESS
[0xEF69:1:0x0006]: reporting 'on_off' attr on 'on_off' cluster: 0/900/1: Result: '[ConfigureReportingResponseRecord(status=0)]'
[0xEF69:1:0x0000]: finished channel configuration
[0xEF69:1:0x0001]: bound 'power' cluster: Status.SUCCESS
[0xEF69:1:0x0001]: reporting 'battery_voltage' attr on 'power' cluster: 3600/10800/1: Result: '[ConfigureReportingResponseRecord(status=0)]'
[0xEF69:1:0x0019]: finished channel configuration
[0xEF69:1:0x0201]: 'async_configure' stage succeeded
[0xEF69:1:0x0006]: 'async_configure' stage failed: 'ConfigureReportingResponseRecord' object is not subscriptable
[0xEF69:1:0x0000]: 'async_configure' stage succeeded
[0xEF69:1:0x0001]: 'async_configure' stage failed: 'ConfigureReportingResponseRecord' object is not subscriptable
[0xEF69:1:0x0019]: 'async_configure' stage succeeded
[0xEF69](TS0601): completed configuration
[0xEF69](TS0601): stored in registry: ZhaDeviceEntry(name='_TZE200_b6wax7g0 TS0601', ieee='84:fd:27:ff:fe:a5:a9:fb', last_seen=1636306009.3184533)
[0xEF69](TS0601): started initialization
[0xEF69:ZDO](TS0601): 'async_initialize' stage succeeded
[0xEF69:1:0x0201]: initializing channel: from_cache: False
[0xEF69:1:0x0201]: Attribute report 'MoesThermostatNew'[system_mode] = SystemMode.Heat
[0xEF69:1:0x0201]: read attributes: Success: {}. Failed: {'abs_min_heat_setpoint_limit': <Status.UNSUPPORTED_ATTRIBUTE: 134>, 'abs_max_heat_setpoint_limit': <Status.UNSUPPORTED_ATTRIBUTE: 134>, 'abs_min_cool_setpoint_limit': <Status.UNSUPPORTED_ATTRIBUTE: 134>, 'abs_max_cool_setpoint_limit': <Status.UNSUPPORTED_ATTRIBUTE: 134>}
[0xEF69:1:0x0201]: read attributes: Success: {}. Failed: {'max_cool_setpoint_limit': <Status.UNSUPPORTED_ATTRIBUTE: 134>, 'max_heat_setpoint_limit': <Status.UNSUPPORTED_ATTRIBUTE: 134>, 'min_cool_setpoint_limit': <Status.UNSUPPORTED_ATTRIBUTE: 134>, 'min_heat_setpoint_limit': <Status.UNSUPPORTED_ATTRIBUTE: 134>}
[0xEF69:1:0x0201]: Attribute report 'MoesThermostatNew'[ctrl_seqe_of_oper] = ControlSequenceOfOperation.Heating_Only
[... cut ...]
[0x6599:11:0x0000] ZCL request 0x000a: [[Attribute(attrid=1, value=<TypeValue type=uint8_t, value=66>)]]
[0x6599:11:0x0000] Attribute report received: app_version=66
[0xb2d6:1:0x0000] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=GLOBAL_COMMAND manufacturer_specific=False is_reply=True disable_default_response=True> manufacturer=None tsn=12 command_id=Command.Report_Attributes>
[0xb2d6:1:0x0000] ZCL request 0x000a: [[Attribute(attrid=1, value=<TypeValue type=uint8_t, value=69>)]]
[0xb2d6:1:0x0000] Attribute report received: app_version=69
```

jacekk015, 2021-11-08:
OK - that's still wrong. You need to take it from the LOGS, not pairing window.
Values and attributes are better presented there.
You see above from my example - there's DEBUG name.
It shows us better what was mapped where.

Also, since you have the TRV in your hand, and wanna to support **new elements** it would be nice if you play with TRV values and try to find out which attribute is for which function you wanna to have.
Something like with the room temperature:
TRV keeps it in the 515 attribute which is 0x0203 in the HEX version. Then ZHA quirk maps it to local_temp and then HA GUI shows it as a Room temp.
```
[0xef69:1:0xef00] ZCL request 0x0001: [Command(status=0, tsn=15, command_id=515, function=0, data=[4, 0, 0, 0, 190])]
[0xEF69:1:0x0201]: Attribute report 'MoesThermostatNew'[local_temp] = 1900
```

If you have options that no one other has we cannot know which attribute is responsible for which function.

For example:
Check DEBUG logs, enable Child lock, check debug logs for differences. Some attribute will be switched from [0] to [1]

What works for you now are known[I mean attributes]. Any new function you need to give us a hint which attribute use that function on TRV. We will do the rest.
Original code could be helpful - maybe some attributes are on the same numbers:
https://github.com/zigpy/zha-device-handlers/blob/dev/zhaquirks/tuya/ts0601_trv.py#L136
```
MOES_TARGET_TEMP_ATTR = 0x0202  # target room temp (decidegree)
MOES_TEMPERATURE_ATTR = 0x0203  # current room temp (decidegree)
MOES_MODE_ATTR = 0x0404  # [0] away [1] scheduled [2] manual [3] comfort [4] eco [5] boost [6] complex
MOES_CHILD_LOCK_ATTR = 0x0107  # [0] unlocked [1] child-locked
MOES_VALVE_DETECT_ATTR = 0x0114  # [0] do not report [1] report
MOES_TEMP_CALIBRATION_ATTR = 0x022C  # temperature calibration (decidegree)
MOES_MIN_TEMPERATURE_ATTR = 0x0266  # minimum limit of temperature setting (decidegree)
MOES_MAX_TEMPERATURE_ATTR = 0x0267  # maximum limit of temperature setting (decidegree)
MOES_WINDOW_DETECT_ATTR = 0x0068  # [0,35,5] on/off, temperature, operating time (min)
MOES_BOOST_TIME_ATTR = 0x0269  # BOOST mode operating time in (sec)
MOES_FORCE_VALVE_ATTR = 0x046A  # [0] normal [1] open [2] close
MOES_COMFORT_TEMP_ATTR = 0x026B  # comfort mode temperaure (decidegree)
MOES_ECO_TEMP_ATTR = 0x026C  # eco mode temperature (decidegree)
MOES_VALVE_STATE_ATTR = 0x026D  # opening percentage
MOES_BATTERY_LOW_ATTR = 0x016E  # battery low warning
MOES_WEEK_FORMAT_ATTR = 0x046F  # [0] 5 days [1] 6 days, [2] 7 days
MOES_AWAY_TEMP_ATTR = 0x0272  # away mode temperature (decidegree)
MOES_AUTO_LOCK_ATTR = 0x0174  # [0] auto [1] manual
MOES_AWAY_DAYS_ATTR = 0x0275  # away mode duration (days)
```

Youpimatin, 2021-11-08:
OK, here is the data from the log at the time of pairing (I expurged some obvious off-topic lines)
I posted on [pastebin](https://pastebin.com/VcHe1fgK) since it is too long for the 2^16 byte limit

I'll play with functions on the valve and check the log when i'm back home.

jacekk015, 2021-11-08:
I've filtered all attributes your TRV has presented[if you didn't cut the log too quick]
```
2021-11-07 18:26:51 DEBUG (MainThread) [zhaquirks.tuya] [0xef69:1:0xef00] Received value [0, 0, 0, 25] for attribute 0x0268 (command 0x0001)
2021-11-07 18:26:58 DEBUG (MainThread) [zhaquirks.tuya] [0xef69:1:0xef00] Received value [0, 0, 0, 50] for attribute 0x0268 (command 0x0001)
2021-11-07 18:27:05 DEBUG (MainThread) [zhaquirks.tuya] [0xef69:1:0xef00] Received value [0, 0, 0, 25] for attribute 0x0268 (command 0x0001)
2021-11-07 18:27:12 DEBUG (MainThread) [zhaquirks.tuya] [0xef69:1:0xef00] Received value [0, 0, 0, 190] for attribute 0x0203 (command 0x0001)
2021-11-07 18:27:12 DEBUG (MainThread) [zhaquirks.tuya] [0xef69:1:0xef00] Received value [0, 0, 0, 61] for attribute 0x020e (command 0x0001)
2021-11-07 18:27:16 DEBUG (MainThread) [zhaquirks.tuya] [0xef69:1:0xef00] Received value [0, 0, 0, 50] for attribute 0x0268 (command 0x0001)
2021-11-07 18:27:26 DEBUG (MainThread) [zhaquirks.tuya] [0xef69:1:0xef00] Received value [0, 0, 0, 21] for attribute 0x0202 (command 0x0001)
2021-11-07 18:27:26 DEBUG (MainThread) [zhaquirks.tuya] [0xef69:1:0xef00] Received value [0, 0, 0, 0] for attribute 0x0205 (command 0x0001)
2021-11-07 18:27:27 DEBUG (MainThread) [zhaquirks.tuya] [0xef69:1:0xef00] Received value [0, 0, 0, 58] for attribute 0x020e (command 0x0001)
2021-11-07 18:27:34 DEBUG (MainThread) [zhaquirks.tuya] [0xef69:1:0xef00] Received value [0, 0, 0, 75] for attribute 0x0268 (command 0x0001)
2021-11-07 18:27:38 DEBUG (MainThread) [zhaquirks.tuya] [0xef69:1:0xef00] Received value [0, 0, 0, 21] for attribute 0x0202 (command 0x0001)
2021-11-07 18:27:38 DEBUG (MainThread) [zhaquirks.tuya] [0xef69:1:0xef00] Received value [0, 0, 0, 0] for attribute 0x0205 (command 0x0001)
2021-11-07 18:27:38 DEBUG (MainThread) [zhaquirks.tuya] [0xef69:1:0xef00] Received value [0, 0, 0, 58] for attribute 0x020e (command 0x0001)
2021-11-07 18:27:46 DEBUG (MainThread) [zhaquirks.tuya] [0xef69:1:0xef00] Received value [0, 0, 0, 20] for attribute 0x0202 (command 0x0001)
2021-11-07 18:27:47 DEBUG (MainThread) [zhaquirks.tuya] [0xef69:1:0xef00] Received value [0, 0, 0, 0] for attribute 0x0205 (command 0x0001)
2021-11-07 18:27:47 DEBUG (MainThread) [zhaquirks.tuya] [0xef69:1:0xef00] Received value [0, 0, 0, 58] for attribute 0x020e (command 0x0001)
2021-11-07 18:27:54 DEBUG (MainThread) [zhaquirks.tuya] [0xef69:1:0xef00] Received value [0, 0, 0, 50] for attribute 0x0268 (command 0x0001)
2021-11-07 18:28:03 DEBUG (MainThread) [zhaquirks.tuya] [0xef69:1:0xef00] Received value [0, 0, 0, 185] for attribute 0x0203 (command 0x0001)
2021-11-07 18:28:04 DEBUG (MainThread) [zhaquirks.tuya] [0xef69:1:0xef00] Received value [0, 0, 0, 61] for attribute 0x020e (command 0x0001)
2021-11-07 18:29:01 DEBUG (MainThread) [zhaquirks.tuya] [0xef69:1:0xef00] Received value [0, 0, 0, 180] for attribute 0x0203 (command 0x0001)
2021-11-07 18:29:01 DEBUG (MainThread) [zhaquirks.tuya] [0xef69:1:0xef00] Received value [0, 0, 0, 58] for attribute 0x020e (command 0x0001)
2021-11-07 18:29:02 DEBUG (MainThread) [zhaquirks.tuya] [0xef69:1:0xef00] Received value [0, 0, 0, 180] for attribute 0x0203 (command 0x0001)
2021-11-07 18:29:02 DEBUG (MainThread) [zhaquirks.tuya] [0xef69:1:0xef00] Received value [0, 0, 0, 61] for attribute 0x020e (command 0x0001)
2021-11-07 18:30:02 DEBUG (MainThread) [zhaquirks.tuya] [0xef69:1:0xef00] Received value [0, 0, 0, 180] for attribute 0x0203 (command 0x0001)
2021-11-07 18:30:02 DEBUG (MainThread) [zhaquirks.tuya] [0xef69:1:0xef00] Received value [0, 0, 0, 58] for attribute 0x020e (command 0x0001)
2021-11-07 18:31:02 DEBUG (MainThread) [zhaquirks.tuya] [0xef69:1:0xef00] Received value [0, 0, 0, 180] for attribute 0x0203 (command 0x0001)
2021-11-07 18:31:02 DEBUG (MainThread) [zhaquirks.tuya] [0xef69:1:0xef00] Received value [0, 0, 0, 61] for attribute 0x020e (command 0x0001)
2021-11-07 18:31:03 DEBUG (MainThread) [zhaquirks.tuya] [0xef69:1:0xef00] Received value [0, 0, 0, 180] for attribute 0x0203 (command 0x0001)
2021-11-07 18:31:03 DEBUG (MainThread) [zhaquirks.tuya] [0xef69:1:0xef00] Received value [0, 0, 0, 58] for attribute 0x020e (command 0x0001)
2021-11-07 18:31:08 DEBUG (MainThread) [zhaquirks.tuya] [0xef69:1:0xef00] Received value [1] for attribute 0x010d (command 0x0001)
2021-11-07 18:31:09 DEBUG (MainThread) [zhaquirks.tuya] [0xef69:1:0xef00] Received value [1] for attribute 0x0401 (command 0x0001)
2021-11-07 18:31:10 DEBUG (MainThread) [zhaquirks.tuya] [0xef69:1:0xef00] Received value [0, 0, 0, 20] for attribute 0x0202 (command 0x0001)
2021-11-07 18:31:10 DEBUG (MainThread) [zhaquirks.tuya] [0xef69:1:0xef00] Received value [0, 0, 0, 180] for attribute 0x0203 (command 0x0001)
2021-11-07 18:31:10 DEBUG (MainThread) [zhaquirks.tuya] [0xef69:1:0xef00] Received value [0] for attribute 0x0104 (command 0x0001)
2021-11-07 18:31:11 DEBUG (MainThread) [zhaquirks.tuya] [0xef69:1:0xef00] Received value [0, 0, 0, 0] for attribute 0x0205 (command 0x0001)
2021-11-07 18:31:11 DEBUG (MainThread) [zhaquirks.tuya] [0xef69:1:0xef00] Received value [0] for attribute 0x0407 (command 0x0001)
2021-11-07 18:31:11 DEBUG (MainThread) [zhaquirks.tuya] [0xef69:1:0xef00] Received value [0] for attribute 0x0108 (command 0x0001)
2021-11-07 18:31:12 DEBUG (MainThread) [zhaquirks.tuya] [0xef69:1:0xef00] Received value [1] for attribute 0x0409 (command 0x0001)
[... cut ...]
2021-11-07 18:31:27 DEBUG (MainThread) [zhaquirks.tuya] [0xef69:1:0xef00] Received value [0, 0, 0, 19] for attribute 0x0202 (command 0x0001)
2021-11-07 18:31:27 DEBUG (MainThread) [zhaquirks.tuya] [0xef69:1:0xef00] Received value [0, 0, 0, 0] for attribute 0x0205 (command 0x0001)
2021-11-07 18:31:27 DEBUG (MainThread) [zhaquirks.tuya] [0xef69:1:0xef00] Received value [0, 0, 0, 58] for attribute 0x020e (command 0x0001)
2021-11-07 18:31:34 DEBUG (MainThread) [zhaquirks.tuya] [0xef69:1:0xef00] Received value [0, 0, 0, 25] for attribute 0x0268 (command 0x0001)
```

Youpimatin, 2021-11-08:
Nice ! thanks.
Let's see if i did cut the log too quick :
```
2021-11-07 18:31:34 DEBUG (MainThread) [zhaquirks.tuya] [0xef69:1:0xef00] Received value [0, 0, 0, 25] for attribute 0x0268 (command 0x0001)
2021-11-07 18:37:03 DEBUG (MainThread) [zhaquirks.tuya] [0xef69:1:0xef00] Received value [0, 0, 0, 180] for attribute 0x0203 (command 0x0001)
2021-11-07 18:53:04 DEBUG (MainThread) [zhaquirks.tuya] [0xef69:1:0xef00] Received value [0, 0, 0, 170] for attribute 0x0203 (command 0x0001)
2021-11-07 18:53:04 DEBUG (MainThread) [zhaquirks.tuya] [0xef69:1:0xef00] Received value [0, 0, 0, 50] for attribute 0x0268 (command 0x0001)
2021-11-07 20:07:54 DEBUG (MainThread) [zhaquirks.tuya] [0xef69:1:0xef00] Received value [0, 0, 0, 61] for attribute 0x020e (command 0x0001)
2021-11-07 20:07:55 DEBUG (MainThread) [zhaquirks.tuya] [0xef69:1:0xef00] Received value [1] for attribute 0x0409 (command 0x0001)
2021-11-07 20:07:55 DEBUG (MainThread) [zhaquirks.tuya] [0xef69:1:0xef00] Received value [0] for attribute 0x0104 (command 0x0001)
2021-11-07 21:17:23 DEBUG (MainThread) [zhaquirks.tuya] [0xef69:1:0xef00] Received value [0, 0, 0, 75] for attribute 0x0268 (command 0x0001)
2021-11-07 21:22:47 DEBUG (MainThread) [zhaquirks.tuya] [0xef69:1:0xef00] Received value [0, 0, 0, 160] for attribute 0x0203 (command 0x0001)
2021-11-07 21:34:42 DEBUG (MainThread) [zhaquirks.tuya] [0xef69:1:0xef00] Received value [0, 0, 0, 20] for attribute 0x0202 (command 0x0001)
2021-11-07 21:34:51 DEBUG (MainThread) [zhaquirks.tuya] [0xef69:1:0xef00] Received value [0, 0, 0, 18] for attribute 0x0202 (command 0x0001)
2021-11-07 21:35:05 DEBUG (MainThread) [zhaquirks.tuya] [0xef69:1:0xef00] Received value [0, 0, 0, 19] for attribute 0x0202 (command 0x0001)
2021-11-07 21:35:05 DEBUG (MainThread) [zhaquirks.tuya] [0xef69:1:0xef00] Received value [0, 0, 0, 19] for attribute 0x0202 (command 0x0001)
2021-11-07 21:35:20 DEBUG (MainThread) [zhaquirks.tuya] [0xef69:1:0xef00] Received value [0, 0, 0, 18] for attribute 0x0202 (command 0x0001)
2021-11-07 21:41:10 DEBUG (MainThread) [zhaquirks.tuya] [0xef69:1:0xef00] Received value [0, 0, 0, 170] for attribute 0x0203 (command 0x0001)
2021-11-07 21:41:11 DEBUG (MainThread) [zhaquirks.tuya] [0xef69:1:0xef00] Received value [0, 0, 0, 25] for attribute 0x0268 (command 0x0001)
2021-11-07 21:52:47 DEBUG (MainThread) [zhaquirks.tuya] [0xef69:1:0xef00] Received value [0, 0, 0, 21] for attribute 0x0202 (command 0x0001)
2021-11-07 22:15:29 DEBUG (MainThread) [zhaquirks.tuya] [0xef69:1:0xef00] Received value [0, 0, 0, 19] for attribute 0x0202 (command 0x0001)
2021-11-07 22:21:19 DEBUG (MainThread) [zhaquirks.tuya] [0xef69:1:0xef00] Received value [0, 0, 0, 180] for attribute 0x0203 (command 0x0001)
2021-11-07 22:21:20 DEBUG (MainThread) [zhaquirks.tuya] [0xef69:1:0xef00] Received value [0, 0, 0, 25] for attribute 0x0268 (command 0x0001)
2021-11-07 23:25:25 DEBUG (MainThread) [zhaquirks.tuya] [0xef69:1:0xef00] Received value [0, 0, 0, 170] for attribute 0x0203 (command 0x0001)
2021-11-07 23:25:25 DEBUG (MainThread) [zhaquirks.tuya] [0xef69:1:0xef00] Received value [0, 0, 0, 50] for attribute 0x0268 (command 0x0001)
2021-11-08 00:18:53 DEBUG (MainThread) [zhaquirks.tuya] [0xef69:1:0xef00] Received value [0, 0, 0, 61] for attribute 0x020e (command 0x0001)
2021-11-08 00:18:53 DEBUG (MainThread) [zhaquirks.tuya] [0xef69:1:0xef00] Received value [1] for attribute 0x0409 (command 0x0001)
2021-11-08 00:18:53 DEBUG (MainThread) [zhaquirks.tuya] [0xef69:1:0xef00] Received value [0] for attribute 0x0104 (command 0x0001)
2021-11-08 01:49:42 DEBUG (MainThread) [zhaquirks.tuya] [0xef69:1:0xef00] Received value [0, 0, 0, 75] for attribute 0x0268 (command 0x0001)
2021-11-08 01:55:06 DEBUG (MainThread) [zhaquirks.tuya] [0xef69:1:0xef00] Received value [0, 0, 0, 160] for attribute 0x0203 (command 0x0001)
2021-11-08 04:19:25 DEBUG (MainThread) [zhaquirks.tuya] [0xef69:1:0xef00] Received value [0, 0, 0, 58] for attribute 0x020e (command 0x0001)
2021-11-08 04:19:25 DEBUG (MainThread) [zhaquirks.tuya] [0xef69:1:0xef00] Received value [1] for attribute 0x0409 (command 0x0001)
2021-11-08 04:19:25 DEBUG (MainThread) [zhaquirks.tuya] [0xef69:1:0xef00] Received value [0] for attribute 0x0104 (command 0x0001)
2021-11-08 05:29:00 DEBUG (MainThread) [zhaquirks.tuya] [0xef69:1:0xef00] Received value [0, 0, 0, 100] for attribute 0x0268 (command 0x0001)
2021-11-08 05:34:24 DEBUG (MainThread) [zhaquirks.tuya] [0xef69:1:0xef00] Received value [0, 0, 0, 150] for attribute 0x0203 (command 0x0001)
2021-11-08 07:00:04 DEBUG (MainThread) [zhaquirks.tuya] [0xef69:1:0xef00] Received value [0, 0, 0, 160] for attribute 0x0203 (command 0x0001)
2021-11-08 07:00:04 DEBUG (MainThread) [zhaquirks.tuya] [0xef69:1:0xef00] Received value [0, 0, 0, 75] for attribute 0x0268 (command 0x0001)
2021-11-08 08:20:21 DEBUG (MainThread) [zhaquirks.tuya] [0xef69:1:0xef00] Received value [0, 0, 0, 58] for attribute 0x020e (command 0x0001)
2021-11-08 08:20:22 DEBUG (MainThread) [zhaquirks.tuya] [0xef69:1:0xef00] Received value [1] for attribute 0x0409 (command 0x0001)
2021-11-08 08:20:22 DEBUG (MainThread) [zhaquirks.tuya] [0xef69:1:0xef00] Received value [0] for attribute 0x0104 (command 0x0001)
2021-11-08 08:47:25 DEBUG (MainThread) [zhaquirks.tuya] [0xef69:1:0xef00] Received value [0, 0, 0, 18] for attribute 0x0202 (command 0x0001)
```

jacekk015, 2021-11-08:
I've used https://onlinetexttools.com/filter-text
firstly used "for attribute 0x" pattern then copied Filtered text to Text field and then used the pattern "0xef69" to filter only that device.
Each time you pair device ID is changing.

Youpimatin, 2021-11-08:
From the log above, I can identify attributes :
0x0202 # target room temp (in degree, NOT in decidegree)
0x0203 # current room temp (decidegree)

and maybe :
0x0268 # opening percentage ? (values of 25 / 50 / 75 / 100 influenced by target and room temperatures)
0x020e # battery percentage ? (58 or 61 for one TRV, 55 or 58 for my second one, updated every 4 hours )

unidentified attributes : 
0x0205 ( [0, 0, 0, 0] )
0x010d (1 or 0)
0x0401 (1)
0x0104 (0)
0x0407 (0)
0x0108 (0)
0x0409 (1)
0x0065 ( [6, 0, 40, 11, 30, 42, 13, 30, 44, 17, 30, 46, 6, 0, 48, 12, 0, 46, 14, 30, 44, 17, 30, 42, 6, 0, 38, 12, 30, 40, 14, 30, 42, 18, 30, 40] )
0x016a (0)

jacekk015, 2021-11-08:
Do you use newest HomeAssistant version?? I see differences in DEBUG LOGS between yours and mine.
Mine logs are giving DPs from Tuya, and I can find those DPs i Zigbee2Mqtt to check which attribute is which.
I has to be some difference.

Since they declare kind of support so I could just easily match those DPs to attributes.
https://github.com/Koenkk/zigbee-herdsman-converters/blob/master/devices/moes.js#L193

My log:
```
2021-09-24 15:01:13 DEBUG (MainThread) [zigpy.zcl] [0xddeb:1:0xef00] ZCL request 0x0002: [Command(status=0, tsn=24, command_id=536, function=0, data=[4, 0, 0, 0, 233])]
2021-09-24 15:01:13 DEBUG (MainThread) [zhaquirks.tuya] [0xddeb:1:0xef00] Received value [0, 0, 0, 233] for attribute 0x0218 (command 0x0002)
```
tsn=24 = DP 24 from Z2M which is current room temp, which is attribute 0x0218

MattWestb, 2021-11-08:
@jacekk015 You is having debug for zigpy.zcl that is doing more raw commands but still is `tsn` _Tansaction Numbe_ in the underlying network that is for flirting duplicate messages or reseeding one that is lost (missed ack or default response).

I was thinking the `attribute 0x0218` from the tuya quirk is the "decoded" DP from the string you is getting from the zigpyc.zcl data string:
 `[Command(status=0, tsn=24, command_id=536, function=0, data=[4, 0, 0, 0, 233])]`

Youpimatin, 2021-11-08:
No I'm not using the latest version of HA, I'm using core-2021.6.6
(because I loose too much usability and functionality with newer versions :
- [this](https://github.com/home-assistant/frontend/issues/9547) (Graph colors and huge legends with hideous colors above)
- I'm loosing digits in new history (now x.xx, and before x.xxx), and it's important for some of my custom devices
- number being changed is hidden by the previous entitity when you use an input slider
- NMAP yaml deprecated, when the integration is horrible, I loose some functionality, it's much more work that a basic text file, I can no longer see easily in which order devices were connected (they could let the yaml working, they choose to break things)
- history loadings takes an "infinite" time (and sometimes it never loads)
- other things I don't have in mind for the moment...)

If it helps by providing more data, I can set the last version of HA to debug (but I will definitely downgrade after to the last good version for me.)

jacekk015, 2021-11-08:
@MattWestb Funny is that this is wrong. In reality tsn returns for debug a DP
```
2021-09-24 15:01:03 DEBUG (MainThread) [zigpy.zcl] [0xddeb:1:0xef00] ZCL request 0x0002: [Command(status=0, tsn=16, command_id=528, function=0, data=[4, 0, 0, 0, 34])]
2021-09-24 15:01:03 DEBUG (MainThread) [zhaquirks.tuya] [0xddeb:1:0xef00] Received value [0, 0, 0, 34] for attribute 0x0210 (command 0x0002)
2021-09-24 15:01:13 DEBUG (MainThread) [zigpy.zcl] [0xddeb:1:0xef00] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=CLUSTER_COMMAND manufacturer_specific=False is_reply=True disable_default_response=False> manufacturer=None tsn=8 command_id=2>
2021-09-24 15:01:13 DEBUG (MainThread) [zigpy.zcl] [0xddeb:1:0xef00] ZCL request 0x0002: [Command(status=0, tsn=24, command_id=536, function=0, data=[4, 0, 0, 0, 233])]
2021-09-24 15:01:13 DEBUG (MainThread) [zhaquirks.tuya] [0xddeb:1:0xef00] Received value [0, 0, 0, 233] for attribute 0x0218 (command 0x0002)
2021-09-24 15:01:14 DEBUG (MainThread) [zigpy.zcl] [0xddeb:1:0xef00] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=CLUSTER_COMMAND manufacturer_specific=False is_reply=True disable_default_response=False> manufacturer=None tsn=9 command_id=2>
2021-09-24 15:01:14 DEBUG (MainThread) [zigpy.zcl] [0xddeb:1:0xef00] ZCL request 0x0002: [Command(status=0, tsn=30, command_id=286, function=0, data=[1, 0])]
2021-09-24 15:01:14 DEBUG (MainThread) [zhaquirks.tuya] [0xddeb:1:0xef00] Received value [0] for attribute 0x011e (command 0x0002)
2021-09-24 15:01:14 DEBUG (MainThread) [zigpy.zcl] [0xddeb:1:0xef00] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=CLUSTER_COMMAND manufacturer_specific=False is_reply=True disable_default_response=False> manufacturer=None tsn=10 command_id=2>
2021-09-24 15:01:14 DEBUG (MainThread) [zigpy.zcl] [0xddeb:1:0xef00] ZCL request 0x0002: [Command(status=0, tsn=34, command_id=546, function=0, data=[4, 0, 0, 0, 131])]
2021-09-24 15:01:14 DEBUG (MainThread) [zhaquirks.tuya] [0xddeb:1:0xef00] Received value [0, 0, 0, 131] for attribute 0x0222 (command 0x0002)
2021-09-24 15:01:25 DEBUG (MainThread) [zigpy.zcl] [0xddeb:1:0xef00] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=CLUSTER_COMMAND manufacturer_specific=False is_reply=True disable_default_response=False> manufacturer=None tsn=11 command_id=2>
2021-09-24 15:01:25 DEBUG (MainThread) [zigpy.zcl] [0xddeb:1:0xef00] ZCL request 0x0002: [Command(status=0, tsn=101, command_id=613, function=0, data=[4, 0, 0, 0, 42])]
2021-09-24 15:01:25 DEBUG (MainThread) [zhaquirks.tuya] [0xddeb:1:0xef00] Received value [0, 0, 0, 42] for attribute 0x0265 (command 0x0002)
2021-09-24 15:01:35 DEBUG (MainThread) [zigpy.zcl] [0xddeb:1:0xef00] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=CLUSTER_COMMAND manufacturer_specific=False is_reply=True disable_default_response=False> manufacturer=None tsn=12 command_id=2>
2021-09-24 15:01:35 DEBUG (MainThread) [zigpy.zcl] [0xddeb:1:0xef00] ZCL request 0x0002: [Command(status=0, tsn=102, command_id=614, function=0, data=[4, 0, 0, 0, 34])]
2021-09-24 15:01:35 DEBUG (MainThread) [zhaquirks.tuya] [0xddeb:1:0xef00] Received value [0, 0, 0, 34] for attribute 0x0266 (command 0x0002)
2021-09-24 15:01:35 DEBUG (MainThread) [homeassistant.components.zha.core.channels.base] [0xDDEB:1:0x0201]: Attribute report 'MoesThermostat'[min_heat_setpoint_limit] = 3400
2021-09-24 15:01:35 DEBUG (MainThread) [homeassistant.components.zha.entity] climate.tze200_qc4fpmcn_ts0601_3e02dbfe_thermostat: Attribute 'min_heat_setpoint_limit' = 3400 update
2021-09-24 15:01:46 DEBUG (MainThread) [zigpy.zcl] [0xddeb:1:0xef00] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=CLUSTER_COMMAND manufacturer_specific=False is_reply=True disable_default_response=False> manufacturer=None tsn=13 command_id=2>
2021-09-24 15:01:46 DEBUG (MainThread) [zigpy.zcl] [0xddeb:1:0xef00] ZCL request 0x0002: [Command(status=0, tsn=103, command_id=103, function=0, data=[8, 19, 1, 1, 0, 0, 34, 0, 0])]
2021-09-24 15:01:46 DEBUG (MainThread) [zhaquirks.tuya] [0xddeb:1:0xef00] Received value [19, 1, 1, 0, 0, 34, 0, 0] for attribute 0x0067 (command 0x0002)
2021-09-24 15:01:56 DEBUG (MainThread) [zigpy.zcl] [0xddeb:1:0xef00] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=CLUSTER_COMMAND manufacturer_specific=False is_reply=True disable_default_response=False> manufacturer=None tsn=14 command_id=2>
2021-09-24 15:01:56 DEBUG (MainThread) [zigpy.zcl] [0xddeb:1:0xef00] ZCL request 0x0002: [Command(status=0, tsn=104, command_id=616, function=0, data=[4, 0, 0, 0, 0])]
2021-09-24 15:01:56 DEBUG (MainThread) [zhaquirks.tuya] [0xddeb:1:0xef00] Received value [0, 0, 0, 0] for attribute 0x0268 (command 0x0002)
2021-09-24 15:02:01 DEBUG (MainThread) [zigpy.device] [0xddeb] Delivery error for seq # 0x0e, on endpoint id 1 cluster 0xef00: message send failure
2021-09-24 15:02:07 DEBUG (MainThread) [zigpy.zcl] [0xddeb:1:0xef00] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=CLUSTER_COMMAND manufacturer_specific=False is_reply=True disable_default_response=False> manufacturer=None tsn=15 command_id=2>
2021-09-24 15:02:07 DEBUG (MainThread) [zigpy.zcl] [0xddeb:1:0xef00] ZCL request 0x0002: [Command(status=0, tsn=105, command_id=617, function=0, data=[4, 0, 0, 0, 42])]
2021-09-24 15:02:07 DEBUG (MainThread) [zhaquirks.tuya] [0xddeb:1:0xef00] Received value [0, 0, 0, 42] for attribute 0x0269 (command 0x0002)
[... cut ...]
DEBUG: Received Zigbee message from '0x5c0272fffedb023e', type 'commandSetDataResponse', cluster 'manuSpecificTuya', data '{"data":{"data":[0,0,0,48],"type":"Buffer"},"datatype":2,"dp":16,"fn":0,"status":0,"transid":16}' from endpoint 1 with groupID 0
DEBUG: Received Zigbee message from '0x5c0272fffedb023e', type 'commandSetDataResponse', cluster 'manuSpecificTuya', data '{"data":{"data":[0,0,0,44],"type":"Buffer"},"datatype":2,"dp":105,"fn":0,"status":0,"transid":105}' from endpoint 1 with groupID 0
```

Compare DP with tsn above. Funny but that's true.
That's why it's easy to migrate code from Z2M, if there is a support. You can easily map DPs to attributes

MattWestb, 2021-11-08:
Can being that the "tuya converter" is doing that in the conversion but then it good that is not being used for other devices then its being little crazy like this:
```
2021-11-08 01:10:34 DEBUG (MainThread) [zigpy.zcl] [0xde47:1:0x0020] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=CLUSTER_COMMAND manufacturer_specific=False is_reply=True disable_default_response=True> manufacturer=None tsn=105 command_id=0>
2021-11-08 01:10:34 DEBUG (MainThread) [zigpy.zcl] [0xde47:1:0x0020] ZCL request 0x0000: []
2021-11-08 01:10:34 DEBUG (MainThread) [zigpy.zcl] [0xde47:1:0x0020] No handler for cluster command 0
2021-11-08 01:10:34 DEBUG (MainThread) [homeassistant.components.zha.core.channels.base] [0xDE47:1:0x0020]: Received 105 tsn command 'checkin': []
2021-11-08 01:10:34 DEBUG (MainThread) [zigpy.device] [0xde47] Extending timeout for 0x69 request
2021-11-08 01:10:36 DEBUG (MainThread) [zigpy.zcl] [0xde47:1:0x0020] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=GLOBAL_COMMAND manufacturer_specific=False is_reply=True disable_default_response=False> manufacturer=None tsn=105 command_id=Command.Default_Response>
2021-11-08 01:10:36 DEBUG (MainThread) [homeassistant.components.zha.core.channels.base] [0xDE47:1:0x0020]: executed 'checkin_response' command with args: '(True, 8)' kwargs: '{'tsn': 105}' result: [0, <Status.SUCCESS: 0>]
2021-11-08 01:10:36 DEBUG (MainThread) [zigpy.device] [0xde47] Extending timeout for 0xe2 request
2021-11-08 01:10:38 DEBUG (MainThread) [zigpy.zcl] [0xde47:1:0x0020] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=GLOBAL_COMMAND manufacturer_specific=False is_reply=True disable_default_response=False> manufacturer=None tsn=226 command_id=Command.Default_Response>
2021-11-08 01:10:38 DEBUG (MainThread) [homeassistant.components.zha.core.channels.base] [0xDE47:1:0x0020]: executed 'fast_poll_stop' command with args: '()' kwargs: '{}' result: [1, <Status.TIMEOUT: 148>]
```
Its one old IKEA Dimmer (the round one with gyro) that is doing on checking to its parent.
And its using DP 226 . . . .

If the tuya decoder is putting the tsn as tuya DPs then i understand way you is liking it very much but is on the border to sheeting !!
(i think its little bad using the same name for different system fuctions)

And then it shall having `zigpy.zcl` and `zhaquirks.tuya` as debug for getting the right information.

I shall doing it on the test network with the TRV and looking little more :-)))

Is this TRV sending wrong data or is Zpgpy using it wrong and loosing the preposition of the setpoint  ?
I think its possible doing one if / else for the setpoint in the quirk and not need doing on new device class for it like is being done but its looks working.

jacekk015, 2021-11-08:
OK - I got the calculation for attribute -> DP
Example - local temp for Maxsmart:
tsn=24, command_id=536 attribute 0x0218
0x0218 -> 0x18 -> 24 decimal :-))
us as a HEX only last two elements

Maxsmart Window detect:
tsn=107, command_id=363 attribute 0x016b
0x016b -> 0x6b -> 107 decimal

SITERWELL_TARGET_TEMP_ATTR = 0x0202
0x0202 -> 0x02 -> 2 decimal
Let's go to Z2M Siterwell
https://github.com/Koenkk/zigbee-herdsman-converters/blob/master/devices/siterwell.js#L13
https://github.com/Koenkk/zigbee-herdsman-converters/blob/master/devices/siterwell.js#L21
fz.tuya_thermostat
https://github.com/Koenkk/zigbee-herdsman-converters/blob/master/converters/fromZigbee.js#L3856
case tuya.dataPoints.heatingSetpoint:
https://github.com/Koenkk/zigbee-herdsman-converters/blob/master/converters/fromZigbee.js#L3906
heatingSetpoint: 2,
https://github.com/Koenkk/zigbee-herdsman-converters/blob/master/lib/tuya.js#L190

MattWestb, 2021-11-08:
Not importing but i was playing little with away mode temperature up and down and was getting one log:
```
2021-11-08 22:28:57 DEBUG (MainThread) [zigpy.zcl] [0xa419:1:0xef00] ZCL request 0x0001: [Command(status=2, tsn=67, command_id=514, function=0, data=[4, 0, 0, 0, 50])]
2021-11-08 22:28:57 DEBUG (MainThread) [zhaquirks.tuya] [0xa419:1:0xef00] Received value [0, 0, 0, 50] for attribute 0x0202 (command 0x0001)
2021-11-08 22:28:57 DEBUG (MainThread) [zigpy.zcl] [0xa419:1:0xef00] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=CLUSTER_COMMAND manufacturer_specific=False is_reply=True disable_default_response=False> manufacturer=None tsn=10 command_id=1>
2021-11-08 22:28:57 DEBUG (MainThread) [zigpy.zcl] [0xa419:1:0xef00] ZCL request 0x0001: [Command(status=2, tsn=68, command_id=514, function=0, data=[4, 0, 0, 0, 50])]
2021-11-08 22:28:57 DEBUG (MainThread) [zhaquirks.tuya] [0xa419:1:0xef00] Received value [0, 0, 0, 50] for attribute 0x0202 (command 0x0001)
2021-11-08 22:29:01 DEBUG (MainThread) [zigpy.zcl] [0xa419:1:0xef00] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=CLUSTER_COMMAND manufacturer_specific=False is_reply=True disable_default_response=False> manufacturer=None tsn=11 command_id=1>
2021-11-08 22:29:01 DEBUG (MainThread) [zigpy.zcl] [0xa419:1:0xef00] ZCL request 0x0001: [Command(status=2, tsn=69, command_id=514, function=0, data=[4, 0, 0, 0, 50])]
2021-11-08 22:29:01 DEBUG (MainThread) [zhaquirks.tuya] [0xa419:1:0xef00] Received value [0, 0, 0, 50] for attribute 0x0202 (command 0x0001)
2021-11-08 22:29:02 DEBUG (MainThread) [zigpy.zcl] [0xa419:1:0xef00] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=CLUSTER_COMMAND manufacturer_specific=False is_reply=True disable_default_response=False> manufacturer=None tsn=12 command_id=1>
2021-11-08 22:29:02 DEBUG (MainThread) [zigpy.zcl] [0xa419:1:0xef00] ZCL request 0x0001: [Command(status=2, tsn=70, command_id=514, function=0, data=[4, 0, 0, 0, 50])]
2021-11-08 22:29:02 DEBUG (MainThread) [zhaquirks.tuya] [0xa419:1:0xef00] Received value [0, 0, 0, 50] for attribute 0x0202 (command 0x0001)
2021-11-08 22:29:03 DEBUG (MainThread) [zigpy.zcl] [0xa419:1:0xef00] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=CLUSTER_COMMAND manufacturer_specific=False is_reply=True disable_default_response=False> manufacturer=None tsn=13 command_id=1>
2021-11-08 22:29:03 DEBUG (MainThread) [zigpy.zcl] [0xa419:1:0xef00] ZCL request 0x0001: [Command(status=2, tsn=70, command_id=514, function=0, data=[4, 0, 0, 0, 50])]
2021-11-08 22:29:03 DEBUG (MainThread) [zhaquirks.tuya] [0xa419:1:0xef00] Received value [0, 0, 0, 50] for attribute 0x0202 (command 0x0001)
2021-11-08 22:29:04 DEBUG (MainThread) [zigpy.zcl] [0xa419:1:0xef00] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=CLUSTER_COMMAND manufacturer_specific=False is_reply=True disable_default_response=False> manufacturer=None tsn=14 command_id=1>
2021-11-08 22:29:04 DEBUG (MainThread) [zigpy.zcl] [0xa419:1:0xef00] ZCL request 0x0001: [Command(status=2, tsn=70, command_id=514, function=0, data=[4, 0, 0, 0, 50])]
2021-11-08 22:29:04 DEBUG (MainThread) [zhaquirks.tuya] [0xa419:1:0xef00] Received value [0, 0, 0, 50] for attribute 0x0202 (command 0x0001)
2021-11-08 22:29:05 DEBUG (MainThread) [zigpy.zcl] [0xa419:1:0xef00] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=CLUSTER_COMMAND manufacturer_specific=False is_reply=True disable_default_response=False> manufacturer=None tsn=15 command_id=1>
2021-11-08 22:29:05 DEBUG (MainThread) [zigpy.zcl] [0xa419:1:0xef00] ZCL request 0x0001: [Command(status=2, tsn=71, command_id=514, function=0, data=[4, 0, 0, 0, 50])]
2021-11-08 22:29:05 DEBUG (MainThread) [zhaquirks.tuya] [0xa419:1:0xef00] Received value [0, 0, 0, 50] for attribute 0x0202 (command 0x0001)
2021-11-08 22:29:06 DEBUG (MainThread) [zigpy.zcl] [0xa419:1:0xef00] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=CLUSTER_COMMAND manufacturer_specific=False is_reply=True disable_default_response=False> manufacturer=None tsn=16 command_id=1>
2021-11-08 22:29:06 DEBUG (MainThread) [zigpy.zcl] [0xa419:1:0xef00] ZCL request 0x0001: [Command(status=2, tsn=72, command_id=514, function=0, data=[4, 0, 0, 0, 55])]
2021-11-08 22:29:06 DEBUG (MainThread) [zhaquirks.tuya] [0xa419:1:0xef00] Received value [0, 0, 0, 55] for attribute 0x0202 (command 0x0001)
2021-11-08 22:29:07 DEBUG (MainThread) [zigpy.zcl] [0xa419:1:0xef00] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=CLUSTER_COMMAND manufacturer_specific=False is_reply=True disable_default_response=False> manufacturer=None tsn=17 command_id=1>
2021-11-08 22:29:07 DEBUG (MainThread) [zigpy.zcl] [0xa419:1:0xef00] ZCL request 0x0001: [Command(status=2, tsn=72, command_id=514, function=0, data=[4, 0, 0, 0, 55])]
2021-11-08 22:29:07 DEBUG (MainThread) [zhaquirks.tuya] [0xa419:1:0xef00] Received value [0, 0, 0, 55] for attribute 0x0202 (command 0x0001)
2021-11-08 22:29:08 DEBUG (MainThread) [zigpy.zcl] [0xa419:1:0xef00] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=CLUSTER_COMMAND manufacturer_specific=False is_reply=True disable_default_response=False> manufacturer=None tsn=18 command_id=1>
2021-11-08 22:29:08 DEBUG (MainThread) [zigpy.zcl] [0xa419:1:0xef00] ZCL request 0x0001: [Command(status=2, tsn=73, command_id=514, function=0, data=[4, 0, 0, 0, 55])]
[... cut ...]
Can being that my TRV is using one old Zigbee module (TYST11) that is doing it "normal Zigbee way" and newer is making it tuya way = tsn = DP that i think it not bad and shall working well then sending different DP but if sending the same DP (changing the temp and the TRV is sending multiple different temperatures with same DP) it can being little strange for the Zigbee network stack and the underlying IEEE network that is using tsn logic for acking frames and reseeding them is they is being lost and double transmitting.

If its working with TZE200 then its great but dont forgetting its looks not working with devices that having TYST11 Zigbee modules.

tuya is still tuya !! 
:-))))

jacekk015, 2021-11-08:
You have ZCL request 0x0001:
I have ZCL request 0x0002:
The only difference I see

Youpimatin, 2021-11-08:
Identified attributes :
0x0065 # programmable schedule (Received value [6, 0, 40, 11, 30, 42, 13, 30, 44, 17, 30, 46, 6, 0, 48, 12, 0, 46, 14, 30, 44, 17, 30, 42, 6, 0, 38, 12, 30, 40, 14, 30, 42, 18, 30, 40] for :
 Monday to friday :  06:00 20°C / 11:30 21°C / 13:30 22°C / 17:30 23°C
 Saturday : 06:00 24°C / 12:00 23°C / 14:30 22°C / 17:30 21°C
 Sunday : 06:00 19°C / 12:30 20°C / 14:30 21°C / 18:30 20°C )
0x0104 # Boost heating (Received value 0:off / 1:in progress)	
0x0108 # Open window detection (Received value 0:off / 1:on)
0x010d # child lock ( Received value 0:unlocked / 1:locked)
0x016a # Energy saving mode (Received value 0:off / 1:on)
0x0202 # target room temp (in degree, NOT in decidegree)
0x0203 # current room temp (decidegree)
0x020e # battery percentage (updated every 4 hours )
0x0267 # Boost heating countdown in second (Received value [0, 0, 1, 44] for 300)
0x0268 # opening percentage (values of 0 / 25 / 50 / 75 / 100)
0x0269 # Temperature compensation ( Received value [255, 255, 255, 255] for -1)
0x026b # Energy saving mode temperature ( Received value [0, 0, 0, 15] )
0x026c # Max target temp ( Received value [0, 0, 0, 35] )
0x026d # Min target temp (Received value [0, 0, 0, 5])

Unidentified attributes :
0x0205 ( [0, 0, 0, 0] / [0, 0, 0, 5] )
0x0401 (1)
0x0407 (0)
0x0409 (1)

MattWestb, 2021-11-08:
0x205 looks have one temperature and one more data, Away with temperature and date or holiday ?
0x40x frost, window, valve jammed alarm ?

jacekk015, 2021-11-08:
moesSheatingSetpoint: 2,
0x0202 # target room temp (in degree, NOT in decidegree)
2021-11-07 18:27:38 DEBUG (MainThread) [zigpy.zcl] [0xef69:1:0xef00] ZCL request 0x0001: [Command(status=0, tsn=18, command_id=514, function=0, data=[4, 0, 0, 0, 21])]
2021-11-07 18:27:38 DEBUG (MainThread) [zhaquirks.tuya] [0xef69:1:0xef00] Received value [0, 0, 0, 21] for attribute 0x0202 (command 0x0001)
****************************************

moesSlocalTemp: 3,
0x0203 # current room temp (decidegree)
2021-11-07 18:27:48 DEBUG (MainThread) [zigpy.zcl] [0x6937:1:0xef00] ZCL request 0x0001: [Command(status=0, tsn=190, command_id=515, function=0, data=[4, 0, 0, 0, 170])]
2021-11-07 18:27:48 DEBUG (MainThread) [zhaquirks.tuya] [0x6937:1:0xef00] Received value [0, 0, 0, 170] for attribute 0x0203 (command 0x0001)
****************************************

moesSvalvePosition: 104,
0x0268 # opening percentage (values of 0 / 25 / 50 / 75 / 100)
2021-11-07 18:27:48 DEBUG (MainThread) [zigpy.zcl] [0x6937:1:0xef00] ZCL request 0x0001: [Command(status=0, tsn=191, command_id=616, function=0, data=[4, 0, 0, 0, 50])]
2021-11-07 18:27:48 DEBUG (MainThread) [zhaquirks.tuya] [0x6937:1:0xef00] Received value [0, 0, 0, 50] for attribute 0x0268 (command 0x0001)
****************************************

moesSbattery: 14,
0x020e # battery percentage (updated every 4 hours )
2021-11-07 18:28:04 DEBUG (MainThread) [zigpy.zcl] [0xef69:1:0xef00] ZCL request 0x0001: [Command(status=0, tsn=24, command_id=526, function=0, data=[4, 0, 0, 0, 61])]
2021-11-07 18:28:04 DEBUG (MainThread) [zhaquirks.tuya] [0xef69:1:0xef00] Received value [0, 0, 0, 61] for attribute 0x020e (command 0x0001)
****************************************

moesSschedule: 101,
0x0065 # programmable schedule (Received value [6, 0, 40, 11, 30, 42, 13, 30, 44, 17, 30, 46, 6, 0, 48, 12, 0, 46, 14, 30, 44, 17, 30, 42, 6, 0, 38, 12, 30, 40, 14, 30, 42, 18, 30, 40] for :
Monday to friday : 06:00 20°C / 11:30 21°C / 13:30 22°C / 17:30 23°C
Saturday : 06:00 24°C / 12:00 23°C / 14:30 22°C / 17:30 21°C
Sunday : 06:00 19°C / 12:30 20°C / 14:30 21°C / 18:30 20°C )
2021-11-07 18:31:12 DEBUG (MainThread) [zigpy.zcl] [0xef69:1:0xef00] ZCL request 0x0001: [Command(status=0, tsn=44, command_id=101, function=0, data=[36, 6, 0, 40, 11, 30, 42, 13, 30, 44, 17, 30, 46, 6, 0, 48, 12, 0, 46, 14, 30, 44, 17, 30, 42, 6, 0, 38, 12, 30, 40, 14, 30, 42, 18, 30, 40])]
2021-11-07 18:31:12 DEBUG (MainThread) [zhaquirks.tuya] [0xef69:1:0xef00] Received value [6, 0, 40, 11, 30, 42, 13, 30, 44, 17, 30, 46, 6, 0, 48, 12, 0, 46, 14, 30, 44, 17, 30, 42, 6, 0, 38, 12, 30, 40, 14, 30, 42, 18, 30, 40] for attribute 0x0065 (command 0x0001)
****************************************

moesSboostHeating: 4,
0x0104 # Boost heating (Received value 0:off / 1:in progress)
2021-11-07 18:31:18 DEBUG (MainThread) [zigpy.zcl] [0xef69:1:0xef00] ZCL request 0x0001: [Command(status=0, tsn=55, command_id=260, function=0, data=[1, 0])]
2021-11-07 18:31:18 DEBUG (MainThread) [zhaquirks.tuya] [0xef69:1:0xef00] Received value [0] for attribute 0x0104 (command 0x0001)
****************************************

moesSwindowDetectionFunktion_A2: 8,
0x0108 # Open window detection (Received value 0:off / 1:on)
2021-11-07 18:31:11 DEBUG (MainThread) [zigpy.zcl] [0xef69:1:0xef00] ZCL request 0x0001: [Command(status=0, tsn=40, command_id=264, function=0, data=[1, 0])]
2021-11-07 18:31:11 DEBUG (MainThread) [zhaquirks.tuya] [0xef69:1:0xef00] Received value [0] for attribute 0x0108 (command 0x0001)
****************************************

moesSchildLock: 13,
0x010d # child lock ( Received value 0:unlocked / 1:locked)
2021-11-07 18:31:12 DEBUG (MainThread) [zigpy.zcl] [0xef69:1:0xef00] ZCL request 0x0001: [Command(status=0, tsn=42, command_id=269, function=0, data=[1, 1])]
2021-11-07 18:31:12 DEBUG (MainThread) [zhaquirks.tuya] [0xef69:1:0xef00] Received value [1] for attribute 0x010d (command 0x0001)
****************************************

moesSecoMode: 106,
0x016a # Energy saving mode (Received value 0:off / 1:on)
2021-11-07 18:31:13 DEBUG (MainThread) [zigpy.zcl] [0xef69:1:0xef00] ZCL request 0x0001: [Command(status=0, tsn=48, command_id=362, function=0, data=[1, 0])]
2021-11-07 18:31:13 DEBUG (MainThread) [zhaquirks.tuya] [0xef69:1:0xef00] Received value [0] for attribute 0x016a (command 0x0001)
****************************************

moesSboostHeatingCountdownTimeSet: 103,
0x0267 # Boost heating countdown in second (Received value [0, 0, 1, 44] for 300)
2021-11-07 18:31:12 DEBUG (MainThread) [zigpy.zcl] [0xef69:1:0xef00] ZCL request 0x0001: [Command(status=0, tsn=45, command_id=615, function=0, data=[4, 0, 0, 1, 44])]
2021-11-07 18:31:12 DEBUG (MainThread) [zhaquirks.tuya] [0xef69:1:0xef00] Received value [0, 0, 1, 44] for attribute 0x0267 (command 0x0001)
****************************************

moesScompensationTempSet: 105,
0x0269 # Temperature compensation ( Received value [255, 255, 255, 255] for -1)
2021-11-07 18:31:13 DEBUG (MainThread) [zigpy.zcl] [0xef69:1:0xef00] ZCL request 0x0001: [Command(status=0, tsn=47, command_id=617, function=0, data=[4, 0, 0, 0, 0])]
2021-11-07 18:31:13 DEBUG (MainThread) [zhaquirks.tuya] [0xef69:1:0xef00] Received value [0, 0, 0, 0] for attribute 0x0269 (command 0x0001)
****************************************

moesSecoModeTempSet: 107,
[... 14 lines cut ...]
2021-11-07 18:31:14 DEBUG (MainThread) [zhaquirks.tuya] [0xef69:1:0xef00] Received value [0, 0, 0, 5] for attribute 0x026d (command 0x0001)
[... cut ...]
2021-11-07 18:31:11 DEBUG (MainThread) [zigpy.zcl] [0xef69:1:0xef00] ZCL request 0x0001: [Command(status=0, tsn=39, command_id=1031, function=0, data=[1, 0])]
2021-11-07 18:31:11 DEBUG (MainThread) [zhaquirks.tuya] [0xef69:1:0xef00] Received value [0] for attribute 0x0407 (command 0x0001)
****************************************

moesSwindowDetection: 9,
0x0409 (1)
2021-11-07 18:31:12 DEBUG (MainThread) [zigpy.zcl] [0xef69:1:0xef00] ZCL request 0x0001: [Command(status=0, tsn=41, command_id=1033, function=0, data=[1, 1])]
2021-11-07 18:31:12 DEBUG (MainThread) [zhaquirks.tuya] [0xef69:1:0xef00] Received value [1] for attribute 0x0409 (command 0x0001)

MattWestb, 2021-11-08:
The 0x0409 is the alarm and the 0x0108 is the enable / disabling the windows detection like on the Siterwell is doing.

jacekk015, 2021-11-08:
User needs to confirm this. Tricky, because manual says something else?

[image]
[BRT-100 User manual.pdf](https://github.com/zigpy/zha-device-handlers/files/7500592/BRT-100.User.manual.pdf)

Youpimatin, 2021-11-08:
I edited, 0x0108 is the mode (function A2 in the manual)
0x0108 # Open window detection MODE (Received value 0:function disabled / 1:function enabled)

jacekk015, 2021-11-08:
So 0x0409 could be the window alarm, but pastebin logs shows already [1] there
also
0x0401 needs explanation which mode has which value

Youpimatin, 2021-11-08:
0x0401 :
0 : Manual Mode
1 : Holiday Mode
3 : Schedule Mode

jacekk015, 2021-11-09:
Ok - to SUM it up, since I'm at work now ;-)
I should prepare a quirk for you, maybe today, maybe tomorrow. It will be 1 revision so maybe not 100% will be OK.
Last few questions:
1) 0x0202 # target room temp -  what step you have there 1 or 0.5??
2) 0x0268 # opening percentage - I will create a edit field, but I don't know if writing will change that on TRV
3) 0x020e # battery percentage - Z2M treat that value as a remaining battery level in % - are you able to confirm that?
```
battery: () => new Numeric('battery', access.STATE).withUnit('%').withDescription('Remaining battery in %').withValueMin(0).withValueMax(100),
```
4) 0x0065 # programmable schedule - since there's no support for component on the ZHA side we are omitting that - there will be attribute only, without entity
5) moesSboostHeatingCountdown- minutes, seconds??
6)moesSreset: 7, 0x0407 (0) - probably with omit this??

Youpimatin, 2021-11-09:
**0x0202 # target room temp -  what step you have there 1 or 0.5??**
1°C steps. (0.5°C setting on the TRV itself, rounded for zigbee transmission)

**0x0268 # opening percentage - I will create a edit field, but I don't know if writing will change that on TRV**
I'll report if writing to this parameter works or not, to adjust quirk if necessary.

**0x020e # battery percentage - Z2M treat that value as a remaining battery level in % - are you able to confirm that?**
Yes ! mean value has slowly decremented in the last two days

**0x0065 # programmable schedule - since there's no support for component on the ZHA side we are omitting that - there will be attribute only, without entity**
OK...

**moesSboostHeatingCountdown- minutes, seconds?? moesSreset: 7, 0x0407 (0) - probably with omit this??**
0x0407 (0 / 1) changes when current temp crosses target temp... maybe just a flag to report if the TRV needs to heat the room or not... if it's only that, it's not very useful since we can compare current temp and target temp

jacekk015, 2021-11-09:
0x0202 # target room temp - what step you have there 1 or 0.5??
1°C steps. (0.5°C setting on the TRV itself, rounded for zigbee interface)
That's strange. If you have a possibility to remove the quirk from the folder and repair device. Then search for 0x0202 in the HA logs. Maybe we could see what really is send to 0x0202 if the quirk isn't available.
It's possible that we see an INT value because quirk is in action and the value is mapped to uint.
Or maybe you could just comment out 0x0202 in your quirk. Try please both ways if possible!
```
#MOES_TARGET_TEMP_ATTR = 0x0202  # target room temp (decidegree)
```
It is mapped to MOES_TARGET_TEMP_ATTR: ("target_temperature", t.uint32_t),
so quirk modifies it to unit32_t
We would like to see RAW sent value.

5. moesSboostHeatingCountdown- minutes, seconds??
In boost mode TRV counts[often] form some value to zero[counter]. Often these are seconds. Just want to confirm this.

6. moesSreset: 7, 0x0407 (0) - probably with omit this??
Probably switch to reset the TRV is not needed as a entity, or maybe it is??

Youpimatin, 2021-11-09:
(I edited my previous message for the battery level topic)

- OK, yes it's possible, I'll try the two quirk methods tonight.

- For the countdown, it is said in the manual that the app shows a countdown, but since I saw nothing in the data when in boost mode, I imagined that the application only used the last hour of changing 0x0104 from 0 to 1 and the last setting of 0x0267, in reality I don't know.

- 0x0407 changes between 0 and 1 with the temperatures, without resetting the TRV from the menu.

-  I did'nt try reset, since valve would change zigbee ID... but I can test if you want (I don't know if it would be a 'dangerous' function if used without knowing)

Youpimatin, 2021-11-10:
So I tried to comment "MOES_TARGET_TEMP_ATTR = 0x0202" line in the quirk and restart HA, but ZHA doesn't want to initialize in this case 'configuration problem"
I deleted the quirk from the folder and paired the TRV, here is the (splitted) log :
```
2021-11-10 09:27:08 INFO (MainThread) [zigpy.application] Device 0xef69 (84:fd:27:ff:fe:a5:a9:fb) left the network
2021-11-10 09:27:10 DEBUG (MainThread) [zigpy.zcl] [0xf153:11:0x0000] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=GLOBAL_COMMAND manufacturer_specific=False is_reply=True disable_default_response=True> manufacturer=None tsn=76 command_id=Command.Report_Attributes>
2021-11-10 09:27:10 DEBUG (MainThread) [zigpy.zcl] [0xf153:11:0x0000] ZCL request 0x000a: [[Attribute(attrid=1, value=<TypeValue type=uint8_t, value=66>)]]
2021-11-10 09:27:10 DEBUG (MainThread) [zigpy.zcl] [0xf153:11:0x0000] Attribute report received: app_version=66
2021-11-10 09:27:13 DEBUG (MainThread) [zigpy.device] [0x3bcf] Delivery error for seq # 0xb8, on endpoint id 0 cluster 0x0034: message send failure
2021-11-10 09:27:13 DEBUG (MainThread) [bellows.ezsp.protocol] Send command sendBroadcast: (<BroadcastAddress.ALL_ROUTERS_AND_COORDINATOR: 65532>, EmberApsFrame(profileId=0, clusterId=54, sourceEndpoint=0, destinationEndpoint=0, options=<EmberApsOption.APS_OPTION_NONE: 0>, groupId=0, sequence=189), 0, 190, b'\xbd<\x00')
2021-11-10 09:27:13 DEBUG (MainThread) [bellows.ezsp.protocol] Send command addTransientLinkKey: (ff:ff:ff:ff:ff:ff:ff:ff, [90, 105, 103, 66, 101, 101, 65, 108, 108, 105, 97, 110, 99, 101, 48, 57])
2021-11-10 09:27:13 DEBUG (bellows.thread_0) [bellows.uart] Sending: b'37e02157541ce94d59947c25aa5592499c4e9aab53cddab7fdec817e'
2021-11-10 09:27:13 DEBUG (bellows.thread_0) [bellows.uart] Data frame: b'74e0a157541c150fc4737e'
2021-11-10 09:27:13 DEBUG (bellows.thread_0) [bellows.uart] Sending: b'8070787e'
2021-11-10 09:27:13 DEBUG (bellows.thread_0) [bellows.uart] Data frame: b'04e0b157546f10b259a24a25aa5592499cf3d8abedce9874fe7b5f89b59f7e'
2021-11-10 09:27:13 DEBUG (bellows.thread_0) [bellows.uart] Sending: b'8160597e'
2021-11-10 09:27:13 DEBUG (bellows.thread_0) [bellows.uart] Sending: b'41e121575485ea4da66bb5da55aac820fb0c42ceaca20be29ca800eccc4769897e'
2021-11-10 09:27:13 DEBUG (bellows.thread_0) [bellows.uart] Data frame: b'15e1a157548515f15e7e'
2021-11-10 09:27:13 DEBUG (bellows.thread_0) [bellows.uart] Sending: b'82503a7e'
2021-11-10 09:27:13 DEBUG (MainThread) [bellows.ezsp.protocol] Application frame 54 (sendBroadcast) received: b'00bd'
2021-11-10 09:27:13 DEBUG (MainThread) [bellows.ezsp.protocol] Application frame 69 (incomingMessageHandler) received: b'0500003600000000000000bdff000000ffff03bd3c00'
2021-11-10 09:27:13 DEBUG (MainThread) [bellows.zigbee.application] Received incomingMessageHandler frame with [<EmberIncomingMessageType.INCOMING_BROADCAST_LOOPBACK: 5>, EmberApsFrame(profileId=0, clusterId=54, sourceEndpoint=0, destinationEndpoint=0, options=<EmberApsOption.APS_OPTION_NONE: 0>, groupId=0, sequence=189), 255, 0, 0x0000, 255, 255, b'\xbd<\x00']
2021-11-10 09:27:13 DEBUG (MainThread) [zigpy.zdo] [0x0000:zdo] ZDO request ZDOCmd.Mgmt_Permit_Joining_req: [60, <Bool.false: 0>]
2021-11-10 09:27:13 DEBUG (MainThread) [bellows.ezsp.protocol] Application frame 175 (addTransientLinkKey) received: b'00'
2021-11-10 09:27:13 DEBUG (MainThread) [homeassistant.components.zha.core.device] [0x0000](EZSP): started initialization
2021-11-10 09:27:13 DEBUG (MainThread) [homeassistant.components.zha.core.channels.base] [0x0000:ZDO](EZSP): 'async_initialize' stage succeeded
2021-11-10 09:27:13 DEBUG (MainThread) [homeassistant.components.zha.core.device] [0x0000](EZSP): power source: Mains
2021-11-10 09:27:13 DEBUG (MainThread) [homeassistant.components.zha.core.device] [0x0000](EZSP): completed initialization
2021-11-10 09:27:14 DEBUG (bellows.thread_0) [bellows.uart] Data frame: b'25e1b1575415134ea6944a13aa5592499c4e271653ce676b827e'
2021-11-10 09:27:14 DEBUG (bellows.thread_0) [bellows.uart] Sending: b'83401b7e'
2021-11-10 09:27:14 DEBUG (MainThread) [bellows.ezsp.protocol] Application frame 63 (messageSentHandler) received: b'06fcff00003600000000000000bdbe0000'
2021-11-10 09:27:14 DEBUG (MainThread) [bellows.zigbee.application] Received messageSentHandler frame with [<EmberOutgoingMessageType.OUTGOING_BROADCAST: 6>, 65532, EmberApsFrame(profileId=0, clusterId=54, sourceEndpoint=0, destinationEndpoint=0, options=<EmberApsOption.APS_OPTION_NONE: 0>, groupId=0, sequence=189), 190, <EmberStatus.SUCCESS: 0>, b'']
2021-11-10 09:27:14 DEBUG (MainThread) [bellows.ezsp.protocol] Send command permitJoining: (60,)
2021-11-10 09:27:14 DEBUG (bellows.thread_0) [bellows.uart] Sending: b'53e62157540829fd1d7e'
2021-11-10 09:27:14 DEBUG (bellows.thread_0) [bellows.uart] Data frame: b'36e6a157540815e2af7e'
2021-11-10 09:27:14 DEBUG (bellows.thread_0) [bellows.uart] Sending: b'8430fc7e'
2021-11-10 09:27:14 DEBUG (MainThread) [bellows.ezsp.protocol] Application frame 34 (permitJoining) received: b'00'
2021-11-10 09:27:16 DEBUG (MainThread) [bellows.ezsp.protocol] Send command nop: ()
2021-11-10 09:27:16 DEBUG (bellows.thread_0) [bellows.uart] Sending: b'64e72157542f86237e'
2021-11-10 09:27:16 DEBUG (bellows.thread_0) [bellows.uart] Data frame: b'47e7a157542fa0f37e'
2021-11-10 09:27:16 DEBUG (bellows.thread_0) [bellows.uart] Sending: b'8520dd7e'
2021-11-10 09:27:16 DEBUG (MainThread) [bellows.ezsp.protocol] Application frame 5 (nop) received: b''
[... cut ...]
2021-11-10 09:27:36 DEBUG (MainThread) [bellows.ezsp.protocol] Application frame 63 (messageSentHandler) received: b'0069ef00003400000040000000bcbc6600'
2021-11-10 09:27:36 DEBUG (MainThread) [bellows.zigbee.application] Received messageSentHandler frame with [<EmberOutgoingMessageType.OUTGOING_DIRECT: 0>, 61289, EmberApsFrame(profileId=0, clusterId=52, sourceEndpoint=0, destinationEndpoint=0, options=<EmberApsOption.APS_OPTION_RETRY: 64>, groupId=0, sequence=188), 188, <EmberStatus.DELIVERY_FAILED: 102>, b'']
```

Youpimatin, 2021-11-10:
```
2021-11-10 09:27:36 DEBUG (bellows.thread_0) [bellows.uart] Data frame: b'41edb157546f15b65894a524ab5593499c018b6c4d609874f7cf3788fc1b3ea3ebccdf6b806f7e'
2021-11-10 09:27:36 DEBUG (bellows.thread_0) [bellows.uart] Sending: b'8520dd7e'
2021-11-10 09:27:36 DEBUG (bellows.thread_0) [bellows.uart] Data frame: b'51edb15754738cd7e31e27db55f9a1f518f327251b7e'
2021-11-10 09:27:36 DEBUG (bellows.thread_0) [bellows.uart] Sending: b'8610be7e'
2021-11-10 09:27:36 DEBUG (bellows.thread_0) [bellows.uart] Data frame: b'61edb157546f15b658944a2eab1593499c1ba31674ab9874fade0d83fd7e1fe5e9b2337e'
2021-11-10 09:27:36 DEBUG (bellows.thread_0) [bellows.uart] Sending: b'87009f7e'
2021-11-10 09:27:36 DEBUG (MainThread) [bellows.ezsp.protocol] Application frame 69 (incomingMessageHandler) received: b'00040100ef0101000100004facc7a0aeffff0a0954010065010400010104'
2021-11-10 09:27:36 DEBUG (MainThread) [bellows.zigbee.application] Received incomingMessageHandler frame with [<EmberIncomingMessageType.INCOMING_UNICAST: 0>, EmberApsFrame(profileId=260, clusterId=61184, sourceEndpoint=1, destinationEndpoint=1, options=<EmberApsOption.APS_OPTION_ENABLE_ROUTE_DISCOVERY: 256>, groupId=0, sequence=79), 172, -57, 0xaea0, 255, 255, b'\tT\x01\x00e\x01\x04\x00\x01\x01']
2021-11-10 09:27:36 DEBUG (MainThread) [zigpy.zcl] [0xaea0:1:0xef00] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=CLUSTER_COMMAND manufacturer_specific=False is_reply=True disable_default_response=False> manufacturer=None tsn=84 command_id=1>
2021-11-10 09:27:36 WARNING (MainThread) [zigpy.zcl] [0xaea0:1:0xef00] Unknown cluster-specific command 1
2021-11-10 09:27:36 DEBUG (MainThread) [zigpy.zcl] [0xaea0:1:0xef00] ZCL request 0x0001: b'\x00e\x01\x04\x00\x01\x01'
2021-11-10 09:27:36 DEBUG (MainThread) [zigpy.zcl] [0xaea0:1:0xef00] No handler for cluster command 1
2021-11-10 09:27:36 DEBUG (MainThread) [bellows.ezsp.protocol] Application frame 89 (incomingRouteRecordHandler) received: b'9965ba8a6dfeffac33bc84bd00'
2021-11-10 09:27:36 DEBUG (MainThread) [bellows.zigbee.application] Received incomingRouteRecordHandler frame with [0x6599, bc:33:ac:ff:fe:6d:8a:ba, 132, -67, []]
2021-11-10 09:27:36 DEBUG (MainThread) [bellows.zigbee.application] Processing route record request: (0x6599, bc:33:ac:ff:fe:6d:8a:ba, 132, -67, [])
2021-11-10 09:27:36 DEBUG (MainThread) [zigpy.device] [0xaea0] Delivery error for seq # 0xb9, on endpoint id 0 cluster 0x0034: message send failure
2021-11-10 09:27:36 DEBUG (MainThread) [bellows.ezsp.protocol] Application frame 69 (incomingMessageHandler) received: b'00040100000b01400100005584bd9965ffff07186e0a0100204202'
2021-11-10 09:27:36 DEBUG (MainThread) [bellows.zigbee.application] Received incomingMessageHandler frame with [<EmberIncomingMessageType.INCOMING_UNICAST: 0>, EmberApsFrame(profileId=260, clusterId=0, sourceEndpoint=11, destinationEndpoint=1, options=<EmberApsOption.APS_OPTION_ENABLE_ROUTE_DISCOVERY|APS_OPTION_RETRY: 320>, groupId=0, sequence=85), 132, -67, 0x6599, 255, 255, b'\x18n\n\x01\x00 B']
2021-11-10 09:27:36 DEBUG (MainThread) [zigpy.zcl] [0x6599:11:0x0000] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=GLOBAL_COMMAND manufacturer_specific=False is_reply=True disable_default_response=True> manufacturer=None tsn=110 command_id=Command.Report_Attributes>
2021-11-10 09:27:36 DEBUG (MainThread) [zigpy.zcl] [0x6599:11:0x0000] ZCL request 0x000a: [[Attribute(attrid=1, value=<TypeValue type=uint8_t, value=66>)]]
2021-11-10 09:27:36 DEBUG (MainThread) [zigpy.zcl] [0x6599:11:0x0000] Attribute report received: app_version=66
2021-11-10 09:27:36 DEBUG (MainThread) [zigpy.application] Sending 'zdo_leave_req' failed: [0xaea0:0:0x0034]: Message send failure
2021-11-10 09:27:36 DEBUG (bellows.thread_0) [bellows.uart] Data frame: b'71edb157546f15b65894a524ab5593499c1e8b6c4d609874f7cf3788fc1b3ea3ebccdf6bd6c47e'
2021-11-10 09:27:36 DEBUG (bellows.thread_0) [bellows.uart] Sending: b'8070787e'
2021-11-10 09:27:36 DEBUG (MainThread) [bellows.ezsp.protocol] Application frame 69 (incomingMessageHandler) received: b'00040100ef01010001000050acc7a0aeffff0a0954010065010400010104'
2021-11-10 09:27:36 DEBUG (MainThread) [bellows.zigbee.application] Received incomingMessageHandler frame with [<EmberIncomingMessageType.INCOMING_UNICAST: 0>, EmberApsFrame(profileId=260, clusterId=61184, sourceEndpoint=1, destinationEndpoint=1, options=<EmberApsOption.APS_OPTION_ENABLE_ROUTE_DISCOVERY: 256>, groupId=0, sequence=80), 172, -57, 0xaea0, 255, 255, b'\tT\x01\x00e\x01\x04\x00\x01\x01']
2021-11-10 09:27:36 DEBUG (MainThread) [bellows.zigbee.application] No such device 0xaea0
2021-11-10 09:27:37 DEBUG (bellows.thread_0) [bellows.uart] Data frame: b'01edb157546f15b65894a524ab5593499c1f8b6c4d609874f7cf3788fc1b3ea3ebccdf6b35067e'
2021-11-10 09:27:37 DEBUG (bellows.thread_0) [bellows.uart] Sending: b'8160597e'
2021-11-10 09:27:37 DEBUG (MainThread) [bellows.ezsp.protocol] Application frame 69 (incomingMessageHandler) received: b'00040100ef01010001000051acc7a0aeffff0a0954010065010400010104'
2021-11-10 09:27:37 DEBUG (MainThread) [bellows.zigbee.application] Received incomingMessageHandler frame with [<EmberIncomingMessageType.INCOMING_UNICAST: 0>, EmberApsFrame(profileId=260, clusterId=61184, sourceEndpoint=1, destinationEndpoint=1, options=<EmberApsOption.APS_OPTION_ENABLE_ROUTE_DISCOVERY: 256>, groupId=0, sequence=81), 172, -57, 0xaea0, 255, 255, b'\tT\x01\x00e\x01\x04\x00\x01\x01']
[... cut ...]
2021-11-10 09:28:58 DEBUG (MainThread) [bellows.ezsp.protocol] Application frame 5 (nop) received: b''
2021-11-10 09:29:08 DEBUG (MainThread) [bellows.ezsp.protocol] Send command nop: ()
2021-11-10 09:29:08 DEBUG (bellows.thread_0) [bellows.uart] Sending: b'20fb2157542f65e37e'
2021-11-10 09:29:08 DEBUG (bellows.thread_0) [bellows.uart] Data frame: b'03fba157542f43337e'
2021-11-10 09:29:08 DEBUG (bellows.thread_0) [bellows.uart] Sending: b'8160597e'
2021-11-10 09:29:08 DEBUG (MainThread) [bellows.ezsp.protocol] Application frame 5 (nop) received: b''
```

Youpimatin, 2021-11-10:
This morning with two 1.2V NIMH :
```2021-11-10 09:35:20 DEBUG (MainThread) [zhaquirks.tuya] [0x4170:1:0xef00] Received value [0, 0, 0, 58] for attribute 0x020e (command 0x0001)
2021-11-10 09:36:19 DEBUG (MainThread) [zhaquirks.tuya] [0x4170:1:0xef00] Received value [0, 0, 0, 61] for attribute 0x020e (command 0x0001)
2021-11-10 09:36:21 DEBUG (MainThread) [zhaquirks.tuya] [0x4170:1:0xef00] Received value [0, 0, 0, 61] for attribute 0x020e (command 0x0001)
```
This morning, after changing them with two 1.5V alkaline :
```
2021-11-10 09:39:15 DEBUG (MainThread) [zhaquirks.tuya] [0x4170:1:0xef00] Received value [0, 0, 0, 100] for attribute 0x020e (command 0x0001)
2021-11-10 09:40:00 DEBUG (MainThread) [zhaquirks.tuya] [0x4170:1:0xef00] Received value [0, 0, 0, 91] for attribute 0x020e (command 0x0001)
2021-11-10 09:40:01 DEBUG (MainThread) [zhaquirks.tuya] [0x4170:1:0xef00] Received value [0, 0, 0, 94] for attribute 0x020e (command 0x0001)
2021-11-10 09:51:18 DEBUG (MainThread) [zhaquirks.tuya] [0x4170:1:0xef00] Received value [0, 0, 0, 100] for attribute 0x020e (command 0x0001)
```

jacekk015, 2021-11-10:
For 0x0202 - we will leave it for now as it is. I will try to make a debug quirk so it could send all data possible - I will try it on my Tuya TRV. We will track this later.

For battery it looks like it's battery left %, from 100 down to "0"

MattWestb, 2021-11-10:
I think making one debug quirk is great but it need 2 different Zigbee modules and one with time cluster as in and one with it as out cluster.
Is is possible doing the `MODELS_INFO:` as one variable and having 5 or more signatures the system can trying ?

I can putting the signature together from different ts0601 quirks for you and its being little easier.

Youpimatin, 2021-11-16:
I've installed 8 of these on my radiators now, and there is another problem with the thermostats entities in Home Assistant :
6 of my TRVs are shown as "Heating"
2 of them are shown as "Off"

If I select them, they all are in "Heat" mode (in the operation dropdown)
(and if I change the operation mode to "Off", they return to "Heat", which is OK, since they can't go off)

Is there anything I can do to help you with the debugging of the quirk ?

jacekk015, 2021-11-16:
I've started to write code already. But after some thinking I've decided to extend all Moes devices.
So you all will receive ChildLock, WindowDetect etc.
Maybe later today I'll try to upload first dev version for overall test of the quirk.
BTW have you enabled custom quirks functionality in configuration.yaml??
https://github.com/zigpy/zha-device-handlers/discussions/693#discussioncomment-857274

Youpimatin, 2021-11-16:
Nice !
Yes I've enabled custom quirks

```
logger:
  default: info
  logs:
    homeassistant.components.zha: debug
    zigpy: debug
    zhaquirks: debug

zha:
  custom_quirks_path: /config/custom_zha_quirks/
```

With "ts0601_trv.py" in the custom_zha_quirks folder.
(not the last one with added "Tuya Battery 2* AA attributes", the previous one)

While waiting the quirk in the past days I wondered where can we find the dev version when you upload it ?

MattWestb, 2021-11-16:
Some tuya TRVs is supporting one real off mode but most is not doing that and i think its little strange implanting one off mode is the try is not supporting it in the firmware.
Most have one Away mode or Antifreeze mode that shall is not on system mode off and its good for not getting all destroyed by frozen water pipes.

If not having real off mode the TRV is always in system mode heating and only changing presets is the most logic and clean to do.

If user like having one off mode they can doing one automation that is setting the setpoint to one very low temperature.

jacekk015, 2021-11-16:
> While waiting the quirk in the past days I wondered where can we find the dev version when you upload it ?
Here, as a ZIP file. You will be notified.

Youpimatin, 2021-11-16:
> If user like having one off mode they can doing one automation that is setting the setpoint to one very low temperature.

Yes it's what I do.

I was just amazed by this display problem :

[image]

jacekk015, 2021-11-16:
@Youpimatin Here's test dev file. Remove your present quirk file for testing
[ts0601_trv_moes.py.zip](https://github.com/zigpy/zha-device-handlers/files/7550006/ts0601_trv_moes.py.zip)

Youpimatin, 2021-11-16:
Downloaded zip ; installed "ts0601_trv_moes.py" ; deleted previous quirk and "__pycache__" folder ; reloaded ZHA integration ; deleted 1 TRV ; re-paired this device.

Quirk: ts0601_trv_moes.MoesHY368_Type3
[image]

Added to lovelace :
[image]

(I had already renamed the child lock switch when I took the screenshots)

So : 
- 3 switches
- 1 analog_output Valve State
- 1 thermostat
- 1 power level

The child lock is working writing from HA to the TRV, but not reported to HA when changed on the TRV
The thermostat set point is working writing from HA to the TRV, but not reported to HA when changed on the TRV
...

jacekk015, 2021-11-17:
Error is a mistake in code copy/paste - will be corrected.
You have[for now] 3 switches: Child lock, Window detect alarm, Window detect A2 function[enabled/disabled] - from PDF manual.
I've decided to create a switch for Window open alarm. If the TRV do something on the Window open alarm, we then could automate this like for example: you put a Zigbee window magnetic sensor, and when sensor detects window opening the you could automatically enable Window Alarm on the TRV. Binary sensor is only one way TRV -> HA.

BTW you always need to restart HA after quirk update, only on boot time the quirk is loaded.

jacekk015, 2021-11-17:
Presets are also working, but now for them to work you need to modify some code in the HA climate.py, or wait for the official update later.
Your TRV has modes:
```
Away = 0x00
Schedule = 0x01
Manual = 0x02
```
@MattWestb could probably help you with this

MattWestb, 2021-11-17:
I think its not needed for testing then this TRV have the preset added for "normal" Moes TRVs so it shall working but the extra is very likely getting errors then being used.
```
            if record.value == 0:
                self._preset = PRESET_AWAY
            if record.value == 1:
                self._preset = PRESET_SCHEDULE
            if record.value == 2:
                self._preset = PRESET_NONE
            if record.value == 3:
                self._preset = PRESET_COMFORT
            if record.value == 4:
                self._preset = PRESET_ECO
            if record.value == 5:
                self._preset = PRESET_BOOST
            if record.value == 6:
                self._preset = PRESET_COMPLEX
```
I was thinking doing one standard with this 3 presets then its looks like most TRV is using them and can adding more TRVs without making now classes for them.

I doing one new climate.py with 3 presets for  it but its takes some minutes to making it not braking the formatting.

One more thing is that i think we shall renaming this device then its not one Moes (more right CN-Hysen) then its made of Baca and is having many different DPs and shall not being mixed with Moes (that have making white labeled of nearly all tuya TRVs and not making them self).
Also getting one PR thru is little more tricky then its braking the old implementation but i see its only making the device working better = shall being dome so user can using all possible function in there device.

jacekk015, 2021-11-17:
> but the extra is very likely getting errors then being use

I can overcome this by making last statement just ELSE. So anything else not known will be mapped to, for example, Schedule.

MattWestb, 2021-11-17:
No problem i have making one new here: [climate_beca.zip](https://github.com/zigpy/zha-device-handlers/files/7554933/climate_beca.zip)
Instruction: Unzip to HA configuration folder and use this command for copy it in to the HA container from docker CLI in your HA container:
```
cp /config/climate.py /usr/src/homeassistant/homeassistant/components/zha/climate.py
```
I have deleting the `"_TZE200_b6wax7g0"` from Moes and adding it  to Beca type.

The `"_TYST11_b6wax7g0"` shall being deleted in the quirk in the future then its looks like Beca Smart is not  releasing devices with different Zigbee mudules (like CN-Hysen is doing all the time).

jacekk015, 2021-11-17:
All will be done.

> Also A4 window open countdown.
Wold being nice in ZHA GUI to having the temperature and time and also the countdown.

User didn't found any attribute for this.

> Can being nice setting the night temperature and only switching with commands in ZHA GUI or automation. Candidate for presets or one night version or manual / none ?

I've planned to mix it to presets, seems there's an ECO mode there.

Most important. What's your decision???
We keep it in Moes as a Type3 or we create a new file with beca suffix??

MattWestb, 2021-11-17:
Perhaps its only implanted in the TRV and dont have any DP for setting it from the host system.

The Moes type 3 was one hack for getting the TRV working then the setpoint temperature was not OK compared with the Moes quirk.

But then making one quirk for this device type its being one braking thing for user that have doing automation with the original quirk, so i think its better renaming it and saying its one new quirk and moving the devices to the new for getting it working OK then its have some broken functionality as Moes type. If in one new quirk or one new class for its type in the old its not so important for the function but i think its better changing the name to beca for showing its not one Moes TRV and need the new class for working OK.

My feeling is that the normal TRV quirk is being very large then all is inside (exude the Sas TRV) it shall being better making different files then all types is needing match code for working.
But if doing one new quirk its also moderate making one working test for the quirk that is the problem i having for the moment then the test for the` ts0602_trv` is made for my TRV and its braking the test then making changes or deleting it from the original quirk and its not allowed leaving not working code in the system.

So if putting it in` ts0601_trv` you (hopefully) dont need doing test code and is lesser work but i dont knowing if its better for the system but its not my responsibility.
On the other hand its not so large working moving device types to one new quirk file later but its make no sense doing work that you knowing  its not working in the long run and must being redone very soon.

In the end is you that is doing the coding and the maintainer (you have experiences of them in the extra attribute request PR) have the last word how they like have it for getting PR merged and in the end its up to you and if you dont do the coding the system is getting no updates so its not good for our users that like have good supported device.

And its only 7.9° C outside so all TRV is very needed !!

jacekk015, 2021-11-17:
@Youpimatin Please retest newer version. The quirk was renamed to Beca, like we've agreed with @MattWestb , so you have to delete/move previous file from the folder. Remember to restart HA after copying.
All things are implemented, so test it good, Eco and Boost mode are mapped to Presets, so if you choose Preset Boost, then Boost mode should be enabled. Same for Eco Preset.
Boost countdown is also available as a read-only field. I assume it's in minutes, since time was set 300 seconds.

Preset should be available under those 3 dots, in Lovelace.
[image]

[ts0601_trv_beca.py.zip](https://github.com/zigpy/zha-device-handlers/files/7557684/ts0601_trv_beca.py.zip)

MattWestb, 2021-11-17:
Sounds great !!
If not getting errors then HA is selecting wrong preset its OK but i think its looks nice have the right resets in the GIU and not some extra that  is not working in your model.

I think its not so much work fixing the presets in the final version so if you have time i think its good doing it nice and clean and not letting leaving lose ends that is confusing users.
```
        Away = 0x00
        Schedule = 0x01 Delete ?
        Manual = 0x02
        Comfort = 0x03 Delete
        Eco = 0x04
        Boost = 0x05
        Complex = 0x06 Delete
```
and renumbering the presets then i dont knowing if ZHA is liking not having missing numbers. Only saying functions and number and i making on new version of climate.py.

And i think the Eco / Night mode is great for user to have :-)))

PS: i must looking if i can steeling more nice code ;-)))

For @Youpimatin 
Picture of selecting presets in ZHA
[image]
:

Youpimatin, 2021-11-17:
Here is what I have (entities renamed) : 
[image]

I'll test further tomorrow, but I think there is a little bug with negative values for the temperature calibration : you cannot write a negative value from HA to the TRV, but if changed from the TRV, negative values are correctly reported.

[image]
:

MattWestb, 2021-11-17:
Looks great @Youpimatin but you is missing the presets in ZHA.
You is only having operating mode On/Off.
Is you using the original (not patched) or the first patched climate.py ?
With both you shall getting presets but more ones with the original one but some not used.

MattWestb, 2021-11-17:
I have looking 3 times in the climate.py and i cant see any wrong compared with other TRVs added in it.

You can trying the one for MaxSmart and you shall getting the "normal" Moes TRV presets.
[SMclimate.zip](https://github.com/zigpy/zha-device-handlers/files/7558171/SMclimate.zip)

Copy it to the HA container as in the first instruction and restart HA and you shall getting some presets in the ZHA GUI.

Youpimatin, 2021-11-18:
I'm using core-2021.6.6, with the original climate.py

As I use an older version of HA, can I update with your climate.py version without breaking something ?

jacekk015, 2021-11-18:
@Youpimatin 
>  you cannot write a negative value from HA to the TRV, but if changed from the TRV, negative values are correctly reported.

Give me LOGS of that communication to/from TRV
Look for attribute 0x0269
```
0x0269 # Temperature compensation ( Received value [255, 255, 255, 255] for -1)
2021-11-07 18:31:13 DEBUG (MainThread) [zigpy.zcl] [0xef69:1:0xef00] ZCL request 0x0001: [Command(status=0, tsn=47, command_id=617, function=0, data=[4, 0, 0, 0, 0])]
2021-11-07 18:31:13 DEBUG (MainThread) [zhaquirks.tuya] [0xef69:1:0xef00] Received value [0, 0, 0, 0] for attribute 0x0269 (command 0x0001)
```

[313 more replies]

closed 2024-07-31 by github-actions[bot]